asus-linux-drivers/asus-numberpad-driver #148: [Layout Request] Layout Name

Forceres, 2023-11-16:
### Layout Name

UM3402

### Your device (sudo dmidecode -s system-product-name)

Zenbook UM3402YAR_UM3402YA

kamack38, 2023-11-16:
It is impossible for us to add a layout that we don't know what it looks like. Could you please supply a layout photo.

ldrahnik, 2023-11-16:
@Forceres Thank you for creating the issue but as wrote @kamack38 we need more info. Layout looks like the same as is named `up5401ea`, so please try this one. If any problem occurs do not hesitate add comment here.

ldrahnik, 2023-11-16:
Can be reopened.

closed 2023-11-16 by ldrahnik