asus-linux-drivers/asus-numberpad-driver #149: [Bug]: toggle and uninstall not working properly

Mtillmann, 2023-11-21:
### Describe the bug

Installed the driver on my zenbook 14 oled using the up5401ea + unicode option. After reboot the keypad was permanently lit up and I could enter numbers BUT I could not toggle it. After uninstall the keypad is permanently lit up. Is there anything I can do to turn off the light?

Thanks

### Expected behavior

Working toggle and light off after uninstall

### Desktop

- Program version 4.1.1
- Distro ubuntu
- Kernel 6.5.7

Mtillmann, 2023-11-21:
FWIW: running the _install_ script again disabled the backlight. I cancelled the install script, then ran uninstall again and rebooted. This has fixed it

ldrahnik, 2023-11-21:
Hello there,
thank you for creating an issue. Uninstall script should be used when is NumberPad deactivated. The question is why you had permanently lit up after reboot. Can you please provide driver logs. Thank you.

Mtillmann, 2023-11-21:
Hi @ldrahnik , sure - how can I produce those logs?

ldrahnik, 2023-11-21:
@Mtillmann `/var/log/asus-numberpad-driver/*`

Mtillmann, 2023-11-21:
[uninstall-21-11-2023-15-56-36.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/13427740/uninstall-21-11-2023-15-56-36.log)
[error.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/13427741/error.log)
[install-21-11-2023-15-54-20.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/13427742/install-21-11-2023-15-54-20.log)

The files are attached. This time the touchpad light up on the lockscreen after install and reboot, then did nothing. 

I also experience something similar to #127 : my super key wont work and I cant hit ctrl+alt+t for a new terminal etc

ldrahnik, 2023-11-21:
@Mtillmann > This time the touchpad light up on the lockscreen after install and reboot, then did nothing.

If was NumLock pressed on loginScreen or automatically enabled it is correct behaviour because by default NumberPad reflects NumLock status and then is enabled aswell.

Mtillmann, 2023-11-21:
shortcuts again when I uninstall.... I also just saw in my logs that I installed a different variant than the first time (up5401ea-31B9). I'll try again. Thanks for your patience

Mtillmann, 2023-11-21:
Ok - initially I installed "up5401ea + unicode", which is broken on my machine. Now I tried only "up5401ea" and to my amazement I can toggle the numpad by holding the symbol for like 2seconds. However, the shortcuts still dont work. I'll try and reinstall the driver without the keyboard options and try again

Mtillmann, 2023-11-21:
Ok, this fixes it for me:

- install "up5401ea"
- install service - yes
- rule for external keyboard - no
- xf86calculator - no

My exact laptop model as reported by `dmidecode` : 
```
	Manufacturer: ASUSTeK COMPUTER INC.
	Product Name: Zenbook UM3402YAR_UM3402YA
```

Thanks for your work

ldrahnik, 2023-11-21:
@Mtillmann Try to reinstall now (uninstall, install with installing calc shortcut too, reboot). Calc shortcut should work, for testing try activate numberpad and with one finger to slide from left corner to the opposite corner right bottom atleast 50% width and height. My mistake. Missing separator at the end when adding new shortcut path (https://github.com/asus-linux-drivers/asus-numberpad-driver/commit/609e833e1eba3e0f71c106290287d86e5da1dd0b). Then please let me know. Thank you :)

Mtillmann, 2023-11-21:
@ldrahnik that commit fixes my problem. I've installed with all options (service, calculator, keyboard rule) and I can toggle the numpad and open/close the calculator with the gesture. I think you should make a new release with that typo fixed. Thanks

ldrahnik, 2023-11-22:
@Mtillmann This happened every install attempt?

```
cat: laptop_numberpad__layouts: No such file or directory
cat: laptop_numberpad__layouts: No such file or directory
cat: laptop_numberpad__layouts: No such file or directory
```

You deleted that file manually?

Mtillmann, 2023-11-22:
This happened with every install and I didn't delete anything

ldrahnik, 2023-11-22:
@Mtillmann I have added your laptop to the file for offline autosuggestion layout and fixed online autosuggestion. No more `cat: laptop_numberpad__layouts: No such file or directory`. Thank you :)

closed 2023-11-22 by ldrahnik

ldrahnik, 2023-11-22:
@Mtillmann Key for percents character works for you? In case no I have added non-unicode variants of all layouts to repository (shortcut left shift + 5 for percent char) which can be handy when does not work shortcut Ctrl+Shift+U+<numbers>. Please, let me know.

Mtillmann, 2023-11-22:
Hi @ldrahnik , % works although it enters differently than the others. ctrl+shit+u enters an underlined u on my keyboard
[Screencast from 22.11.2023 21:27:42.webm](https://github.com/asus-linux-drivers/asus-numberpad-driver/assets/289963/9f003c67-43f3-44f4-b501-5f506ce6c295)

Mtillmann, 2023-11-22:
In the video I just realized that  underline u  is entered and then converted to `%`

ldrahnik, 2023-11-23:
@Mtillmann No more ```rm: cannot remove '/usr/share/asus-numberpad-driver/scripts/calculator_toggle.sh': Permission denied
rm: cannot remove '/usr/share/asus-numberpad-driver/udev/external_keyboard_is_disconnected.sh': Permission denied
rm: cannot remove '/usr/share/asus-numberpad-driver/udev/external_keyboard_is_connected.sh': Permission denied
Something went wrong when removing files from the /usr/share/asus-numberpad-driver
Asus numberpad driver removed```, uninstall process should run properly now.

ldrahnik, 2023-11-23:
@Mtillmann If works '%' via `Ctrl+Shift+U` shortcut, also should work full unicode layout (`up5401ea-unicode`) without any problems. Could you confirm that please?

Mtillmann, 2023-11-23:
Hi @ldrahnik , I've tested the unicode variant and it works as well, although with the unicode variant every character/digit is entered a way where it is convert from something underlined to the final character, see attachment.
[Screencast from 23.11.2023 23:34:11.webm](https://github.com/asus-linux-drivers/asus-numberpad-driver/assets/289963/62317142-edbf-4b81-b716-fced521d51fe)