asus-linux-drivers/asus-numberpad-driver #151: [Bug]: Log directory used by systemd service is not created

SelfRef, 2023-11-29:
### Describe the bug

After installing AUR package [asus-numberpad-driver-up5401ea-git](https://aur.archlinux.org/packages/asus-numberpad-driver-up5401ea-git) I couldn't start the `asus_numberpad_driver.service` system service.

It failed with error:
```
(python)[60652]: asus_numberpad_driver.service: Failed to set up standard output: No such file or directory
(python)[60652]: asus_numberpad_driver.service: Failed at step STDOUT spawning /usr/bin/python: No such file or direct>
systemd[1]: Started Asus NumberPad Driver.
systemd[1]: asus_numberpad_driver.service: Main process exited, code=exited, status=209/STDOUT
systemd[1]: asus_numberpad_driver.service: Failed with result 'exit-code'.
```

It turns out the path to log file `/var/log/asus-numberpad-driver/error.log` cannot be written because there's no `/var/log/asus-numberpad-driver/` directory. After creating it manually, the service is working as intended.

### Expected behavior

The `/var/log/asus-numberpad-driver/` directory should be created by installation script.

### Desktop

- Program version: r654.6693e3b-1
- Distro: Arch Linux
- Kernel: 6.6.2-arch1-1

ldrahnik, 2023-11-29:
@SelfRef Hi, I understand well the issue installation directly from git repository works without any problems?

SelfRef, 2023-11-29:
> @SelfRef Hi, I understand well the issue that installation directly from git repository works without any problems?

Yes. From what I've checked, the installation script is using this directory (as `LOGS_DIR_PATH`) to put installation logs there, so it's created in this case. When installing through AUR package it's not created.

I assume a simple `mkdir` in PKGBUILD would resolve this issue:
```diff
_package() {
    _config_dir="/usr/share/${_pkgbase}"
    _log_dir="/var/log/asus-numberpad-driver"
+   mkdir "${pkgdir}${_log_dir}"
    cd "${srcdir}/${_pkgbase}"
```

ldrahnik, 2023-11-29:
@kamack38 ping

ldrahnik, 2023-12-06:
@kamack38 Normally it should be like that. Unfortunatelly current design does not allow do that because is required sudo to create log directory inside `/var/log`. Btw, I do not know why would anyone delete directory for logs of software which is currently using but can happens, then is easy to reinstall entirely driver or call `bash install_logs.sh` when is identified systemctl error does why driver service does not start.