asus-linux-drivers/asus-numberpad-driver #153: [Bug]: ImportError: cannot import name 'AsyncNotifier' from 'pyinotify'

Jailoris, 2023-12-28:
### Describe the bug

Fail to import 'AsyncNotifier' class from 'pyinotify' on package python-pynotify.0.9.6-32.fc39 (Fedora 39)

### Expected behavior

I believe it worked correctly on Fedora 38, might want to ask the maintainers at [](https://packages.fedoraproject.org/pkgs/python-inotify/python3-inotify/)

### Relevant log output

```shell
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 16, in <module>
    from pyinotify import WatchManager, IN_CLOSE_WRITE, IN_IGNORED, IN_MOVED_TO, AsyncNotifier
ImportError: cannot import name 'AsyncNotifier' from 'pyinotify' (/usr/lib/python3.12/site-packages/pyinotify.py). Did you mean: 'AsyncioNotifier'?
```

### Desktop

- Program version [d54b8f5]
- Distro [Fedora 39]
- Kernel [6.6.8-200.fc39.x86_64]

ldrahnik, 2023-12-28:
Hello there,
thank you for report. Please attach the log from installing too.

ldrahnik, 2023-12-28:
Could you please post equivalent for your package manager?

```
$ sudo apt show python3-pyinotify
Package: python3-pyinotify
Version: 0.9.6-1.3
Priority: optional
Section: python
Source: pyinotify
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Mikhail Gusarov <dottedmag@debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 110 kB
Depends: python3:any
Suggests: python-pyinotify-doc
Homepage: https://github.com/seb-m/pyinotify
Download-Size: 24.8 kB
APT-Manual-Installed: yes
APT-Sources: http://archive.ubuntu.com/ubuntu jammy/main amd64 Packages
Description: simple Linux inotify Python bindings
 pyinotify is a simple wrapper for the Linux inotify mechanism.
 .
 inotify is a Linux Kernel feature available since 2.6.13. inotify makes
 it possible for applications to easily be notified of filesystem changes.
 .
 This is the Python 3 version of the package.
```

ldrahnik, 2023-12-28:
Please try run driver with previous version of `python3` if you can like this, replace `python3`: 

```
LOG=DEBUG sudo python3 ./numberpad.py "up5401ea" "" ""
```

Jailoris, 2023-12-29:
I tried on a Python 3.11 virtualenv since installing older python libraries of the previous distro iteration isn't possible with the package manager and it seems to work correctly. I do believe that Fedora maintainers had a hand into this since it worked correctly on Fedora 38 for me with Python 3.11 and python-inotify.0.9.6-29.fc38. For the moment I modified the systemd service to run the script in a virtualenv.

ldrahnik, 2023-12-29:
Could you please copy paste content o file in which one can not be found `AsyncNotifier`

Jailoris, 2023-12-29:
`numberpad.py`
```
#!/usr/bin/env python3

import configparser
import importlib
import logging
import math
import os
import re
import subprocess
import sys
import threading
from time import sleep, time
from typing import Optional
import numpy as np
from libevdev import EV_ABS, EV_KEY, EV_LED, EV_MSC, EV_SYN, Device, InputEvent, const, device
from pyinotify import WatchManager, IN_CLOSE_WRITE, IN_IGNORED, IN_MOVED_TO, AsyncNotifier
import Xlib.display
import Xlib.X
import Xlib.XK
```

```
[daniel@fedora ~] $ LOG=DEBUG sudo python3 /usr/share/asus-numberpad-driver/numberpad.py up5401ea-31B9 /usr/share/asus-numberpad-driver/
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 16, in <module>
    from pyinotify import WatchManager, IN_CLOSE_WRITE, IN_IGNORED, IN_MOVED_TO, AsyncNotifier
ImportError: cannot import name 'AsyncNotifier' from 'pyinotify' (/usr/lib/python3.12/site-packages/pyinotify.py). Did you mean: 'AsyncioNotifier'?

```

I looked at the `pyinotify.py` site package in `/usr/lib/python3.12/site-packages` and weirdly enough the AsyncNotifier class is present but won't import :

`pyinotify.py (python-inotify.0.9.6-32.fc39)`
```
try:
    import asyncore

    class AsyncNotifier(asyncore.file_dispatcher, Notifier):
        """
        This notifier inherits from asyncore.file_dispatcher in order to be able to
        use pyinotify along with the asyncore framework.

        """
        def __init__(self, watch_manager, default_proc_fun=None, read_freq=0,
                     threshold=0, timeout=None, channel_map=None):
            """
            Initializes the async notifier. The only additional parameter is
            'channel_map' which is the optional asyncore private map. See
            Notifier class for the meaning of the others parameters.

            """
            Notifier.__init__(self, watch_manager, default_proc_fun, read_freq,
                              threshold, timeout)
            asyncore.file_dispatcher.__init__(self, self._fd, channel_map)

        def handle_read(self):
            """
            When asyncore tells us we can read from the fd, we proceed processing
            events. This method can be overridden for handling a notification
            differently.

            """
            self.read_events()
            self.process_events()
```

Jailoris, 2023-12-29:
I can confirm that the script now run under Python3.12 by installing the compatibility package `python3-pyasyncore-1.0.2-1.fc39`

`sudo dnf install python3-pyasyncore`

ldrahnik, 2023-12-30:
Great news (y) 

Could you please test whether works my solution in this [branch](https://github.com/asus-linux-drivers/asus-numberpad-driver/commit/0688873df2d9d90fdfc495bed47323e896a0ff45) which uses pip?

I used this problem as an opportunity to go back to pip but called with `--user` or as `venv`, I am undecided yet. In the branch is implemented variant with `--user` flag.

Great is when is detected python version lower like in my case not `3.12` is installed nothing:

```
...
Requirement already satisfied: python-xlib in /usr/lib/python3/dist-packages (from -r requirements.txt (line 4)) (0.29)
Collecting pyasyncore
  Downloading pyasyncore-1.0.2-py2.py30.py31.py32.py33.py34.py35.py36.py37.py38.py39.py310.py311-none-any.whl (3.5 kB)
Requirement already satisfied: nose in /home/ldrahnik/.local/lib/python3.10/site-packages (from inotify->-r 
...
```

ldrahnik, 2023-12-31:
In the master branch fixed by installing package by current system package manager https://github.com/asus-linux-drivers/asus-numberpad-driver/commit/ded12fe3e2c909b9c7526254f3a8479ed338983a

ldrahnik, 2023-12-31:
>Could you please test whether works my solution in this [branch](https://github.com/asus-linux-drivers/asus-numberpad-driver/commit/0688873df2d9d90fdfc495bed47323e896a0ff45) which uses pip?

>I used this problem as an opportunity to migrate back to using pip but called with --user or as venv, I am undecided yet. In the branch is implemented variant with --user flag.

>When is detected python version lower like in my case not 3.12 is installed nothing:

```Requirement already satisfied: python-xlib in /usr/lib/python3/dist-packages (from -r requirements.txt (line 4)) (0.29)
Collecting pyasyncore
  Downloading pyasyncore-1.0.2-py2.py30.py31.py32.py33.py34.py35.py36.py37.py38.py39.py310.py311-none-any.whl (3.5 kB)
Requirement already satisfied: nose in /home/ldrahnik/.local/lib/python3.10/site-packages (from inotify->-r 
```
The current branch implementation with ```pip --user``` works on Ubuntu 22.04.3 LTS but does not work on Debian (tested in the branch https://github.com/asus-linux-drivers/asus-numberpad-driver/issues/154)

```
Installation with this new branch brings another error message:
flipo% bash install.sh
[sudo] Mot de passe de flipo :
groupadd : le groupe 'numberpad' existe déjà
Added group numberpad to current user

Lecture des listes de paquets…
Construction de l'arbre des dépendances…
Lecture des informations d'état…
ibus est déjà la version la plus récente (1.5.27-5).
libevdev2 est déjà la version la plus récente (1.13.0+dfsg-1).
curl est déjà la version la plus récente (7.88.1-10+deb12u5).
xinput est déjà la version la plus récente (1.6.3-1).
i2c-tools est déjà la version la plus récente (4.3-2+b3).
python3-dev est déjà la version la plus récente (3.11.2-1+b1).
python3-pip est déjà la version la plus récente (23.0.1+dfsg-1).
libxml2-utils est déjà la version la plus récente (2.9.14+dfsg-1.3~deb12u1).
0 mis à jour, 0 nouvellement installés, 0 à enlever et 0 non mis à jour.

error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
python3-xyz, where xyz is the package you are trying to
install.

If you wish to install a non-Debian-packaged Python package,
create a virtual environment using python3 -m venv path/to/venv.
Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
sure you have python3-full installed.

If you wish to install a non-Debian packaged Python application,
it may be easiest to use pipx install xyz, which will manage a
virtual environment for you. Make sure you have pipx installed.

See /usr/share/doc/python3.11/README.venv for more information.
```

ldrahnik, 2024-01-02:
Pip packages will be eventually implemented in other issues/branches.

closed 2024-01-02 by ldrahnik