asus-linux-drivers/asus-numberpad-driver #154: [Bug]: driver fails to start

dflipo, 2023-12-29:
### Describe the bug

After apparently successful installation, the driver doesn't start as shown below.

### Expected behavior

What am I missing?

### Relevant log output

```shell
flipo% systemctl restart --user asus_numberpad_driver@flipo.service
flipo% systemctl status --user asus_numberpad_driver@flipo.service
× asus_numberpad_driver@flipo.service - Asus NumberPad Driver
     Loaded: loaded (/usr/lib/systemd/user/asus_numberpad_driver@.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Fri 2023-12-29 17:11:09 CET; 1s ago
   Duration: 8ms
    Process: 23060 ExecStart=/usr/bin/env python3 /usr/share/asus-numberpad-driver/numberpad.py up5401ea /usr/share/asus-numberpad-driver/ (code=exited, status=2)
   Main PID: 23060 (code=exited, status=2)
        CPU: 7ms

déc. 29 17:11:09 givry systemd[3273]: asus_numberpad_driver@flipo.service: Scheduled restart job, restart counter is at 5.
déc. 29 17:11:09 givry systemd[3273]: Stopped asus_numberpad_driver@flipo.service - Asus NumberPad Driver.
déc. 29 17:11:09 givry systemd[3273]: asus_numberpad_driver@flipo.service: Start request repeated too quickly.
déc. 29 17:11:09 givry systemd[3273]: asus_numberpad_driver@flipo.service: Failed with result 'exit-code'.
déc. 29 17:11:09 givry systemd[3273]: Failed to start asus_numberpad_driver@flipo.service - Asus NumberPad Driver.
```

### Desktop

- Program version 4.1.5
- Distro Debian 12 Bookworm
- Kernel 6.6.8
- Asus Zenbook UX3402ZA  French keyboard
-

ldrahnik, 2023-12-29:
@dflipo Please attach the driver error log.

dflipo, 2023-12-29:
Sorry, I do not see any error, I just notice that the touchpad doesn't work (as numberpad only) and check the status of the `systemctl` command which says the driver didn't start... Should I enable a debug mode or so?

ldrahnik, 2023-12-29:
@dflipo Can you please post that error file here because usually Is service restarted when driver returns error code 1, thank you

dflipo, 2023-12-30:
Thanks for your help!  When I run
`flipo% /usr/bin/env python3 /usr/share/asus-numberpad-driver/numberpad.py up5401ea /usr/share/asus-numberpad-driver/
python3:`
the error is
 `can't open file '/usr/share/asus-numberpad-driver/numberpad.py': [Errno 2] No such file or directory`

For some unknown reason the file `numberpad.py` was not copied, the directory only includes 2 subdirectories:
flipo% ll /usr/share/asus-numberpad-driver/
drwxr-xr-x   2 root root  4096 29 déc.  12:38 scripts/
drwxr-xr-x   2 root root  4096 29 déc.  12:45 udev/

So I add  `numberpad.py` manually, then
flipo% ls -la /usr/share/asus-numberpad-driver/
-rwxr-xr-x   1 flipo flipo 65644 10 déc.  08:04 numberpad.py*
drwxr-xr-x   2 root  root   4096 29 déc.  12:38 scripts/
drwxr-xr-x   2 root  root   4096 29 déc.  12:45 udev/

and run the command again:
flipo% /usr/bin/env python3 /usr/share/asus-numberpad-driver/numberpad.py up5401ea /usr/share/asus-numberpad-driver/
python3:
2023-12-30 09:54:47,435 ERROR    Numpad layout *.py from dir layouts is required as first argument. Re-run install script or add missing first argument (valid value is b7402, e210ma, g533, gx551, gx701, up5401ea, ..).

I rerun the install script and answer 'y' to all questions, i end up with
Installation finished succesfully

Reboot is required. Do you want reboot now? [y/N]

I will reboot and tell you what happens, THANKS again!

dflipo, 2023-12-30:
After rebooting, the touchpad still doesn't work as numberpad (only as traditional touchpad, touching the upper right corner doesn't swith the number pad on). So I run again

flipo% /usr/bin/env python3 /usr/share/asus-numberpad-driver/numberpad.py up5401ea /usr/share/asus-numberpad-driver/
2023-12-30 10:04:58,823 INFO     Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
2023-12-30 10:04:58,824 INFO     Set keyboard 0 from H: Handlers=kbd leds event0
2023-12-30 10:04:58,836 INFO     Detecting touchpad from string: "N: Name="ASUE140D:00 04F3:31B9 Touchpad""
2023-12-30 10:04:58,836 INFO     Set touchpad device id 1 from S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-ASUE140D:00/0018:04F3:31B9.0001/input/input23
2023-12-30 10:04:58,836 INFO     Set touchpad id 11 from H: Handlers=mouse1 event11
2023-12-30 10:04:58,836 INFO     Touchpad min-max: x 0-3880, y 0-2299
2023-12-30 10:04:58,836 INFO     Numpad min-max: x 200-3680, y 200-2219
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 522, in <module>
    udev = dev.create_uinput_device()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/libevdev/device.py", line 752, in create_uinput_device
    d._uinput = UinputDevice(self._libevdev, uinput_fd)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/libevdev/_clib.py", line 936, in __init__
    raise OSError(-rc, os.strerror(-rc))
PermissionError: [Errno 13] Permission denied

I changed the ownership of contents of /usr/share/asus-numberpad-driver/ to flipo:root but it doesn't help.
flipo% ll /usr/share/asus-numberpad-driver/
drwxr-xr-x   3 flipo root  4096 30 déc.  09:56 layouts/
-rwxr-xr-x   1 flipo root 65644 30 déc.  09:56 numberpad.py*
drwxr-xr-x   2 flipo root  4096 29 déc.  12:38 scripts/
drwxr-xr-x   2 flipo root  4096 29 déc.  12:45 udev/

Thanks for any hints!

ldrahnik, 2023-12-30:
First please try reboot after installing because now it looks like your user is just not in `uninput` group.

But first change ownership back. I do not why you did that.

And then (if was NumberPad started successfully or even when not please repeat the installing), run `bash ./uninstall.sh` and then again `bash install.sh`  + necessarry reboot and then check whether it works and post here entire output from installing. Install attempt is every time even logged to file located in dir `/var/logs/asus-num..`

dflipo, 2023-12-30:
Following your instructions, I first uninstalled, reboot, then install again, reboot. The numberpad still doesn't work, I append the files error.log and install.lo found in `/var/log/asus-numberpad-driver`

[error.log.zip](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/13797659/error.log.zip)
[install-30-12-2023-10-37-02.log.zip](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/13797660/install-30-12-2023-10-37-02.log.zip)

ldrahnik, 2023-12-30:
Put here please output of the command `$ groups`

dflipo, 2023-12-30:
flipo% groups
flipo cdrom floppy sudo audio dip video plugdev users input netdev bluetooth lpadmin scanner i2c numberpad uinput

ldrahnik, 2023-12-30:
And `$ ls -l /dev/uinput`?

dflipo, 2023-12-30:
flipo% ls -l /dev/uinput
crw------- 1 root root 10, 223 30 déc.  10:38 /dev/uinput

ldrahnik, 2023-12-30:
This is the problem why driver does not run. It should be:

```
$ ls -l /dev/uinput
crw-rw---- 1 root uinput 10, 223 Dec 30 11:21 /dev/uinput
```

ldrahnik, 2023-12-30:
In driver is used the same command as is posted and even described why it is needed here https://tkcheng.wordpress.com/2013/11/11/changing-uinput-device-permission/

ldrahnik, 2023-12-30:
Command should solve it for you

```
$ sudo chown :uinput /dev/uinput
```

ldrahnik, 2023-12-30:
Please confirm it works even between reboots.

dflipo, 2023-12-30:
What works on my system (Debian 12):
root# chown :uinput  /dev/uinput
root# chmod g+rw  /dev/uinput
flipo% systemctl restart --user asus_numberpad_driver@flipo.service
The the number pad then works as expected :-) Great thanks for your very efficient support!!!

Btw. you should mention that the user *must* have the directory "/usr/sbin" listed in his/her $PATH (for i2cdetect, i2ctransfer).
I still wonder if this is safe…

What fails on my system: 
after reboot, the permissions on /dev/uinput are wrong, they are reset to:
crw------- 1 root root 10, 223 Dec 30 11:59 /dev/uinput

Adding `/etc/udev/rules.d/99-input.rules` doesn't help on Debian, the rules are in `/usr/lib/udev/rules.d/` *but* adding `99-input.rules` in this directory doesn't provide the expected permissions on /dev/uinput after reboot. What am I missing?

dflipo, 2023-12-30:
Regarding the PATH, i would suggest to change the error message about INTERFACES in `install_device_check.sh` from

"No i2c interface can be found. Make sure you have installed libevdev packages"

to something like

"No i2c interface can be found. Make sure you have installed libevdev packages AND that your PATH includes /usr/bin"

ldrahnik, 2023-12-30:
From [Debian udev wiki](https://wiki.debian.org/udev):

```
...
The rules files (which amount to more configuration for udevd) are taken from /run/udev/rules.d, /etc/udev/rules.d or /lib/udev/rules.d....
...
```

ldrahnik, 2023-12-30:
@dflipo 

> Btw. you should mention that the user must have the directory "/usr/sbin" listed in his/her $PATH (for i2cdetect, i2ctransfer).
I still wonder if this is safe…

Can you be more specific about what risk you see?

ldrahnik, 2023-12-30:
>> Adding /etc/udev/rules.d/99-input.rules doesn't help on Debian, the rules are in /usr/lib/udev/rules.d/ but adding 99-input.rules in this directory doesn't provide the expected permissions on /dev/uinput after reboot. What am I missing?

>From [Debian udev wiki](https://wiki.debian.org/udev):

```
...
The rules files (which amount to more configuration for udevd) are taken from /run/udev/rules.d, /etc/udev/rules.d or /lib/udev/rules.d....
...
```

You can test whether was the rule applied like this (`--verbose`):

```
$ ls -l /dev/uinput
crwxrwxr-x 1 root uinput 10, 223 Dec 30 13:16 /dev/uinpu

$ echo 'KERNEL=="uinput", GROUP="uinput", MODE:="0660"' | sudo tee /etc/udev/rules.d/99-input.rules >/dev/null

$ sudo udevadm control --reload-rules && sudo udevadm trigger --verbose --sysname-match=uinput

$ ls -l /dev/uinput
crw-rw---- 1 root uinput 10, 223 Dec 30 13:23 /dev/uinput
```

dflipo, 2023-12-30:
> Can you be more specific about what risk you see? I think your distribution does have sbin in $PATH by default and you changed it by yourself. Is it documented somewhere in case I am wrong? Can you provide doc link, please?

I have been administrator of  my Uni lab's network (I am retired now), at that time only administrators had `/sbin` and `/usr/sbin` in their PATH. Otherwise, what is the point of splitting binaries between `/usr/bin` and `usr/sbin`? (for me `s` stands for `system`).

As of Debian, sorry, I see nothing in my installation scripts that exclude `/usr/sbin` from the users' PATH. 

In `/usr/sbin` I see programs like `unix_chkpwd`,  `update-grub`,  `update-secureboot-policy`, etc.,  not really meant for standard users imho...

ldrahnik, 2023-12-30:
>> Can you be more specific about what risk you see? I think your distribution does have sbin in $PATH by default and you changed it by yourself. Is it documented somewhere in case I am wrong? Can you provide doc link, please?

> I have been administrator of my Uni lab's network (I am retired now), at that time only administrators had /sbin and /usr/sbin in their PATH. Otherwise, what is the point of splitting binaries between /usr/bin and usr/sbin? (for me s stands for system).

> As of Debian, sorry, I see nothing in my installation scripts that exclude /usr/sbin from the users' PATH.

> In /usr/sbin I see programs like unix_chkpwd, update-grub, update-secureboot-policy, etc., not really meant for standard users imho...

I agree best practice is to use the Python library (`smbus2`) could you please test this [branch](https://github.com/asus-linux-drivers/asus-numberpad-driver/tree/replaced-i2c-transfer-tool-to-usage-of-python-lib) as I now unfortunately do not have my laptop with NumberPad (reclamation).

ldrahnik, 2023-12-30:
>>> Adding /etc/udev/rules.d/99-input.rules doesn't help on Debian, the rules are in /usr/lib/udev/rules.d/ but adding 99-input.rules in this directory doesn't provide the expected permissions on /dev/uinput after reboot. What am I missing?

>>From [Debian udev wiki](https://wiki.debian.org/udev):

```
...
The rules files (which amount to more configuration for udevd) are taken from /run/udev/rules.d, /etc/udev/rules.d or /lib/udev/rules.d....
...
```

> You can test whether was the rule applied like this (`--verbose`, `crwxrwxr-x` has been changed to `crw-rw----`):

```
$ ls -l /dev/uinput
crwxrwxr-x 1 root uinput 10, 223 Dec 30 13:16 /dev/uinpu

$ echo 'KERNEL=="uinput", GROUP="uinput", MODE:="0660"' | sudo tee /etc/udev/rules.d/99-input.rules >/dev/null

$ sudo udevadm control --reload-rules && sudo udevadm trigger --verbose --sysname-match=uinput

$ ls -l /dev/uinput
crw-rw---- 1 root uinput 10, 223 Dec 30 13:23 /dev/uinput
```

Have you had time to check where (how) can be the udev rule applied to your distro?

dflipo, 2023-12-30:
> I agree best practice is to use the Python library (smbus2), could you please test this [branch](https://github.com/asus-linux-drivers/asus-numberpad-driver/tree/replaced-i2c-transfer-tool-to-usage-of-python-lib) (where I removed usage of i2c-transfer from sbin, important is [this commit](https://github.com/asus-linux-drivers/asus-numberpad-driver/commit/922241e5471dd3e221fd4f5db47b3fe0af9188b8), first one only adds pip3) because unfortunately I do not have my laptop with NumberPad now (reclamation).

Installation with this new branch brings another error message:
flipo% bash install.sh 
[sudo] Mot de passe de flipo : 
groupadd : le groupe 'numberpad' existe déjà
Added group numberpad to current user

Lecture des listes de paquets…
Construction de l'arbre des dépendances…
Lecture des informations d'état…
ibus est déjà la version la plus récente (1.5.27-5).
libevdev2 est déjà la version la plus récente (1.13.0+dfsg-1).
curl est déjà la version la plus récente (7.88.1-10+deb12u5).
xinput est déjà la version la plus récente (1.6.3-1).
i2c-tools est déjà la version la plus récente (4.3-2+b3).
python3-dev est déjà la version la plus récente (3.11.2-1+b1).
python3-pip est déjà la version la plus récente (23.0.1+dfsg-1).
libxml2-utils est déjà la version la plus récente (2.9.14+dfsg-1.3~deb12u1).
0 mis à jour, 0 nouvellement installés, 0 à enlever et 0 non mis à jour.

error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
pip dependencies via file requirements.txt cannot be loaded correctly.

I don't understand what is going on (I have no experience at all with python),  I don't know which "environment is externally managed"...

I have installed the Debian package `python3-smbus2`, and tried again,  => same error message.

dflipo, 2023-12-30:
> Have you had time to check where (how) can be the udev rule applied to your distro?

Yes, but with no success yet :-(

flipo% ls -l /dev/uinput 
crw------- 1 root root 10, 223 30 déc.  18:57 /dev/uinput
flipo% echo 'KERNEL=="uinput", GROUP="uinput", MODE:="0660"' | sudo tee /etc/udev/rules.d/99-input.rules >/dev/null
flipo% sudo udevadm control --reload-rules && sudo udevadm trigger --verbose --sysname-match=uinput
flipo% ls -l /dev/uinput
crw------- 1 root root 10, 223 30 déc.  18:57 /dev/uinput
flipo% sudo udevadm control --reload-rules && sudo udevadm trigger --verbose --sysname-match=uinput
flipo% ls -l /dev/uinput
crw------- 1 root root 10, 223 30 déc.  18:57 /dev/uinput

I have noticed in other rules on Debian that the MODE is defined with `MODE="0660"` instead of  `MODE="0660"` (in your command), so I tried it again with `MODE="0660"`, same result... I have to investigate what goes wrong on my system.

ldrahnik, 2023-12-31:
> I agree best practice is to use the Python library (smbus2), could you please test this [...]

> Installation with this new branch brings another error message:
flipo% bash install.sh

Bad idea to mix two branches. Can you please try it now? Thank you and sorry!

ldrahnik, 2023-12-31:
>Could you please test whether works my solution in this [branch](https://github.com/asus-linux-drivers/asus-numberpad-driver/commit/0688873df2d9d90fdfc495bed47323e896a0ff45) which uses pip?

>I used this problem as an opportunity to migrate back to using pip but called with --user or as venv, I am undecided yet. In the branch is implemented variant with --user flag.

>When is detected python version lower like in my case not 3.12 is installed nothing:

```Requirement already satisfied: python-xlib in /usr/lib/python3/dist-packages (from -r requirements.txt (line 4)) (0.29)
Collecting pyasyncore
  Downloading pyasyncore-1.0.2-py2.py30.py31.py32.py33.py34.py35.py36.py37.py38.py39.py310.py311-none-any.whl (3.5 kB)
Requirement already satisfied: nose in /home/ldrahnik/.local/lib/python3.10/site-packages (from inotify->-r 
```
Does not work on Debian (tested in the branch https://github.com/asus-linux-drivers/asus-numberpad-driver/issues/154)

```
Installation with this new branch brings another error message:
flipo% bash install.sh
[sudo] Mot de passe de flipo :
groupadd : le groupe 'numberpad' existe déjà
Added group numberpad to current user

Lecture des listes de paquets…
Construction de l'arbre des dépendances…
Lecture des informations d'état…
ibus est déjà la version la plus récente (1.5.27-5).
libevdev2 est déjà la version la plus récente (1.13.0+dfsg-1).
curl est déjà la version la plus récente (7.88.1-10+deb12u5).
xinput est déjà la version la plus récente (1.6.3-1).
i2c-tools est déjà la version la plus récente (4.3-2+b3).
python3-dev est déjà la version la plus récente (3.11.2-1+b1).
python3-pip est déjà la version la plus récente (23.0.1+dfsg-1).
libxml2-utils est déjà la version la plus récente (2.9.14+dfsg-1.3~deb12u1).
0 mis à jour, 0 nouvellement installés, 0 à enlever et 0 non mis à jour.

error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
python3-xyz, where xyz is the package you are trying to
install.

If you wish to install a non-Debian-packaged Python package,
create a virtual environment using python3 -m venv path/to/venv.
Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
sure you have python3-full installed.

If you wish to install a non-Debian packaged Python application,
it may be easiest to use pipx install xyz, which will manage a
virtual environment for you. Make sure you have pipx installed.

See /usr/share/doc/python3.11/README.venv for more information.
```

dflipo, 2023-12-31:
I tried it right now.  My udev rules problem being unsolved, after rebooting I had to do
root# chown :uinput  /dev/uinput
root# chmod g+rw  /dev/uinput
to get
root# ll /dev/uinput
crw-rw---- 1 root uinput 10, 223 Dec 31 17:04 /dev/uinput
then I tried to start your driver:
root# systemctl restart --user asus_numberpad_driver@flipo.service
Failed to connect to bus: No medium found

I append the error.log file (zipped) in case it helps.
[error.log.zip](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/13801687/error.log.zip)

ldrahnik, 2024-01-03:
As summary what has to be done in this issue:

- finish implementation of i2c lib `smbus2` whether is possible, so do not have to be called `i2ctransfer` tool from `sbin`
- `dev/uinput` udev rule does not work on debian

ldrahnik, 2024-01-03:
@dflipo Could you please look here? https://github.com/chrippa/ds4drv/issues/93#issuecomment-265300511

dflipo, 2024-01-03:
> @dflipo Could you please look here? https://github.com/chrippa/ds4drv/issues/93#issuecomment-265300511

I added file `/etc/modules-load.d/uinput.conf`, it seems to fix the udev rules permissions: after reboot I get
flipo% ll /dev/uinput 
crw-rw---- 1 root uinput 10, 223  3 janv. 16:34 /dev/uinput

Unfortunately installation through [https://github.com/asus-linux-drivers/asus-numberpad-driver/tree/replaced-i2c-transfer-tool-to-usage-of-python-lib](url) still fails for `./install_device_check.sh`: 
ligne 5: i2cdetect : command not found
No i2c interface can be found. Make sure you have installed libevdev packages

If I add PATH="${PATH}:/usr/sbin" in line4, the installation completes successfully but with an error in error.log:
2024-01-03 16:34:18,011 INFO     Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
2024-01-03 16:34:18,011 INFO     Set keyboard 0 from H: Handlers=sysrq kbd leds event0
2024-01-03 16:34:18,036 INFO     Detecting touchpad from string: "N: Name="ASUE140D:00 04F3:31B9 Touchpad""
2024-01-03 16:34:18,036 INFO     Set touchpad device id 1 from S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-ASUE140D:00/0018:04F3:31B9.0001/input/input24
2024-01-03 16:34:18,036 INFO     Set touchpad id 12 from H: Handlers=mouse1 event12
2024-01-03 16:34:18,036 ERROR    Can't open the I2C bus connection (id: 1)

ldrahnik, 2024-01-06:
> Unfortunately installation through [https://github.com/asus-linux-drivers/asus-numberpad-driver/tree/replaced-i2c-transfer-tool-to-usage-of-python-lib](https://github.com/asus-linux-drivers/asus-numberpad-driver/issues/url) still fails for ./install_device_check.sh:
ligne 5: i2cdetect : command not found
No i2c interface can be found. Make sure you have installed libevdev packages

Commit https://github.com/asus-linux-drivers/asus-numberpad-driver/commit/ca4ef6cb31bc406cbc4754f50eee182c02abc516

ldrahnik, 2024-01-06:
> I added file /etc/modules-load.d/uinput.conf, it seems to fix the udev rules permissions: after reboot I get
flipo% ll /dev/uinput
crw-rw---- 1 root uinput 10, 223 3 janv. 16:34 /dev/uinput

Commit https://github.com/asus-linux-drivers/asus-numberpad-driver/commit/bf9df36a4a63d846917e2c1993fc18ba051384d9

ldrahnik, 2024-01-06:
> Adding /etc/udev/rules.d/99-input.rules doesn't help on Debian, the rules are in /usr/lib/udev/rules.d/

Commit (now it is consistent with udev rule for external keyboard, was already installed to `/usr/lib/udev/rules.d/`) https://github.com/asus-linux-drivers/asus-numberpad-driver/commit/9ed83f7180db49ad8b0a939e76f6bd347ce444b1

ldrahnik, 2024-01-06:
> I have noticed in other rules on Debian that the MODE is defined with MODE="0660" instead of MODE:="0660"

Commit https://github.com/asus-linux-drivers/asus-numberpad-driver/commit/cc2e39c4ce780298691723ca58e4f86701fb9256

ldrahnik, 2024-01-06:
>2024-01-03 16:34:18,036 ERROR Can't open the I2C bus connection (id: 1)

Commit https://github.com/asus-linux-drivers/asus-numberpad-driver/commit/089fe6a8f7ebb3f959c0321bce5cb55eea3db2f6

ldrahnik, 2024-01-06:
@dflipo ping

ldrahnik, 2024-01-07:
@dflipo Merged, can be reopened

closed 2024-01-07 by ldrahnik

dflipo, 2024-01-07:
Thanks!!! version 4.2.0 works perfectly for me.