asus-linux-drivers/asus-numberpad-driver #155: [Bug]: Permissions problem

Sporknife, 2024-01-07:
### Describe the bug

The python script is ran with unsufficient permissions. If I run the scipt with sudo, it runs without issues.

### Expected behavior

It should run the script with sufficient permissions.

### Relevant log output

```shell
2024-01-07 22:43:18,384 INFO     Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
2024-01-07 22:43:18,384 INFO     Set keyboard 0 from H: Handlers=sysrq kbd leds event0
2024-01-07 22:43:18,410 INFO     Detecting touchpad from string: "N: Name="ASUE140D:00 04F3:31B9 Touchpad""
2024-01-07 22:43:18,411 INFO     Set touchpad device id 1 from S: Sysfs=/devices/platform/AMDI0010:03/i2c-1/i2c-ASUE140D:00/0018:04F3:31B9.0002/input/input14
2024-01-07 22:43:18,411 INFO     Set touchpad id 9 from H: Handlers=mouse3 event9
2024-01-07 22:43:18,411 ERROR    Can't open the I2C bus connection (id: 1)
2024-01-07 22:43:18,789 INFO     Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
2024-01-07 22:43:18,789 INFO     Set keyboard 0 from H: Handlers=sysrq kbd leds event0
2024-01-07 22:43:18,814 INFO     Detecting touchpad from string: "N: Name="ASUE140D:00 04F3:31B9 Touchpad""
2024-01-07 22:43:18,814 INFO     Set touchpad device id 1 from S: Sysfs=/devices/platform/AMDI0010:03/i2c-1/i2c-ASUE140D:00/0018:04F3:31B9.0002/input/input14
2024-01-07 22:43:18,815 INFO     Set touchpad id 9 from H: Handlers=mouse3 event9
2024-01-07 22:43:18,815 ERROR    Can't open the I2C bus connection (id: 1)
2024-01-07 22:43:19,290 INFO     Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
2024-01-07 22:43:19,290 INFO     Set keyboard 0 from H: Handlers=sysrq kbd leds event0
2024-01-07 22:43:19,314 INFO     Detecting touchpad from string: "N: Name="ASUE140D:00 04F3:31B9 Touchpad""
2024-01-07 22:43:19,314 INFO     Set touchpad device id 1 from S: Sysfs=/devices/platform/AMDI0010:03/i2c-1/i2c-ASUE140D:00/0018:04F3:31B9.0002/input/input14
2024-01-07 22:43:19,314 INFO     Set touchpad id 9 from H: Handlers=mouse3 event9
2024-01-07 22:43:19,315 ERROR    Can't open the I2C bus connection (id: 1)
2024-01-07 22:43:19,788 INFO     Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
2024-01-07 22:43:19,788 INFO     Set keyboard 0 from H: Handlers=sysrq kbd leds event0
2024-01-07 22:43:19,817 INFO     Detecting touchpad from string: "N: Name="ASUE140D:00 04F3:31B9 Touchpad""
2024-01-07 22:43:19,817 INFO     Set touchpad device id 1 from S: Sysfs=/devices/platform/AMDI0010:03/i2c-1/i2c-ASUE140D:00/0018:04F3:31B9.0002/input/input14
2024-01-07 22:43:19,817 INFO     Set touchpad id 9 from H: Handlers=mouse3 event9
2024-01-07 22:43:19,817 ERROR    Can't open the I2C bus connection (id: 1)
2024-01-07 22:43:20,118 INFO     Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
2024-01-07 22:43:20,118 INFO     Set keyboard 0 from H: Handlers=sysrq kbd leds event0
2024-01-07 22:43:20,157 INFO     Detecting touchpad from string: "N: Name="ASUE140D:00 04F3:31B9 Touchpad""
2024-01-07 22:43:20,157 INFO     Set touchpad device id 1 from S: Sysfs=/devices/platform/AMDI0010:03/i2c-1/i2c-ASUE140D:00/0018:04F3:31B9.0002/input/input14
2024-01-07 22:43:20,157 INFO     Set touchpad id 9 from H: Handlers=mouse3 event9
2024-01-07 22:43:20,157 ERROR    Can't open the I2C bus connection (id: 1)
```

### Desktop

- Program version [master]
Operating System: openSUSE Tumbleweed 20240106
KDE Plasma Version: 5.27.10
KDE Frameworks Version: 5.113.0
Qt Version: 5.15.12
Kernel Version: 6.6.9-1-default (64-bit)
Graphics Platform: Wayland
Processors: 16 × AMD Ryzen 7 5800H with Radeon Graphics
Memory: 15.0 GiB of RAM
Graphics Processor: AMD Radeon Graphics
Manufacturer: ASUSTeK COMPUTER INC.
Product Name: Zenbook UN5401QAB_UN5401QA
System Version: 1.0

ldrahnik, 2024-01-07:
@Sporknife Does help installing package `i2c-tools`? Reinstall the latest master and let me know please, thank you.

Sporknife, 2024-01-07:
I still get the error pasted below. It probably doesn't have the necessary permissions. 
```
2024-01-07 23:09:53,924 ERROR    Can't open the I2C bus connection (id: 1)
```

ldrahnik, 2024-01-07:
@Sporknife Please, try to reinstall again latest master branch, reboot. Because without that package, I had the same error.

Sporknife, 2024-01-07:
It shall be done.

Sporknife, 2024-01-07:
It shall be done.

ldrahnik, 2024-01-07:
Fixed? In case not what is your output of command `$ groups`? And what is the output of command `$ ls -ll /dev/i2c-1`?

ldrahnik, 2024-01-08:
Can be reopened.

closed 2024-01-08 by ldrahnik

Sporknife, 2024-01-08:
Still doesn't work.
```
2024-01-08 11:11:24,378 INFO     Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
2024-01-08 11:11:24,379 INFO     Set keyboard 0 from H: Handlers=sysrq kbd leds event0
2024-01-08 11:11:24,413 INFO     Detecting touchpad from string: "N: Name="ASUE140D:00 04F3:31B9 Touchpad""
2024-01-08 11:11:24,414 INFO     Set touchpad device id 1 from S: Sysfs=/devices/platform/AMDI0010:03/i2c-1/i2c-ASUE140D:00/0018:04F3:31B9.0002/input/input15
2024-01-08 11:11:24,414 INFO     Set touchpad id 2 from H: Handlers=mouse1 event2
2024-01-08 11:11:24,414 ERROR    Can't open the I2C bus connection (id: 1)
```

ldrahnik, 2024-01-08:
I asked a question. Without that, I can do nothing and will not be reopened.

Sporknife, 2024-01-08:
crw------- 1 root root 89, 1 Jan  8 11:11 /dev/i2c-1

ldrahnik, 2024-01-08:
Also please attach the log from installing the latest master.

Sporknife, 2024-01-08:
[error.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/13859515/error.log)
[install-08-01-2024-11-08-57.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/13859517/install-08-01-2024-11-08-57.log)

ldrahnik, 2024-01-08:
I am sorry your distro is not supported at this moment and I can not provide support for your distro but pull request to code from your side is welcome :) What I see in log and have to be fixed in case you decide improve the code:

- Your distro does not use `gsettings` for setting custom shortcuts? `No such schema “org.gnome.settings-daemon.plugins.media-keys”`?
- In case default calculator is installed may be added to this driver too `Not detected any supported calculator (gnome-calculator && io.elementary.calculator)`
- What is command for reboot in your distro?  `line 134: reboot: command not found`
- Does exist tool `i2cdetect` in `/usr/sbin` folder? and `/usr/sbin` is only not in `$PATH` or it is not installed? `The i2cdetect tool not found to proceed initial test whether any i2c device react like NumberPad` But it can cause also missing package `i2c-tools`, not only missing folder `sbin` in `$PATH`

- Package manager is not identify so I do not see what packages you have already installed and which ones you not installed (`Not detected package manager. Driver may not work properly because required packages have not been installed. Please create an issue (https://github.com/asus-linux-drivers/asus-numberpad-driver/issues).`)

Sporknife, 2024-01-08:
- gsettings doesn't exist because I am not on gnome.
- i just selected 'yes' here because I wasnt paying attention
- i2cdetect exists in /usr/sbin but it seems /usr/sbin wasn't in $PATH
- i am on opensuse

ldrahnik, 2024-01-08:
@Sporknife Please try the latest master, I have added the udev rule which should on boot change group `root` of `/dev/i2c-xy` to `i2c`

ldrahnik, 2024-01-08:
@Sporknife 

> What is the command for reboot in your distro? line 134: reboot: command not found

Does work?

```
$ sudo /sbin/shutdown -r now
```

Sporknife, 2024-01-08:
@ldrahnik Use systemctl restart instead.

ldrahnik, 2024-01-08:
> Please try the latest master, I have added the udev rule which should on boot change group `root` of `/dev/i2c-xy` to `i2c`

@Sporknife Solved issue?

Sporknife, 2024-01-08:
Works now.