asus-linux-drivers/asus-numberpad-driver #156: [Bug]: Doesn't install properly on Arch due to non-existent package

soleofthesea, 2024-01-09:
### Describe the bug

The script attempts to install _python3-smbus2_ which does not exist in the Arch repos, instead it is found in the AUR (under _python-smbus_).

Also, despite `pacman` returning an error the script continues.

### Expected behavior

Either the script auto-installs the package or aborts with an error message.

### Desktop

- Program version [master]
- Distro [e.g. Arch Linux]
- Kernel [6.6.9]

ldrahnik, 2024-01-09:
@soleofthesea Thank you for creating the issue. It is the wrong dependency. But is not the equivalent package `[python-smbus2](https://aur.archlinux.org/packages/python-smbus2)`?

ldrahnik, 2024-01-09:
Can be reopened

dvdoll, 2024-01-09:
trying to run `install.sh` on my zenbook ux3404 q420v on arch gives the same error aforementioned. installing `python-smbus2` manually and running the script again, results in the error once again.

ldrahnik, 2024-01-09:
Please post here output of install attempt. Package des not exist `python-smbus2`?

soleofthesea, 2024-01-09:
It's because `pacman` doesn't look at the AUR so as far as it's concerned `python-smbus2` still doesn't exist.

Cleanest solution would probably be to build straight from source, something like
`pacman -S git
git clone https://github.com/kplindegaard/smbus2
cd smbus2
python - pip install .`

dvdoll, 2024-01-09:
the error im referring to is the following
```
error: target not found: python-smbus2
Something went wrong when installing packages via package manager
```
also, i've tried building from source like @soleofthesea mentioned but it seems to bounce back an invalid syntax error.

soleofthesea, 2024-01-09:
Might also need `python-setuptools`.

ldrahnik, 2024-01-09:
The master branch has been hard reset.

dvdoll, 2024-01-09:
the script seems to still give the error, but ignores it and continues on with the script. the script preforms normally and installs, but doesn't seem to work for me. (no backlighting & no numpad functions). might just be an issue on my end though, someone else should give it a try.

dvdoll, 2024-01-09:
```
>>> pip install .
  File "<stdin>", line 1
    pip install .
        ^^^^^^^
SyntaxError: invalid syntax
>>> 
```

soleofthesea, 2024-01-09:
*Ahem*

So about that syntax error...apologies, typo.
Correct command should be `python -m pip install .`

dvdoll, 2024-01-09:
no worries. it happens. after running that command, it said pip wasn't found, so i ran `sudo pacman -S python-pip`, ran `python -m pip install .` and received the following.
```
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try 'pacman -S
    python-xyz', where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Arch-packaged Python package,
    create a virtual environment using 'python -m venv path/to/venv'.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip.
    
    If you wish to install a non-Arch packaged Python application,
    it may be easiest to use 'pipx install xyz', which will manage a
    virtual environment for you. Make sure you have python-pipx
    installed via pacman.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
```

ldrahnik, 2024-01-09:
Yes, because od this `pip` Is not the way how solve this problem.  What about to use previous package?

dvdoll, 2024-01-09:
the only packages i can find are `python-smbus2` and `python-smbus-git` which are both aur packages.

soleofthesea, 2024-01-09:
What @dvdoll said. https://archlinux.org/packages/?sort=&q=smbus&maintainer=&flagged=

As I understand it `error: externally-managed-environment` is only of concern if there is an equivalent package in Arch's repos, because a `pacman -Syu` might override `pip` and vice versa. So while a bit alarmist it should be safe to ignore it for this instance... at least until a proper substitute can be found, anyway.

ldrahnik, 2024-01-09:
@soleofthesea Could you please post here exactly the same commands what I can use? Or do not you want make PR? With `pip` solution.

soleofthesea, 2024-01-09:
Well, about that... it might be OK to ignore the error for now just to get the driver to run but you're not really supposed to do that.
(This is recent change in Linux so I forgot, so please ignore my earlier comment about it being a "solution")

ldrahnik, 2024-01-10:
@dvdoll Could you please test whether [this](https://linux-packages.com/aur/package/python-smbus2) helps. Please try first way via `yay`, try driver then uninstall via `yay -Rns python-smbus2`, and please try even the second option via `$ makepkg`

ldrahnik, 2024-01-10:
@dvdoll @soleofthesea Please, test the [branch](https://github.com/asus-linux-drivers/asus-numberpad-driver/tree/python-smbus2-arch-fix)

ldrahnik, 2024-01-10:
@soleofthesea @dvdoll How I can check whether `$ yay` is installed? Is it possible like this `$ yay 2>/dev/null`? Then `$ echo $?` returns `0` in the case is installed and `1` when is not?

ldrahnik, 2024-01-10:
> Probably

Probably? Could you please open the terminal and try that? Thank you

ldrahnik, 2024-01-10:
> I think it might be simpler to just create a Python venv at this point.

Pull request is welcome

soleofthesea, 2024-01-10:
> Probably? Could you please open the terminal and try that? Thank you

If the script is going to manually build from AUR via `makepkg -si` then writing additional code to use `yay` is redundant. Like buying instant coffee when you already have a coffee maker.

ldrahnik, 2024-01-10:
> If the script is going to manually build from AUR via makepkg -si then writing additional code to use yay is redundant. Like buying instant coffee when you already have a coffee maker.

Did you check the code in mentioned branch?

soleofthesea, 2024-01-10:
Alright, fair point. But there should be handling for `paru` and `pamac` too; those are pretty popular helpers as well.
Also I'm currently on mobile, so I'll respond later once I'm able to test the code.

ldrahnik, 2024-01-10:
I have to agree with you, it looks like there is no obstacle why not use Python3 virtual environment. Init for your test is pushed to this [branch](https://github.com/asus-linux-drivers/asus-numberpad-driver/pull/158).

My insights:

- `apt` package `python3-virtualenv` have dependency `python3-pip` & `python3-setuptools`
- `apt` package `python3-virtualenv` has advantage over `python3.10-env` because has no specific version in the name
- Because this driver currently depends on the latest Python3, a virtual environment has to be created like this:

```
$ virtualenv --python=$(python3 --version | cut -d" " -f2) /usr/share/asus-numberpad-driver/.env
```

ldrahnik, 2024-01-11:
@soleofthesea @dvdoll Merged to the master. Can be reopened.

closed 2024-01-11 by ldrahnik