asus-linux-drivers/asus-numberpad-driver #159: [Bug]: Tapping on button sends mouse click, and tapping % types "25 "

dvdoll, 2024-01-12:
### Describe the bug

When enabling the numpad and tapping on any of the displayed buttons, not only is its input sent, but so is a left click. Furthermore, when tapping on the displayed "%" button, an input of `25 ` is sent.

### Expected behavior

Tap-to-click automatically disable, tapping % inputs %.

### Desktop

- Program version [master]
- Distro [Arch Linux, KDE DE]
- Kernel [6.6.10-arch1-1]

ldrahnik, 2024-01-12:
@dvdoll Thank you a lot for reporting this. Tap-to-click should be disabled. Please, can you attach the driver's error log file located here `/var/log/asus-numberpad-driver`?

ldrahnik, 2024-01-12:
@dvdoll Do you use Wayland? (`$ echo $XDG_SESSION_TYPE`) Which keyboard layout do you use? (e.g. CZ, FR, EN, ..) Which NumberPad layout do you use, unicode or non unicode or standard?

```
Select layout:

1) b7402-non-unicode.py		   8) g513-unicode.py		    15) gx701.py		      22) up5401ea.py			29) ux581l.py
2) b7402.py			   9) g533-non-unicode.py	    16) gx701-unicode.py	      23) up5401ea-unicode-3145.py	30) ux581l-unicode.py
3) b7402-unicode.py		  10) g533.py			    17) up5401ea-3145.py	      24) up5401ea-unicode-31B9.py	31) Quit
4) e210ma-non-unicode.py	  11) g533-unicode.py		    18) up5401ea-31B9.py	      25) up5401ea-unicode.py
5) e210ma.py			  12) gx551-non-unicode.py	    19) up5401ea-non-unicode-3145.py  26) ux433fa-non-unicode.py
6) e210ma-unicode.py		  13) gx551-unicode.py		    20) up5401ea-non-unicode-31B9.py  27) ux433fa-unicode.py
7) g513-non-unicode.py		  14) gx701-non-unicode.py	    21) up5401ea-non-unicode.py	      28) ux581l-non-unicode.py
```

dvdoll, 2024-01-12:
I use X11, not wayland. I use an EN keyboard layout. I believe I'm only using the `up5401ea` layout.

Error log ;
[error.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/13919613/error.log)

ldrahnik, 2024-01-12:
@dvdoll Thank you, I see the problem:

```
2024-01-11 18:27:32,916 INFO     Pressed numpad key
2024-01-11 18:27:32,916 INFO     %
2024-01-11 18:27:32,944 ERROR    Error during trying to find key for modifier of found compose shortcut Control_L
2024-01-11 18:27:32,944 ERROR    Error during trying to find key for modifier of found compose shortcut Shift_L
2024-01-11 18:27:32,944 INFO     Unpressed numpad key
2024-01-11 18:27:32,944 INFO     %
```

ldrahnik, 2024-01-12:
@dvdoll What is output of this command? `$ gsettings get org.freedesktop.ibus.panel.emoji unicode-hotkey`

dvdoll, 2024-01-12:
```
$ gsettings get org.freedesktop.ibus.panel.emoji unicode-hotkey
['<Control><Shift>u']
```

ldrahnik, 2024-01-12:
Please, try [this](https://github.com/asus-linux-drivers/asus-numberpad-driver/tree/159-test-branch) branch and post here error log

dvdoll, 2024-01-12:
[error.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/13920592/error.log)

ldrahnik, 2024-01-12:
```
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.11/site-packages/Xlib/display.py", line 71, in __init__
    protocol_display.Display.__init__(self, *args, **keys)
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.11/site-packages/Xlib/protocol/display.py", line 166, in __init__
    raise error.DisplayConnectionError(self.display_name, r.reason)
Xlib.error.DisplayConnectionError: Can't connect to display ":0": b'Authorization required, but no authorization protocol specified\n'
```

What is an output of the command `$DISPLAY`?

ldrahnik, 2024-01-12:
Do you use external monitor?

dvdoll, 2024-01-12:
No, just the laptop monitor. 

```
$ echo $DISPLAY
:0
```

ldrahnik, 2024-01-12:
Does help temporary workaround:

```
$ xhost +
```

mentioned [here](https://stackoverflow.com/questions/31902846/how-to-fix-error-xlib-error-displayconnectionerror-cant-connect-to-display-0) ?

dvdoll, 2024-01-12:
```
$ xhost +
bash: xhost: command not found
$ xhost +local:$USER
bash: xhost: command not found
```
Also just to mention, using the branch you sent earlier, [this one](https://github.com/asus-linux-drivers/asus-numberpad-driver/tree/159-test-branch), the numberpad turns off and "crashes" when I tap on the "%". The tap to click is still enabled as well.

ldrahnik, 2024-01-12:
> Also just to mention, using the branch you sent earlier, [this one](https://github.com/asus-linux-drivers/asus-numberpad-driver/tree/159-test-branch?rgh-link-date=2024-01-12T15%3A33%3A06Z), the numberpad turns off and "crashes" when I tap on the "%".

In newly created branch was crash expected, Sorry, I have no idea what is wrong. I wouldrecommend to you try driver on clean installation if that happens too.

> The tap to click is still enabled as well.

What is and output of the command:

```
$ gsettings get org.gnome.desktop.peripherals.touchpad tap-to-click
true
```

# are you able disable tap-to-click?
$ gsettings set org.gnome.desktop.peripherals.touchpad tap-to-click false
# and reneable
$ gsettings set org.gnome.desktop.peripherals.touchpad tap-to-click true
```

dvdoll, 2024-01-12:
```
$ gsettings get org.gnome.desktop.peripherals.touchpad tap-to-click
true
$ gsettings set org.gnome.desktop.peripherals.touchpad tap-to-click false
$ gsettings set org.gnome.desktop.peripherals.touchpad tap-to-click true
```
It looks like it can get, but it can not set, I get no output when trying to set.

ldrahnik, 2024-01-12:
Did you tried tap on Touchpad when was disabled like this?

```
$ gsettings set org.gnome.desktop.peripherals.touchpad tap-to-click false
```

dvdoll, 2024-01-12:
Oh yes, I see. It will set it properly, but it doesnt actually apply the changes. Even when it is set to `false`, I am still able to tap to click.

```
$ gsettings get org.gnome.desktop.peripherals.touchpad tap-to-click
true
$ gsettings set org.gnome.desktop.peripherals.touchpad tap-to-click false
$ gsettings get org.gnome.desktop.peripherals.touchpad tap-to-click
false
$ gsettings set org.gnome.desktop.peripherals.touchpad tap-to-click true
$ gsettings get org.gnome.desktop.peripherals.touchpad tap-to-click
true
```

ldrahnik, 2024-01-12:
Sorry, my bad it does not work because you use `KDE` and these settings work only for `gnome`. I missed that information on the top. Please, try the latest master, disabling tap-to-click should work.

dvdoll, 2024-01-12:
Installed latest master, both issues still persist. Tap to click is still enabled, and % inputs `25 `

dvdoll, 2024-01-12:
Using that works, I'm able to enable and disable tapping, and it applies properly.

ldrahnik, 2024-01-12:
Great. Weird it does not work in latest master. Please reinstall the driver and post here the error log again please, would be best to clean the error log first, starting to be a little messy.

dvdoll, 2024-01-12:
[error.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/13922114/error.log)

ldrahnik, 2024-01-12:
Try to remove `xauth` file and then `reboot`, will be autoconfigured in boot, do not worry, it should help I hope. What I did when I had this problem:

```
$ xauth
Using authority file /home/ldrahnik/.Xauthority
xauth>^C
$ rm /home/ldrahnik/.Xauthority
$ reboot
```

dvdoll, 2024-01-12:
Okay, I did what you said and it's still behaving the same. Interestingly enough, the xauth file is a temporary file, for me.

```
$ xauth
Using authority file /tmp/xauth_UeMFCG
xauth> 
```

ldrahnik, 2024-01-12:
I am closing this issue as the issue is solved on the driver side and the problem that is behind both of your mentioned erorrs is now not related to the drivers code. What I would do be you? I would thread [this thread](https://forums.opensuse.org/t/authorization-required-but-no-authorization-protocol-specified/167149/17), in summary I would try also create the new user, check whether installing NumberPad throws the same problem and then try `$ xhost +si:localuser:$USER &`

closed 2024-01-12 by ldrahnik

ldrahnik, 2024-01-14:
@dvdoll Solved? Create a new discussion if not [here](https://github.com/orgs/asus-linux-drivers/discussions)