asus-linux-drivers/asus-numberpad-driver #160: [Bug]: fails for ASUS Vivobook 17 inch laptop

AubreyJaffer, 2024-01-12:
### Describe the bug

The log output is appended.  It says it didn't work.

### Expected behavior

I expected that numeric keypad would start working.

### Relevant log output

```shell
Added group numberpad to current user

Reading package lists...
Building dependency tree...
Reading state information...
ibus is already the newest version (1.5.26-4).
libevdev2 is already the newest version (1.12.1+dfsg-1).
python3-pyinotify is already the newest version (0.9.6-1.3).
xinput is already the newest version (1.6.3-1build2).
i2c-tools is already the newest version (4.3-2build1).
python3-libevdev is already the newest version (0.5-3).
python3-xlib is already the newest version (0.29-1).
curl is already the newest version (7.81.0-1ubuntu1.15).
libxml2-utils is already the newest version (2.9.13+dfsg-1ubuntu0.3).
python3-dev is already the newest version (3.10.6-1~22.04).
python3-numpy is already the newest version (1:1.21.5-1ubuntu22.04.1).
The following package was automatically installed and is no longer required:
  libsdl2-image-2.0-0
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.

i2c-dev module loaded
i2c-dev module added to auto loaded modules

Testing interface i2c-1: failed
Testing interface i2c-2: success
The detection was successful. Touchpad with numberpad found: i2c-2

groupadd: group 'input' already exists
groupadd: group 'i2c' already exists
groupadd: group 'uinput' already exists
Added groups input, i2c, uinput, numberpad to current user

Default config will be autocreated during the first run and available for futher modifications here:
/usr/share/asus-numberpad-driver/numberpad_dev

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100 55280    0 55280    0     0  78534      0 --:--:-- --:--:-- --:--:-- 78634
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100 84376    0 84376    0     0   112k      0 --:--:-- --:--:-- --:--:--  112k
Could not automatically detect numberpad layout for your laptop. Please create an issue (https://github.com/asus-linux-drivers/asus-numberpad-driver/issues).

Is recommended layout wrong please? Please create an issue (https://github.com/asus-linux-drivers/asus-numberpad-driver/issues).

NumberPad layout

Are predefined 3 variants of numberpad layouts for each laptop:
 - Non unicode variant does not send any character via shortcut Ctrl+Shift+U. Uses numeric keys and key combinations (Shift + number) for percent and hashtag characters. Because of this is vulnerable to overbindings (custom overbindings any key which is used in layout or is enough change to other language keyboard layout e.g. French). Recommended when you use only lang layouts where can be percent, hashtag char printed via Shift+5/3 and you do not have overbinded any key used in layout (manual change overbinded key directly in layout can anyway make this layout usable)
 - Standard (without unicode and non-unicode postfix). The keys except percent and hashtag characters (these 2 uses unicode Ctrl+Shift+U shortcut) are send directly so is not vulnerable when is changed lang keyboard layout (to e.g. French) but layout at all is still not resistant to overbinding other keys to something else and that is reason why exist last variant
 - Last unicode variant send keys like unicode chars except backspace and enter using shortcut Ctrl+Shift+U+<sequence of number keys>. Is resistant to overbinding any key or lang layout change (to e.g. French) but sending multiple keys instead of one, max. 2 can be overkill if you do not need it.

Select layout:

1) b7402-non-unicode.py		   9) g533-non-unicode.py	    17) up5401ea-3145.py	      25) up5401ea-unicode.py
2) b7402.py			  10) g533.py			    18) up5401ea-31B9.py	      26) ux433fa-non-unicode.py
3) b7402-unicode.py		  11) g533-unicode.py		    19) up5401ea-non-unicode-3145.py  27) ux433fa-unicode.py
4) e210ma-non-unicode.py	  12) gx551-non-unicode.py	    20) up5401ea-non-unicode-31B9.py  28) ux581l-non-unicode.py
5) e210ma.py			  13) gx551-unicode.py		    21) up5401ea-non-unicode.py	      29) ux581l.py
6) e210ma-unicode.py		  14) gx701-non-unicode.py	    22) up5401ea.py		      30) ux581l-unicode.py
7) g513-non-unicode.py		  15) gx701.py			    23) up5401ea-unicode-3145.py      31) Quit
8) g513-unicode.py		  16) gx701-unicode.py		    24) up5401ea-unicode-31B9.py
Please enter your choice
```

### Desktop

- Program version [e.g. 57c742a]
- Distro [e.g. Arch Linux]
- Kernel [e.g. 5.18.15.arch1-2]

ldrahnik, 2024-01-12:
Hello, thanks a lot for letting us know! Could you please specify your laptop model? For example by command:

```
$ sudo dmidecode -s system-product-name
```

Without that, I can not improve auto-detection for your laptop. Anyway, you can continue in the install script, did you try that? Just pick up the right layout recognized by eyes from the file [Readme.md](https://github.com/asus-linux-drivers/asus-numberpad-driver#layouts)

AubreyJaffer, 2024-01-12:
sudo dmidecode -s system-product-name
 Vivobook_ASUSLaptop X1703ZA_K1703ZA

gx551 looks like the keypad.  The NumLk key toggles the circle 9 thing
in the top screen pane.  but it did that before the install.

Neither gx551 mode works.  Is reboot really necessary after changing
the model?  I did reboot.

-event2   DEVICE_ADDED            Power Button
seat0 default group1  cap:k
-event6   DEVICE_ADDED            Video Bus
seat0 default group2  cap:k
-event0   DEVICE_ADDED            Lid Switch
seat0 default group3  cap:S
-event1   DEVICE_ADDED            Power Button
seat0 default group4  cap:k
-event7   DEVICE_ADDED            Logitech M325
seat0 default group5  cap:p left scroll-nat scroll-button
-event5   DEVICE_ADDED            USB2.0 HD UVC WebCam: USB2.0 HD
seat0 default group6  cap:k
-event21  DEVICE_ADDED            ASUE1305:00 04F3:3212 Mouse
seat0 default group7  cap:p left scroll-nat scroll-button
-event22  DEVICE_ADDED            ASUE1305:00 04F3:3212 Touchpad
seat0 default group7  cap:pg  size 129x83mm tap(dl off) left
scroll-nat scroll-2fg-edge click-buttonareas-clickfinger dwt-on
-event4   DEVICE_ADDED            Asus WMI hotkeys
seat0 default group8  cap:k
-event3   DEVICE_ADDED            AT Translated Set 2 keyboard
seat0 default group9  cap:k
-event23  DEVICE_ADDED            ASUE1305:00 04F3:3212 NumberPad
seat0 default group10 cap:kp left scroll-nat
-event3   KEYBOARD_KEY            +0.000s	*** (-1) pressed
 event3   KEYBOARD_KEY            +0.126s	*** (-1) released
 event3   KEYBOARD_KEY            +3.207s	*** (-1) pressed
 event3   KEYBOARD_KEY            +3.392s	*** (-1) released
 event3   KEYBOARD_KEY            +5.314s	*** (-1) pressed
 event3   KEYBOARD_KEY            +5.609s	*** (-1) released
 event3   KEYBOARD_KEY            +6.034s	*** (-1) pressed
 event3   KEYBOARD_KEY            +6.099s	*** (-1) pressed
 event3   KEYBOARD_KEY            +6.290s	*** (-1) released
 event3   KEYBOARD_KEY            +6.390s	*** (-1) released
 event3   KEYBOARD_KEY            +6.660s	*** (-1) pressed
 event3   KEYBOARD_KEY            +6.829s	*** (-1) pressed
 event3   KEYBOARD_KEY            +6.834s	*** (-1) pressed
 event3   KEYBOARD_KEY            +6.965s	*** (-1) released
 event3   KEYBOARD_KEY            +6.981s	*** (-1) released
 event3   KEYBOARD_KEY            +7.105s	*** (-1) released
 event3   KEYBOARD_KEY            +7.250s	*** (-1) pressed
 event3   KEYBOARD_KEY            +7.430s	*** (-1) pressed
 event3   KEYBOARD_KEY            +7.455s	*** (-1) pressed
 event3   KEYBOARD_KEY            +7.565s	*** (-1) released
 event3   KEYBOARD_KEY            +7.625s	*** (-1) released
 event3   KEYBOARD_KEY            +7.680s	*** (-1) released
 event3   KEYBOARD_KEY            +10.937s	*** (-1) pressed
 event3   KEYBOARD_KEY            +11.143s	*** (-1) released
 event3   KEYBOARD_KEY            +11.428s	*** (-1) pressed
 event3   KEYBOARD_KEY            +11.598s	*** (-1) released
 event3   KEYBOARD_KEY            +11.899s	KEY_KPENTER (96) pressed

 event3   KEYBOARD_KEY            +12.069s	KEY_KPENTER (96) released

ldrahnik, 2024-01-12:
Are you sure your laptop does have NumberPad inside Touchpad? Can you send me link with your product where it is obvious?

ldrahnik, 2024-01-12:
@AubreyJaffer Because I believe your laptop does not have NumberPad - https://www.asus.com/supportonly/x1703za/helpdesk_download/

ldrahnik, 2024-01-12:
@AubreyJaffer 

> The NumLk key toggles the circle 9 thing in the top screen pane.  but it did that before the install.

What do you mean by this?

AubreyJaffer, 2024-01-12:
Attached are a photo of the keyboard, and the events log of pressing each
key on the keypad, scanning left to right, starting at top left.  The keys
pressed were:
NumLk / * -
7 8 9 +
4 5 6 +
1 2 3 Enter
0 . Enter

ldrahnik, 2024-01-12:
@AubreyJaffer Please attach that photo of NumberPad on Github

ldrahnik, 2024-01-12:
Closing as expectation is to fix numeric keypad but driver is about Touchpad's NumberPad.

closed 2024-01-12 by ldrahnik

ldrahnik, 2024-01-14:
@AubreyJaffer You can open discussion [here](https://github.com/orgs/asus-linux-drivers/discussions) and I will try to help you.