asus-linux-drivers/asus-numberpad-driver #161: [Layout Request] G614

joezhouchenye, 2024-01-19:
### Layout Name

G614

### Your device (sudo dmidecode -s system-product-name)

ROG Strix G16 G614JVR_G614JVR

### Numpad Layout (photo/picture or code)

[image]

joezhouchenye, 2024-01-19:
Using g533 works but the light is off.

joezhouchenye, 2024-01-19:
Yes. It seems this part should be changed.

[image]

[image]

ldrahnik, 2024-01-19:
@joezhouchenye Design of laptop is IMHO amazing 🙂 Could you please try whether these commands light up device?
Only the previous last value is changed - 0x01 and 0x60. Device id has to be touchpad i2c device id: $ cat /proc/bus/input/devices. Please attach here entire `/proc/bus/input/devices` and also `$ sudo libinput debug-events`.

 
$ i2ctransfer -f -y <device_id> w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x60 0xad
$ i2ctransfer -f -y <device_id> w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad

ldrahnik, 2024-01-19:
@joezhouchenye `ASUF` device is not supported now. Looks like a new product line. I will add it now :)

ldrahnik, 2024-01-19:
@joezhouchenye The reason why was not found `grep TouchPad` is because grep is case sensitive and the name is with small letter `p` I guess? Can you confirm my assumption?

joezhouchenye, 2024-01-19:
Yes. It should be `grep Touchpad`.

joezhouchenye, 2024-01-19:
It still doesn't work. I get `Remote I/O error` by directly executing the `test_birghtness.py`.

[image]

joezhouchenye, 2024-01-19:
```
I: Bus=0019 Vendor=0000 Product=0005 Version=0000
N: Name="Lid Switch"
P: Phys=PNP0C0D/button/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:08/PNP0C0D:01/input/input0
U: Uniq=
H: Handlers=event0 
B: PROP=0
B: EV=21
B: SW=1

I: Bus=0019 Vendor=0000 Product=0003 Version=0000
N: Name="Sleep Button"
P: Phys=PNP0C0E/button/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0E:00/input/input1
U: Uniq=
H: Handlers=kbd event1 
B: PROP=0
B: EV=3
B: KEY=4000 0 0

I: Bus=0019 Vendor=0000 Product=0001 Version=0000
N: Name="Power Button"
P: Phys=PNP0C0C/button/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0C:00/input/input2
U: Uniq=
H: Handlers=kbd event2 
B: PROP=0
B: EV=3
B: KEY=10000000000000 0

I: Bus=0011 Vendor=0001 Product=0001 Version=abba
N: Name="AT Translated Set 2 keyboard"
P: Phys=isa0060/serio0/input0
S: Sysfs=/devices/platform/i8042/serio0/input/input3
U: Uniq=
H: Handlers=sysrq kbd event3 leds 
B: PROP=0
B: EV=120013
B: KEY=402000000 3803078f800d001 feffffdfffefffff fffffffffffffffe
B: MSC=10
B: LED=7

I: Bus=0019 Vendor=0000 Product=0000 Version=0000
N: Name="Asus WMI hotkeys"
P: Phys=asus-nb-wmi/input0
S: Sysfs=/devices/platform/asus-nb-wmi/input/input9
U: Uniq=
H: Handlers=rfkill kbd event4 
B: PROP=0
B: EV=100013
B: KEY=400000000000000 1000000080000 400000 0 0 0 81606f00900c00 c280027801701000 e000000000000 0
B: MSC=10

I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="HDA NVidia HDMI/DP,pcm=3"
P: Phys=ALSA
S: Sysfs=/devices/pci0000:00/0000:00:01.0/0000:01:00.1/sound/card0/input10
U: Uniq=
H: Handlers=event5 
B: PROP=0
B: EV=21
B: SW=140

I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="HDA NVidia HDMI/DP,pcm=7"
P: Phys=ALSA
S: Sysfs=/devices/pci0000:00/0000:00:01.0/0000:01:00.1/sound/card0/input11
U: Uniq=
H: Handlers=event6 
[... 190 lines cut ...]
I: Bus=0003 Vendor=0b05 Product=19b6 Version=0110
N: Name="Asus Keyboard"
P: Phys=usb-0000:00:14.0-3/input0
S: Sysfs=/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/0003:0B05:19B6.0004/input/input38
U: Uniq=
H: Handlers=sysrq rfkill kbd mouse2 event19 leds 
B: PROP=0
B: EV=12001f
B: KEY=33eff 0 0 483ffff17aff32d bfd4444600000000 1f0001 1b30c730b17cc07 ffbf7bfad971dfff febeffdfffefffff fffffffffffffffe
B: REL=1943
B: ABS=10100000000
B: MSC=10
B: LED=1f

I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="ASUF1416:00 2808:0108 NumberPad"
P: Phys=
S: Sysfs=/devices/virtual/input/input39
U: Uniq=
H: Handlers=kbd event25 
B: PROP=0
B: EV=3
B: KEY=70000 0 1000 5000fffa0 2814403f0444ffc

```

ldrahnik, 2024-01-19:
@joezhouchenye It picked the right one, Touchpad's one. I need to see your DSDT table:

```
$ sudo apt-get install iasl
$ cd <work-dir>
$ sudo cp /sys/firmware/acpi/tables/DSDT DSDT
$ sudo iasl -d DSDT
$ sudo gedit DSDT.dsl # post here entire
```

joezhouchenye, 2024-01-19:
[dsdt.txt](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/13989775/dsdt.txt)

ldrahnik, 2024-01-19:
Try these commands (I suppose touchpad i2c is still `0`):

```
$ i2ctransfer -f -y 0 w13@0x00 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x60 0xad
$ i2ctransfer -f -y 0 w13@0x00 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad
```

ldrahnik, 2024-01-19:
What is the output of the command?

```
$ i2cdetect -y 0 # I suppose touchpad device i2c id is still 0
```

ldrahnik, 2024-01-19:
Try these commands (I suppose touchpad i2c is still `0`):

$ i2ctransfer -f -y 0 w13@0x38 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x60 0xad
$ i2ctransfer -f -y 0 w13@0x38 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad

joezhouchenye, 2024-01-19:
OK. Thanks. I will check it tomorrow. I am already at home and can not see the state of the Numpad. The commands are successfully executed without any error.

joezhouchenye, 2024-01-20:
The commands work.

[image]

joezhouchenye, 2024-01-20:
I have downloaded the latest commit in the pull request. The Numberpad works well now. Thanks again.

ldrahnik, 2024-01-20:
Nice to hear. Merged to the master. Should not be lighted up even dot above NumLock spot top right corner? Did you try multiple brightness levels? Whether are supported.

closed 2024-01-20 by ldrahnik

joezhouchenye, 2024-01-20:
The entire top right corner can be touched to light up the number pad (The area shown in the figure below).

[image]

I didn't try multiple brightness levels.

When the number pad is lighted up, the launching calculator feature works.

ldrahnik, 2024-01-20:
> I didn't try multiple brightness levels.

Try to hold the top left icon with finger and standarly used levels for other laptops should rotate (be send via i2c, `0x41-0x48`)

ldrahnik, 2024-01-20:
> Should not be lighted up even dot above NumLock spot top right corner?

Because on the first picture you attached to the issue during creating is lighted up dot top right corner but is not lighted up in last picture on linux. I am right?

joezhouchenye, 2024-01-20:
> Try to hold the top left icon spot area (corner) with finger and standardly used brightness levels for other laptops should rotate after 1s (be send via i2c, `0x41-0x48`)

I don't see any difference in brightness.

> Because on the first picture you attached to the issue during creating is lighted up dot top right corner (on windows) but is not lighted up in last picture (on linux). I am right?

It's just the protection cover. The NUMLK can not be lighted up.

Compared with Windows, it has the following differences:

- I need to hold NUMLK for a longer time in Linux
- In Windows, the NUMLK can always be enabled. The light is off after 2 minutes, but as soon as you touch the touchpad again, it will be lit up immediately. However, I have to reenable it by holding at NUMLK in Linux.
- When NUMLK is enabled, I can not use the touchpad in Windows. It's a numpad only. (For this one, I prefer the Linux behavior which I can use the touchpad and numpad at the same time.)

ldrahnik, 2024-01-20:
@joezhouchenye 

> I need to hold NUMLK for a longer time in Linux

IMHO default 1s (value can be changed in the config file) is better to avoid activation by mistake. Also, you can activate NumberPad via the slide function to the center from the top right icon (NumLock key area) - Windows does not support it.

> In Windows, the NUMLK can always be enabled. The light is off after 2 minutes, but as soon as you touch the touchpad again, it will be lit up immediately. However, I have to reenable it by holding at NUMLK in Linux.

Agree, committed.

> When NUMLK is enabled, I can not use the touchpad in Windows. It's a numpad only. (For this one, I prefer the Linux behavior which I can use the touchpad and numpad at the same time.)

In this driver it is also configurable. Anyway, you are the first person who says is not possible to control Touchpad when is NumberPad activated on Windows, probably Asus changed that with this series. Is for that config value somewhere in MyAsus?

joezhouchenye, 2024-01-20:
I cannot find any configuration options in Windows. Only the option to disable or enable.

获取Outlook for Android<https://aka.ms/AAb9ysg>