asus-linux-drivers/asus-numberpad-driver #164: [Bug]: Numpad keys (top left and top right) not functional on Zenbook 14 UX3405MA_UX3405MA

PellattLlo, 2024-02-08:
### Describe the bug

Top left and top right buttons do not produce the desired effects on ubuntu, on wayland nor x11 desktop. Top right button produces error log output: 

`2024-02-08 16:08:20,821 INFO     Touched top_right_icon (numlock) in time: 1707408500.8219445`
`2024-02-08 16:08:21,098 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1707408501.0987236`

Top left button does not produce any log output.

Numpad can be activated by pressing numlock on an external keyboard.

### Expected behavior

Top left button should change keyboard backlight brightness, top right button should activate / deactivate numpad.

### Relevant log output

```shell
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ASUP1415:00 093A:300C Touchpad
2024-02-08 16:00:35,494 INFO     Setting up for config file key: "enabled" with value: "True"
2024-02-08 16:00:35,495 INFO     check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/numberpad_dev wd=1 >
2024-02-08 16:00:35,595 INFO     Numpad activated
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ASUP1415:00 093A:300C Touchpad
2024-02-08 16:00:36,135 INFO     Setting up for config file key: "enabled" with value: "False"
2024-02-08 16:00:36,136 INFO     check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/numberpad_dev wd=1 >
2024-02-08 16:00:36,236 INFO     Numpad deactivated
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ASUP1415:00 093A:300C Touchpad
2024-02-08 16:00:36,784 INFO     Setting up for config file key: "enabled" with value: "True"
2024-02-08 16:00:36,785 INFO     check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/numberpad_dev wd=1 >
2024-02-08 16:00:36,885 INFO     Numpad activated
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ASUP1415:00 093A:300C Touchpad
2024-02-08 16:00:37,425 INFO     Setting up for config file key: "enabled" with value: "False"
2024-02-08 16:00:37,426 INFO     check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/numberpad_dev wd=1 >
2024-02-08 16:00:37,526 INFO     Numpad deactivated
2024-02-08 16:00:38,723 INFO     Touched top_right_icon (numlock) in time: 1707408038.7232833
2024-02-08 16:00:38,986 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1707408038.9867997
2024-02-08 16:00:39,026 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1707408039.0267549
2024-02-08 16:00:39,066 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1707408039.0666623
2024-02-08 16:00:39,106 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1707408039.106802
2024-02-08 16:00:39,146 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1707408039.1467745
2024-02-08 16:00:39,306 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1707408039.306759
2024-02-08 16:00:39,346 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1707408039.3467703
2024-02-08 16:00:39,507 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1707408039.5069737
2024-02-08 16:00:39,627 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1707408039.627101
2024-02-08 16:00:39,667 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1707408039.6671083
2024-02-08 16:00:39,707 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1707408039.7070696
2024-02-08 16:00:39,747 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1707408039.747432
2024-02-08 16:00:39,827 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1707408039.8271015
2024-02-08 16:00:39,867 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1707408039.867086
2024-02-08 16:00:43,127 INFO     Touched top_right_icon (numlock) in time: 1707408043.1273718
2024-02-08 16:00:43,403 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1707408043.4031234
2024-02-08 16:00:43,487 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1707408043.486992
2024-02-08 16:00:43,527 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1707408043.5274632
2024-02-08 16:00:43,567 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1707408043.5669854
2024-02-08 16:00:43,607 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1707408043.6074326
2024-02-08 16:00:43,647 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1707408043.647124
2024-02-08 16:00:43,687 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1707408043.6871138
2024-02-08 16:00:43,727 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1707408043.7270966
2024-02-08 16:00:43,767 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1707408043.767266
2024-02-08 16:00:43,807 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1707408043.8074334
2024-02-08 16:00:43,847 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1707408043.8470836
2024-02-08 16:00:43,886 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1707408043.8868537
[... 50 lines cut ...]
2024-02-08 16:02:35,942 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1707408155.9427655
2024-02-08 16:02:36,866 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1707408156.8667514
2024-02-08 16:02:36,907 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1707408156.9077666
2024-02-08 16:08:20,821 INFO     Touched top_right_icon (numlock) in time: 1707408500.8219445
2024-02-08 16:08:21,098 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1707408501.0987236
2024-02-08 16:08:21,120 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1707408501.120691
2024-02-08 16:08:21,448 INFO     Touched top_right_icon (numlock) in time: 1707408501.4487326
2024-02-08 16:08:21,608 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1707408501.6088562
2024-02-08 16:08:54,442 INFO     Touched top_right_icon (numlock) in time: 1707408534.442924
2024-02-08 16:08:54,510 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1707408534.510143
2024-02-08 16:09:28,550 INFO     Setting up for config file key: "enabled" with value: "True"
2024-02-08 16:09:28,551 INFO     check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/numberpad_dev wd=1 >
2024-02-08 16:09:28,651 INFO     Numpad activated
2024-02-08 16:09:30,232 INFO     Setting up for config file key: "enabled" with value: "False"
2024-02-08 16:09:30,233 INFO     check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/numberpad_dev wd=1 >
2024-02-08 16:09:30,333 INFO     Numpad deactivated
```

### Desktop

- Program version 4.4.0
- Distro Ubuntu 23.10
- Kernel 6.7.1-060701-generic

ldrahnik, 2024-02-09:
Hi, please post here entire output from installing

ldrahnik, 2024-02-10:
@PellattLlo Can be reopened and be more investigated. At this moment by provided logs because by default is activation time 1s nothing looks wrong.

closed 2024-02-10 by ldrahnik