asus-linux-drivers/asus-numberpad-driver #165: [Bug]: Numberpad instantly turns off on latest Fedora KDE Spin

Kedlub, 2024-02-19:
### Describe the bug

After activating the Numberpad it deactivates itself again in a split second. It started happening sometime after the latest Fedora update a week ago. Tried older kernel version, so probably caused by Plasma Desktop, as I can't think of anything else that could interfere with numlock or the Numberpad itself.

### Expected behavior

After activating it should be kept active

### Relevant log output

```shell
2024-02-19 17:37:45,734 DEBUG    Started new slot
2024-02-19 17:37:45,735 DEBUG    finger down at x 3804 y 154
2024-02-19 17:37:45,735 INFO     Touched top_right_icon (numlock) in time: 1708360665.7356646
2024-02-19 17:37:45,941 INFO     The numpad numlock was pressed longer than the activation time: 0.20541954040527344
2024-02-19 17:37:45,941 INFO     Activation time: 0.2
2024-02-19 17:37:45,966 INFO     System numlock activated
2024-02-19 17:37:45,966 INFO     Numpad activated
2024-02-19 17:37:45,966 INFO     grab
2024-02-19 17:37:45,969 INFO     Setting up for config file key: "enabled" with value: "True"
2024-02-19 17:37:45,970 DEBUG    Writting to config file: "<_io.TextIOWrapper name='/usr/share/asus-numberpad-driver/numberpad_dev' mode='w' encoding='UTF-8'>"
2024-02-19 17:37:45,970 INFO     check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/numberpad_dev wd=1 >
2024-02-19 17:37:46,265 INFO     un-grab
2024-02-19 17:37:46,289 INFO     Setting up for config file key: "enabled" with value: "False"
2024-02-19 17:37:46,290 DEBUG    Writting to config file: "<_io.TextIOWrapper name='/usr/share/asus-numberpad-driver/numberpad_dev' mode='w' encoding='UTF-8'>"
2024-02-19 17:37:46,290 INFO     check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/numberpad_dev wd=1 >
2024-02-19 17:37:46,390 INFO     Numpad deactivated
2024-02-19 17:37:46,680 DEBUG    Ended existing slot
2024-02-19 17:37:46,682 DEBUG    finger down at x 3784 y 160
2024-02-19 17:37:46,682 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1708360666.6821616
```

### Desktop

- Program version 4.4.1
- Distro Fedora 39 (KDE Spin)
- Kernel 6.7.4-200.fc39.x86_64

ldrahnik, 2024-02-19:
Hi there! I see `DEBUG` log entry so I guess you run driver directly with allowed DEBUG log level without disabling installed service that usually lead to this weird instant inactivating. I am right?

```
# first
$ systemctl stop --user asus_numberpad_driver@<$USER>.service
# then debug e.g.
$ cd asus-numberpad-driver
$ LOG=DEBUG ./numberpad.py "up5401ea" ""
```

Kedlub, 2024-02-19:
Hello!
I have followed the [troubleshooting](https://github.com/asus-linux-drivers/asus-numberpad-driver?tab=readme-ov-file#troubleshooting) section from the README. Before running it manually, I stopped the systemd service, and then ran it manually with the DEBUG flag.
So only one instance of the driver was running at a time. This bug happens even while running normally via systemd.

ldrahnik, 2024-02-19:
Do not you have running systemctl service for root user from previous version?
Could you please attach here entire output from install script?

Kedlub, 2024-02-19:
This is a relatively fresh installation of Fedora, so no, I didn't have it installed as root.

Here is the install log:
`groupadd: skupina „numberpad“ již existuje
Added group numberpad to current user

Poslední kontrola metadat: před 3:52:09, Po 19. února 2024, 14:55:40.
Balíček ibus-1.5.29-1.fc39.x86_64 je již nainstalován.
Balíček libevdev-1.13.1-2.fc39.x86_64 je již nainstalován.
Balíček curl-8.2.1-4.fc39.x86_64 je již nainstalován.
Balíček xinput-1.6.3-7.fc39.x86_64 je již nainstalován.
Balíček i2c-tools-4.3-7.fc39.x86_64 je již nainstalován.
Balíček python3-devel-3.12.1-2.fc39.x86_64 je již nainstalován.
Balíček python3-virtualenv-20.21.1-5.fc39.noarch je již nainstalován.
Balíček libxml2-2.10.4-3.fc39.x86_64 je již nainstalován.
Balíček libxml2-2.10.4-3.fc39.i686 je již nainstalován.
Závislosti vyřešeny.
Není co dělat.
Hotovo!

groupadd: skupina „input“ již existuje
groupadd: skupina „i2c“ již existuje
groupadd: skupina „uinput“ již existuje
Added groups input, i2c, uinput, numberpad to current user
uinput module loaded
i2c-dev module loaded
uinput module added to auto loaded modules
Udev rules reloaded and triggered

Testing interface i2c-0: success (adr 0x15)
The detection was successful. Touchpad with numberpad found: i2c-0

Default config will be autocreated during the first run and available for futher modifications here:
/usr/share/asus-numberpad-driver/numberpad_dev

created virtual environment CPython3.12.1.final.0-64 in 368ms
  creator CPython3Posix(dest=/usr/share/asus-numberpad-driver/.env, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(extra_search_dir=/usr/share/python-wheels,download=False, pip=bundle, via=copy, app_data_dir=/home/arzyk/.local/share/virtualenv)
    added seed packages: pip==23.2.1
  activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
Requirement already satisfied: pip in /usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages (23.2.1)
Collecting pip
  Obtaining dependency information for pip from https://files.pythonhosted.org/packages/8a/6a/19e9fe04fca059ccf770861c7d5721ab4c2aebc539889e97c7977528a53b/pip-24.0-py3-none-any.whl.metadata
  Using cached pip-24.0-py3-none-any.whl.metadata (3.6 kB)
Using cached pip-24.0-py3-none-any.whl (2.1 MB)
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 23.2.1
    Uninstalling pip-23.2.1:
      Successfully uninstalled pip-23.2.1
Successfully installed pip-24.0
Collecting setuptools
  Using cached setuptools-69.1.0-py3-none-any.whl.metadata (6.1 kB)
Using cached setuptools-69.1.0-py3-none-any.whl (819 kB)
Installing collected packages: setuptools
Successfully installed setuptools-69.1.0
Collecting libevdev (from -r requirements.txt (line 1))
  Using cached libevdev-0.11-py3-none-any.whl
Collecting numpy (from -r requirements.txt (line 2))
  Using cached numpy-1.26.4-cp312-cp312-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (61 kB)
Collecting pyinotify (from -r requirements.txt (line 3))
  Using cached pyinotify-0.9.6-py3-none-any.whl
Collecting python-xlib (from -r requirements.txt (line 4))
  Using cached python_xlib-0.33-py2.py3-none-any.whl.metadata (6.2 kB)
Collecting smbus2 (from -r requirements.txt (line 5))
  Using cached smbus2-0.4.3-py2.py3-none-any.whl.metadata (7.1 kB)
Collecting pyasyncore (from -r requirements.txt (line 6))
  Using cached pyasyncore-1.0.3-py3-none-any.whl.metadata (3.4 kB)
Collecting six>=1.10.0 (from python-xlib->-r requirements.txt (line 4))
  Using cached six-1.16.0-py2.py3-none-any.whl (11 kB)
[... 46 lines cut ...]

The application of this rule results in the following changes if an external keyboard is connected:

 - Numlock key does not activate NumberPad (config value is set to sys_numlock_enables_numpad=0)
 - Numberpad disactivation does not disable Numlock (config value is set to numpad_disables_sys_numlock=0)

In summary when an external keyboard is connected then NumberPad activation is not linked to Numlock state and vice versa.

Do you want install the rule for external keyboard? [y/N]n

Calculator app

Do you want try to install toggling script for XF86Calculator key? [y/N]n

Power supply saver

By default is idle functionality disabled and may be manually enabled in config file later. Or by installing this rule can be enabled by every detected change of power supply mode to battery mode.

Idle mode is configured to be enabled after 10s of inactivity and to decrease 30% of brightness.

Do you want install the rule for idle functionality? [y/N]n

Installation finished succesfully

Reboot is required. Do you want reboot now? [y/N]`

ldrahnik, 2024-02-19:
@Kedlub Please, post here current driver config

Kedlub, 2024-02-19:
This is currently the basic vanilla post installation config, but I normally change the `activation_time` and `press_key_when_is_done_untouch` after installation

```
[main]
numpad_disables_sys_numlock = 1
disable_due_inactivity_time = 0
touchpad_disables_numpad = 1
key_repetitions = 0
multitouch = 0
one_touch_key_rotation = 0
activation_time = 1
sys_numlock_enables_numpad = 1
top_left_icon_activation_time = 1
top_left_icon_slide_func_activation_x_ratio = 0.3
top_left_icon_slide_func_activation_y_ratio = 0.3
top_right_icon_slide_func_activation_x_ratio = 0.3
top_right_icon_slide_func_activation_y_ratio = 0.3
enabled_touchpad_pointer = 3
press_key_when_is_done_untouch = 1
enabled = 0
default_backlight_level = 0x01
brightness = 0x48
top_left_icon_brightness_func_disabled = 0
distance_to_move_only_pointer = 0
idled = 0
idle_brightness = 30
idle_enabled = 1
idle_time = 10
```

ldrahnik, 2024-02-19:
@Kedlub Please reinstall from the latest master with additional debug logging, should help find out what is wrong

Kedlub, 2024-02-19:
```
2024-02-19 19:42:00,125 INFO     Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
2024-02-19 19:42:00,125 INFO     Set keyboard 3 from H: Handlers=sysrq kbd leds event3
2024-02-19 19:42:00,140 INFO     Detecting touchpad from string: "N: Name="ASUE140D:00 04F3:31B9 Touchpad""
2024-02-19 19:42:00,141 INFO     Set touchpad device id 0 from S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-ASUE140D:00/0018:04F3:31B9.0001/input/input8
2024-02-19 19:42:00,141 INFO     Set touchpad id 5 from H: Handlers=mouse1 event5
2024-02-19 19:42:00,142 INFO     Touchpad min-max: x 0-3880, y 0-2299
2024-02-19 19:42:00,142 INFO     Numpad min-max: x 200-3680, y 200-2219
2024-02-19 19:42:01,177 DEBUG    load_all_config_values: config_lock.acquire will be called
2024-02-19 19:42:01,177 DEBUG    load_all_config_values: config_lock.acquire called succesfully
2024-02-19 19:42:01,179 DEBUG    Writting to config file: "<_io.TextIOWrapper name='./numberpad_dev' mode='w' encoding='UTF-8'>"
2024-02-19 19:42:01,282 DEBUG    check_system_numlock_vs_local: numlock_lock.acquire will be called
2024-02-19 19:42:01,283 DEBUG    check_system_numlock_vs_local: numlock_lock.acquire called succesfully
2024-02-19 19:42:01,285 DEBUG    check_numpad_automatical_disable_or_idle_due_inactivity: numlock_lock.acquire will be called
2024-02-19 19:42:01,296 DEBUG    check_numpad_automatical_disable_or_idle_due_inactivity: numlock_lock.acquire called succesfully
2024-02-19 19:42:01,296 DEBUG    check_numpad_automatical_disable_or_idle_due_inactivity: numlock_lock.acquire will be called
2024-02-19 19:42:01,296 DEBUG    check_numpad_automatical_disable_or_idle_due_inactivity: numlock_lock.acquire called succesfully
2024-02-19 19:42:01,649 DEBUG    Started new slot
2024-02-19 19:42:01,650 DEBUG    finger down at x 3782 y 73
2024-02-19 19:42:01,651 INFO     Touched top_right_icon (numlock) in time: 1708368121.6510804
2024-02-19 19:42:01,796 DEBUG    check_system_numlock_vs_local: numlock_lock.acquire will be called
2024-02-19 19:42:01,796 DEBUG    check_system_numlock_vs_local: numlock_lock.acquire called succesfully
2024-02-19 19:42:02,297 DEBUG    check_numpad_automatical_disable_or_idle_due_inactivity: numlock_lock.acquire will be called
2024-02-19 19:42:02,297 DEBUG    check_numpad_automatical_disable_or_idle_due_inactivity: numlock_lock.acquire called succesfully
2024-02-19 19:42:02,297 DEBUG    check_numpad_automatical_disable_or_idle_due_inactivity: numlock_lock.acquire will be called
2024-02-19 19:42:02,297 DEBUG    check_numpad_automatical_disable_or_idle_due_inactivity: numlock_lock.acquire called succesfully
2024-02-19 19:42:02,308 DEBUG    check_system_numlock_vs_local: numlock_lock.acquire will be called
2024-02-19 19:42:02,309 DEBUG    check_system_numlock_vs_local: numlock_lock.acquire called succesfully
2024-02-19 19:42:02,823 DEBUG    check_system_numlock_vs_local: numlock_lock.acquire will be called
2024-02-19 19:42:02,823 DEBUG    check_system_numlock_vs_local: numlock_lock.acquire called succesfully
2024-02-19 19:42:03,108 DEBUG    Ended existing slot
2024-02-19 19:42:03,109 DEBUG    finger down at x 3751 y 134
2024-02-19 19:42:03,109 INFO     The numpad numlock was pressed longer than the activation time: 1.4585788249969482
2024-02-19 19:42:03,109 INFO     Activation time: 1.0
2024-02-19 19:42:03,110 INFO     Un-touched with NumberPad activation top_right_icon (numlock) in time: 1708368123.1100214
2024-02-19 19:42:03,110 DEBUG    local_numlock_pressed: numlock_lock.acquire will be called
2024-02-19 19:42:03,110 DEBUG    local_numlock_pressed: numlock_lock.acquire called succesfully
2024-02-19 19:42:03,133 INFO     System numlock activated
2024-02-19 19:42:03,134 INFO     Numpad activated
2024-02-19 19:42:03,134 DEBUG    ['gsettings', 'set', 'org.gnome.desktop.peripherals.touchpad', 'tap-to-click', 'false']
2024-02-19 19:42:03,161 DEBUG    ['xinput', 'set-prop', 'ASUE140D:00 04F3:31B9 Touchpad', 'libinput Tapping Enabled', '0']
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ASUE140D:00 04F3:31B9 Touchpad
2024-02-19 19:42:03,170 DEBUG    config_set: config_lock.acquire will be called
2024-02-19 19:42:03,170 DEBUG    config_set: config_lock.acquire called succesfully
2024-02-19 19:42:03,170 INFO     Setting up for config file key: "enabled" with value: "True"
2024-02-19 19:42:03,171 DEBUG    Writting to config file: "<_io.TextIOWrapper name='./numberpad_dev' mode='w' encoding='UTF-8'>"
2024-02-19 19:42:03,171 INFO     check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/home/arzyk/sources/asus-numberpad-driver pathname=/home/arzyk/sources/asus-numberpad-driver/numberpad_dev wd=1 >
2024-02-19 19:42:03,286 DEBUG    check_touchpad_status: numlock_lock.acquire will be called
2024-02-19 19:42:03,287 DEBUG    check_touchpad_status: numlock_lock.acquire called succesfully
2024-02-19 19:42:03,297 DEBUG    check_numpad_automatical_disable_or_idle_due_inactivity: numlock_lock.acquire will be called
2024-02-19 19:42:03,298 DEBUG    check_numpad_automatical_disable_or_idle_due_inactivity: numlock_lock.acquire called succesfully
2024-02-19 19:42:03,298 DEBUG    check_numpad_automatical_disable_or_idle_due_inactivity: numlock_lock.acquire will be called
[... cut ...]
2024-02-19 19:42:04,299 DEBUG    check_numpad_automatical_disable_or_idle_due_inactivity: numlock_lock.acquire called succesfully
2024-02-19 19:42:04,483 DEBUG    check_system_numlock_vs_local: numlock_lock.acquire will be called
2024-02-19 19:42:04,483 DEBUG    check_system_numlock_vs_local: numlock_lock.acquire called succesfully
2024-02-19 19:42:04,994 DEBUG    check_system_numlock_vs_local: numlock_lock.acquire will be called
2024-02-19 19:42:04,994 DEBUG    check_system_numlock_vs_local: numlock_lock.acquire called succesfully
```

ldrahnik, 2024-02-19:
@Kedlub The same but please use the branch named `165-bug`. It looks like to me problem is reading numlock value from your keyboard, lets check it.

ldrahnik, 2024-02-19:
@Kedlub Hm, faster way how to check it is probably:

```
$ sudo apt install xdotool
$ xdotool key Num_Lock
# numberpad should be activated
$ xdotool key Num_Lock
# numberpad should be deactivated
```

Kedlub, 2024-02-19:
Okay, it works completely fine in X11 (even using the `xdotool` to toggle numlock). It's only broken in Wayland, which I am using, as that's the default for Fedora. I tried `ydotool` to toggle numlock on Wayland, but it does nothing.

According to the log using the `165-bug` branch, the numlock isn't changed at all:
```
2024-02-19 22:23:58,776 INFO     Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
2024-02-19 22:23:58,776 INFO     Set keyboard 3 from H: Handlers=sysrq kbd leds event3
2024-02-19 22:23:58,783 INFO     Detecting touchpad from string: "N: Name="ASUE140D:00 04F3:31B9 Touchpad""
2024-02-19 22:23:58,784 INFO     Set touchpad device id 0 from S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-ASUE140D:00/0018:04F3:31B9.0001/input/input8
2024-02-19 22:23:58,784 INFO     Set touchpad id 5 from H: Handlers=mouse1 event5
2024-02-19 22:23:58,784 INFO     Touchpad min-max: x 0-3880, y 0-2299
2024-02-19 22:23:58,784 INFO     Numpad min-max: x 200-3680, y 200-2219
2024-02-19 22:23:59,816 DEBUG    load_all_config_values: config_lock.acquire will be called
2024-02-19 22:23:59,816 DEBUG    load_all_config_values: config_lock.acquire called succesfully
2024-02-19 22:23:59,818 DEBUG    Writting to config file: "<_io.TextIOWrapper name='./numberpad_dev' mode='w' encoding='UTF-8'>"
2024-02-19 22:23:59,920 DEBUG    check_system_numlock_vs_local: numlock_lock.acquire will be called
2024-02-19 22:23:59,921 DEBUG    check_system_numlock_vs_local: numlock_lock.acquire called succesfully
2024-02-19 22:23:59,921 DEBUG    get_system_numlock:
2024-02-19 22:23:59,921 DEBUG    0
2024-02-19 22:23:59,922 DEBUG    check_numpad_automatical_disable_or_idle_due_inactivity: numlock_lock.acquire will be called
2024-02-19 22:23:59,934 DEBUG    check_numpad_automatical_disable_or_idle_due_inactivity: numlock_lock.acquire called succesfully
2024-02-19 22:23:59,934 DEBUG    check_numpad_automatical_disable_or_idle_due_inactivity: numlock_lock.acquire will be called
2024-02-19 22:23:59,934 DEBUG    check_numpad_automatical_disable_or_idle_due_inactivity: numlock_lock.acquire called succesfully
2024-02-19 22:24:00,434 DEBUG    check_system_numlock_vs_local: numlock_lock.acquire will be called
2024-02-19 22:24:00,434 DEBUG    check_system_numlock_vs_local: numlock_lock.acquire called succesfully
2024-02-19 22:24:00,434 DEBUG    get_system_numlock:
2024-02-19 22:24:00,435 DEBUG    0
2024-02-19 22:24:00,935 DEBUG    check_numpad_automatical_disable_or_idle_due_inactivity: numlock_lock.acquire will be called
2024-02-19 22:24:00,935 DEBUG    check_numpad_automatical_disable_or_idle_due_inactivity: numlock_lock.acquire called succesfully
2024-02-19 22:24:00,935 DEBUG    check_numpad_automatical_disable_or_idle_due_inactivity: numlock_lock.acquire will be called
2024-02-19 22:24:00,936 DEBUG    check_numpad_automatical_disable_or_idle_due_inactivity: numlock_lock.acquire called succesfully
2024-02-19 22:24:00,941 DEBUG    check_system_numlock_vs_local: numlock_lock.acquire will be called
2024-02-19 22:24:00,942 DEBUG    check_system_numlock_vs_local: numlock_lock.acquire called succesfully
2024-02-19 22:24:00,942 DEBUG    get_system_numlock:
2024-02-19 22:24:00,942 DEBUG    0
2024-02-19 22:24:01,448 DEBUG    check_system_numlock_vs_local: numlock_lock.acquire will be called
2024-02-19 22:24:01,449 DEBUG    check_system_numlock_vs_local: numlock_lock.acquire called succesfully
2024-02-19 22:24:01,449 DEBUG    get_system_numlock:
2024-02-19 22:24:01,450 DEBUG    0
2024-02-19 22:24:01,450 DEBUG    Started new slot
2024-02-19 22:24:01,452 DEBUG    finger down at x 3785 y 109
2024-02-19 22:24:01,452 INFO     Touched top_right_icon (numlock) in time: 1708377841.452684
2024-02-19 22:24:01,936 DEBUG    check_numpad_automatical_disable_or_idle_due_inactivity: numlock_lock.acquire will be called
2024-02-19 22:24:01,936 DEBUG    check_numpad_automatical_disable_or_idle_due_inactivity: numlock_lock.acquire called succesfully
2024-02-19 22:24:01,936 DEBUG    check_numpad_automatical_disable_or_idle_due_inactivity: numlock_lock.acquire will be called
2024-02-19 22:24:01,937 DEBUG    check_numpad_automatical_disable_or_idle_due_inactivity: numlock_lock.acquire called succesfully
2024-02-19 22:24:01,959 DEBUG    check_system_numlock_vs_local: numlock_lock.acquire will be called
2024-02-19 22:24:01,959 DEBUG    check_system_numlock_vs_local: numlock_lock.acquire called succesfully
2024-02-19 22:24:01,960 DEBUG    get_system_numlock:
2024-02-19 22:24:01,960 DEBUG    0
2024-02-19 22:24:02,466 DEBUG    check_system_numlock_vs_local: numlock_lock.acquire will be called
2024-02-19 22:24:02,466 DEBUG    check_system_numlock_vs_local: numlock_lock.acquire called succesfully
2024-02-19 22:24:02,466 DEBUG    get_system_numlock:
2024-02-19 22:24:02,467 DEBUG    0
2024-02-19 22:24:02,881 DEBUG    Ended existing slot
2024-02-19 22:24:02,882 DEBUG    finger down at x 3754 y 149
2024-02-19 22:24:02,882 INFO     The numpad numlock was pressed longer than the activation time: 1.429630994796753
2024-02-19 22:24:02,882 INFO     Activation time: 1.0
2024-02-19 22:24:02,882 INFO     Un-touched with NumberPad activation top_right_icon (numlock) in time: 1708377842.8826947
2024-02-19 22:24:02,882 DEBUG    local_numlock_pressed: numlock_lock.acquire will be called
2024-02-19 22:24:02,882 DEBUG    local_numlock_pressed: numlock_lock.acquire called succesfully
2024-02-19 22:24:02,891 DEBUG    get_system_numlock:
[... cut ...]
2024-02-19 22:24:04,202 DEBUG    check_system_numlock_vs_local: numlock_lock.acquire will be called
2024-02-19 22:24:04,203 DEBUG    check_system_numlock_vs_local: numlock_lock.acquire called succesfully
2024-02-19 22:24:04,203 DEBUG    get_system_numlock:
2024-02-19 22:24:04,204 DEBUG    0
2024-02-19 22:24:04,713 DEBUG    check_system_numlock_vs_local: numlock_lock.acquire will be called
2024-02-19 22:24:04,713 DEBUG    check_system_numlock_vs_local: numlock_lock.acquire called succesfully
2024-02-19 22:24:04,713 DEBUG    get_system_numlock:
2024-02-19 22:24:04,714 DEBUG    0
```

ldrahnik, 2024-02-19:
@Kedlub Please try this:

- press capslock or any other key with led
- run `$ evtest` select Translate keyboard id

You should see

```
    Event code 0 (LED_NUML) state 0
    Event code 1 (LED_CAPSL) state 1
    Event code 2 (LED_SCROLLL) state 0
```

Kedlub, 2024-02-19:
@ldrahnik Yes this works for the caps lock.

```
Event type 17 (EV_LED)
    Event code 0 (LED_NUML) state 0
    Event code 1 (LED_CAPSL) state 1
    Event code 2 (LED_SCROLLL) state 0
```

ldrahnik, 2024-02-20:
@Kedlub And for `LED_NUML`?

Kedlub, 2024-02-20:
Okay, the NumberPad actually lights up when toggling the numlock using `ydotool`. Sorry, I was using the wrong keycode when testing before.
But for some reason, it only works once. I toggled it, the Numberpad lit up, and then the idle functionality toggled it off. And now I can't light it up again. The Numberpad also wasn't really working, it was just lit up but not responding to any touches, it was still acting only as a basic touchpad.

But it was visible in the `evtest`:
```
Event type 17 (EV_LED)
    Event code 0 (LED_NUML) state 1
    Event code 1 (LED_CAPSL) state 0
    Event code 2 (LED_SCROLLL) state 0
```

ldrahnik, 2024-02-20:
@Kedlub For this testing please disable idle functionality in config

Kedlub, 2024-02-20:
I tried activating it using the numberpad button, where it instantly deactivated. Then with the ydotool where it is now kept active, and can't be deactivated. It seems to be registering number touches according to the log, but it doesn't actually input any of these numbers.

[error.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/14342109/error.log)

ldrahnik, 2024-02-20:
@Kedlub Could you please try to run `$ sudo evtest` and then `$ ydotool key Num_Lock` and post here captured output?

ldrahnik, 2024-02-20:
@Kedlub And then please run `$ sudo evtest` (Translated keyboard select)  and `LOG=DEBUG ./numberpad.py "up5401ea" ""` and post here output.

Kedlub, 2024-02-21:
@ldrahnik It seems that the numlock doesn't get toggled at all from the driver.

ydotool toggle:
```
Event: time 1708498693.198332, type 4 (EV_MSC), code 4 (MSC_SCAN), value 1c
Event: time 1708498693.198332, type 1 (EV_KEY), code 28 (KEY_ENTER), value 0
Event: time 1708498693.198332, -------------- SYN_REPORT ------------
Event: time 1708498695.535356, type 4 (EV_MSC), code 4 (MSC_SCAN), value 1c
Event: time 1708498695.535356, type 1 (EV_KEY), code 28 (KEY_ENTER), value 1
Event: time 1708498695.535356, -------------- SYN_REPORT ------------
Event: time 1708498695.566343, type 17 (EV_LED), code 0 (LED_NUML), value 1
Event: time 1708498695.566343, -------------- SYN_REPORT ------------
Event: time 1708498695.634066, type 4 (EV_MSC), code 4 (MSC_SCAN), value 1c
Event: time 1708498695.634066, type 1 (EV_KEY), code 28 (KEY_ENTER), value 0
Event: time 1708498695.634066, -------------- SYN_REPORT ------------
```

NumberPad button toggle:
```
Event: time 1708498989.134448, type 4 (EV_MSC), code 4 (MSC_SCAN), value 1c
Event: time 1708498989.134448, type 1 (EV_KEY), code 28 (KEY_ENTER), value 0
Event: time 1708498989.134448, -------------- SYN_REPORT ------------
Event: time 1708498990.958242, type 4 (EV_MSC), code 4 (MSC_SCAN), value 1c
Event: time 1708498990.958242, type 1 (EV_KEY), code 28 (KEY_ENTER), value 1
Event: time 1708498990.958242, -------------- SYN_REPORT ------------
Event: time 1708498991.018902, type 4 (EV_MSC), code 4 (MSC_SCAN), value 1c
Event: time 1708498991.018902, type 1 (EV_KEY), code 28 (KEY_ENTER), value 0
Event: time 1708498991.018902, -------------- SYN_REPORT ------------
```

ldrahnik, 2024-02-21:
@Kedlub Please run `$ xev` and then`$ ydotool key Num_Lock`, equivalent for wayland I think is `wev`? I need to see equivalent for rows below:

```
$ xev

...
KeyRelease event, serial 38, synthetic NO, window 0x7600001,
    root 0x786, subw 0x0, time 36034048, (-25,634), root:(1326,1475),
    state 0x10, keycode 77 (keysym 0xff7f, Num_Lock), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False
...
```

ldrahnik, 2024-02-21:
@Kedlub Please try again branch `165-bug`

Kedlub, 2024-02-21:
@ldrahnik It still doesn't work even with the latest commit from `165-bug`, no change from previous logs, if the numlock was activated before it always thinks it's activated and vice versa.

And this was the only related event to numlock. This is hard to try without physical numlock, as Wayland passes events to `wev` only if I am hovering over the window that it creates.

```
[14:     wl_keyboard] enter: serial: 1296; surface: 3
                      sym: Num_Lock     (65407), utf8: ''
[14:     wl_keyboard] modifiers: serial: 1287; group: 0
                      depressed: 00000010: Mod2
                      latched: 00000000
                      locked: 00000010: Mod2
```

Log from the newest commit

```
2024-02-21 11:19:51,717 INFO     The numpad numlock was pressed longer than the activation time: 1.1830694675445557
2024-02-21 11:19:51,717 INFO     Activation time: 1.0
2024-02-21 11:19:51,717 INFO     Un-touched with NumberPad activation top_right_icon (numlock) in time: 1708510791.7177613
2024-02-21 11:19:51,717 DEBUG    local_numlock_pressed: numlock_lock.acquire will be called
2024-02-21 11:19:51,720 DEBUG    local_numlock_pressed: numlock_lock.acquire called succesfully
2024-02-21 11:19:51,728 DEBUG    get_system_numlock:
2024-02-21 11:19:51,728 DEBUG    0
2024-02-21 11:19:51,740 INFO     System numlock activated
2024-02-21 11:19:51,740 INFO     Numpad activated
2024-02-21 11:19:51,741 DEBUG    ['gsettings', 'set', 'org.gnome.desktop.peripherals.touchpad', 'tap-to-click', 'false']
2024-02-21 11:19:51,763 DEBUG    ['xinput', 'set-prop', 'ASUE140D:00 04F3:31B9 Touchpad', 'libinput Tapping Enabled', '0']
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ASUE140D:00 04F3:31B9 Touchpad
2024-02-21 11:19:51,771 DEBUG    config_set: config_lock.acquire will be called
2024-02-21 11:19:51,771 DEBUG    config_set: config_lock.acquire called succesfully
2024-02-21 11:19:51,771 INFO     Setting up for config file key: "enabled" with value: "True"
2024-02-21 11:19:51,771 DEBUG    Writting to config file: "<_io.TextIOWrapper name='/usr/share/asus-numberpad-driver/numberpad_dev' mode='w' encoding='UTF-8'>"
2024-02-21 11:19:51,771 INFO     check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/numberpad_dev wd=1 >
2024-02-21 11:19:51,990 DEBUG    check_touchpad_status: numlock_lock.acquire will be called
2024-02-21 11:19:51,991 DEBUG    check_touchpad_status: numlock_lock.acquire called succesfully
2024-02-21 11:19:51,997 DEBUG    check_numpad_automatical_disable_or_idle_due_inactivity: numlock_lock.acquire will be called
2024-02-21 11:19:51,999 DEBUG    check_numpad_automatical_disable_or_idle_due_inactivity: numlock_lock.acquire called succesfully
2024-02-21 11:19:51,999 DEBUG    check_numpad_automatical_disable_or_idle_due_inactivity: numlock_lock.acquire will be called
2024-02-21 11:19:51,999 DEBUG    check_numpad_automatical_disable_or_idle_due_inactivity: numlock_lock.acquire called succesfully
2024-02-21 11:19:52,220 DEBUG    check_system_numlock_vs_local: numlock_lock.acquire will be called
2024-02-21 11:19:52,221 DEBUG    check_system_numlock_vs_local: numlock_lock.acquire called succesfully
2024-02-21 11:19:52,221 DEBUG    get_system_numlock:
2024-02-21 11:19:52,221 DEBUG    0
2024-02-21 11:19:52,231 DEBUG    ['gsettings', 'set', 'org.gnome.desktop.peripherals.touchpad', 'tap-to-click', 'true']
2024-02-21 11:19:52,253 DEBUG    ['xinput', 'set-prop', 'ASUE140D:00 04F3:31B9 Touchpad', 'libinput Tapping Enabled', '1']
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ASUE140D:00 04F3:31B9 Touchpad
2024-02-21 11:19:52,260 DEBUG    config_set: config_lock.acquire will be called
2024-02-21 11:19:52,261 DEBUG    config_set: config_lock.acquire called succesfully
2024-02-21 11:19:52,261 INFO     Setting up for config file key: "enabled" with value: "False"
2024-02-21 11:19:52,261 DEBUG    Writting to config file: "<_io.TextIOWrapper name='/usr/share/asus-numberpad-driver/numberpad_dev' mode='w' encoding='UTF-8'>"
2024-02-21 11:19:52,261 INFO     check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/numberpad_dev wd=1 >
2024-02-21 11:19:52,362 INFO     Numpad deactivated
```

ldrahnik, 2024-02-21:
@Kedlub

> This is hard to try without physical numlock, as Wayland passes events to wev only if I am hovering over the window that it creates.

```
$ sleep 3 && ydotool key Num_Lock
```

ldrahnik, 2024-02-21:
@Kedlub Please try again `165-bug`

Kedlub, 2024-02-21:
@ldrahnik Newest branch still nothing, now it doesn't even detect the manually activated numlock with `ydotool`, `get_system_numlock` is always zero.

Here is the output of `wev` with the delayed ydotool:
```
[14:     wl_keyboard] key: serial: 759; time: 288577; key: 77; state: 1 (pressed)
                      sym: Num_Lock     (65407), utf8: ''
[14:     wl_keyboard] modifiers: serial: 760; group: 0
                      depressed: 00000010: Mod2
                      latched: 00000000
                      locked: 00000010: Mod2
```

Kedlub, 2024-02-21:
Hm, it seems that ydotool works very differently from xdotool, I had to manually "unpress" the numlock key, to be able to deactivate it. `ydotool key 69:1 69:0`
Where 69 is the Numlock key according to `/usr/include/linux/input-event-codes.h`
While running the command now with the driver running, it toggles the numberpad successfully. But the Numberpad still can't type any symbols, and still can't be toggled with the Numberpad button.

Kedlub, 2024-02-21:
This is how the full press of the numlock key looks like according to `wev`

```
[14:     wl_keyboard] key: serial: 4928; time: 1409176; key: 77; state: 1 (pressed)
                      sym: Num_Lock     (65407), utf8: ''
[14:     wl_keyboard] modifiers: serial: 4929; group: 0
                      depressed: 00000010: Mod2
                      latched: 00000000
                      locked: 00000010: Mod2
[14:     wl_keyboard] key: serial: 4930; time: 1409188; key: 77; state: 0 (released)
                      sym: Num_Lock     (65407), utf8: ''
[14:     wl_keyboard] modifiers: serial: 4931; group: 0
                      depressed: 00000000
                      latched: 00000000
                      locked: 00000010: Mod2
```

No keys that are clicked on the NumberPad are seen in the `wev` output

ldrahnik, 2024-02-21:
@Kedlub So for summary problem is every key send by this driver via created NumberPad udev device because in posted log keys are sent but not printed / used. Is this correctlink for live ISO? https://fedoraproject.org/spins/kde/download

ldrahnik, 2024-02-21:
I installed mentioned Fedora live ISO, installed updates using `$ dnf update`:

KDE plasma version `5.27.8`
KDE Framework version `5.110.0`
Qt Vrsion `5.15.10`
Kernel 6.5.6-300
Wayland

and it works. Described behaviour I had only because of running duplicated instances of driver (debugging directly twice). So, please post here entire output of command `$ sudo libinput debug-events` and output of these commands: `$ systemctl status --user | grep asus`. if it is not this case, I need to know how to reproduce your system state with steps starting with clean Fedora, without that I am unfortunatelly not able help you. It may be problem of driver but I have not confirmed yet.

ldrahnik, 2024-02-21:
Can be reopened.

closed 2024-02-21 by ldrahnik

Kedlub, 2024-02-22:
It seems that the live ISO didn't update to the newest packages.
These are the versions in my system:

- KDE Plasma Version: 5.27.10
- KDE Frameworks Version: 5.115.0
- Qt Version: 5.15.12
- Kernel Version: 6.7.5-200.fc39.x86_64 (64-bit)

Output of `systemctl status --user | grep asus`
```
│ ├─app-asus_numberpad_driver.slice
           │ │ └─asus_numberpad_driver@arzyk.service
           │ │   └─7755 /usr/share/asus-numberpad-driver/.env/bin/python3 /usr/share/asus-numberpad-driver/numberpad.py up5401ea-31B9 /usr/share/asus-numberpad-driver/
```
The systemd service is running right now, but while troubleshooting, I stop this service and start the script manually to see the logs directly.

Output of `sudo libinput debug-events`, I tried pressing the button for activating NumberPad while running
```
-event2   DEVICE_ADDED            Power Button                      seat0 default group1  cap:k
-event7   DEVICE_ADDED            Video Bus                         seat0 default group2  cap:k
-event0   DEVICE_ADDED            Lid Switch                        seat0 default group3  cap:S
-event1   DEVICE_ADDED            Power Button                      seat0 default group4  cap:k
-event4   DEVICE_ADDED            ASUE140D:00 04F3:31B9 Mouse       seat0 default group5  cap:p left scroll-nat scroll-button
-event5   DEVICE_ADDED            ASUE140D:00 04F3:31B9 Touchpad    seat0 default group5  cap:pg  size 125x72mm tap(dl off) left scroll-nat scroll-2fg-edge click-buttonareas-clickfinger dwt-on dwtp-on
-event6   DEVICE_ADDED            ASUE140D:00 04F3:31B9 Keyboard    seat0 default group5  cap:k
-event8   DEVICE_ADDED            Intel HID events                  seat0 default group6  cap:k
-event9   DEVICE_ADDED            Intel HID 5 button array          seat0 default group7  cap:k
-event11  DEVICE_ADDED            Asus WMI hotkeys                  seat0 default group8  cap:k
-event3   DEVICE_ADDED            AT Translated Set 2 keyboard      seat0 default group9  cap:k
-event16  DEVICE_ADDED            ASUE140D:00 04F3:31B9 NumberPad   seat0 default group10 cap:kp left scroll-nat
 event16  KEYBOARD_KEY            +1.856s	*** (-1) pressed
 event16  KEYBOARD_KEY            +1.856s	*** (-1) released
 event16  KEYBOARD_KEY            +3.950s	*** (-1) pressed
 event16  KEYBOARD_KEY            +3.950s	*** (-1) released
```

If there is something wrong with my system, then I don't really know where to start looking, as most of the time I install flatpak apps, and the only more "invasive" packages that I installed is the Virtualization group (with QEMU and LibVirt) and Waydroid. Apart from doing a SSDT workaround to get sound working on my laptop (which is also no longer needed as 6.7 kernel fixed it) I think I didn't touch any hardware or input related configurations. I also installed the RPMFusion repository, from which I got intel-media-driver and h264 codecs, but that should be completely unrelated.

I'm not sure if it could be related, but some apps (such as the game streaming app Parsec, at least the Flatpak version) started recognizing the touchpad as a game pad, where the entire touchpad works as a left stick on a game pad, even if the touchpad is disabled. I can't really say when it started happening, but on my old Fedora installation which I had about 2–3 months ago, both Parsec and this driver were working fine, and I had probably the same configuration as now.

I will probably try a fresh install this weekend, and slowly try installing the packages I use to find out what could have caused it.

Kedlub, 2024-02-22:
Okay, never mind, I found it out. Probably while I was debugging the issues with Parsec, I disabled both "ASUE140D:00 04F3:31B9 Mouse" and "ASUE140D:00 04F3:31B9 NumberPad" devices in Plasma mouse settings, and forgot to revert it, and having these mouse devices disabled breaks the driver.
Thank you for your help.

ldrahnik, 2024-02-22:
@Kedlub Nice (y) I am not familiar with KDE so this possibility to disable specific devices is new to me. Exist terminal command to list disabled devices like this?

Kedlub, 2024-02-22:
@ldrahnik The device disabling looks like it applies only to Mice and Touchpads, I don't see it anywhere else. And on Wayland it seems that it gets saved into KWin, which is taking care of the input events I presume.
[Relevant KWin source code](https://github.com/KDE/plasma-desktop/blob/master/kcms/mouse/backends/kwin_wl/kwin_wl_backend.cpp)
The settings applet itself is interacting with KWin using DBus.

These settings are then stored inside `~/.config/kcminputrc`, which looks like this on my system:
```
[$Version]
update_info=delete_cursor_old_default_size.upd:DeleteCursorOldDefaultSize,kcminputrc_fix_botched_5_21_0.upd:kcminputrc_fix_botched_5_21_0_pre,kcminputrc_fix_botched_5_21_0.upd:kcminputrc_fix_botched_5_21_0,kcminputrc_repeat.upd:kcminputrc_migrate_key_repeat

[Libinput][0][0][ASUE140D:00 04F3:31B9 NumberPad]
Enabled=true

[Libinput][1267][12729][ASUE140D:00 04F3:31B9 Mouse]
Enabled=true
PointerAcceleration=0.000
PointerAccelerationProfile=2

[Libinput][1267][12729][ASUE140D:00 04F3:31B9 Touchpad]
ClickMethod=2
Enabled=true
NaturalScroll=true
ScrollFactor=0.3
TapToClick=true

[Libinput][9354][33382][Trust BT Mouse]
PointerAcceleration=-0.200
PointerAccelerationProfile=1

[Mouse]
X11LibInputXAccelProfileFlat=true

[Tmp]
update_info=delete_cursor_old_default_size.upd:DeleteCursorOldDefaultSize
```

Where the most important is this one:
```
[Libinput][0][0][ASUE140D:00 04F3:31B9 NumberPad]
Enabled=true
```

The NumberPad "mouse" device needs to be `Enabled` for the driver to work.