asus-linux-drivers/asus-numberpad-driver #168: [Layout Request] ux7602-unicode

FrancoisChaumont, 2024-03-13:
### Layout Name

ux7602-unicode

### Your device (cat /sys/devices/virtual/dmi/id/product_name)

Zenbook UX7602VI_UX7602VI

kamack38, 2024-03-14:
For the future we would really appreciate, if you had read the issue template and filled numpad layout field (just take a picture of your numpad or find one on the Internet). Anyway here's one:
[image]

kamack38, 2024-03-14:
Wait, I think `up5401ea` is the correct one, it looks identical.

FrancoisChaumont, 2024-03-15:
> For the future we would really appreciate, if you had read the issue template and filled numpad layout field (just take a picture of your numpad or find one on the Internet). Anyway here's one: [...]

Apologies, I wasn't sure how to go about all those fields.

FrancoisChaumont, 2024-03-15:
> Wait, I think `up5401ea` is the correct one, it looks identical.

Amazing! This is the one. It works perfectly! 
Thank you so much!

closed 2024-03-15 by FrancoisChaumont

kamack38, 2024-03-15:
> Apologies, I wasn't sure how to go about all those fields.

Do you think that there's something we could do about that to make it easier for the future users?

FrancoisChaumont, 2024-03-19:
> Do you think that there's something we could do about that to make it easier for the future users?

Maybe add some examples for values to provide based of an already existing layout for guidance/reference?

I had no clue what to put under:
- Supported Devices
- All the offsets
- All icon data

My bad on the Numpad Layout, I could have sent a picture.