asus-linux-drivers/asus-numberpad-driver #169: [Bug]: can not toggle numpad to touchpad

rudrab, 2024-03-21:
### Describe the bug

I have bought asus 3405 today. I have installed this driver. If I do "enabled=1", then the numberpad gers enabled. But, the calculator icon in top right corner is not working to toggle between numberpad and touchpad. 

### Expected behavior

Touching the calculator icon for few seconds should enable the numberpad, or change the status from numberpad to touchpad

### Desktop

- Program version Latest from git
- Distro Fedora 40
- Kernel 6.8
-

rudrab, 2024-03-21:
This is my numpad_dev:
[main]
numpad_disables_sys_numlock = 1
disable_due_inactivity_time = 0
touchpad_disables_numpad = 1
key_repetitions = 0
multitouch = 0
one_touch_key_rotation = 0
activation_time = 1
sys_numlock_enables_numpad = 1
top_left_icon_activation_time = 1
top_left_icon_slide_func_activation_x_ratio = 0.3
top_left_icon_slide_func_activation_y_ratio = 0.3
top_right_icon_slide_func_activation_x_ratio = 0.3
top_right_icon_slide_func_activation_y_ratio = 0.3
enabled_touchpad_pointer = 3
press_key_when_is_done_untouch = 1
enabled = 0
default_backlight_level = 0x01
brightness = 0x48
top_left_icon_brightness_func_disabled = 0
distance_to_move_only_pointer = 0
idled = 0
idle_brightness = 30
idle_enabled = 0
idle_time = 10

ldrahnik, 2024-03-21:
@rudrab Hello, thank you for creating an issue. Please post here entire driver log file.

rudrab, 2024-03-21:
Hi @ldrahnik,
Thanks for your reply. 
How I can get the log file?

ldrahnik, 2024-03-21:
`/var/log/asus-numberpad-driver/*`

rudrab, 2024-03-22:
Kindly find the log files.
Regards,
[error.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/14717194/error.log)
[install-21-03-2024-20-43-11.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/14717196/install-21-03-2024-20-43-11.log)

ldrahnik, 2024-03-22:
@rudrab Was the driver installed on the same user for you try to activate?

rudrab, 2024-03-22:
Yes.

ldrahnik, 2024-03-22:
Could you please clean the content of `error` log file and then try to activate using enabled, ctrl+c here, clean or remember the last line number and then use top right icon. I do not know what is the problem yet.

rudrab, 2024-03-22:
At the installation, I have made everything "yes". Is that OK?

ldrahnik, 2024-03-22:
@rudrab Hm, how do you activate the NumberPad? You have to hold by default the top right icon for 1s or do a slide gesture to the center from top right icon at least half width and half height, because I see in log file a lot of unsuccessful activation. So you maybe do not know only how to activate.

ldrahnik, 2024-03-22:
@rudrab In config file is value `activation_time = 1` which means hold top right icon for at least 1s and these values: ```top_right_icon_slide_func_activation_x_ratio = 0.3
top_right_icon_slide_func_activation_y_ratio = 0.3``` means at least 30% of width and height, you can change it for own better experience

rudrab, 2024-03-22:
Yes, that part I know. I have already done this for zenbook 13. Never faced this issue.

rudrab, 2024-03-22:
If I put enabled=1, the numpad activates. But then I can not deactivate it using the icon.

ldrahnik, 2024-03-22:
@rudrab Please uninstall and then install again (answering all Y is ok), and hold the top right icon for at least 1s and post here config log, by log you never pressed longer then 1s. You do not have to even uninstall, you can run these commands:

```
$ cd <asus-numberpad-driver-git-repo>
$ systemctl --user stop asus_numberpad_driver@<your username>.service
$ usr/share/asus-numberpad-driver/.env/bin/python3 /usr/share/asus-numberpad-driver/numberpad.py "up5401ea"
```

rudrab, 2024-03-22:
```
systemctl --user status asus_numberpad_driver@rudra.service 
● asus_numberpad_driver@rudra.service - Asus NumberPad Driver
     Loaded: loaded (/usr/lib/systemd/user/asus_numberpad_driver@.service; enabled; preset: disabled)
    Drop-In: /usr/lib/systemd/user/service.d
             └─10-timeout-abort.conf
     Active: active (running) since Fri 2024-03-22 15:17:56 IST; 6h ago
   Main PID: 2407 (python3)
      Tasks: 22 (limit: 18565)
     Memory: 62.7M (peak: 64.9M)
        CPU: 11.477s
     CGroup: /user.slice/user-1000.slice/user@1000.service/app.slice/app-asus_numberpad_driver.slice/asus_numberpad_driver@rudra.service
             └─2407 /usr/share/asus-numberpad-driver/.env/bin/python3 /usr/share/asus-numberpad-driver/numberpad.py up5401ea /usr/share/asus-numberpa>

Mar 22 15:17:56 roddur systemd[2375]: Started asus_numberpad_driver@rudra.service - Asus NumberPad Driver.
```

ldrahnik, 2024-03-22:
@rudrab Not status, stop

rudrab, 2024-03-22:
```
systemctl --user stop asus_numberpad_driver@rudra.service 
cd Utility/asus-numberpad-driver/
 /usr/share/asus-numberpad-driver/.env/bin/python3 /usr/share/asus-numberpad-driver/numberpad.py "up5401ea"
2024-03-22 18:41:06,390 INFO     Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
2024-03-22 18:41:06,390 INFO     Set keyboard 3 from H: Handlers=sysrq kbd leds event3
2024-03-22 18:41:06,399 INFO     Detecting touchpad from string: "N: Name="ASUP1415:00 093A:300C Touchpad""
2024-03-22 18:41:06,399 INFO     Set touchpad device id 1 from S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-1/i2c-ASUP1415:00/0018:093A:300C.0001/input/input8
2024-03-22 18:41:06,399 INFO     Set touchpad id 5 from H: Handlers=mouse1 event5
2024-03-22 18:41:06,400 INFO     Touchpad min-max: x 0-3996, y 0-2242
2024-03-22 18:41:06,400 INFO     Numpad min-max: x 200-3796, y 200-2162
2024-03-22 18:41:07,423 INFO     Setting up for config file key: "brightness" with value: "0x48"
2024-03-22 18:41:19,206 INFO     Touched top_right_icon (numlock) in time: 1711113079.20628
2024-03-22 18:41:19,460 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1711113079.4609237
2024-03-22 18:41:19,529 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1711113079.5292096
2024-03-22 18:41:19,597 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1711113079.5973325
2024-03-22 18:41:19,733 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1711113079.7331758
2024-03-22 18:41:27,315 INFO     Touched top_right_icon (numlock) in time: 1711113087.315701
2024-03-22 18:41:27,565 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1711113087.5654747
2024-03-22 18:41:27,633 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1711113087.633197
2024-03-22 18:41:27,667 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1711113087.6671894
2024-03-22 18:41:27,701 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1711113087.7012005
2024-03-22 18:41:27,735 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1711113087.735178
2024-03-22 18:41:28,108 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1711113088.1088998
2024-03-22 18:41:28,176 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1711113088.1768672
2024-03-22 18:41:28,211 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1711113088.2109947
2024-03-22 18:41:28,279 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1711113088.2789812
2024-03-22 18:41:28,380 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1711113088.3809674
2024-03-22 18:41:28,551 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1711113088.5509796
2024-03-22 18:41:28,686 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1711113088.6869552
2024-03-22 18:41:33,045 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1711113093.0443566
2024-03-22 18:41:33,078 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1711113093.0782745
2024-03-22 18:41:51,312 INFO     Pressed touchpad pointer button
2024-03-22 18:41:51,958 INFO     Released touchpad pointer button
^C2024-03-22 18:41:56,862 ERROR    Listening touchpad events unexpectedly failed

```

ldrahnik, 2024-03-22:
@rudrab How many seconds did you hold the finger on pad? 0.2s as is written in log or not?

rudrab, 2024-03-22:
I am holding more than 5s.

ldrahnik, 2024-03-22:
@rudrab And when you activate Numberpad via `enabled=1` then keys are sent without any problem? It just works?

ldrahnik, 2024-03-22:
@rudrab Please do the same as previously but with running command `$ sudo libinput debug-events` and post here output, I do not know how to replicate your problem yet so this could help

rudrab, 2024-03-22:
> @rudrab And when you activate Numberpad via `enabled=1` then keys are send without any problem? It just works?

In that case, it is enabled. It can not be toggled

rudrab, 2024-03-22:
> sudo libinput debug-events | grep <eventxyoftouchpad>

sudo libinput debug-events | grep touchpad
[sudo] password for rudra: 
(standard input):                  6:	-event5   DEVICE_ADDED            ASUP1415:00 093A:300C Touchpad    seat0 default group5  cap:pg  size 129x72mm tap(dl off) left scroll-nat scroll-2fg-edge click-buttonareas-clickfinger dwt-on dwtp-on

rudrab, 2024-03-22:
```
sudo libinput debug-events | grep touchpad
[sudo] password for rudra: 
(standard input):                  6:	-event5   DEVICE_ADDED            ASUP1415:00 093A:300C Touchpad    seat0 default group5  cap:pg  size 129x72mm tap(dl off) left scroll-nat scroll-2fg-edge click-buttonareas-clickfinger dwt-on dwtp-on
(standard input):               1102:	event5  - ASUP1415:00 093A:300C Touchpad: kernel bug: Touch jump detected and discarded.
(standard input):               1103:	See https://wayland.freedesktop.org/libinput/doc/1.25.0/touchpad-jumping-cursors.html for details
(standard input):               1122:	event5  - ASUP1415:00 093A:300C Touchpad: kernel bug: Touch jump detected and discarded.
(standard input):               1123:	See https://wayland.freedesktop.org/libinput/doc/1.25.0/touchpad-jumping-cursors.html for details
(standard input):               1158:	event5  - ASUP1415:00 093A:300C Touchpad: kernel bug: Touch jump detected and discarded.
(standard input):               1159:	See https://wayland.freedesktop.org/libinput/doc/1.25.0/touchpad-jumping-cursors.html for details
(standard input):               1348:	event5  - ASUP1415:00 093A:300C Touchpad: kernel bug: Touch jump detected and discarded.
(standard input):               1349:	See https://wayland.freedesktop.org/libinput/doc/1.25.0/touchpad-jumping-cursors.html for details
(standard input):               1470:	event5  - ASUP1415:00 093A:300C Touchpad: kernel bug: Touch jump detected and discarded.
(standard input):               1471:	See https://wayland.freedesktop.org/libinput/doc/1.25.0/touchpad-jumping-cursors.html for details
(standard input):               1472:	event5  - ASUP1415:00 093A:300C Touchpad: WARNING: log rate limit exceeded (5 msgs per 24h). Discarding future messages.

```

rudrab, 2024-03-22:
I am taking time to check how to run libinput record. It is installed, but showing not installed.
 ```
libinput record
libinput: record is not installed
 rudra roddur ~ libinput --help
Usage: libinput [--help|--version] <command> [<args>]

Global options:
  --help ...... show this help and exit
  --version ... show version information and exit

Commands:
  list-devices
	List all devices with their default configuration options

  debug-events
	Print events to stdout

  measure <feature>
	Measure various device properties. See the man page for more info

  analyze <feature>
	Analyze device events. See the man page for more info

  record
	Record event stream from a device node. See the man page for more info

  replay
	Replay a previously recorded event stream. See the man page for more info

```

ldrahnik, 2024-03-22:
```
$ libinput --version
1.20.0
ldrahnik@Zenbook-UP5401EA:~$ libinput record
Available devices:
/dev/input/event0:	Lid Switch
...
Select the device event number: 
```

ldrahnik, 2024-03-22:
@rudrab Or `$ sudo evtest`

rudrab, 2024-03-22:
```
sudo evtest
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	Lid Switch
/dev/input/event1:	Power Button
/dev/input/event10:	PC Speaker
/dev/input/event11:	Asus WMI hotkeys
/dev/input/event12:	sof-hda-dsp Headphone
/dev/input/event13:	sof-hda-dsp HDMI/DP,pcm=3
/dev/input/event14:	sof-hda-dsp HDMI/DP,pcm=4
/dev/input/event15:	sof-hda-dsp HDMI/DP,pcm=5
/dev/input/event16:	ASUP1415:00 093A:300C NumberPad
/dev/input/event17:	ASUP1415:00 093A:300C NumberPad
/dev/input/event2:	Power Button
/dev/input/event3:	AT Translated Set 2 keyboard
/dev/input/event4:	ASUP1415:00 093A:300C Mouse
/dev/input/event5:	ASUP1415:00 093A:300C Touchpad
/dev/input/event6:	ASUP1415:00 093A:300C Keyboard
/dev/input/event7:	Video Bus
/dev/input/event8:	Intel HID events
/dev/input/event9:	Intel HID 5 button array
Select the device event number [0-17]: 17
...
  Event code 272 (BTN_LEFT)
    Event code 273 (BTN_RIGHT)
    Event code 274 (BTN_MIDDLE)
Properties:
Testing ... (interrupt to exit)

```

ldrahnik, 2024-03-22:
@rudrab 16 or 17 because one is stopped installed systemctl service

rudrab, 2024-03-22:
Yes. I have tried to choose both, getting the same result as testing.

ldrahnik, 2024-03-22:
@rudrab Anyway when you do not have activated NumberPad yet you have to listen Touchpad event instead

rudrab, 2024-03-22:
What should I do?

ldrahnik, 2024-03-22:
@rudrab Listen touchpad event instead and try to activate and post here captired output in case both was like I tried both NumberPad events

rudrab, 2024-03-22:
Using the evtest only, right?

ldrahnik, 2024-03-22:
Yeah, might be enough for us, run simultanously with NumberPad and try to activate. I need to see events what your distro consider as jumping cursor.

ldrahnik, 2024-03-23:
Looks like kernel bug because this driver only measures slide gesture and key touch and then is activated - can be reopened.

rudrab, 2024-03-23:
```
/usr/share/asus-numberpad-driver/.env/bin/python3 /usr/share/asus-numberpad-driver/numberpad.py "up5401ea"
2024-03-23 19:39:18,961 INFO     Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
2024-03-23 19:39:18,961 INFO     Set keyboard 3 from H: Handlers=sysrq kbd leds event3
2024-03-23 19:39:18,975 INFO     Detecting touchpad from string: "N: Name="ASUP1415:00 093A:300C Touchpad""
2024-03-23 19:39:18,976 INFO     Set touchpad device id 1 from S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-1/i2c-ASUP1415:00/0018:093A:300C.0001/input/input8
2024-03-23 19:39:18,976 INFO     Set touchpad id 5 from H: Handlers=mouse1 event5
2024-03-23 19:39:18,976 INFO     Touchpad min-max: x 0-3996, y 0-2242
2024-03-23 19:39:18,976 INFO     Numpad min-max: x 200-3796, y 200-2162
2024-03-23 19:40:14,653 INFO     Touched top_right_icon (numlock) in time: 1711203014.6533058
2024-03-23 19:40:14,908 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1711203014.9085107
2024-03-23 19:40:14,976 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1711203014.976372
2024-03-23 19:40:15,316 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1711203015.3159814
2024-03-23 19:40:18,611 INFO     Touched top_right_icon (numlock) in time: 1711203018.6119227
2024-03-23 19:40:18,868 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1711203018.8680298
2024-03-23 19:40:19,480 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1711203019.4799836
2024-03-23 19:40:19,514 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1711203019.5140665
2024-03-23 19:40:20,875 INFO     Touched top_right_icon (numlock) in time: 1711203020.8750446
2024-03-23 19:40:21,132 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1711203021.1321723
2024-03-23 19:40:22,519 INFO     Touched top_right_icon (numlock) in time: 1711203022.5190053
2024-03-23 19:40:23,567 INFO     Slided from top_right_icon taken NOT longer then is required. X, y:
2024-03-23 19:40:23,567 INFO     3910
2024-03-23 19:40:23,567 INFO     1857
2024-03-23 19:40:23,567 INFO     Required is min x, y:
2024-03-23 19:40:23,567 INFO     1198.8
2024-03-23 19:40:23,567 INFO     672.6
2024-03-23 19:40:24,300 INFO     Touched top_right_icon (numlock) in time: 1711203024.3005147
2024-03-23 19:40:24,556 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1711203024.5561721
2024-03-23 19:40:26,732 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1711203026.732101
2024-03-23 19:40:27,344 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1711203027.344127
2024-03-23 19:40:28,126 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1711203028.1263053
2024-03-23 19:40:28,330 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1711203028.3304024
2024-03-23 19:40:28,738 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1711203028.7380958
2024-03-23 19:40:29,245 INFO     Un-touched without NumberPad activation top_right_icon (numlock) in time: 1711203029.2454336
^C2024-03-23 19:40:52,386 ERROR    Listening touchpad events unexpectedly failed
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 1935, in <module>
    listen_touchpad_events()
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 1552, in listen_touchpad_events
    for e in d_t.events():
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages/libevdev/device.py", line 542, in events
    ev = self._libevdev.next_event(flags)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages/libevdev/_clib.py", line 889, in next_event
    rc = self._next_event(self._ctx, flags, ctypes.byref(ev))
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
KeyboardInterrupt
^CTraceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 1962, in <module>
    thread.join()
  File "/usr/lib64/python3.12/threading.py", line 1147, in join
    self._wait_for_tstate_lock()
  File "/usr/lib64/python3.12/threading.py", line 1167, in _wait_for_tstate_lock
    if lock.acquire(block, timeout):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
KeyboardInterrupt

```
This is evtest output. 

```
sudo evtest
[sudo] password for rudra: 
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	Lid Switch
/dev/input/event1:	Power Button
/dev/input/event10:	PC Speaker
/dev/input/event11:	Asus WMI hotkeys
/dev/input/event12:	sof-hda-dsp Headphone
[... 46 lines cut ...]
    Event code 73 (KEY_KP9)
    Event code 74 (KEY_KPMINUS)
    Event code 75 (KEY_KP4)
    Event code 76 (KEY_KP5)
    Event code 77 (KEY_KP6)
    Event code 78 (KEY_KPPLUS)
    Event code 79 (KEY_KP1)
    Event code 80 (KEY_KP2)
    Event code 81 (KEY_KP3)
    Event code 82 (KEY_KP0)
    Event code 83 (KEY_KPDOT)
    Event code 96 (KEY_KPENTER)
    Event code 98 (KEY_KPSLASH)
    Event code 117 (KEY_KPEQUAL)
    Event code 140 (KEY_CALC)
    Event code 272 (BTN_LEFT)
    Event code 273 (BTN_RIGHT)
    Event code 274 (BTN_MIDDLE)
Properties:
Testing ... (interrupt to exit)
expected 24 bytes, got -1

evtest: error reading: No such device

```

rudrab, 2024-03-23:
@ldrahnik I am extremly grateful that you have spent so much time on this. I was afk for 1 day so was unable to post the evtest output.  I can open a bug in bugzilla, but I am not sure what specific point I should raise.

ldrahnik, 2024-03-23:
@rudrab https://github.com/bulletmark/libinput-gestures/issues/236
after reading this it looks like to me current state of your laptop is somehow broken, get me unworking clean distro (.iso) and I able help you. What I would do be you? I would try figure out why does not work `$ libinput record`, maybe associated with the problem somehow. Anyway this issue Is not related to this driver but let mě know about your progress I am interested what caused the problém.

Johnmcenroyy, 2024-09-20:
Also the same problem with Asus UX3405MA.
Linux kernel is 6.11, I am running from terminal for testing:
```python3 /usr/share/asus-numberpad-driver/numberpad.py up5401ea ~/.config/asus-numberpad-driver/ /usr/share/asus-numberpad-driver```

Can't toggle numpad with corner, only with enable=1 it works and so can't disable it.

ldrahnik, 2024-09-20:
@Johnmcenroyy Could you please try `git clone` this repository and post here clean error log?

ldrahnik, 2024-09-20:
@Johnmcenroyy I noticed that previous @rudrab had installed the driver two times, so when he activated the NumberPad, NumberPad was immediately inactivated. Don you have the same problem? 

```
...
/dev/input/event16:	ASUP1415:00 093A:300C NumberPad
/dev/input/event17:	ASUP1415:00 093A:300C NumberPad
...
```

Johnmcenroyy, 2024-09-20:
@ldrahnik For now I disabled all systemd services, beside only user service is launched, not from aur. So if I suddenly forgotten something seems there are no others services.
My debug log (nearly 5 seconds of click on right top corner of touchpad):

```
2024-09-21 00:29:31,880 INFO X11 detected and connected succesfully to the display :0.0
2024-09-21 00:29:31,880 INFO Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
2024-09-21 00:29:31,880 INFO Set keyboard 3 from H: Handlers=sysrq kbd leds event3
2024-09-21 00:29:31,937 INFO Detecting touchpad from string: "N: Name="ASUP1415:00 093A:300C Touchpad""
2024-09-21 00:29:31,937 INFO Set touchpad device id 14 from S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-14/i2c-ASUP1415:00/0018:093A:300C.0001/input/input13
2024-09-21 00:29:31,937 INFO Set touchpad id 10 from H: Handlers=event10 mouse1
2024-09-21 00:29:31,938 INFO Touchpad min-max: x 0-3996, y 0-2242
2024-09-21 00:29:31,938 INFO Numpad min-max: x 200-3796, y 200-2162
2024-09-21 00:29:31,938 DEBUG X11 will try to load keymap
2024-09-21 00:29:31,943 DEBUG X11 loaded keymap succesfully
2024-09-21 00:29:31,943 DEBUG {'Num_Lock': KEY_NUMLOCK:69, '0': KEY_0:11, '1': KEY_1:2, '2': KEY_2:3, '3': KEY_3:4, '4': KEY_4:5, '5': KEY_5:6, '6': KEY_6:7, '7': KEY_7:8, '8': KEY_8:9, '9': KEY_9:10, 'a': KEY_A:30, 'b': KEY_B:48, 'c': KEY_C:46, 'd': KEY_D:32, 'e': KEY_E:18, 'f': KEY_F:33, 'Shift_L': KEY_LEFTSHIFT:42, 'Control_L': KEY_LEFTCTRL:29, 'u': KEY_U:22, 'space': KEY_SPACE:57, 'slash': KEY_SLASH:53, 'BackSpace': KEY_BACKSPACE:14, 'asterisk': [KEY_LEFTSHIFT:42, KEY_8:9], 'minus': KEY_MINUS:12, 'percent': [KEY_LEFTSHIFT:42, KEY_5:6], 'period': KEY_DOT:52, 'Return': KEY_ENTER:28, 'plus': [KEY_LEFTSHIFT:42, KEY_EQUAL:13], 'equal': KEY_EQUAL:13}
2024-09-21 00:29:32,446 DEBUG Writting to config file: "<_io.TextIOWrapper name='/home/nikita/.config/asus-numberpad-driver/numberpad_dev' mode='w' encoding='UTF-8'>"
2024-09-21 00:29:38,518 DEBUG set_tracking_id starts
2024-09-21 00:29:38,518 DEBUG 75
2024-09-21 00:29:38,518 DEBUG Started new slot
2024-09-21 00:29:38,518 DEBUG None
2024-09-21 00:29:38,519 DEBUG finger down at x 3926 y 142
2024-09-21 00:29:38,519 INFO Touched top_right_icon area (numlock) in time: 1726867778.5192108
2024-09-21 00:29:38,777 DEBUG finger down at x 3922 y 149
2024-09-21 00:29:38,777 INFO Un-touched top_right_icon area (representing numlock key) in time: 1726867778.777857
2024-09-21 00:29:38,814 DEBUG finger down at x -1 y 150
2024-09-21 00:29:38,851 DEBUG finger down at x 3921 y 150
2024-09-21 00:29:38,851 INFO Un-touched top_right_icon area (representing numlock key) in time: 1726867778.8510954
2024-09-21 00:29:38,887 DEBUG finger down at x -1 y -1
2024-09-21 00:29:38,924 DEBUG finger down at x 3920 y -1
2024-09-21 00:29:38,960 DEBUG finger down at x 3920 y 151
2024-09-21 00:29:38,961 INFO Un-touched top_right_icon area (representing numlock key) in time: 1726867778.961069
2024-09-21 00:29:38,997 DEBUG finger down at x -1 y 152
2024-09-21 00:29:39,034 DEBUG finger down at x -1 y 153
2024-09-21 00:29:39,071 DEBUG finger down at x -1 y 154
2024-09-21 00:29:39,107 DEBUG finger down at x -1 y 154
2024-09-21 00:29:39,144 DEBUG finger down at x -1 y 154
2024-09-21 00:29:39,181 DEBUG finger down at x -1 y 154
2024-09-21 00:29:39,217 DEBUG finger down at x -1 y 155
2024-09-21 00:29:39,254 DEBUG finger down at x -1 y 155
2024-09-21 00:29:39,291 DEBUG finger down at x -1 y 156
2024-09-21 00:29:39,327 DEBUG finger down at x -1 y 156
2024-09-21 00:29:39,364 DEBUG finger down at x -1 y 156
2024-09-21 00:29:39,401 DEBUG finger down at x -1 y 156
2024-09-21 00:29:39,437 DEBUG finger down at x -1 y 156
2024-09-21 00:29:39,474 DEBUG finger down at x 3921 y 158
2024-09-21 00:29:39,474 INFO Un-touched top_right_icon area (representing numlock key) in time: 1726867779.4744854
2024-09-21 00:29:39,511 DEBUG finger down at x -1 y -1
2024-09-21 00:29:39,547 DEBUG finger down at x -1 y 159
2024-09-21 00:29:39,584 DEBUG finger down at x -1 y 160
2024-09-21 00:29:39,624 DEBUG finger down at x 3922 y 160
2024-09-21 00:29:39,624 INFO Un-touched top_right_icon area (representing numlock key) in time: 1726867779.624812
2024-09-21 00:29:39,660 DEBUG finger down at x -1 y -1
2024-09-21 00:29:39,697 DEBUG finger down at x -1 y 162
2024-09-21 00:29:39,734 DEBUG finger down at x -1 y 162
2024-09-21 00:29:39,771 DEBUG finger down at x -1 y 162
2024-09-21 00:29:39,807 DEBUG finger down at x -1 y 162
2024-09-21 00:29:39,844 DEBUG finger down at x -1 y 162
2024-09-21 00:29:39,881 DEBUG finger down at x -1 y 162
2024-09-21 00:29:39,917 DEBUG finger down at x -1 y 162
2024-09-21 00:29:39,954 DEBUG finger down at x -1 y 162
2024-09-21 00:29:39,991 DEBUG finger down at x -1 y 162
2024-09-21 00:29:40,027 DEBUG finger down at x -1 y 162
2024-09-21 00:29:40,064 DEBUG finger down at x -1 y 162
2024-09-21 00:29:40,101 DEBUG finger down at x -1 y 162
2024-09-21 00:29:40,137 DEBUG finger down at x -1 y 164
2024-09-21 00:29:40,177 DEBUG finger down at x -1 y 164
2024-09-21 00:29:40,214 DEBUG finger down at x -1 y 165
2024-09-21 00:29:40,251 DEBUG finger down at x -1 y 165
2024-09-21 00:29:40,287 DEBUG finger down at x -1 y 165
2024-09-21 00:29:40,324 DEBUG finger down at x -1 y 165
[... 178 lines cut ...]
2024-09-21 00:29:46,921 DEBUG finger down at x -1 y 190
2024-09-21 00:29:46,961 DEBUG finger down at x -1 y 190
2024-09-21 00:29:46,997 DEBUG finger down at x -1 y 190
[... cut ...]
2024-09-21 00:29:47,364 DEBUG finger down at x 3923 y 190
2024-09-21 00:29:47,364 INFO Un-touched top_right_icon area (representing numlock key) in time: 1726867787.364289
2024-09-21 00:29:47,366 DEBUG set_tracking_id starts
2024-09-21 00:29:47,366 DEBUG -1
2024-09-21 00:29:47,366 DEBUG Ended existing slot
2024-09-21 00:29:47,366 DEBUG finger down at x -1 y -1
```
Maybe it something that with this model UX3405MA

Manual backlight works also:
```
i2ctransfer -f -y 14 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad
```

Johnmcenroyy, 2024-09-20:
```
-event2   DEVICE_ADDED            Power Button                      seat0 default group1  cap:k
-event4   DEVICE_ADDED            Video Bus                         seat0 default group2  cap:k
-event0   DEVICE_ADDED            Lid Switch                        seat0 default group3  cap:S
-event1   DEVICE_ADDED            Power Button                      seat0 default group4  cap:k
-event9   DEVICE_ADDED            ASUP1415:00 093A:300C Mouse       seat0 default group5  cap:p left scroll-nat scroll-button
-event10  DEVICE_ADDED            ASUP1415:00 093A:300C Touchpad    seat0 default group5  cap:pg  size 129x72mm tap(dl off) left scroll-nat scroll-2fg-edge click-buttonareas-clickfinger dwt-on dwtp-on
-event11  DEVICE_ADDED            ASUP1415:00 093A:300C Keyboard    seat0 default group5  cap:k
-event5   DEVICE_ADDED            Intel HID events                  seat0 default group6  cap:k
-event6   DEVICE_ADDED            Intel HID 5 button array          seat0 default group7  cap:k
-event8   DEVICE_ADDED            Asus WMI hotkeys                  seat0 default group8  cap:k
-event3   DEVICE_ADDED            AT Translated Set 2 keyboard      seat0 default group9  cap:k
```

Johnmcenroyy, 2024-09-20:
With driver
```
-event2   DEVICE_ADDED            Power Button                      seat0 default group1  cap:k
-event4   DEVICE_ADDED            Video Bus                         seat0 default group2  cap:k
-event0   DEVICE_ADDED            Lid Switch                        seat0 default group3  cap:S
-event1   DEVICE_ADDED            Power Button                      seat0 default group4  cap:k
-event9   DEVICE_ADDED            ASUP1415:00 093A:300C Mouse       seat0 default group5  cap:p left scroll-nat scroll-button
-event10  DEVICE_ADDED            ASUP1415:00 093A:300C Touchpad    seat0 default group5  cap:pg  size 129x72mm tap(dl off) left scroll-nat scroll-2fg-edge click-buttonareas-clickfinger dwt-on dwtp-on
-event11  DEVICE_ADDED            ASUP1415:00 093A:300C Keyboard    seat0 default group5  cap:k
-event5   DEVICE_ADDED            Intel HID events                  seat0 default group6  cap:k
-event6   DEVICE_ADDED            Intel HID 5 button array          seat0 default group7  cap:k
-event8   DEVICE_ADDED            Asus WMI hotkeys                  seat0 default group8  cap:k
-event3   DEVICE_ADDED            AT Translated Set 2 keyboard      seat0 default group9  cap:k
-event16  DEVICE_ADDED            ASUP1415:00 093A:300C NumberPad   seat0 default group10 cap:kp left scroll-nat
```

ldrahnik, 2024-09-21:
@Johnmcenroyy And your config looks like? Located `/usr/share/asus-numberpad-driver/numberpad_dev`.

rudrab, 2024-09-21:
this is my config:
```
[main]
numpad_disables_sys_numlock = 1
disable_due_inactivity_time = 0
touchpad_disables_numpad = 1
key_repetitions = 0
multitouch = 0
one_touch_key_rotation = 0
activation_time = 1
sys_numlock_enables_numpad = 1
top_left_icon_activation_time = 1
top_left_icon_slide_func_activation_radius = 1200
top_left_icon_slide_func_activates_numpad = 1
top_right_icon_slide_func_activation_radius = 1200
enabled_touchpad_pointer = 3
press_key_when_is_done_untouch = 1
enabled = 0
default_backlight_level = 0x01
brightness = 0x48
top_left_icon_brightness_func_disabled = 0
distance_to_move_only_pointer = 0
idled = 0
idle_brightness = 30
idle_enabled = 0
idle_time = 5
```

error log file is attached. 
[error.log](https://github.com/user-attachments/files/17084355/error.log)

ldrahnik, 2024-09-21:
@rudrab And in your case activation works but inactivation do nothing?

rudrab, 2024-09-21:
Time bound inactivation is not doing anything. 
I have found the gesture: for activation, i have to bring my finger from top right towars '-' symbol; for inactivation, I have to bring my finger from top right to straight down. Just touching the symbol at top right is not doing anything,

Johnmcenroyy, 2024-09-21:
@ldrahnik It was by mistake, my config is in home folder. Anyway it didn't work - toggle button didn't work. Maybe I have done something wrong, I read documentation but how it is for now. I have edited directly numberpad.py and now it works. So I am running driver (edited version) from autostart in XFCE:
```/usr/bin/python3 ~/bin/asus-numberpad-driver/numberpad-fixed.py up5401ea ~/.config/asus-numberpad-driver/```

This is how I have edited the file:
```
--- ./numberpad.py	2024-09-20 16:19:30.531552740 +0300
+++ ./numberpad.py	2024-09-21 17:09:33.993572707 +0300
@@ -5,6 +5,7 @@
 import logging
 import math
 import os
+import psutil
 import re
 import subprocess
 import sys
@@ -1860,6 +1861,13 @@
         log.info("Un-touched numlock key (not top_right_icon) in time: %s", time())
 
 
+def process_status(process_name):
+    for process in psutil.process_iter(['pid', 'name']):
+        if process.info['name'] == process_name:
+            return True
+    return False
+
+
 def pressed_touchpad_top_right_icon(value):
     global top_right_icon_touch_start_time, numlock_touch_start_time, abs_mt_slot_numpad_key
 
@@ -1871,6 +1879,18 @@
         numlock_touch_start_time = time()
 
         abs_mt_slot_numpad_key[abs_mt_slot_value] = get_evdev_key_for_char('Num_Lock')
+
+        ############################################
+        # MY ADDITION
+        ############################################
+        command = 'xdotool key Num_Lock'
+        subprocess.check_call(command.split())
+        process_name = "gnome-calculator"
+        if process_status(process_name):
+            subprocess.Popen('killall gnome-calculator', shell=True)
+        else:
+            subprocess.Popen('gnome-calculator', shell=True)
+        ############################################
     else:
         top_right_icon_touch_start_time = 0
         numlock_touch_start_time = 0
```
Now I am clicking on top right corner of touchpad - backlight on and gnome-calculator appears, 
and off after clicking again.

Anyway thanks for your time and writing such a useful driver. If I know something more I will write here.

rudrab, 2024-09-21:
One possible problem is gnome 47 is a complete wayland inplementation...rhere is no gnome on xorg version

Johnmcenroyy, 2024-09-21:
ydotool is an alternative to xdotool that works on both X11 and Wayland.
It can be used on anything as long as it accepts keyboard/mouse/whatever input. 
For example, X11, text console, "RetroArch OS", fbdev apps (fbterm/mplayer/SDL1/LittleVGL/Qt Embedded), etc.

ldrahnik, 2024-09-21:
In case I replicated this problem on my laptop (i have not replicated yet) I would try figure out when is filled -1 to x and y axis when is touched finger.

rudrab, 2024-09-21:
Thanks a lot for your support

ldrahnik, 2024-09-22:
@rudrab @Johnmcenroyy Could you please record touch on top right corner on your Touchpad (NumberPad do not have to be installed) and post here result? Like me, I picked touchpad `ASUE140D:00 04F3:31B9 Touchpad`:

```
evtest
No device specified, trying to scan all of /dev/input/event*
Not running as root, no devices may be available.
Available devices:
/dev/input/event0:	Lid Switch
/dev/input/event1:	Power Button
/dev/input/event2:	Power Button
/dev/input/event3:	AT Translated Set 2 keyboard
/dev/input/event4:	Asus WMI hotkeys
/dev/input/event5:	ELAN9009:00 04F3:2E36
/dev/input/event6:	ELAN9009:00 04F3:2E36 UNKNOWN
/dev/input/event7:	ELAN9009:00 04F3:2E36 UNKNOWN
/dev/input/event8:	ELAN9009:00 04F3:2E36 Stylus
/dev/input/event9:	ELAN9009:00 04F3:2E36 UNKNOWN
/dev/input/event10:	ELAN9009:00 04F3:2E36 UNKNOWN
/dev/input/event11:	ASUE140D:00 04F3:31B9 Mouse
/dev/input/event12:	ASUE140D:00 04F3:31B9 Touchpad
/dev/input/event13:	ASUE140D:00 04F3:31B9 Keyboard
/dev/input/event14:	Video Bus
/dev/input/event15:	E-Signal/A-One USB Gaming Mouse
/dev/input/event16:	E-Signal/A-One USB Gaming Mouse Keyboard
/dev/input/event17:	Intel HID events
/dev/input/event18:	ELAN9009:00 04F3:2E36 UNKNOWN
/dev/input/event19:	ELAN9009:00 04F3:2E36 Stylus asus-stylus-driver
/dev/input/event20:	sof-hda-dsp Headphone
/dev/input/event21:	sof-hda-dsp HDMI/DP,pcm=3
/dev/input/event22:	sof-hda-dsp HDMI/DP,pcm=4
/dev/input/event23:	sof-hda-dsp HDMI/DP,pcm=5
/dev/input/event24:	Asus WMI hotkeys
/dev/input/event25:	ASUE140D:00 04F3:31B9 NumberPad
Select the device event number [0-25]: 12
Input driver version is 1.0.1
Input device ID: bus 0x18 vendor 0x4f3 product 0x31b9 version 0x100
Input device name: "ASUE140D:00 04F3:31B9 Touchpad"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 272 (BTN_LEFT)
    Event code 325 (BTN_TOOL_FINGER)
    Event code 328 (BTN_TOOL_QUINTTAP)
    Event code 330 (BTN_TOUCH)
    Event code 333 (BTN_TOOL_DOUBLETAP)
    Event code 334 (BTN_TOOL_TRIPLETAP)
    Event code 335 (BTN_TOOL_QUADTAP)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value   3880
      Min        0
      Max     3880
      Resolution      31
    Event code 1 (ABS_Y)
      Value      0
      Min        0
      Max     2299
      Resolution      32
    Event code 47 (ABS_MT_SLOT)
      Value      0
      Min        0
      Max        4
    Event code 53 (ABS_MT_POSITION_X)
      Value      0
      Min        0
      Max     3880
      Resolution      31
    Event code 54 (ABS_MT_POSITION_Y)
      Value      0
      Min        0
[... 322 lines cut ...]
Event: time 1726994645.007879, -------------- SYN_REPORT ------------
Event: time 1726994645.014834, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 3838
Event: time 1726994645.014834, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 66
Event: time 1726994645.014834, type 3 (EV_ABS), code 0 (ABS_X), value 3838
Event: time 1726994645.014834, type 3 (EV_ABS), code 1 (ABS_Y), value 66
Event: time 1726994645.014834, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 833000
Event: time 1726994645.014834, -------------- SYN_REPORT ------------
Event: time 1726994645.021469, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 3841
Event: time 1726994645.021469, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 61
Event: time 1726994645.021469, type 3 (EV_ABS), code 0 (ABS_X), value 3841
Event: time 1726994645.021469, type 3 (EV_ABS), code 1 (ABS_Y), value 61
Event: time 1726994645.021469, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 840000
Event: time 1726994645.021469, -------------- SYN_REPORT ------------
Event: time 1726994645.028428, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 3843
Event: time 1726994645.028428, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 57
Event: time 1726994645.028428, type 3 (EV_ABS), code 0 (ABS_X), value 3843
Event: time 1726994645.028428, type 3 (EV_ABS), code 1 (ABS_Y), value 57
Event: time 1726994645.028428, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 847000
Event: time 1726994645.028428, -------------- SYN_REPORT ------------
Event: time 1726994645.035084, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value -1
Event: time 1726994645.035084, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1726994645.035084, type 1 (EV_KEY), code 325 (BTN_TOOL_FINGER), value 0
Event: time 1726994645.035084, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 854000
Event: time 1726994645.035084, -------------- SYN_REPORT ------------
```

Johnmcenroyy, 2024-09-22:
@ldrahnik One click on top right corner of touchpad without Numberpad driver installed:

```
$ evtest
No device specified, trying to scan all of /dev/input/event*
Not running as root, no devices may be available.
Available devices:
/dev/input/event0:	Lid Switch
/dev/input/event1:	Power Button
/dev/input/event10:	ASUP1415:00 093A:300C Touchpad
/dev/input/event11:	ASUP1415:00 093A:300C Keyboard
/dev/input/event12:	sof-hda-dsp Headphone
/dev/input/event13:	sof-hda-dsp HDMI/DP,pcm=3
/dev/input/event14:	sof-hda-dsp HDMI/DP,pcm=4
/dev/input/event15:	sof-hda-dsp HDMI/DP,pcm=5
/dev/input/event2:	Power Button
/dev/input/event3:	AT Translated Set 2 keyboard
/dev/input/event4:	Video Bus
/dev/input/event5:	Intel HID events
/dev/input/event6:	Intel HID 5 button array
/dev/input/event7:	PC Speaker
/dev/input/event8:	Asus WMI hotkeys
/dev/input/event9:	ASUP1415:00 093A:300C Mouse
Select the device event number [0-15]: 10
Input driver version is 1.0.1
Input device ID: bus 0x18 vendor 0x93a product 0x300c version 0x100
Input device name: "ASUP1415:00 093A:300C Touchpad"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 272 (BTN_LEFT)
    Event code 325 (BTN_TOOL_FINGER)
    Event code 328 (BTN_TOOL_QUINTTAP)
    Event code 330 (BTN_TOUCH)
    Event code 333 (BTN_TOOL_DOUBLETAP)
    Event code 334 (BTN_TOOL_TRIPLETAP)
    Event code 335 (BTN_TOOL_QUADTAP)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value   3920
      Min        0
      Max     3996
      Resolution      31
    Event code 1 (ABS_Y)
      Value    138
      Min        0
      Max     2242
      Resolution      31
    Event code 47 (ABS_MT_SLOT)
      Value      0
      Min        0
      Max        4
    Event code 53 (ABS_MT_POSITION_X)
      Value      0
      Min        0
      Max     3996
      Resolution      31
    Event code 54 (ABS_MT_POSITION_Y)
      Value      0
      Min        0
      Max     2242
      Resolution      31
    Event code 55 (ABS_MT_TOOL_TYPE)
      Value      0
      Min        0
      Max        2
    Event code 57 (ABS_MT_TRACKING_ID)
      Value      0
      Min        0
      Max    65535
  Event type 4 (EV_MSC)
    Event code 5 (MSC_TIMESTAMP)
Key repeat handling:
  Repeat type 20 (EV_REP)
    Repeat code 0 (REP_DELAY)
      Value    250
    Repeat code 1 (REP_PERIOD)
      Value     33
Properties:
  Property type 0 (INPUT_PROP_POINTER)
  Property type 2 (INPUT_PROP_BUTTONPAD)
Testing ... (interrupt to exit)
Event: time 1726998849.378650, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 38
Event: time 1726998849.378650, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 3944
Event: time 1726998849.378650, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 144
Event: time 1726998849.378650, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1726998849.378650, type 1 (EV_KEY), code 325 (BTN_TOOL_FINGER), value 1
Event: time 1726998849.378650, type 3 (EV_ABS), code 0 (ABS_X), value 3944
Event: time 1726998849.378650, type 3 (EV_ABS), code 1 (ABS_Y), value 144
Event: time 1726998849.378650, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 0
Event: time 1726998849.378650, -------------- SYN_REPORT ------------
Event: time 1726998849.388737, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 7100
Event: time 1726998849.388737, -------------- SYN_REPORT ------------
Event: time 1726998849.395882, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 14500
Event: time 1726998849.395882, -------------- SYN_REPORT ------------
Event: time 1726998849.402983, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 21400
Event: time 1726998849.402983, -------------- SYN_REPORT ------------
Event: time 1726998849.410146, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 28500
Event: time 1726998849.410146, -------------- SYN_REPORT ------------
Event: time 1726998849.417262, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 35500
Event: time 1726998849.417262, -------------- SYN_REPORT ------------
Event: time 1726998849.424393, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 42600
Event: time 1726998849.424393, -------------- SYN_REPORT ------------
Event: time 1726998849.431546, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 48800
Event: time 1726998849.431546, -------------- SYN_REPORT ------------
Event: time 1726998849.438681, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value -1
Event: time 1726998849.438681, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1726998849.438681, type 1 (EV_KEY), code 325 (BTN_TOOL_FINGER), value 0
Event: time 1726998849.438681, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 55600
Event: time 1726998849.438681, -------------- SYN_REPORT ------------
```

rudrab, 2024-09-22:
My evtest report (the driver is already installed. Do you want me to uninstall and post the evtest?)
```
evtest 
No device specified, trying to scan all of /dev/input/event*
Not running as root, no devices may be available.
Available devices:
/dev/input/event0:	Lid Switch
/dev/input/event1:	Power Button
/dev/input/event10:	PC Speaker
/dev/input/event11:	Asus WMI hotkeys
/dev/input/event12:	sof-hda-dsp Headphone
/dev/input/event13:	sof-hda-dsp HDMI/DP,pcm=3
/dev/input/event14:	sof-hda-dsp HDMI/DP,pcm=4
/dev/input/event15:	sof-hda-dsp HDMI/DP,pcm=5
/dev/input/event16:	ASUP1415:00 093A:300C NumberPad
/dev/input/event2:	Power Button
/dev/input/event3:	AT Translated Set 2 keyboard
/dev/input/event4:	ASUP1415:00 093A:300C Mouse
/dev/input/event5:	ASUP1415:00 093A:300C Touchpad
/dev/input/event6:	ASUP1415:00 093A:300C Keyboard
/dev/input/event7:	Video Bus
/dev/input/event8:	Intel HID events
/dev/input/event9:	Intel HID 5 button array
Select the device event number [0-16]: 5
Input driver version is 1.0.1
Input device ID: bus 0x18 vendor 0x93a product 0x300c version 0x100
Input device name: "ASUP1415:00 093A:300C Touchpad"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 272 (BTN_LEFT)
    Event code 325 (BTN_TOOL_FINGER)
    Event code 328 (BTN_TOOL_QUINTTAP)
    Event code 330 (BTN_TOUCH)
    Event code 333 (BTN_TOOL_DOUBLETAP)
    Event code 334 (BTN_TOOL_TRIPLETAP)
    Event code 335 (BTN_TOOL_QUADTAP)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value   1935
      Min        0
      Max     3996
      Resolution      31
    Event code 1 (ABS_Y)
      Value    981
      Min        0
      Max     2242
      Resolution      31
    Event code 47 (ABS_MT_SLOT)
      Value      0
      Min        0
      Max        4
    Event code 53 (ABS_MT_POSITION_X)
      Value      0
      Min        0
      Max     3996
      Resolution      31
    Event code 54 (ABS_MT_POSITION_Y)
      Value      0
      Min        0
      Max     2242
      Resolution      31
    Event code 55 (ABS_MT_TOOL_TYPE)
      Value      0
      Min        0
      Max        2
    Event code 57 (ABS_MT_TRACKING_ID)
      Value      0
      Min        0
      Max    65535
[... 269 lines cut ...]
Event: time 1726998884.862589, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 731100
Event: time 1726998884.862589, -------------- SYN_REPORT ------------
Event: time 1726998884.869777, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 738100
Event: time 1726998884.869777, -------------- SYN_REPORT ------------
Event: time 1726998884.877011, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 745100
Event: time 1726998884.877011, -------------- SYN_REPORT ------------
Event: time 1726998884.884162, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 752100
Event: time 1726998884.884162, -------------- SYN_REPORT ------------
Event: time 1726998884.891391, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 759200
Event: time 1726998884.891391, -------------- SYN_REPORT ------------
Event: time 1726998884.898579, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 136
Event: time 1726998884.898579, type 3 (EV_ABS), code 1 (ABS_Y), value 136
Event: time 1726998884.898579, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 766200
Event: time 1726998884.898579, -------------- SYN_REPORT ------------
Event: time 1726998884.905815, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 773200
Event: time 1726998884.905815, -------------- SYN_REPORT ------------
Event: time 1726998884.913013, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 779500
Event: time 1726998884.913013, -------------- SYN_REPORT ------------
Event: time 1726998884.920225, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value -1
Event: time 1726998884.920225, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1726998884.920225, type 1 (EV_KEY), code 325 (BTN_TOOL_FINGER), value 0
Event: time 1726998884.920225, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 786300
Event: time 1726998884.920225, -------------- SYN_REPORT ------------

```

ldrahnik, 2024-09-22:
@rudrab Now please install/run this branch https://github.com/asus-linux-drivers/asus-numberpad-driver/tree/169-not-toggling-numpad-to-touchpad and post here error log (cleaned before) with enabled logging of each event, it could provide insight into what is going wrong.

ldrahnik, 2024-09-22:
Mine

```
cd /home/ldrahnik/projects/asus-numberpad-driver ; /usr/bin/env /bin/python3 /home/ldrahnik/.vscode/extensions/ms-python.debugpy-2024.10.0-linux-x64/bundled/libs/debugpy/adapter/../../debugpy/launcher 52279 -- /home/ldrahnik/projects/asus-numberpad-driver/numberpad.py up5401ea 
2024-09-22 12:47:55,959 INFO X11 detected and connected succesfully to the display :0
2024-09-22 12:47:55,961 INFO Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
2024-09-22 12:47:55,961 INFO Set keyboard 3 from H: Handlers=sysrq kbd event3 leds
2024-09-22 12:47:55,969 INFO Detecting touchpad from string: "N: Name="ASUE140D:00 04F3:31B9 Touchpad""
2024-09-22 12:47:55,969 INFO Set touchpad device id 2 from S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-2/i2c-ASUE140D:00/0018:04F3:31B9.0002/input/input31
2024-09-22 12:47:55,969 INFO Set touchpad id 12 from H: Handlers=mouse3 event12
2024-09-22 12:47:55,970 INFO Touchpad min-max: x 0-3880, y 0-2299
2024-09-22 12:47:55,970 INFO Numpad min-max: x 200-3680, y 200-2219
2024-09-22 12:47:55,971 DEBUG X11 will try to load keymap
2024-09-22 12:47:55,977 DEBUG X11 loaded keymap succesfully
2024-09-22 12:47:55,977 DEBUG {'Num_Lock': KEY_NUMLOCK:69, '0': KEY_0:11, '1': KEY_1:2, '2': KEY_2:3, '3': KEY_3:4, '4': KEY_4:5, '5': KEY_5:6, '6': KEY_6:7, '7': KEY_7:8, '8': KEY_8:9, '9': KEY_9:10, 'a': KEY_A:30, 'b': KEY_B:48, 'c': KEY_C:46, 'd': KEY_D:32, 'e': KEY_E:18, 'f': KEY_F:33, 'Shift_L': KEY_LEFTSHIFT:42, 'Control_L': KEY_LEFTCTRL:29, 'u': KEY_U:22, 'space': KEY_SPACE:57, 'slash': KEY_SLASH:53, 'BackSpace': KEY_BACKSPACE:14, 'asterisk': [KEY_LEFTSHIFT:42, KEY_8:9], 'minus': KEY_MINUS:12, 'percent': [KEY_LEFTSHIFT:42, KEY_5:6], 'period': KEY_DOT:52, 'Return': KEY_ENTER:28, 'plus': [KEY_LEFTSHIFT:42, KEY_EQUAL:13], 'equal': KEY_EQUAL:13}
2024-09-22 12:47:56,489 DEBUG Writting to config file: "<_io.TextIOWrapper name='./numberpad_dev' mode='w' encoding='UTF-8'>"
2024-09-22 12:47:57,432 INFO InputEvent(EV_ABS, ABS_MT_TRACKING_ID, 51)
2024-09-22 12:47:57,432 DEBUG set_tracking_id starts
2024-09-22 12:47:57,432 DEBUG 51
2024-09-22 12:47:57,432 DEBUG Started new slot
2024-09-22 12:47:57,432 DEBUG None
2024-09-22 12:47:57,432 INFO InputEvent(EV_ABS, ABS_MT_POSITION_X, 3843)
2024-09-22 12:47:57,432 INFO InputEvent(EV_ABS, ABS_MT_POSITION_Y, 100)
2024-09-22 12:47:57,433 INFO InputEvent(EV_KEY, BTN_TOUCH, 1)
2024-09-22 12:47:57,433 INFO InputEvent(EV_KEY, BTN_TOOL_FINGER, 1)
2024-09-22 12:47:57,433 DEBUG finger down at x 3843 y 100
2024-09-22 12:47:57,433 INFO Touched top_right_icon area (numlock) in time: 1727002077.4337506
2024-09-22 12:47:57,433 INFO InputEvent(EV_ABS, ABS_X, 3843)
2024-09-22 12:47:57,434 INFO InputEvent(EV_ABS, ABS_Y, 100)
2024-09-22 12:47:57,434 INFO InputEvent(EV_MSC, MSC_TIMESTAMP, 0)
2024-09-22 12:47:57,434 INFO InputEvent(EV_SYN, SYN_REPORT, 0)
2024-09-22 12:47:57,435 INFO InputEvent(EV_MSC, MSC_TIMESTAMP, 14000)
2024-09-22 12:47:57,435 INFO InputEvent(EV_SYN, SYN_REPORT, 0)
2024-09-22 12:47:57,442 INFO InputEvent(EV_MSC, MSC_TIMESTAMP, 21000)
2024-09-22 12:47:57,442 INFO InputEvent(EV_SYN, SYN_REPORT, 0)
2024-09-22 12:47:57,449 INFO InputEvent(EV_MSC, MSC_TIMESTAMP, 28000)
2024-09-22 12:47:57,449 INFO InputEvent(EV_SYN, SYN_REPORT, 0)
2024-09-22 12:47:57,456 INFO InputEvent(EV_MSC, MSC_TIMESTAMP, 35000)
2024-09-22 12:47:57,456 INFO InputEvent(EV_SYN, SYN_REPORT, 0)
2024-09-22 12:47:57,462 INFO InputEvent(EV_ABS, ABS_MT_POSITION_X, 3841)
2024-09-22 12:47:57,463 INFO InputEvent(EV_ABS, ABS_MT_POSITION_Y, 101)
2024-09-22 12:47:57,463 INFO InputEvent(EV_ABS, ABS_X, 3841)
2024-09-22 12:47:57,463 INFO InputEvent(EV_ABS, ABS_Y, 101)
2024-09-22 12:47:57,463 INFO InputEvent(EV_MSC, MSC_TIMESTAMP, 42000)
2024-09-22 12:47:57,464 INFO InputEvent(EV_SYN, SYN_REPORT, 0)
2024-09-22 12:47:57,469 INFO InputEvent(EV_ABS, ABS_MT_POSITION_X, 3839)
2024-09-22 12:47:57,470 INFO InputEvent(EV_ABS, ABS_MT_POSITION_Y, 102)
2024-09-22 12:47:57,470 INFO InputEvent(EV_ABS, ABS_X, 3839)
2024-09-22 12:47:57,470 INFO InputEvent(EV_ABS, ABS_Y, 102)
2024-09-22 12:47:57,470 INFO InputEvent(EV_MSC, MSC_TIMESTAMP, 49000)
2024-09-22 12:47:57,470 INFO InputEvent(EV_SYN, SYN_REPORT, 0)
2024-09-22 12:47:57,476 INFO InputEvent(EV_ABS, ABS_MT_POSITION_X, 3837)
2024-09-22 12:47:57,476 INFO InputEvent(EV_ABS, ABS_MT_POSITION_Y, 103)
2024-09-22 12:47:57,477 INFO InputEvent(EV_ABS, ABS_X, 3837)
2024-09-22 12:47:57,477 INFO InputEvent(EV_ABS, ABS_Y, 103)
2024-09-22 12:47:57,477 INFO InputEvent(EV_MSC, MSC_TIMESTAMP, 56000)
2024-09-22 12:47:57,477 INFO InputEvent(EV_SYN, SYN_REPORT, 0)
2024-09-22 12:47:57,483 INFO InputEvent(EV_ABS, ABS_MT_POSITION_X, 3835)
2024-09-22 12:47:57,483 INFO InputEvent(EV_ABS, ABS_MT_POSITION_Y, 104)
2024-09-22 12:47:57,484 INFO InputEvent(EV_ABS, ABS_X, 3835)
2024-09-22 12:47:57,484 INFO InputEvent(EV_ABS, ABS_Y, 104)
2024-09-22 12:47:57,484 INFO InputEvent(EV_MSC, MSC_TIMESTAMP, 63000)
2024-09-22 12:47:57,484 INFO InputEvent(EV_SYN, SYN_REPORT, 0)
2024-09-22 12:47:57,490 INFO InputEvent(EV_ABS, ABS_MT_POSITION_X, 3833)
2024-09-22 12:47:57,490 INFO InputEvent(EV_ABS, ABS_MT_POSITION_Y, 105)
2024-09-22 12:47:57,490 INFO InputEvent(EV_ABS, ABS_X, 3833)
2024-09-22 12:47:57,490 INFO InputEvent(EV_ABS, ABS_Y, 105)
2024-09-22 12:47:57,490 INFO InputEvent(EV_MSC, MSC_TIMESTAMP, 70000)
2024-09-22 12:47:57,491 INFO InputEvent(EV_SYN, SYN_REPORT, 0)
2024-09-22 12:47:57,496 INFO InputEvent(EV_ABS, ABS_MT_POSITION_X, 3831)
[... 647 lines cut ...]
[... cut ...]
2024-09-22 12:48:05,057 DEBUG Writting to config file: "<_io.TextIOWrapper name='./numberpad_dev' mode='w' encoding='UTF-8'>"
2024-09-22 12:48:05,058 INFO check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/home/ldrahnik/projects/asus-numberpad-driver pathname=/home/ldrahnik/projects/asus-numberpad-driver/numberpad_dev wd=1 >
2024-09-22 12:48:05,158 INFO Numpad deactivated
```

rudrab, 2024-09-22:
`evtest` log
```
evtest
No device specified, trying to scan all of /dev/input/event*
Not running as root, no devices may be available.
Available devices:
/dev/input/event0:	Lid Switch
/dev/input/event1:	Power Button
/dev/input/event10:	PC Speaker
/dev/input/event11:	Asus WMI hotkeys
/dev/input/event12:	sof-hda-dsp Headphone
/dev/input/event13:	sof-hda-dsp HDMI/DP,pcm=3
/dev/input/event14:	sof-hda-dsp HDMI/DP,pcm=4
/dev/input/event15:	sof-hda-dsp HDMI/DP,pcm=5
/dev/input/event16:	ASUP1415:00 093A:300C NumberPad
/dev/input/event2:	Power Button
/dev/input/event3:	AT Translated Set 2 keyboard
/dev/input/event4:	ASUP1415:00 093A:300C Mouse
/dev/input/event5:	ASUP1415:00 093A:300C Touchpad
/dev/input/event6:	ASUP1415:00 093A:300C Keyboard
/dev/input/event7:	Video Bus
/dev/input/event8:	Intel HID events
/dev/input/event9:	Intel HID 5 button array
Select the device event number [0-16]: 5
Input driver version is 1.0.1
Input device ID: bus 0x18 vendor 0x93a product 0x300c version 0x100
Input device name: "ASUP1415:00 093A:300C Touchpad"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 272 (BTN_LEFT)
    Event code 325 (BTN_TOOL_FINGER)
    Event code 328 (BTN_TOOL_QUINTTAP)
    Event code 330 (BTN_TOUCH)
    Event code 333 (BTN_TOOL_DOUBLETAP)
    Event code 334 (BTN_TOOL_TRIPLETAP)
    Event code 335 (BTN_TOOL_QUADTAP)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value   3900
      Min        0
      Max     3996
      Resolution      31
    Event code 1 (ABS_Y)
      Value   1580
      Min        0
      Max     2242
      Resolution      31
    Event code 47 (ABS_MT_SLOT)
      Value      0
      Min        0
      Max        4
    Event code 53 (ABS_MT_POSITION_X)
      Value      0
      Min        0
      Max     3996
      Resolution      31
    Event code 54 (ABS_MT_POSITION_Y)
      Value      0
      Min        0
      Max     2242
      Resolution      31
    Event code 55 (ABS_MT_TOOL_TYPE)
      Value      0
      Min        0
      Max        2
    Event code 57 (ABS_MT_TRACKING_ID)
      Value      0
      Min        0
      Max    65535
[... 207 lines cut ...]
Event: time 1727002288.303350, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 1753800
Event: time 1727002288.303350, -------------- SYN_REPORT ------------
Event: time 1727002288.310780, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 1760800
Event: time 1727002288.310780, -------------- SYN_REPORT ------------
Event: time 1727002288.317711, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 1767800
Event: time 1727002288.317711, -------------- SYN_REPORT ------------
Event: time 1727002288.324813, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 1774800
Event: time 1727002288.324813, -------------- SYN_REPORT ------------
Event: time 1727002288.332028, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value -1
Event: time 1727002288.332028, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1727002288.332028, type 1 (EV_KEY), code 325 (BTN_TOOL_FINGER), value 0
Event: time 1727002288.332028, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 1781700
Event: time 1727002288.332028, -------------- SYN_REPORT ------------

```
The git branch:
```
git branch 
* 169-not-toggling-numpad-to-touchpad
  master
```
I am also attaching the uninstall, install and error log (sorry that error log has grown up too big)
[uninstall-21-09-2024-18-44-52.log](https://github.com/user-attachments/files/17089314/uninstall-21-09-2024-18-44-52.log)
[uninstall-22-09-2024-16-17-49.log](https://github.com/user-attachments/files/17089315/uninstall-22-09-2024-16-17-49.log)
[Uploading error.log…]()

[11 more replies]

closed 2024-09-22 by ldrahnik