asus-linux-drivers/asus-numberpad-driver #170: [Feature Request] Missing support for the 2023 asus GX650 and upper versions

NicolasDepoilly, 2024-03-25:
### Short Description

Hi,
I've done all your steps of installation, but on the script to discover device's numberpad, I havent found my asus numberpad for a GX650RS computer.

I've tried the following steps, but it was unsuccessful and nothing happened.

Please keep me up to date when it is done, so I can try it.

Thanks

PS : nothing related, but I ve found this awsome package, and it could be cool to integrate officially in asus linux driver with the permission of author : https://github.com/wroberts/rogauracore

ldrahnik, 2024-03-25:
@NicolasDepoilly Hi,

> I've done all your steps of installation, but on the script to discover device's numberpad, I havent found my asus numberpad for a GX650RS computer.

you should be fine with selecting the layout named `gx551` during install, check https://github.com/asus-linux-drivers/asus-numberpad-driver/blob/master/images/Asus-GX551.jpg. Asus recycles NumberPad layouts. In case, I misunderstood your problem please post the output of the install script and driver's error log.

> PS : nothing related, but I ve found this awsome package, and it could be cool to integrate officially in asus linux driver with the permission of author : [wroberts/rogauracore](https://github.com/wroberts/rogauracore)

What do you mean by integrate? Create an empty repository with link to the project?

NicolasDepoilly, 2024-03-25:
My bad, I thought your project was more official from asus, forget this latest  comment  and i edit my post ( main question is still releavant! :) )

ldrahnik, 2024-03-25:
@NicolasDepoilly 

> I've done all your steps of installation, but on the script to discover device's numberpad, I havent found my asus numberpad for a GX650RS computer.

you should be fine with selecting the layout named `gx551` during install, check https://github.com/asus-linux-drivers/asus-numberpad-driver/blob/master/images/Asus-GX551.jpg. Asus recycles NumberPad layouts. In case, I misunderstood your problem please post the output of the install script and driver's error log.

NicolasDepoilly, 2024-03-25:
Hi,
I ve successfully made the installation with GX551 selected, but after reboot nothing s done when I press on the key for numberpad.
I ve read in your doc that "run driver now and every time that user logs in " so I guessed I had to redo this command ( maybe I missunderstood) 
So i ve retried to do this command bash install_service.sh only, but I have following error and i didnt want to do it as sudo as you said not to do it.
```
bash install_service.sh
Systemctl service

Do you want install systemctl service? [y/N]y

LAYOUT_NAME: up5401ea
CONFIG_FILE_DIR_PATH: /usr/share/asus-numberpad-driver

env var DISPLAY: :1
env var AUTHORITY: /run/user/1000/gdm/Xauthority
env var XDG_RUNTIME_DIR: /run/user/1000
env var DBUS_SESSION_BUS_ADDRESS: unix:path=/run/user/1000/bus
env var XDG_SESSION_TYPE: x11

ERROR LOG FILE: /var/log/asus-numberpad-driver/error.log

[sudo] password for nicolas: 
sudo: a password is required
Something went wrong when moving the asus_numberpad_driver.service

````

Should I redo something after install and reboot?
Thanks for quick answers btw

ldrahnik, 2024-03-25:
@NicolasDepoilly When the install script asks for the password just enter the sudo password. The script is not designed to run entirely under sudo but even if you install something for the current user you have to enter the password, that is ok, this is the case.

NicolasDepoilly, 2024-03-25:
I ve had following lines when I just did enter password, but sadly, still nothing on click on numpad key : 
Asus numberpad driver service placed
Systemctl daemon reloaded
Asus numberpad driver service enabled
Asus numberpad driver service started

ldrahnik, 2024-03-25:
@NicolasDepoilly The install script you can run many times (e.g. when you misstype `Y/N` answer). Nothing wrong will happen.

ldrahnik, 2024-03-25:
@NicolasDepoilly Run `$ bash install.sh`, layout select as I wrote and then in case it does not work post here please log files located here: `cat /var/log/asus-numberpad-driver/` and service status using this command: `systemctl --user status asus_numberpad..`

NicolasDepoilly, 2024-03-25:
this is still not working, i paste here the logs. fyi i chose to not reload config and redo it properly, and i chose "no" on idle script because I d prefer it permanent
```
nicolas@pop-os:~/Downloads/asus-numberpad-driver$ cat /var/log/asus-numberpad-driver/install-25-03-2024-23-17-08.log
Reading package lists...
Building dependency tree...
Reading state information...
ibus is already the newest version (1.5.26-4).
libevdev2 is already the newest version (1.12.1+dfsg-1).
xinput is already the newest version (1.6.3-1build2).
i2c-tools is already the newest version (4.3-2build1).
python3-virtualenv is already the newest version (20.13.0+ds-2).
curl is already the newest version (7.81.0-1ubuntu1.15).
libxml2-utils is already the newest version (2.9.13+dfsg-1ubuntu0.4).
python3-dev is already the newest version (3.10.6-1~22.04).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
From https://github.com/asus-linux-drivers/asus-numberpad-driver
   d1efb9d..afc5323  master     -> origin/master

groupadd: group 'input' already exists
groupadd: group 'i2c' already exists
groupadd: group 'uinput' already exists
Added groups input, i2c, uinput, numberpad to current user
uinput module loaded
i2c-dev module loaded
uinput module added to auto loaded modules
Udev rules reloaded and triggered

Testing interface i2c-0: success (adr 0x15)
The detection was successful. Touchpad with numberpad found: i2c-0

In system remains config file from previous installation. Do you want replace that config with default config? [y/N]
created virtual environment CPython3.10.12.final.0-64 in 233ms
  creator CPython3Posix(dest=/usr/share/asus-numberpad-driver/.env, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=/home/nicolas/.local/share/virtualenv)
    added seed packages: libevdev==0.11, numpy==1.26.4, pip==22.0.2, pyasyncore==1.0.4, pyinotify==0.9.6, python_xlib==0.33, setuptools==59.6.0, six==1.16.0, smbus2==0.4.3, wheel==0.37.1
  activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
Requirement already satisfied: pip in /usr/share/asus-numberpad-driver/.env/lib/python3.10/site-packages (22.0.2)
Collecting pip
  Using cached pip-24.0-py3-none-any.whl (2.1 MB)
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 22.0.2
    Uninstalling pip-22.0.2:
      Successfully uninstalled pip-22.0.2
Successfully installed pip-24.0
Requirement already satisfied: setuptools in /usr/share/asus-numberpad-driver/.env/lib/python3.10/site-packages (59.6.0)
Collecting setuptools
  Using cached setuptools-69.2.0-py3-none-any.whl.metadata (6.3 kB)
Using cached setuptools-69.2.0-py3-none-any.whl (821 kB)
Installing collected packages: setuptools
  Attempting uninstall: setuptools
    Found existing installation: setuptools 59.6.0
    Uninstalling setuptools-59.6.0:
      Successfully uninstalled setuptools-59.6.0
Successfully installed setuptools-69.2.0
Requirement already satisfied: libevdev in /usr/share/asus-numberpad-driver/.env/lib/python3.10/site-packages (from -r requirements.txt (line 1)) (0.11)
Requirement already satisfied: numpy in /usr/share/asus-numberpad-driver/.env/lib/python3.10/site-packages (from -r requirements.txt (line 2)) (1.26.4)
Requirement already satisfied: pyinotify in /usr/share/asus-numberpad-driver/.env/lib/python3.10/site-packages (from -r requirements.txt (line 3)) (0.9.6)
Requirement already satisfied: python-xlib in /usr/share/asus-numberpad-driver/.env/lib/python3.10/site-packages (from -r requirements.txt (line 4)) (0.33)
Requirement already satisfied: smbus2 in /usr/share/asus-numberpad-driver/.env/lib/python3.10/site-packages (from -r requirements.txt (line 5)) (0.4.3)
Requirement already satisfied: pyasyncore in /usr/share/asus-numberpad-driver/.env/lib/python3.10/site-packages (from -r requirements.txt (line 6)) (1.0.4)
Requirement already satisfied: six>=1.10.0 in /usr/share/asus-numberpad-driver/.env/lib/python3.10/site-packages (from python-xlib->-r requirements.txt (line 4)) (1.16.0)

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 52279    0 52279    0     0   131k      0 --:--:-- --:--:-- --:--:--  131k
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 84499    0 84499    0     0   208k      0 --:--:-- --:--:-- --:--:--  208k
Could not automatically detect NumberPad layout for your laptop.
[... 65 lines cut ...]
Calculator app

Do you want try to install toggling script for XF86Calculator key? [y/N]
Detected gnome-calculator
Setting up for gnome-calculator
Toggling script for calculator app gnome-calculator has been installed.

Power supply saver

By default is idle functionality disabled and may be manually enabled in config file later. Or by installing this rule can be enabled by every detected change of power supply mode to battery mode.

Idle mode is configured to be enabled after 10s of inactivity and to decrease 30% of brightness.

Do you want install the rule for idle functionality? [y/N]
Installation finished succesfully

Reboot is required. Do you want reboot now? [y/N]
```

```
systemctl --user status asus_numberpad..
Unit asus_numberpad...service could not be found.

```
PS : i tried same command without double dot, same result fyi

ldrahnik, 2024-03-25:
@NicolasDepoilly You have to edit the command with your username: `$ systemctl status --user asus_numberpad_driver@<username>.service`

ldrahnik, 2024-03-25:
@NicolasDepoilly What is the output of error log: `$ cat /var/log/asus-numberpad-driver/error.log`

ldrahnik, 2024-03-25:
@NicolasDepoilly My username is for example ldrahnik: `ldrahnik@Zenbook-UP5401EA:~/$ echo "something"`. So I run:```$ systemctl status --user asus_numberpad_driver@ldrahnik.service```

NicolasDepoilly, 2024-03-25:
> @NicolasDepoilly You have to run the command with your username: `$ systemctl status --user asus_numberpad_driver@<username>.service`

I don't get what you mean maybe, but I tried this and same result : 
```
nicolas@pop-os:~/Downloads/asus-numberpad-driver$ systemctl --user status asus_numberpad@nicolas.service
Unit asus_numberpad@nicolas.service could not be found.
```

the key seems to be detected, but is "just" making clicks

```
nicolas@pop-os:~/Downloads/asus-numberpad-driver$ cat /var/log/asus-numberpad-driver/error.log
2024-03-25 22:48:55,175 INFO     Detecting keyboard from string: "N: Name="Asus Keyboard""
2024-03-25 22:48:55,226 INFO     Detecting keyboard from string: "N: Name="Asus Keyboard""
2024-03-25 22:48:55,226 INFO     Set keyboard 20 from H: Handlers=sysrq kbd event20 leds
2024-03-25 22:48:55,262 INFO     Detecting touchpad from string: "N: Name="ASUE140F:00 04F3:31F7 Touchpad""
2024-03-25 22:48:55,263 INFO     Set touchpad device id 0 from S: Sysfs=/devices/platform/AMDI0010:00/i2c-0/i2c-ASUE140F:00/0018:04F3:31F7.0001/input/input32
2024-03-25 22:48:55,263 INFO     Set touchpad id 5 from H: Handlers=mouse1 event5
2024-03-25 22:48:55,263 INFO     Touchpad min-max: x 0-1710, y 0-2925
2024-03-25 22:48:55,263 INFO     Numpad min-max: x 20-1690, y 0-2925
2024-03-25 22:48:56,269 INFO     Setting up for config file key: "top_left_icon_brightness_func_disabled" with value: "True"
2024-03-25 22:49:02,973 INFO     Touched numlock key (not top_right_icon) in time: 1711403342.9735432
2024-03-25 22:49:03,539 INFO     Un-touched without NumberPad activation numlock key (not top_right_icon) in time: 1711403343.5397897
2024-03-25 22:49:03,582 INFO     Touched numlock key (not top_right_icon) in time: 1711403343.582226
2024-03-25 22:49:03,952 INFO     Un-touched without NumberPad activation numlock key (not top_right_icon) in time: 1711403343.9521668
2024-03-25 22:49:04,540 INFO     Touched numlock key (not top_right_icon) in time: 1711403344.5402296
2024-03-25 22:49:04,986 INFO     Un-touched without NumberPad activation numlock key (not top_right_icon) in time: 1711403344.9865031
2024-03-25 22:49:47,240 INFO     Touched numlock key (not top_right_icon) in time: 1711403387.2404406
2024-03-25 22:49:47,718 INFO     Un-touched without NumberPad activation numlock key (not top_right_icon) in time: 1711403387.7185714
2024-03-25 22:49:48,097 INFO     Touched numlock key (not top_right_icon) in time: 1711403388.0975764
2024-03-25 22:49:48,677 INFO     Un-touched without NumberPad activation numlock key (not top_right_icon) in time: 1711403388.6779687
2024-03-25 22:51:45,574 INFO     Touched numlock key (not top_right_icon) in time: 1711403505.5740864
2024-03-25 22:51:46,052 INFO     Un-touched without NumberPad activation numlock key (not top_right_icon) in time: 1711403506.052311
2024-03-25 22:54:42,980 INFO     Touched numlock key (not top_right_icon) in time: 1711403682.9807496
2024-03-25 22:54:43,452 INFO     Un-touched without NumberPad activation numlock key (not top_right_icon) in time: 1711403683.4519894
2024-03-25 22:54:43,729 INFO     Touched numlock key (not top_right_icon) in time: 1711403683.7298322
2024-03-25 22:54:43,877 INFO     Un-touched without NumberPad activation numlock key (not top_right_icon) in time: 1711403683.8776808
2024-03-25 22:54:44,851 INFO     Touched numlock key (not top_right_icon) in time: 1711403684.8518362
2024-03-25 22:54:45,392 INFO     Un-touched without NumberPad activation numlock key (not top_right_icon) in time: 1711403685.3924031
2024-03-25 22:54:45,467 INFO     Touched numlock key (not top_right_icon) in time: 1711403685.4672608
2024-03-25 22:54:45,709 INFO     Un-touched without NumberPad activation numlock key (not top_right_icon) in time: 1711403685.7099075
2024-03-25 22:54:45,756 INFO     Touched numlock key (not top_right_icon) in time: 1711403685.7567747
2024-03-25 22:54:46,097 INFO     Un-touched without NumberPad activation numlock key (not top_right_icon) in time: 1711403686.0974832
2024-03-25 23:02:03,205 INFO     Detecting keyboard from string: "N: Name="Asus Keyboard""
2024-03-25 23:02:03,205 INFO     Set keyboard 20 from H: Handlers=sysrq kbd event20 leds
2024-03-25 23:02:03,234 INFO     Detecting touchpad from string: "N: Name="ASUE140F:00 04F3:31F7 Touchpad""
2024-03-25 23:02:03,235 INFO     Set touchpad device id 0 from S: Sysfs=/devices/platform/AMDI0010:00/i2c-0/i2c-ASUE140F:00/0018:04F3:31F7.0001/input/input32
2024-03-25 23:02:03,235 INFO     Set touchpad id 5 from H: Handlers=mouse1 event5
2024-03-25 23:02:03,236 INFO     Touchpad min-max: x 0-1710, y 0-2925
2024-03-25 23:02:03,236 INFO     Numpad min-max: x 200-1510, y 200-2845
2024-03-25 23:02:04,257 INFO     Setting up for config file key: "brightness" with value: "0x48"
2024-03-25 23:17:11,303 INFO     check_config_values_changes: detected external change of config file -> loading changes
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad.py path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/numberpad.py wd=1 >
2024-03-25 23:17:43,854 INFO     Detecting keyboard from string: "N: Name="Asus Keyboard""
2024-03-25 23:17:43,855 INFO     Set keyboard 20 from H: Handlers=sysrq kbd event20 leds
2024-03-25 23:17:43,870 INFO     Detecting touchpad from string: "N: Name="ASUE140F:00 04F3:31F7 Touchpad""
[... cut ...]
2024-03-25 23:47:25,280 INFO     Un-touched without NumberPad activation numlock key (not top_right_icon) in time: 1711406845.2807593
2024-03-25 23:47:25,406 INFO     Touched numlock key (not top_right_icon) in time: 1711406845.4064405
2024-03-25 23:47:25,524 INFO     Un-touched without NumberPad activation numlock key (not top_right_icon) in time: 1711406845.5248148
2024-03-25 23:47:26,433 INFO     Touched numlock key (not top_right_icon) in time: 1711406846.4333618
2024-03-25 23:47:27,162 INFO     Un-touched without NumberPad activation numlock key (not top_right_icon) in time: 1711406847.1628287

```

ldrahnik, 2024-03-25:
@NicolasDepoilly Please run these commands `$ libinput record`, `$ sudo libinput debug-events` in terminal and then try to activate NumberPad via top right icon > 1s press and copy output here

NicolasDepoilly, 2024-03-25:
```
nicolas@pop-os:~/Downloads/asus-numberpad-driver$ libinput record
Available devices:
/dev/input/event0:	Lid Switch
/dev/input/event1:	Power Button
/dev/input/event2:	Sleep Button
/dev/input/event3:	Asus WMI hotkeys
/dev/input/event4:	ASUE140F:00 04F3:31F7 Mouse
/dev/input/event5:	ASUE140F:00 04F3:31F7 Touchpad
/dev/input/event6:	HD-Audio Generic HDMI/DP,pcm=3
/dev/input/event7:	HDA NVidia HDMI/DP,pcm=3
/dev/input/event8:	ELAN9009:00 04F3:2F32
/dev/input/event9:	ELAN9009:00 04F3:2F32 UNKNOWN
/dev/input/event10:	ELAN9009:00 04F3:2F32 UNKNOWN
/dev/input/event11:	Video Bus
/dev/input/event12:	Video Bus
/dev/input/event13:	Razer Razer DeathAdder V2 Pro
/dev/input/event14:	Razer Razer DeathAdder V2 Pro Keyboard
/dev/input/event15:	Razer Razer DeathAdder V2 Pro Mouse
/dev/input/event16:	Razer Razer DeathAdder V2 Pro
/dev/input/event17:	Razer Razer DeathAdder V2 Pro
/dev/input/event18:	Razer Razer DeathAdder V2 Pro Consumer Control
/dev/input/event19:	Asus Keyboard
/dev/input/event20:	Asus Keyboard
/dev/input/event21:	ASUE140F:00 04F3:31F7 Keyboard
/dev/input/event22:	ELAN9009:00 04F3:2F32 Stylus
/dev/input/event23:	ELAN9009:00 04F3:2F32 Touchpad
/dev/input/event24:	HDA NVidia HDMI/DP,pcm=7
/dev/input/event25:	HDA NVidia HDMI/DP,pcm=8
/dev/input/event26:	HDA NVidia HDMI/DP,pcm=9
/dev/input/event27:	HD-Audio Generic Headphone
/dev/input/event28:	ASUE140F:00 04F3:31F7 NumberPad
Select the device event number: 28
Recording to 'stdout'.
# libinput record
version: 1
ndevices: 1
libinput:
  version: "1.20.0"
  git: "unknown"
system:
os: "pop:22.04"
kernel: "6.8.0-76060800daily20240311-generic"
dmi: "dmi:bvnAmericanMegatrendsInternational,LLC.:bvrGX650RS.319:bd08/11/2023:br5.24:efr3.6:svnASUSTeKCOMPUTERINC.:pnROGZephyrusDuo16GX650RS_GX650RS:pvr1.0:rvnASUSTeKCOMPUTERINC.:rnGX650RS:rvr1.0:cvnASUSTeKCOMPUTERINC.:ct10:cvr1.0:sku:"
devices:
- node: /dev/input/event28
  evdev:
    # Name: ASUE140F:00 04F3:31F7 NumberPad
    # ID: bus 00 vendor 00 product 00 version 00
    # Supported Events:
    # Event type 0 (EV_SYN)
    # Event type 1 (EV_KEY)
    #   Event code 2 (KEY_1)
    #   Event code 3 (KEY_2)
    #   Event code 4 (KEY_3)
    #   Event code 5 (KEY_4)
    #   Event code 6 (KEY_5)
    #   Event code 7 (KEY_6)
    #   Event code 8 (KEY_7)
    #   Event code 9 (KEY_8)
    #   Event code 10 (KEY_9)
    #   Event code 11 (KEY_0)
    #   Event code 18 (KEY_E)
    #   Event code 22 (KEY_U)
    #   Event code 28 (KEY_ENTER)
    #   Event code 29 (KEY_LEFTCTRL)
    #   Event code 30 (KEY_A)
    #   Event code 31 (KEY_S)
    #   Event code 32 (KEY_D)
    #   Event code 33 (KEY_F)
[... 150 lines cut ...]
 event5   POINTER_MOTION          +9.056s	 -0.34/  1.02 ( -1.00/ +3.00)
 event5   POINTER_MOTION          +9.062s	 -0.34/  0.68 ( -1.00/ +2.00)
 event5   POINTER_MOTION          +9.445s	  0.64/ -0.64 ( +2.00/ -2.00)
 event5   POINTER_MOTION          +9.453s	  0.64/ -0.64 ( +2.00/ -2.00)
 event5   GESTURE_HOLD_BEGIN      +9.456s	1
 event5   POINTER_MOTION          +9.460s	  0.68/ -0.68 ( +2.00/ -2.00)
 event5   POINTER_MOTION          +9.468s	  0.34/ -0.68 ( +1.00/ -2.00)
 event5   POINTER_MOTION          +9.475s	  0.34/ -0.68 ( +1.00/ -2.00)
 event5   POINTER_MOTION          +9.482s	  0.00/ -1.02 ( +0.00/ -3.00)
 event5   POINTER_MOTION          +9.490s	 -0.34/ -1.36 ( -1.00/ -4.00)
 event5   GESTURE_HOLD_END        +9.490s	1 cancelled
 event5   POINTER_MOTION          +9.497s	 -0.68/ -2.04 ( -2.00/ -6.00)
 event5   POINTER_MOTION          +9.675s	  0.63/  0.32 ( +2.00/ +1.00)
 event5   POINTER_MOTION          +9.682s	  0.24/  0.00 ( +1.00/ +0.00)
 event5   GESTURE_HOLD_BEGIN      +9.685s	1
 event5   POINTER_MOTION          +9.690s	  0.28/  0.00 ( +1.00/ +0.00)
 event5   POINTER_MOTION          +9.697s	  0.28/  0.00 ( +1.00/ +0.00)
 event5   GESTURE_HOLD_END        +9.736s	1
 event5   POINTER_MOTION          +9.867s	 -0.85/ -0.28 ( -3.00/ -1.00)
-event20  KEYBOARD_KEY            +11.700s	*** (-1) pressed
 event20  KEYBOARD_KEY            +12.036s	*** (-1) pressed

```

it seems that this touch is not detected as another thing than a mouse pointer by my understanding

ldrahnik, 2024-03-26:
> @nicolas@pop-os:~/Downloads/asus-numberpad-driver$ systemctl --user status asus_numberpad@nicolas.service
Unit asus_numberpad@nicolas.service could not be found.

> I don't get what you mean maybe, but I tried this and same result :

I do not need to see the output of that command but you have to run the command with your username as I wrote: `$ systemctl status --user asus_numberpad_driver@<username>.service. Your executed command was not correct.

ldrahnik, 2024-03-26:
@NicolasDepoilly What do you mean by title **Missing support for the 2023 asus GX650 and upper versions?** Do you own any other new Asus laptop where the driver does not work?

ldrahnik, 2024-03-26:
@NicolasDepoilly You have to hold NumLock icon for atleast 1s by default.

ldrahnik, 2024-03-26:
Can be reopened.

closed 2024-03-26 by ldrahnik

NicolasDepoilly, 2024-03-26:
FYI, I ve just re done it one more time and reboot after it, and it seems to be ok now! thanks again for your help and I close this issue!

NicolasDepoilly, 2024-03-26:
FYI I just played a bit more with it, and sadly it is working BUT the map of keys is not matching properly, so for instance when I click on 7 it is always making a 5, and the bottom seems to be hard to reach. To sum up, all keys are mapped upper from their original position.
Maybe if you can tell me how to setup the base position of this map, I could try to play with it a bit!
Thanks again for all your work and help!

ldrahnik, 2024-03-26:
@NicolasDepoilly Before install you can change in folder `layouts` inside your layout by name 4 variables which determine margins (top, left, bottom, right). That Is all. Row Is than calculated the same for each. After install you can change it inside thé same dir `layouts` but located here `/usr/share/asus-numberpad-driver/layouts` - Will be required `systemctl --user restart asus_..`