asus-linux-drivers/asus-numberpad-driver #171: [Layout Request] UX425QA

lare354, 2024-03-26:
### Layout Name

UX425QA

### Your device (cat /sys/devices/virtual/dmi/id/product_name)

ZenBook UX425QA_UM425QA

### Numpad Layout (photo/picture or code)

[image]

### Additional Info

it looks like the "up5401ea" layout, but when I test those the '%' button either produces "25" or "E" depending on the layout. I have also tried most other layouts but they do not produce the correct outputs of the buttons

### Acknowledgements

- [X] This layout doesn't currently exist and I have read the [Layout section](https://github.com/asus-linux-drivers/asus-numberpad-driver?tab=readme-ov-file#layouts)

lare354, 2024-03-26:
additionally, when it outputs '25', a unicode character selector also opens, prompting me to select a character to output. The selected character is printed after 25. "25 %" for example. The menu opens every time the key is pressed

ldrahnik, 2024-03-27:
In case problem is only character `%`and you use Wayland it Is known issue https://github.com/asus-linux-drivers/asus-numberpad-driver/issues/104. In case not provide please driver logs and install output and can ne reopened.

closed 2024-03-27 by ldrahnik

ldrahnik, 2024-03-27:
@lare354 Quick fix for you Is manually change inside folder `layouts` fór your layout key for printing character `%` and then install again. What a language layout do you use? Not EN?