asus-linux-drivers/asus-numberpad-driver #172: [Layout Request] GX650PZ

Nospamty, 2024-03-27:
### Layout Name

GX650PZ

### Your device (cat /sys/devices/virtual/dmi/id/product_name)

ROG Zephyrus Duo 16 GX650PZ_GX650PZ

### Supported Devices

GX551

### Numpad Layout (photo/picture or code)

[image]

### Acknowledgements

- [X] This layout doesn't currently exist and I have read the [Layout section](https://github.com/asus-linux-drivers/asus-numberpad-driver?tab=readme-ov-file#layouts)

ldrahnik, 2024-03-27:
@Nospamty Hello, thank you for creating an issue. Please post here driver error log and output of install script.

Nospamty, 2024-03-28:
Sorry I had not realized it produced an error log. Here are the contents:

2023-11-28 22:46:42,191 INFO     Detecting keyboard from string: "N: Name="Asus Keyboard""
2023-11-28 22:46:42,246 INFO     Detecting keyboard from string: "N: Name="Asus Keyboard""
2023-11-28 22:46:42,246 INFO     Set keyboard 16 from H: Handlers=sysrq kbd event16 leds
2023-11-28 22:46:42,306 INFO     Detecting touchpad from string: "N: Name="ASUE140F:00 04F3:32A2 Touchpad""
2023-11-28 22:46:42,307 INFO     Set touchpad device id 0 from S: Sysfs=/devices/platform/AMDI0010:00/i2c-0/i2c-ASUE140F:00/0018:04F3:32A2.0001/input/input35
2023-11-28 22:46:42,307 INFO     Set touchpad id 23 from H: Handlers=mouse1 event23
2023-11-28 22:46:42,307 INFO     Touchpad min-max: x 0-1710, y 0-2925
2023-11-28 22:46:42,307 INFO     Numpad min-max: x 0-1710, y 80-2925
2023-11-28 22:46:43,319 INFO     Setting up for config file key: "top_left_icon_brightness_func_disabled" with value: "True"
2023-11-28 22:48:48,602 INFO     check_config_values_changes: detected external change of config file -> loading changes
2024-03-26 16:36:08,565 INFO     Detecting keyboard from string: "N: Name="Asus Keyboard""
2024-03-26 16:36:08,603 INFO     Detecting keyboard from string: "N: Name="Asus Keyboard""
2024-03-26 16:36:08,603 INFO     Set keyboard 16 from H: Handlers=sysrq kbd event16 leds
2024-03-26 16:36:08,635 INFO     Detecting touchpad from string: "N: Name="ASUE140F:00 04F3:32A2 Touchpad""
2024-03-26 16:36:08,635 INFO     Set touchpad device id 0 from S: Sysfs=/devices/platform/AMDI0010:00/i2c-0/i2c-ASUE140F:00/0018:04F3:32A2.0001/input/input28
2024-03-26 16:36:08,635 INFO     Set touchpad id 7 from H: Handlers=mouse1 event7
2024-03-26 16:36:08,635 INFO     Touchpad min-max: x 0-1710, y 0-2925
2024-03-26 16:36:08,635 INFO     Numpad min-max: x 20-1690, y 0-2925
2024-03-26 16:36:09,647 INFO     Setting up for config file key: "top_left_icon_brightness_func_disabled" with value: "True"
2024-03-26 16:36:51,686 INFO     check_config_values_changes: detected external change of config file -> loading changes
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=sedyXeiGA path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/sedyXeiGA wd=1 >
2024-03-26 16:36:51,787 INFO     check_config_values_changes: detected external change of config file -> loading changes
<Event cookie=136251 dir=False mask=0x80 maskname=IN_MOVED_TO name=numberpad_dev path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/numberpad_dev wd=1 >
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=sedeyuruN path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/sedeyuruN wd=1 >
<Event cookie=136309 dir=False mask=0x80 maskname=IN_MOVED_TO name=numberpad_dev path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/numberpad_dev wd=1 >
2024-03-26 23:38:42,855 INFO     Detecting keyboard from string: "N: Name="Asus Keyboard""
2024-03-26 23:38:42,908 INFO     Detecting keyboard from string: "N: Name="Asus Keyboard""
2024-03-26 23:38:42,908 INFO     Set keyboard 15 from H: Handlers=sysrq kbd event15 leds
2024-03-26 23:38:42,940 INFO     Detecting touchpad from string: "N: Name="ASUE140F:00 04F3:32A2 Touchpad""
2024-03-26 23:38:42,941 INFO     Set touchpad device id 0 from S: Sysfs=/devices/platform/AMDI0010:00/i2c-0/i2c-ASUE140F:00/0018:04F3:32A2.0001/input/input31
2024-03-26 23:38:42,941 INFO     Set touchpad id 19 from H: Handlers=mouse1 event19
2024-03-26 23:38:42,941 INFO     Touchpad min-max: x 0-1710, y 0-2925
2024-03-26 23:38:42,941 INFO     Numpad min-max: x 20-1690, y 0-2925
2024-03-26 23:38:56,688 INFO     Touched numlock key (not top_right_icon) in time: 1711521536.6887052
2024-03-26 23:38:56,749 INFO     Un-touched without NumberPad activation numlock key (not top_right_icon) in time: 1711521536.7490256
2024-03-26 23:38:57,193 INFO     Touched numlock key (not top_right_icon) in time: 1711521537.193872
2024-03-26 23:38:57,269 INFO     Un-touched without NumberPad activation numlock key (not top_right_icon) in time: 1711521537.2690678
2024-03-26 23:38:57,823 INFO     Touched numlock key (not top_right_icon) in time: 1711521537.8235195
2024-03-26 23:38:57,905 INFO     Un-touched without NumberPad activation numlock key (not top_right_icon) in time: 1711521537.905524
2024-03-26 23:38:57,980 INFO     Touched numlock key (not top_right_icon) in time: 1711521537.9808292
2024-03-26 23:38:58,056 INFO     Un-touched without NumberPad activation numlock key (not top_right_icon) in time: 1711521538.0560682
2024-03-26 23:38:59,130 INFO     Touched numlock key (not top_right_icon) in time: 1711521539.130487
2024-03-26 23:38:59,219 INFO     Un-touched without NumberPad activation numlock key (not top_right_icon) in time: 1711521539.2194107
2024-03-26 23:38:59,349 INFO     Touched numlock key (not top_right_icon) in time: 1711521539.3494718
2024-03-26 23:38:59,452 INFO     Un-touched without NumberPad activation numlock key (not top_right_icon) in time: 1711521539.4520707
2024-03-26 23:38:59,465 INFO     Touched numlock key (not top_right_icon) in time: 1711521539.4658027
2024-03-26 23:38:59,575 INFO     Un-touched without NumberPad activation numlock key (not top_right_icon) in time: 1711521539.5752182
2024-03-26 16:39:41,060 INFO     Touched numlock key (not top_right_icon) in time: 1711496381.06014
[... cut ...]
<Event cookie=28086 dir=False mask=0x80 maskname=IN_MOVED_TO name=numberpad_dev path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/numberpad_dev wd=1 >
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=sedJQLsBC path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/sedJQLsBC wd=1 >
<Event cookie=28144 dir=False mask=0x80 maskname=IN_MOVED_TO name=numberpad_dev path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/numberpad_dev wd=1 >

Nospamty, 2024-03-28:
bash install.sh
[sudo] password for user1:               
groupadd: group 'numberpad' already exists
Added group numberpad to current user

Reading package lists...
Building dependency tree...
Reading state information...
ibus is already the newest version (1.5.26-4).
libevdev2 is already the newest version (1.12.1+dfsg-1).
xinput is already the newest version (1.6.3-1build2).
i2c-tools is already the newest version (4.3-2build1).
python3-virtualenv is already the newest version (20.13.0+ds-2).
curl is already the newest version (7.81.0-1ubuntu1.16).
libxml2-utils is already the newest version (2.9.13+dfsg-1ubuntu0.4).
python3-dev is already the newest version (3.10.6-1~22.04).
0 upgraded, 0 newly installed, 0 to remove and 9 not upgraded.
From https://github.com/asus-linux-drivers/asus-numberpad-driver
   9e5091f..384bf48  master     -> origin/master

groupadd: group 'input' already exists
groupadd: group 'i2c' already exists
groupadd: group 'uinput' already exists
Added groups input, i2c, uinput, numberpad to current user
uinput module loaded
i2c-dev module loaded
uinput module added to auto loaded modules
Udev rules reloaded and triggered

Testing interface i2c-0: success (adr 0x15)
The detection was successful. Touchpad with numberpad found: i2c-0

Default config will be autocreated during the first run and available for futher modifications here:
/usr/share/asus-numberpad-driver/numberpad_dev

created virtual environment CPython3.10.12.final.0-64 in 220ms
  creator CPython3Posix(dest=/usr/share/asus-numberpad-driver/.env, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=/home/user1/.local/share/virtualenv)
    added seed packages: pip==24.0, setuptools==69.1.1, wheel==0.42.0
  activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
Requirement already satisfied: pip in /usr/share/asus-numberpad-driver/.env/lib/python3.10/site-packages (24.0)
Requirement already satisfied: setuptools in /usr/share/asus-numberpad-driver/.env/lib/python3.10/site-packages (69.1.1)
Collecting setuptools
  Using cached setuptools-69.2.0-py3-none-any.whl.metadata (6.3 kB)
Using cached setuptools-69.2.0-py3-none-any.whl (821 kB)
Installing collected packages: setuptools
  Attempting uninstall: setuptools
    Found existing installation: setuptools 69.1.1
    Uninstalling setuptools-69.1.1:
      Successfully uninstalled setuptools-69.1.1
Successfully installed setuptools-69.2.0
Collecting libevdev (from -r requirements.txt (line 1))
  Using cached libevdev-0.11-py3-none-any.whl
Collecting numpy (from -r requirements.txt (line 2))
  Using cached numpy-1.26.4-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (61 kB)
Collecting pyinotify (from -r requirements.txt (line 3))
  Using cached pyinotify-0.9.6-py3-none-any.whl
Collecting python-xlib (from -r requirements.txt (line 4))
  Using cached python_xlib-0.33-py2.py3-none-any.whl.metadata (6.2 kB)
Collecting smbus2 (from -r requirements.txt (line 5))
  Using cached smbus2-0.4.3-py2.py3-none-any.whl.metadata (7.1 kB)
Collecting pyasyncore (from -r requirements.txt (line 6))
  Using cached pyasyncore-1.0.4-py3-none-any.whl.metadata (3.7 kB)
Collecting six>=1.10.0 (from python-xlib->-r requirements.txt (line 4))
  Using cached six-1.16.0-py2.py3-none-any.whl.metadata (1.8 kB)
Using cached numpy-1.26.4-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (18.2 MB)
Using cached python_xlib-0.33-py2.py3-none-any.whl (182 kB)
Using cached smbus2-0.4.3-py2.py3-none-any.whl (11 kB)
Using cached pyasyncore-1.0.4-py3-none-any.whl (10 kB)
Using cached six-1.16.0-py2.py3-none-any.whl (11 kB)
[... 66 lines cut ...]
Rule 90-numberpad-external-keyboard.rules applied
Udev rules reloaded and triggered

Calculator app

Do you want try to install toggling script for XF86Calculator key? [y/N]y

Detected gnome-calculator
Setting up for gnome-calculator
Toggling script for calculator app gnome-calculator has been installed.

Power supply saver

By default is idle functionality disabled and may be manually enabled in config file later. Or by installing this rule can be enabled by every detected change of power supply mode to battery mode.

Idle mode is configured to be enabled after 10s of inactivity and to decrease 30% of brightness.

Do you want install the rule for idle functionality? [y/N]y

Rule 80-numberpad-power-supply applied
Udev rules reloaded and triggered

Installation finished succesfully

Reboot is required. Do you want reboot now? [y/N]y

ldrahnik, 2024-03-28:
@Nospamty You have to hold the key for atleast 1s. In case you do please run simultanously `$ sudo libinput debug-events` and `$ libinput record`

Nospamty, 2024-03-29:
Sorry, at what point am I supposed to hold a key?

ldrahnik, 2024-03-29:
@Nospamty NumLock icon key for NumberPad activation / deactivation, in log Is written you did only touch / untouch. If you hold longger than 1s so log lies then please try to capture activation attempt using mentioned tools

Nospamty, 2024-03-29:
Yes it is incorrect because I never touched the numlock key, it never prompted me to do so. All I input is bash install.sh then y or n to answer prompts.

ldrahnik, 2024-03-29:
@Nospamty I mean NumLock icon on touchpad - attached screen in first post on the top left spot - **Numpad Layout (photo/picture or code)**.  In error log are displayed touches. Not in install log.

ldrahnik, 2024-03-31:
Can be reopened.

closed 2024-03-31 by ldrahnik