asus-linux-drivers/asus-numberpad-driver #176: [Bug]: ERROR Numpad layout *.py from dir layouts is required as first argument

locrianz, 2024-04-17:
### Describe the bug

I installed the script, chose not to use a systemctl service, and restarted my computer. Then I ran

```
/usr/share/asus-numberpad-driver/.env/bin/python3 /usr/share/asus-numberpad-driver/numberpad.py
```

as instructed, and got this error:

```
ERROR    Numpad layout *.py from dir layouts is required as first argument. Re-run install script or add missing first argument (valid value is b7402, e210ma, g533, gx551, gx701, up5401ea, ..).
```

### Expected behavior

I expected this argument to be automatically included in the script I run, since the setup already detected my default numpad layout and asked if I wanted to use it. (And I said yes.)

How do I format the argument to include the layout?

### Desktop

- Program version: v5.1.1
- Distro: Linux Mint
- Kernel: 5.15.0-102-generic x86_64

ldrahnik, 2024-04-17:
For example `up5401ea`, without `.py`.

locrianz, 2024-04-17:
I just wasn't sure how to format it, whether I needed to put, say, "layout=ux581l" or something. I haven't used many Python scripts before. But this worked:

```
/usr/share/asus-numberpad-driver/.env/bin/python3 /usr/share/asus-numberpad-driver/numberpad.py ux581l
```

ldrahnik, 2024-04-17:
@locrianz I have changed Readme a little bit.