asus-linux-drivers/asus-numberpad-driver #177: [Bug]: Script numberpad.py execution failed

darkpastiurs, 2024-04-26:
### Describe the bug

I am experiencing an issue with the `asus_numpad_driver` service, specifically encountering an error within the `numberpad.py` script.

### Expected behavior

Being able to enable the Numpad functionality on my device by using the specified button on the mousepad.

### Relevant log output

```shell
File "/usr/share/asus-numberpad-driver/numberpad.py", line 269
    device_id: Optional[str] = None
    ^
SyntaxError: annotated name 'device_id' can't be global
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 269
    device_id: Optional[str] = None
    ^
SyntaxError: annotated name 'device_id' can't be global
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 269
    device_id: Optional[str] = None
    ^
SyntaxError: annotated name 'device_id' can't be global
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 269
    device_id: Optional[str] = None
    ^
SyntaxError: annotated name 'device_id' can't be global
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 269
    device_id: Optional[str] = None
    ^
SyntaxError: annotated name 'device_id' can't be global
```

### Desktop

- Program version: v5.1.1
- Distro: OpenSUSE Leap 15.5
- Kernel: 5.14.21-150500.55.52-default
- Laptop Model:  ASUS ProArt StudioBook 17

[install-26-04-2024-15-38-11.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/15135878/install-26-04-2024-15-38-11.log)

ldrahnik, 2024-04-27:
@darkpastiurs Python version?

ldrahnik, 2024-04-27:
Hello, could you please turn english on, try install again and post here install log, I see few errors and I do not want use translator.

ldrahnik, 2024-04-27:
What is output of `python3 --version`?

ldrahnik, 2024-04-27:
@darkpastiurs Try the latest master and post log again.

darkpastiurs, 2024-04-28:
[install-27-04-2024-19-16-37.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/15140446/install-27-04-2024-19-16-37.log)

ldrahnik, 2024-04-28:
It should work now. Second error is about analytics only. Are you sure your laptop's hardware supports NumberPad? Can I do not think so.

ldrahnik, 2024-04-28:
Closing as the second error was solved. Can be reopened.

darkpastiurs, 2024-04-28:
Regarding the question about whether my computer supports the NumberPad, the answer is yes. When I had Windows 11, it was working correctly (a few days ago I formatted my computer and left only OpenSUSE). 

The problem persists with the driver. I am sharing the [error.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/15142516/error.log) file. 

I apologize for the delay in my responses.

ldrahnik, 2024-04-29:
@darkpastiurs Could you please try to run driver with newer version of python? I use personaly (distro based on Ubuntu22.04) `Python 3.10.12`. During installation was written to log that you use version `3.6.15`. In case **OpenSUSE Leap 15.5** does not provider package which can be just updated try to install directly or using `pyenv` - in that case read in `Readme.md` **How to install the driver when is used pyenv for managing multiple Python versions? in [FAQ](https://github.com/asus-linux-drivers/asus-numberpad-driver#faq)**

darkpastiurs, 2024-04-29:
I have performed the installation with pyenv using the version you mentioned, but it is still not working. I am sharing the installation log 
[install-29-04-2024-08-19-17.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/15150559/install-29-04-2024-08-19-17.log)

ldrahnik, 2024-04-29:
@darkpastiurs Latest version does not have `bc` command anymore. Do you use the latest version of this driver from the master branch? Anyway it looks like python `3.6.15` is too old for handling virtuale nvironment for version `3.10.12`. As I understand that error.

darkpastiurs, 2024-04-29:
I have managed to make it work, however, I had to install Python version 3.11.8, python311-virtualenv, and modify the install.sh file on line 96. I changed virtualenv --python=$(which python3) /usr/share/asus-numberpad-driver/.env to virtualenv-3.11 --python=$(which python3.11) /usr/share/asus-numberpad-driver/.env. Here is the log of the successful installation. 
[install-29-04-2024-10-03-33.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/15151931/install-29-04-2024-10-03-33.log)

closed 2024-04-29 by ldrahnik