asus-linux-drivers/asus-numberpad-driver #178: [Bug]: Numpad doesn't work and has high CPU usage

irvng-wav, 2024-05-04:
### Describe the bug

I successfully installed v6.0.0 from the releases page, but when I rebooted my system, the numpad didn't work, meaning it doesn't turn on and launching the calculator also doesn't work. On top of that my CPU usage went up like crazy so the fans are always spinning loudly. Here are my logs from /var/log/asus-numberpad-driver:

[error.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/15209143/error.log)
[install-04-05-2024-02-58-05.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/15209144/install-04-05-2024-02-58-05.log)

### Expected behavior

For the numpad to work and not stress my CPU.

### Relevant log output

```shell
error.log:

2024-05-04 02:58:49,509 INFO     Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
2024-05-04 02:58:49,510 INFO     Set keyboard 3 from H: Handlers=sysrq kbd leds event3
2024-05-04 02:58:49,529 INFO     Detecting touchpad from string: "N: Name="ASUE140D:00 04F3:31B9 Touchpad""
2024-05-04 02:58:49,529 INFO     Set touchpad device id 1 from S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-ASUE140D:00/0018:04F3:31B9.0001/input/input12
2024-05-04 02:58:49,529 INFO     Set touchpad id 9 from H: Handlers=event9 mouse1
2024-05-04 02:58:49,529 INFO     Touchpad min-max: x 0-3880, y 0-2299
2024-05-04 02:58:49,529 INFO     Numpad min-max: x 200-3680, y 200-2219
2024-05-04 02:58:50,531 INFO     Setting up for config file key: "brightness" with value: "0x48"
```

### Desktop

- Program version: Release 6.0.0
- Distro: Arch Linux
- Kernel: Linux 6.8.9-arch1-1
- Laptop: Zenbook 14 UX3402ZA
- DE: GNOME on Wayland

ldrahnik, 2024-05-04:
@irvng-wav Thank you for reporting. Until will be fixed please use the master branch fixed to the commit  https://github.com/asus-linux-drivers/asus-numberpad-driver/commit/2aae6152508bc4e169003fb87a24dfc7bc6edfe4 or previous release tag.

irvng-wav, 2024-05-04:
@ldrahnik thank you for the quick response! How do I `git clone` a specific commit?

ldrahnik, 2024-05-04:
https://www.theserverside.com/blog/Coffee-Talk-Java-News-Stories-and-Opinions/git-clone-specific-commit-id-single-branch-reset-depth-head-main-master

ldrahnik, 2024-05-04:
The previous version or mentioned commit does work?

ldrahnik, 2024-05-04:
@irvng-wav Unfortunately, It looks like currently the driver does not support your laptop. I found and fixed a bug that allowed you to continue installation with not supported touchpad. In case you are able activate NumberPad on Windows / your laptop has hardware support for NumberPad I need post here your DSDT table using:

```
$ sudo apt-get install iasl
$ cd <work-dir>
$ sudo cp /sys/firmware/acpi/tables/DSDT DSDT
$ sudo iasl -d DSDT
$ sudo gedit DSDT.dsl
```

irvng-wav, 2024-05-04:
Hello @ldrahnik I just successfully installed the [2aae615](https://github.com/asus-linux-drivers/asus-numberpad-driver/commit/2aae6152508bc4e169003fb87a24dfc7bc6edfe4) commit by following this process:
```shell
git clone https://github.com/asus-linux-drivers/asus-numberpad-driver
cd asus-numberpad-driver
git reset --hard 2aae615
bash install.sh
``` 
and I can confirm the numpad is working and it no longer has high CPU usage.

The things I found now is that 1. Launching the calculator isn't working, and 2. The numpad turns on at it's maximum brightness and has too many brightness levels. By default in Windows it only has two levels, a low and a high brightness setting. Having too many is kind of inconvenient.

Here are the logs from /var/log/asus-numberpad-driver for this installation:
[error.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/15210597/error.log)
[install-04-05-2024-12-16-40.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/15210598/install-04-05-2024-12-16-40.log)

irvng-wav, 2024-05-04:
> In case you are able activate NumberPad on Windows / your laptop has hardware support for NumberPad I need post here your DSDT table using:
> 
> ```
> $ sudo apt-get install iasl
> $ cd <work-dir>
> $ sudo cp /sys/firmware/acpi/tables/DSDT DSDT
> $ sudo iasl -d DSDT
> $ sudo gedit DSDT.dsl # output of this please post here
> ```

[DSDT.zip](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/15210619/DSDT.zip)

irvng-wav, 2024-05-04:
> The things I found now is that 1. Launching the calculator isn't working

```shell
bash /usr/share/asus-numberpad-driver/scripts/calculator_toggle.sh
``` 
Works correctly, running the command above launches the calculator.

```shell
xdotool key XF86Calculator
``` 
Returns nothing.

velito86, 2024-05-04:
Hi, recently I have upgraded my Kubuntu to 24.04 and started to experience the same issue. My laptop is Asus Zenbook UX3402VA and previously the numpad was working as expected using Kubuntu 23.10, kernel 6.5.xxx. What I figured so far is that after reboot I have python3 process which makes my laptop to go crazy. After restart of asus_numberpad_driver service, python3 process is not utilizing the CPU and numpad starts working as expected, except that it is turned on to max brightness which is not adjustable (it was adjustable previously).

Desktop
Program version: Release 6.0.0
Distro: Kubuntu 24.04
Kernel: Linux 6.8.0-31
Laptop: Zenbook 14 UX3402VA
DE: Plasma 5.27.11 on X11

ldrahnik, 2024-05-04:
@irvng-wav 

> The things I found now is that 1. Launching the calculator isn't working

Confirmed. Fixed. Please manually fix (picture below) whether is not missing comma (`,`) at the start of the path for the shortcut with the calculator toggling script and whether is not missing at the end of previous shortcut too.

Wrong:

[image]

Correct:

[image]

ldrahnik, 2024-05-04:
@velito86 Thank you for report.

ldrahnik, 2024-05-05:
@irvng-wav 

> By default in Windows it only has two levels

In case you use default config settings of this driver than you set up prefered brightness only once no? Because than should be used every NumberPad activation recently used brightness.

irvng-wav, 2024-05-05:
Thank you @ldrahnik. I installed the master branch fixed to the commit `66649a9` but then the problematic behavior that I described in the first post returned: numpad didn't work and cpu usage became high.

So uninstalled it and installed it fixed to the commit `2aae615` again, but I edited line 58 of `install_calc_toggle.sh` by adding the missing apostrophes to the comma so it ended up looking like this:
```shell
new_shortcut_string=${filtered_existing_shortcut_string::-2}"', '/org/gnome/settings-daemon/plugins/media-keys/custom-keybindings/custom$NEW_SHORTCUT_INDEX/']"
``` 
Just like you did in your commit.

Then I rebooted my system, installed `dconf-editor`, navigated to `custom-keybindings`, and I found this:
[image]
It doesn't look like yours. So I though the calculator toggle was still not working, but then I had the idea to try the toggle when the numpad is on. To my surprise it worked but you have to swipe your finger down from the toggle button and then press the calculator toggle to open the calculator. Here's a video:

https://github.com/asus-linux-drivers/asus-numberpad-driver/assets/147911588/63c93de8-8303-4ebf-9121-eb8f20bab6bf

In Windows you just have to slide your finger down from the toggle button and it automatically launches the calculator plus turns on the numpad, here's a video:

https://github.com/asus-linux-drivers/asus-numberpad-driver/assets/147911588/f8d3382c-816a-4089-a63d-70e89b082bed

At the end of the video you can see a bug when entering a mathematical operation and then trying to close the calculator with the toggle, so even the official driver is kinda wonky lol.

For me it would be ideal if the behavior of the linux driver did that too, that you can use the calculator toggle without the need to have the numpad already on. I don't know if that is something that can be done, if not I'm ok with how it works right now! so I want to thank you @ldrahnik! Your work is truly appreciated and also your commitment shows when helping us fix these problems. 😄🙏🏻

irvng-wav, 2024-05-05:
> @irvng-wav
> 
> > By default in Windows it only has two levels
> 
> In case you use default config settings of this driver than you set up prefered brightness only once, no? Because than should be used every NumberPad activation recently used brightness. So sounds to me better to have more options for brightness and count 8 does not sound to me be too much.

Yes! I actually realized that it saves your last brightness level, so that's perfect. Thank you!

ldrahnik, 2024-05-05:
@irvng-wav

> It doesn't look like yours.

Because I have had already added a lot of shorcuts and than I have installed this driver which adds only one - that Is why at the end of array with all shortcuts.

ldrahnik, 2024-05-05:
@irvng-wav 

> I installed the master branch fixed to the commit 66649a9 but then the problematic behavior that I described in the first post returned: numpad didn't work and cpu usage became high.

Thank you, was my mistake.

ldrahnik, 2024-05-05:
@irvng-wav 

> In Windows you just have to slide your finger down from the toggle button and it automatically launches the calculator plus turns on the numpad, here's a video:

Agree, at this moment is necessary to cross over the x axis (changeable in config) but y axis too (changeable in config too), both. Could you please create a separate feature issue? In my mind sounds as best option have something like elipse or similar curve preferred over triangle connecting point on x side and y side which serve as threshold line when is slide gesture considered as successfull (so I just check whether is point when user ends gesture inside of this shape).

ldrahnik, 2024-05-05:
@velito86 @irvng-wav 

> Numpad doesn't work and has high CPU usage

Should be fixed. Tested using X11 only.

irvng-wav, 2024-05-05:
> @velito86 @irvng-wav
> 
> > Numpad doesn't work and has high CPU usage
> 
> Should be fixed. Tested using X11 only. Can you confirm that? The latest master.

The numpad still doesn't work but it's no longer stressing the CPU.

irvng-wav, 2024-05-05:
> @irvng-wav
> 
> > In Windows you just have to slide your finger down from the toggle button and it automatically launches the calculator plus turns on the numpad, here's a video:
> 
> I agree, at this moment is necessary to cross over the x axis (changeable in config) but the y axis too (changeable in config too), so both these lines has to crossed. Could you please create a separate issue as feature request? In my mind sounds as best option have something like elipse or similar curve preferred over directly connected points (triangle) which would than serve as threshold line when is slide gesture considered as successful (so I will just check in code whether is point outside of this shape). But depends on performance of mathematics expression for such a shape. Feel free to describe your prefered variant.

Done. Just opened this [feature request](https://github.com/asus-linux-drivers/asus-numberpad-driver/issues/179).

ldrahnik, 2024-05-05:
@irvng-wav Thank you. I have created separated feature request for change activation area of slide gestures [here](https://github.com/asus-linux-drivers/asus-numberpad-driver/issues/180)

ldrahnik, 2024-05-06:
>> @velito86 @irvng-wav
>> 
>> > Numpad doesn't work and has high CPU usage
>> 
>> Should be fixed. Tested using X11 only. Can you confirm that? The latest master.

> Can confirm the numpad still doesn't work but it's no longer stressing the CPU. Tested on Wayland and XOrg.

Does help restarting systemctl service? Then NumberPad works? As was mentioned by @velito86.

irvng-wav, 2024-05-07:
@ldrahnik nope. I tried the following command:
```
systemctl restart --user asus_numberpad_driver@irvng.service
``` 
and it still doesn't work. In error.log the same thing is logged over and over again:
```
2024-05-07 02:10:41,108 INFO Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
2024-05-07 02:10:41,108 INFO Set keyboard 4 from H: Handlers=sysrq kbd leds event4
2024-05-07 02:10:41,144 INFO Detecting touchpad from string: "N: Name="ASUE140D:00 04F3:31B9 Touchpad""
2024-05-07 02:10:41,144 INFO Set touchpad device id 15 from S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-15/i2c-ASUE140D:00/0018:04F3:31B9.0001/input/input13
2024-05-07 02:10:41,144 INFO Set touchpad id 10 from H: Handlers=event10 mouse1
2024-05-07 02:10:41,145 INFO Touchpad min-max: x 0-3880, y 0-2299
2024-05-07 02:10:41,145 INFO Numpad min-max: x 200-3680, y 200-2219
2024-05-07 02:10:50,008 INFO Old device at /dev/input/event16 (/sys/devices/virtual/input/input25)
2024-05-07 02:10:50,440 INFO New device at /dev/input/event17 (/sys/devices/virtual/input/input26)
2024-05-07 02:10:51,555 ERROR Listening touchpad events unexpectedly failed
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 2355, in <module>
    listen_touchpad_events()
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 1948, in listen_touchpad_events
    for e in d_t.events():
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages/libevdev/device.py", line 541, in events
    raise EventsDroppedException()
libevdev.device.EventsDroppedException
``` 
[error.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/15232619/error.log)

ldrahnik, 2024-05-07:
@irvng-wav Weird. What would help me is to move `listen_touchpad_events()` outside of the `try` block, try again and post here log. Could you do that, please?

now:

```
try:
    listen_touchpad_events()
except:
```

to:

```
listen_touchpad_events()

try:
    pass
except:
```

ldrahnik, 2024-05-07:
@irvng-wav The latest fix could help.

irvng-wav, 2024-05-08:
> @irvng-wav What would help me to figure out what is wrong would move `listen_touchpad_events()` outside of the `try` block, try again, and post here error log. Could you do that, please?
> 
> now:
> 
> ```
> try:
>     listen_touchpad_events()
> except:
> ```
> 
> to:
> 
> ```
> listen_touchpad_events()
> 
> try:
>     pass
> except:
> ```

Ok if I understood correctly that would mean editing `/usr/share/asus-numberpad-driver/numberpad.py` and make it like this:
[image]
So I did that, saved the changes and restarted the service with:
```
systemctl restart --user asus_numberpad_driver@irvng.service
``` 
Now the output of `error.log` is this:
```
2024-05-08 00:34:49,632 INFO Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
2024-05-08 00:34:49,632 INFO Set keyboard 4 from H: Handlers=sysrq kbd leds event4
2024-05-08 00:34:49,653 INFO Detecting touchpad from string: "N: Name="ASUE140D:00 04F3:31B9 Touchpad""
2024-05-08 00:34:49,653 INFO Set touchpad device id 15 from S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-15/i2c-ASUE140D:00/0018:04F3:31B9.0001/input/input13
2024-05-08 00:34:49,654 INFO Set touchpad id 10 from H: Handlers=event10 mouse1
2024-05-08 00:34:49,654 INFO Touchpad min-max: x 0-3880, y 0-2299
2024-05-08 00:34:49,654 INFO Numpad min-max: x 200-3680, y 200-2219
2024-05-08 00:34:56,864 INFO Old device at /dev/input/event16 (/sys/devices/virtual/input/input39)
2024-05-08 00:34:57,073 INFO New device at /dev/input/event17 (/sys/devices/virtual/input/input40)
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 2353, in <module>
    listen_touchpad_events()
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 1947, in listen_touchpad_events
    for e in d_t.events():
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages/libevdev/device.py", line 541, in events
    raise EventsDroppedException()
libevdev.device.EventsDroppedException
``` 
All this was done with the latest commit `4b75c12`.

ldrahnik, 2024-05-08:
@irvng-wav  Thank you a lot. Atleast directly without systemctl installed it should work, more in https://github.com/asus-linux-drivers/asus-numberpad-driver/pull/181

ldrahnik, 2024-05-08:
@irvng-wav The latest commits might help too.

ldrahnik, 2024-05-11:
@irvng-wav The latest commit might help too (fixed switching layouts for gnome with older deprecated  `org.gnome.desktop.input-sources`).

irvng-wav, 2024-05-12:
You absolutely did it man! I just installed the last commit and it works like a charm. Thank you so much for your dedication, commitment and of course your time and energy that you put into this project! I will donate to your [ko-fi page](https://ko-fi.com/ldrahnik) as soon as I can. I'm overspent right now so probably when I get my next payment.

I encourage everyone that's benefiting from this project to donate to @ldrahnik. Truly a wonderful service.

Thank you once again. 🙏🏻

ldrahnik, 2024-05-12:
@irvng-wav Could you please post here the entire error log file (`/var/log/asus-numberpad-driver/error.log`) containing these random stops.

irvng-wav, 2024-05-13:
> @irvng-wav Could you please post here the entire error log file (`/var/log/asus-numberpad-driver/error.log`) containing these random stops.

Sure! Here's the log from the installation of the last commit [d2d8487](https://github.com/asus-linux-drivers/asus-numberpad-driver/commit/d2d8487cf93270c67c8e93ae06f3923c96e5c19c):
[error.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/15289861/error.log)

This time the behavior was this:
1. I installed the driver and rebooted, it didn't work for like a minute or two, then it started working, but the backlight level didn't save, so every time I turned off and then on the numpad, the backlight reset to the highest level.
2. Then I rebooted, and it also didn't work for like a minute, then it started working and now the backlight level did save.

ldrahnik, 2024-05-14:
@irvng-wav 

> 2024-05-12 23:27:54,720 ERROR Listening touchpad events unexpectedly failed
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 2386, in <module>
    udev = dev.create_uinput_device()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages/libevdev/device.py", line 764, in create_uinput_device
    d._uinput = UinputDevice(self._libevdev, uinput_fd)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages/libevdev/_clib.py", line 964, in __init__
    raise OSError(-rc, os.strerror(-rc))
PermissionError: [Errno 13] Permission denied

What is the output of `$ groups`?

ldrahnik, 2024-05-14:
@irvng-wav Please, try to run driver directly without systemctl service using `$ /usr/share/asus-numberpad-driver/.env/bin/python3 /usr/share/asus-numberpad-driver/numberpad.py <up5401ea|e210ma|..>`. Does it work? Clean the error log before and attach content please.

irvng-wav, 2024-05-17:
> @irvng-wav Please, try to run driver directly without systemctl service using `$ /usr/share/asus-numberpad-driver/.env/bin/python3 /usr/share/asus-numberpad-driver/numberpad.py <up5401ea|e210ma|..>`. Does it work? Clean the error log file before and post it here please.

I get this (the numpad was already working before I launched the command):
```
[irvng@arch-zenbook ~]$ /usr/share/asus-numberpad-driver/.env/bin/python3 /usr/share/asus-numberpad-driver/numberpad.py up5401ea
2024-05-17 00:07:10,816 INFO Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
2024-05-17 00:07:10,817 INFO Set keyboard 4 from H: Handlers=sysrq kbd leds event4
2024-05-17 00:07:10,846 INFO Detecting touchpad from string: "N: Name="ASUE140D:00 04F3:31B9 Touchpad""
2024-05-17 00:07:10,846 INFO Set touchpad device id 15 from S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-15/i2c-ASUE140D:00/0018:04F3:31B9.0001/input/input13
2024-05-17 00:07:10,846 INFO Set touchpad id 10 from H: Handlers=event10 mouse1
2024-05-17 00:07:10,846 INFO Touchpad min-max: x 0-3880, y 0-2299
2024-05-17 00:07:10,846 INFO Numpad min-max: x 200-3680, y 200-2219
2024-05-17 00:07:18,941 INFO Setting up for config file key: "brightness" with value: "0x48"
2024-05-17 00:07:19,046 ERROR Listening touchpad events unexpectedly failed
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 2435, in <module>
    listen_touchpad_events()
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 1957, in listen_touchpad_events
    for e in d_t.events():
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages/libevdev/device.py", line 541, in events
    raise EventsDroppedException()
libevdev.device.EventsDroppedException
2024-05-17 00:07:19,048 INFO Clean up started
[2024-05-17 00:07:19,092 pyinotify ERROR] Cannot delete unknown watch descriptor '/home/irvng'
2024-05-17 00:07:19,092 ERROR Cannot delete unknown watch descriptor '/home/irvng'
2024-05-17 00:07:19,093 INFO Clean up finished
2024-05-17 00:07:19,093 INFO Exiting
``` 
[error.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/15344377/error.log)

irvng-wav, 2024-05-17:
After my last comment I deleted error.log to start fresh again, rebooted the computer and did the same thing. Now the output of `/usr/share/asus-numberpad-driver/.env/bin/python3 /usr/share/asus-numberpad-driver/numberpad.py up5401ea` is this:
```
[irvng@arch-zenbook ~]$ /usr/share/asus-numberpad-driver/.env/bin/python3 /usr/share/asus-numberpad-driver/numberpad.py up5401ea
2024-05-17 00:11:21,452 INFO Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
2024-05-17 00:11:21,452 INFO Set keyboard 4 from H: Handlers=sysrq kbd leds event4
2024-05-17 00:11:21,483 INFO Detecting touchpad from string: "N: Name="ASUE140D:00 04F3:31B9 Touchpad""
2024-05-17 00:11:21,483 INFO Set touchpad device id 15 from S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-15/i2c-ASUE140D:00/0018:04F3:31B9.0001/input/input13
2024-05-17 00:11:21,483 INFO Set touchpad id 10 from H: Handlers=event10 mouse1
2024-05-17 00:11:21,483 INFO Touchpad min-max: x 0-3880, y 0-2299
2024-05-17 00:11:21,483 INFO Numpad min-max: x 200-3680, y 200-2219
2024-05-17 00:11:52,020 INFO Touched top_right_icon area (numlock) in time: 1715926312.0209637
2024-05-17 00:11:52,480 INFO Un-touched top_right_icon area (representing numlock key) in time: 1715926312.480498
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ASUE140D:00 04F3:31B9 Touchpad
2024-05-17 00:11:52,576 INFO Setting up for config file key: "enabled" with value: "True"
2024-05-17 00:11:52,577 INFO check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/home/irvng pathname=/home/irvng/numberpad_dev wd=1 >
2024-05-17 00:11:52,677 INFO Numpad activated
2024-05-17 00:11:54,593 INFO Touched top_right_icon area (numlock) in time: 1715926314.593668
2024-05-17 00:11:55,185 INFO Un-touched top_right_icon area (representing numlock key) in time: 1715926315.185171
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ASUE140D:00 04F3:31B9 Touchpad
2024-05-17 00:11:55,290 INFO Setting up for config file key: "enabled" with value: "False"
2024-05-17 00:11:55,291 INFO check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/home/irvng pathname=/home/irvng/numberpad_dev wd=1 >
2024-05-17 00:11:55,391 INFO Numpad deactivated
2024-05-17 00:12:00,986 INFO Touched top_right_icon area (numlock) in time: 1715926320.9868817
2024-05-17 00:12:01,543 INFO Un-touched top_right_icon area (representing numlock key) in time: 1715926321.5433066
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ASUE140D:00 04F3:31B9 Touchpad
2024-05-17 00:12:01,639 INFO Setting up for config file key: "enabled" with value: "True"
2024-05-17 00:12:01,639 INFO check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/home/irvng pathname=/home/irvng/numberpad_dev wd=1 >
2024-05-17 00:12:01,739 INFO Numpad activated
2024-05-17 00:12:02,699 INFO Touched top_right_icon area (numlock) in time: 1715926322.6990178
2024-05-17 00:12:03,149 INFO Un-touched top_right_icon area (representing numlock key) in time: 1715926323.1490846
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ASUE140D:00 04F3:31B9 Touchpad
2024-05-17 00:12:03,337 INFO Setting up for config file key: "enabled" with value: "False"
2024-05-17 00:12:03,337 INFO check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/home/irvng pathname=/home/irvng/numberpad_dev wd=1 >
2024-05-17 00:12:03,432 INFO Touched top_right_icon area (numlock) in time: 1715926323.4329174
2024-05-17 00:12:03,437 INFO Numpad deactivated
2024-05-17 00:12:03,647 INFO Un-touched top_right_icon area (representing numlock key) in time: 1715926323.6470108
2024-05-17 00:12:03,958 INFO Touched top_right_icon area (numlock) in time: 1715926323.958899
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ASUE140D:00 04F3:31B9 Touchpad
2024-05-17 00:12:04,499 INFO Setting up for config file key: "enabled" with value: "True"
2024-05-17 00:12:04,499 INFO check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/home/irvng pathname=/home/irvng/numberpad_dev wd=1 >
2024-05-17 00:12:04,511 INFO Un-touched top_right_icon area (representing numlock key) in time: 1715926324.5117722
2024-05-17 00:12:04,599 INFO Numpad activated
2024-05-17 00:12:05,661 INFO Touched top_right_icon area (numlock) in time: 1715926325.6613195
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ASUE140D:00 04F3:31B9 Touchpad
2024-05-17 00:12:06,167 INFO Setting up for config file key: "enabled" with value: "False"
2024-05-17 00:12:06,167 INFO check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
[... cut ...]
[error.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/15344422/error.log)

The numpad was also working before I entered the command. I think because of that now the numpad first turns on to the lowest brightness (I left it there), then it changes to max brightness. Also when I change it back to the lowest brightness it doesn't save the state. Here's an updated error.log after that behavior:

[error.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/15344449/error.log)

ldrahnik, 2024-05-17:
@irvng-wav What is the output of `$ groups?`

irvng-wav, 2024-05-17:
Once again, deleted error.log, rebooted, but this time I ran `systemctl stop --user asus_numberpad_driver@irvng.service` to stop the already running numpad and then entered `/usr/share/asus-numberpad-driver/.env/bin/python3 /usr/share/asus-numberpad-driver/numberpad.py up5401ea`:

```
[irvng@arch-zenbook ~]$ /usr/share/asus-numberpad-driver/.env/bin/python3 /usr/share/asus-numberpad-driver/numberpad.py up5401ea
2024-05-17 00:20:49,221 INFO Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
2024-05-17 00:20:49,221 INFO Set keyboard 4 from H: Handlers=sysrq kbd leds event4
2024-05-17 00:20:49,256 INFO Detecting touchpad from string: "N: Name="ASUE140D:00 04F3:31B9 Touchpad""
2024-05-17 00:20:49,256 INFO Set touchpad device id 15 from S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-15/i2c-ASUE140D:00/0018:04F3:31B9.0001/input/input13
2024-05-17 00:20:49,256 INFO Set touchpad id 9 from H: Handlers=event9 mouse1
2024-05-17 00:20:49,256 INFO Touchpad min-max: x 0-3880, y 0-2299
2024-05-17 00:20:49,256 INFO Numpad min-max: x 200-3680, y 200-2219
2024-05-17 00:21:04,666 INFO Touched top_right_icon area (numlock) in time: 1715926864.666638
2024-05-17 00:21:05,673 INFO The numpad numlock was pressed longer than the activation time and will be activated: 1.0063786506652832
2024-05-17 00:21:05,673 INFO Activation time: 1.0
2024-05-17 00:21:05,713 INFO System numlock activated
2024-05-17 00:21:05,713 INFO Numpad activated
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ASUE140D:00 04F3:31B9 Touchpad
2024-05-17 00:21:05,731 INFO Setting up for config file key: "enabled" with value: "True"
2024-05-17 00:21:05,731 INFO check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/home/irvng pathname=/home/irvng/numberpad_dev wd=1 >
2024-05-17 00:21:06,143 INFO Un-touched top_right_icon area (representing numlock key) in time: 1715926866.143782
2024-05-17 00:21:08,666 INFO Touched top_right_icon area (numlock) in time: 1715926868.6663475
2024-05-17 00:21:09,672 INFO The numpad numlock was pressed longer than the activation time and will be activated: 1.0062620639801025
2024-05-17 00:21:09,672 INFO Activation time: 1.0
2024-05-17 00:21:09,699 INFO System numlock deactivated
2024-05-17 00:21:09,700 INFO Numpad deactivated
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ASUE140D:00 04F3:31B9 Touchpad
2024-05-17 00:21:09,722 INFO Setting up for config file key: "enabled" with value: "False"
2024-05-17 00:21:09,723 INFO check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/home/irvng pathname=/home/irvng/numberpad_dev wd=1 >
2024-05-17 00:21:10,053 INFO Un-touched top_right_icon area (representing numlock key) in time: 1715926870.0538008
2024-05-17 00:21:10,213 INFO Touched top_right_icon area (numlock) in time: 1715926870.212999
2024-05-17 00:21:11,216 INFO The numpad numlock was pressed longer than the activation time and will be activated: 1.0030953884124756
2024-05-17 00:21:11,216 INFO Activation time: 1.0
2024-05-17 00:21:11,236 INFO System numlock activated
2024-05-17 00:21:11,236 INFO Numpad activated
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ASUE140D:00 04F3:31B9 Touchpad
2024-05-17 00:21:11,256 INFO Setting up for config file key: "enabled" with value: "True"
2024-05-17 00:21:11,256 INFO check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/home/irvng pathname=/home/irvng/numberpad_dev wd=1 >
2024-05-17 00:21:11,562 INFO Un-touched top_right_icon area (representing numlock key) in time: 1715926871.5622182
2024-05-17 00:21:12,040 INFO Touched top_left_icon area in time: 1715926872.0401757
2024-05-17 00:21:12,531 INFO Slide from top_left_icon exceeded the activation threshold for x and y.
2024-05-17 00:21:12,531 INFO Activation x 3830.00 (top left corner is 0; required > 776.00 / 3880.00)
2024-05-17 00:21:12,531 INFO Activation y 804.00 (top left corner is 0; required > 459.80 / 2299.00)
2024-05-17 00:21:12,531 INFO Used bindings for touchpad left_icon slide function
2024-05-17 00:21:12,531 INFO Touched top_left_icon area in time: 1715926872.5315113
2024-05-17 00:21:13,285 INFO Un-touched top_right_icon area (representing numlock key) in time: 1715926873.2852216
2024-05-17 00:21:14,088 INFO Touched top_left_icon area in time: 1715926874.0885785
2024-05-17 00:21:14,482 INFO Slide from top_left_icon exceeded the activation threshold for x and y.
2024-05-17 00:21:14,482 INFO Activation x 3852.00 (top left corner is 0; required > 776.00 / 3880.00)
2024-05-17 00:21:14,482 INFO Activation y 729.00 (top left corner is 0; required > 459.80 / 2299.00)
2024-05-17 00:21:14,482 INFO Used bindings for touchpad left_icon slide function
2024-05-17 00:21:14,483 INFO Touched top_left_icon area in time: 1715926874.4830256
2024-05-17 00:21:15,008 INFO Un-touched top_right_icon area (representing numlock key) in time: 1715926875.0086925
2024-05-17 00:21:17,773 INFO Touched top_right_icon area (numlock) in time: 1715926877.7733996
[... cut ...]
2024-05-17 00:21:30,362 INFO check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/home/irvng pathname=/home/irvng/numberpad_dev wd=1 >
2024-05-17 00:21:30,509 INFO Un-touched top_right_icon area (representing numlock key) in time: 1715926890.5096023
``` 
Here's the error.log corresponding to that:
[error.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/15344514/error.log)

irvng-wav, 2024-05-17:
> @irvng-wav What is the output of `$ groups?`
```
[irvng@arch-zenbook ~]$ groups
i2c realtime video input audio wheel irvng numberpad uinput
```

irvng-wav, 2024-05-17:
> Please check rights of parent directory in config file location & config file.

```
[irvng@arch-zenbook asus-numberpad-driver]$ ls -la
total 124
drwxr-xr-x   6 irvng root   4096 may 17 13:18 .
drwxr-xr-x 239 root  root  12288 may 17 00:01 ..
drwxr-xr-x   4 irvng irvng  4096 may 17 00:01 .env
drwxr-xr-x   3 irvng root   4096 may 17 00:02 layouts
-rw-r--r--   1 irvng irvng   719 may 17 13:20 numberpad_dev
-rwxr-xr-x   1 root  root  86973 may 17 00:01 numberpad.py
drwxr-xr-x   2 root  root   4096 may 17 00:02 scripts
drwxr-xr-x   2 root  root   4096 may 17 00:02 udev
```

irvng-wav, 2024-05-17:
I restarted my computer a couple times and the numpad wasn't working, although the service was running. The first time the systemctl service displayed an error but I don't have that log anymore, on the second reboot the status was active and running without any problems.

So I started manually the service with 
```
/usr/share/asus-numberpad-driver/.env/bin/python3 /usr/share/asus-numberpad-driver/numberpad.py up5401ea
``` 
> Lowest saved brightness looks like `0x41` (should be changed each 1s during the time when you hold the top left icon spot):
> Does it change? Thank you for your time. It looks like the problem is only Arch.

You're welcome! Thank you as well. Here's a video:

https://github.com/asus-linux-drivers/asus-numberpad-driver/assets/147911588/eab875e7-99b5-4240-88d8-9a2efd85c940

Weird behavior man. The commit `2aae615` works for me perfectly. Literally no issues at all. What's different from that version compared to this one?

ldrahnik, 2024-05-18:
@irvng-wav Thank you. Problem I replicated and should be fixed by the latest commit.

ldrahnik, 2024-05-18:
@irvng-wav About not receiving keymap through wayland - I have added debug logs to the master branch. Please, start the driver directly with `LOG=DEBUG ..` at the start and post printed output to the terminal here.

ldrahnik, 2024-05-23:
@irvng-wav ping

irvng-wav, 2024-05-24:
Hey @ldrahnik! I saw your ping earlier. Thank you for the reminder, just had time to test the new branch. Here's the log:

```
[irvng@arch-zenbook ~]$ LOG=DEBUG /usr/share/asus-numberpad-driver/.env/bin/python3 /usr/share/asus-numberpad-driver/numberpad.py up5401ea
2024-05-23 20:40:58,791 INFO Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
2024-05-23 20:40:58,791 INFO Set keyboard 4 from H: Handlers=sysrq kbd leds event4
2024-05-23 20:40:58,818 INFO Detecting touchpad from string: "N: Name="ASUE140D:00 04F3:31B9 Touchpad""
2024-05-23 20:40:58,818 INFO Set touchpad device id 15 from S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-15/i2c-ASUE140D:00/0018:04F3:31B9.0001/input/input13
2024-05-23 20:40:58,818 INFO Set touchpad id 10 from H: Handlers=event10 mouse1
2024-05-23 20:40:58,819 INFO Touchpad min-max: x 0-3880, y 0-2299
2024-05-23 20:40:58,819 INFO Numpad min-max: x 200-3680, y 200-2219
2024-05-23 20:40:58,836 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7bddb5307320>
2024-05-23 20:40:58,836 DEBUG 1
2024-05-23 20:40:58,841 DEBUG wl_compositor
2024-05-23 20:40:58,841 DEBUG 6
2024-05-23 20:40:58,842 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7bddb5307320>
2024-05-23 20:40:58,842 DEBUG 2
2024-05-23 20:40:58,842 DEBUG wl_drm
2024-05-23 20:40:58,842 DEBUG 2
2024-05-23 20:40:58,842 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7bddb5307320>
2024-05-23 20:40:58,842 DEBUG 3
2024-05-23 20:40:58,842 DEBUG wl_shm
2024-05-23 20:40:58,842 DEBUG 1
2024-05-23 20:40:58,842 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7bddb5307320>
2024-05-23 20:40:58,842 DEBUG 4
2024-05-23 20:40:58,842 DEBUG wl_output
2024-05-23 20:40:58,842 DEBUG 4
2024-05-23 20:40:58,847 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7bddb5307320>
2024-05-23 20:40:58,852 DEBUG 5
2024-05-23 20:40:58,852 DEBUG zxdg_output_manager_v1
2024-05-23 20:40:58,852 DEBUG 3
2024-05-23 20:40:58,852 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7bddb5307320>
2024-05-23 20:40:58,852 DEBUG 6
2024-05-23 20:40:58,852 DEBUG wl_data_device_manager
2024-05-23 20:40:58,852 DEBUG 3
2024-05-23 20:40:58,852 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7bddb5307320>
2024-05-23 20:40:58,853 DEBUG 7
2024-05-23 20:40:58,853 DEBUG zwp_primary_selection_device_manager_v1
2024-05-23 20:40:58,853 DEBUG 1
2024-05-23 20:40:58,853 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7bddb5307320>
2024-05-23 20:40:58,853 DEBUG 8
2024-05-23 20:40:58,853 DEBUG wl_subcompositor
2024-05-23 20:40:58,853 DEBUG 1
2024-05-23 20:40:58,853 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7bddb5307320>
2024-05-23 20:40:58,853 DEBUG 9
2024-05-23 20:40:58,853 DEBUG xdg_wm_base
2024-05-23 20:40:58,853 DEBUG 6
2024-05-23 20:40:58,853 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7bddb5307320>
2024-05-23 20:40:58,853 DEBUG 10
2024-05-23 20:40:58,853 DEBUG gtk_shell1
2024-05-23 20:40:58,853 DEBUG 5
2024-05-23 20:40:58,853 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7bddb5307320>
2024-05-23 20:40:58,853 DEBUG 11
2024-05-23 20:40:58,853 DEBUG wp_viewporter
2024-05-23 20:40:58,853 DEBUG 1
2024-05-23 20:40:58,853 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7bddb5307320>
2024-05-23 20:40:58,853 DEBUG 12
2024-05-23 20:40:58,853 DEBUG wp_fractional_scale_manager_v1
2024-05-23 20:40:58,853 DEBUG 1
2024-05-23 20:40:58,853 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7bddb5307320>
2024-05-23 20:40:58,853 DEBUG 13
2024-05-23 20:40:58,853 DEBUG zwp_pointer_gestures_v1
2024-05-23 20:40:58,853 DEBUG 3
2024-05-23 20:40:58,853 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7bddb5307320>
2024-05-23 20:40:58,853 DEBUG 14
2024-05-23 20:40:58,853 DEBUG zwp_tablet_manager_v2
2024-05-23 20:40:58,853 DEBUG 1
2024-05-23 20:40:58,853 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7bddb5307320>
2024-05-23 20:40:58,853 DEBUG 15
2024-05-23 20:40:58,853 DEBUG wl_seat
[... 54 lines cut ...]
2024-05-23 20:40:58,908 DEBUG 29
2024-05-23 20:40:58,908 DEBUG wp_linux_drm_syncobj_manager_v1
2024-05-23 20:40:58,908 DEBUG 1
2024-05-23 20:40:58,943 DEBUG Wayland will try to load keymap
2024-05-23 20:41:04,141 DEBUG Wayland loaded keymap succesfully
2024-05-23 20:41:04,141 DEBUG {'Num_Lock': KEY_NUMLOCK:69, '0': KEY_0:11, '1': KEY_1:2, '2': KEY_2:3, '3': KEY_3:4, '4': KEY_4:5, '5': KEY_5:6, '6': KEY_6:7, '7': KEY_7:8, '8': KEY_8:9, '9': KEY_9:10, 'a': KEY_A:30, 'b': KEY_B:48, 'c': KEY_C:46, 'd': KEY_D:32, 'e': KEY_E:18, 'f': KEY_F:33, 'Shift_L': KEY_LEFTSHIFT:42, 'Control_L': KEY_LEFTCTRL:29, 'u': KEY_U:22, 'space': KEY_SPACE:57, 'slash': [KEY_LEFTSHIFT:42, KEY_7:8], 'BackSpace': KEY_BACKSPACE:14, 'asterisk': [KEY_LEFTSHIFT:42, KEY_RIGHTBRACE:27], 'minus': KEY_SLASH:53, 'percent': [KEY_LEFTSHIFT:42, KEY_5:6], 'period': KEY_DOT:52, 'Return': KEY_ENTER:28, 'plus': KEY_RIGHTBRACE:27, 'equal': [KEY_LEFTSHIFT:42, KEY_0:11]}
2024-05-23 20:41:04,644 INFO Setting up for config file key: "brightness" with value: "0x48"
2024-05-23 20:41:04,644 DEBUG Writting to config file: "<_io.TextIOWrapper name='./numberpad_dev' mode='w' encoding='UTF-8'>"
[... cut ...]
2024-05-23 20:41:04,754 INFO Clean up started
[2024-05-23 20:41:04,828 pyinotify ERROR] Cannot delete unknown watch descriptor '/home/irvng'
2024-05-23 20:41:04,828 ERROR Cannot delete unknown watch descriptor '/home/irvng'
2024-05-23 20:41:04,828 INFO Clean up finished
2024-05-23 20:41:04,828 INFO Exiting
[irvng@arch-zenbook ~]$ 
``` 
The driver is working correctly. Although at first I thought it wasn't working but it might have been me not pressing the numpad on/off button for the needed ammount of time. Let me restart and try again.

irvng-wav, 2024-05-24:
Ok just restarted my computer and the numpad is working right from the start. That's great! Here's the error.log from this testing session:
[error.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/15426766/error.log)

So I stopped the systemctl service, then manually started it wth LOG=DEBUG. Here are the logs:
```
[irvng@arch-zenbook ~]$ systemctl stop --user asus_numberpad_driver@irvng.service
[irvng@arch-zenbook ~]$ LOG=DEBUG /usr/share/asus-numberpad-driver/.env/bin/python3 /usr/share/asus-numberpad-driver/numberpad.py up5401ea
2024-05-23 20:57:51,043 INFO Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
2024-05-23 20:57:51,043 INFO Set keyboard 4 from H: Handlers=sysrq kbd leds event4
2024-05-23 20:57:51,066 INFO Detecting touchpad from string: "N: Name="ASUE140D:00 04F3:31B9 Touchpad""
2024-05-23 20:57:51,066 INFO Set touchpad device id 15 from S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-15/i2c-ASUE140D:00/0018:04F3:31B9.0001/input/input13
2024-05-23 20:57:51,066 INFO Set touchpad id 10 from H: Handlers=event10 mouse1
2024-05-23 20:57:51,067 INFO Touchpad min-max: x 0-3880, y 0-2299
2024-05-23 20:57:51,067 INFO Numpad min-max: x 200-3680, y 200-2219
2024-05-23 20:57:51,079 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7316e23f6ea0>
2024-05-23 20:57:51,079 DEBUG 1
2024-05-23 20:57:51,079 DEBUG wl_compositor
2024-05-23 20:57:51,079 DEBUG 6
2024-05-23 20:57:51,079 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7316e23f6ea0>
2024-05-23 20:57:51,079 DEBUG 2
2024-05-23 20:57:51,079 DEBUG wl_drm
2024-05-23 20:57:51,079 DEBUG 2
2024-05-23 20:57:51,079 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7316e23f6ea0>
2024-05-23 20:57:51,079 DEBUG 3
2024-05-23 20:57:51,079 DEBUG wl_shm
2024-05-23 20:57:51,079 DEBUG 1
2024-05-23 20:57:51,079 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7316e23f6ea0>
2024-05-23 20:57:51,079 DEBUG 4
2024-05-23 20:57:51,079 DEBUG wl_output
2024-05-23 20:57:51,079 DEBUG 4
2024-05-23 20:57:51,079 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7316e23f6ea0>
2024-05-23 20:57:51,080 DEBUG 5
2024-05-23 20:57:51,085 DEBUG zxdg_output_manager_v1
2024-05-23 20:57:51,090 DEBUG 3
2024-05-23 20:57:51,090 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7316e23f6ea0>
2024-05-23 20:57:51,095 DEBUG 6
2024-05-23 20:57:51,095 DEBUG wl_data_device_manager
2024-05-23 20:57:51,095 DEBUG 3
2024-05-23 20:57:51,096 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7316e23f6ea0>
2024-05-23 20:57:51,096 DEBUG 7
2024-05-23 20:57:51,096 DEBUG zwp_primary_selection_device_manager_v1
2024-05-23 20:57:51,096 DEBUG 1
2024-05-23 20:57:51,096 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7316e23f6ea0>
2024-05-23 20:57:51,096 DEBUG 8
2024-05-23 20:57:51,096 DEBUG wl_subcompositor
2024-05-23 20:57:51,096 DEBUG 1
2024-05-23 20:57:51,096 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7316e23f6ea0>
2024-05-23 20:57:51,096 DEBUG 9
2024-05-23 20:57:51,096 DEBUG xdg_wm_base
2024-05-23 20:57:51,096 DEBUG 6
2024-05-23 20:57:51,101 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7316e23f6ea0>
2024-05-23 20:57:51,101 DEBUG 10
2024-05-23 20:57:51,101 DEBUG gtk_shell1
2024-05-23 20:57:51,101 DEBUG 5
2024-05-23 20:57:51,101 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7316e23f6ea0>
2024-05-23 20:57:51,101 DEBUG 11
2024-05-23 20:57:51,106 DEBUG wp_viewporter
2024-05-23 20:57:51,106 DEBUG 1
2024-05-23 20:57:51,106 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7316e23f6ea0>
2024-05-23 20:57:51,106 DEBUG 12
2024-05-23 20:57:51,107 DEBUG wp_fractional_scale_manager_v1
2024-05-23 20:57:51,107 DEBUG 1
2024-05-23 20:57:51,107 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7316e23f6ea0>
2024-05-23 20:57:51,107 DEBUG 13
2024-05-23 20:57:51,107 DEBUG zwp_pointer_gestures_v1
2024-05-23 20:57:51,107 DEBUG 3
2024-05-23 20:57:51,112 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7316e23f6ea0>
2024-05-23 20:57:51,117 DEBUG 14
2024-05-23 20:57:51,122 DEBUG zwp_tablet_manager_v2
2024-05-23 20:57:51,128 DEBUG 1
[... 109 lines cut ...]
2024-05-23 20:58:07,284 DEBUG key unpress was already handled because slot is empty
2024-05-23 20:58:07,476 DEBUG Started new slot
2024-05-23 20:58:07,478 DEBUG finger down at x 1689 y 1830
2024-05-23 20:58:07,898 DEBUG Ended existing slot
2024-05-23 20:58:07,900 DEBUG finger down at x 1728 y 1707
2024-05-23 20:58:07,900 DEBUG key unpress was already handled because slot is empty
^C2024-05-23 20:58:08,863 ERROR Listening touchpad events unexpectedly failed
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 2431, in <module>
    listen_touchpad_events()
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 1953, in listen_touchpad_events
    for e in d_t.events():
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages/libevdev/device.py", line 542, in events
    ev = self._libevdev.next_event(flags)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages/libevdev/_clib.py", line 889, in next_event
    rc = self._next_event(self._ctx, flags, ctypes.byref(ev))
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
KeyboardInterrupt
2024-05-23 20:58:08,876 INFO Clean up started
[2024-05-23 20:58:08,896 pyinotify ERROR] Cannot delete unknown watch descriptor '/home/irvng'
2024-05-23 20:58:08,896 ERROR Cannot delete unknown watch descriptor '/home/irvng'
2024-05-23 20:58:08,896 INFO Clean up finished
2024-05-23 20:58:08,897 INFO Exiting
```

ldrahnik, 2024-05-24:
@irvng-wav May I close this issue as completed? or any problem persists?

ldrahnik, 2024-05-24:
@kamack38 Could you please confirm it works too? I tried installing driver on garuda in the virtual box yesterday and it works fine.

irvng-wav, 2024-05-25:
> @irvng-wav May I close this issue as completed? or any problem persists?

Will test in the morning and report back to you. It's sleep time on this side of the world. 😜

irvng-wav, 2024-05-26:
For some reason the numpad service isn't working correctly when I reboot my system. I rebooted twice and had to start the service manually. Here is the log:
```
[irvng@arch-zenbook ~]$ LOG=DEBUG /usr/share/asus-numberpad-driver/.env/bin/python3 /usr/share/asus-numberpad-driver/numberpad.py up5401ea
2024-05-26 01:06:02,436 INFO Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
2024-05-26 01:06:02,436 INFO Set keyboard 4 from H: Handlers=sysrq kbd leds event4
2024-05-26 01:06:02,462 INFO Detecting touchpad from string: "N: Name="ASUE140D:00 04F3:31B9 Touchpad""
2024-05-26 01:06:02,462 INFO Set touchpad device id 15 from S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-15/i2c-ASUE140D:00/0018:04F3:31B9.0001/input/input13
2024-05-26 01:06:02,462 INFO Set touchpad id 10 from H: Handlers=event10 mouse1
2024-05-26 01:06:02,462 INFO Touchpad min-max: x 0-3880, y 0-2299
2024-05-26 01:06:02,462 INFO Numpad min-max: x 200-3680, y 200-2219
2024-05-26 01:06:02,474 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7a6b305faba0>
2024-05-26 01:06:02,479 DEBUG 1
2024-05-26 01:06:02,484 DEBUG wl_compositor
2024-05-26 01:06:02,490 DEBUG 6
2024-05-26 01:06:02,490 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7a6b305faba0>
2024-05-26 01:06:02,490 DEBUG 2
2024-05-26 01:06:02,490 DEBUG wl_drm
2024-05-26 01:06:02,495 DEBUG 2
2024-05-26 01:06:02,500 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7a6b305faba0>
2024-05-26 01:06:02,505 DEBUG 3
2024-05-26 01:06:02,511 DEBUG wl_shm
2024-05-26 01:06:02,511 DEBUG 1
2024-05-26 01:06:02,511 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7a6b305faba0>
2024-05-26 01:06:02,511 DEBUG 4
2024-05-26 01:06:02,511 DEBUG wl_output
2024-05-26 01:06:02,511 DEBUG 4
2024-05-26 01:06:02,511 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7a6b305faba0>
2024-05-26 01:06:02,511 DEBUG 5
2024-05-26 01:06:02,511 DEBUG zxdg_output_manager_v1
2024-05-26 01:06:02,511 DEBUG 3
2024-05-26 01:06:02,516 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7a6b305faba0>
2024-05-26 01:06:02,516 DEBUG 6
2024-05-26 01:06:02,516 DEBUG wl_data_device_manager
2024-05-26 01:06:02,516 DEBUG 3
2024-05-26 01:06:02,516 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7a6b305faba0>
2024-05-26 01:06:02,517 DEBUG 7
2024-05-26 01:06:02,522 DEBUG zwp_primary_selection_device_manager_v1
2024-05-26 01:06:02,522 DEBUG 1
2024-05-26 01:06:02,522 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7a6b305faba0>
2024-05-26 01:06:02,522 DEBUG 8
2024-05-26 01:06:02,522 DEBUG wl_subcompositor
2024-05-26 01:06:02,522 DEBUG 1
2024-05-26 01:06:02,522 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7a6b305faba0>
2024-05-26 01:06:02,522 DEBUG 9
2024-05-26 01:06:02,522 DEBUG xdg_wm_base
2024-05-26 01:06:02,522 DEBUG 6
2024-05-26 01:06:02,522 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7a6b305faba0>
2024-05-26 01:06:02,522 DEBUG 10
2024-05-26 01:06:02,522 DEBUG gtk_shell1
2024-05-26 01:06:02,522 DEBUG 5
2024-05-26 01:06:02,522 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7a6b305faba0>
2024-05-26 01:06:02,522 DEBUG 11
2024-05-26 01:06:02,522 DEBUG wp_viewporter
2024-05-26 01:06:02,522 DEBUG 1
2024-05-26 01:06:02,522 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7a6b305faba0>
2024-05-26 01:06:02,522 DEBUG 12
2024-05-26 01:06:02,522 DEBUG wp_fractional_scale_manager_v1
2024-05-26 01:06:02,522 DEBUG 1
2024-05-26 01:06:02,522 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7a6b305faba0>
2024-05-26 01:06:02,522 DEBUG 13
2024-05-26 01:06:02,522 DEBUG zwp_pointer_gestures_v1
2024-05-26 01:06:02,522 DEBUG 3
2024-05-26 01:06:02,522 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7a6b305faba0>
2024-05-26 01:06:02,522 DEBUG 14
2024-05-26 01:06:02,522 DEBUG zwp_tablet_manager_v2
2024-05-26 01:06:02,522 DEBUG 1
2024-05-26 01:06:02,522 DEBUG <pywayland.protocol.wayland.wl_registry.WlRegistryProxy object at 0x7a6b305faba0>
2024-05-26 01:06:02,522 DEBUG 15
2024-05-26 01:06:02,523 DEBUG wl_seat
2024-05-26 01:06:02,523 DEBUG 8
[... 76 lines cut ...]
2024-05-26 01:06:10,496 DEBUG Writting to config file: "<_io.TextIOWrapper name='./numberpad_dev' mode='w' encoding='UTF-8'>"
2024-05-26 01:06:10,496 INFO check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/home/irvng pathname=/home/irvng/numberpad_dev wd=1 >
2024-05-26 01:06:10,596 INFO Numpad activated
2024-05-26 01:06:10,639 INFO System numlock deactivated
2024-05-26 01:06:10,640 INFO Setting up for config file key: "idled" with value: "False"
2024-05-26 01:06:10,640 DEBUG Writting to config file: "<_io.TextIOWrapper name='./numberpad_dev' mode='w' encoding='UTF-8'>"
2024-05-26 01:06:10,641 INFO check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/home/irvng pathname=/home/irvng/numberpad_dev wd=1 >
2024-05-26 01:06:10,741 INFO Cancelled numpad idle
2024-05-26 01:06:10,742 INFO Cancelled idle
[... cut ...]
2024-05-26 01:06:10,757 INFO check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/home/irvng pathname=/home/irvng/numberpad_dev wd=1 >
2024-05-26 01:06:10,857 INFO Numpad deactivated
[2024-05-26 01:06:10,885 pyinotify ERROR] Cannot delete unknown watch descriptor '/home/irvng'
2024-05-26 01:06:10,885 ERROR Cannot delete unknown watch descriptor '/home/irvng'
2024-05-26 01:06:10,886 INFO Clean up finished
2024-05-26 01:06:10,886 INFO Exiting
```

ldrahnik, 2024-05-26:
@irvng-wav Please, try to add this line `Environment="LOG=DEBUG"` at the end of file `asus_numberpad_driver.wayland.service` and install and post the clean error log here again.

irvng-wav, 2024-05-27:
> @irvng-wav Please, download the latest master, add this line `Environment="LOG=DEBUG"` at the end of file `asus_numberpad_driver.wayland.service` and install & reboot, and post the clean error log here again.

At the end of [Service] or [Install]?
[image]

ldrahnik, 2024-05-27:
Service

irvng-wav, 2024-05-28:
Ok, this time it worked great right after rebooting! Here's the current error.log:
[error.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/15463384/error.log)

irvng-wav, 2024-05-28:
Cleared the log, changed the activation values to 0.3 seconds for on/off and 0.4 seconds for changing brightness, rebooted, and everything is also working perfectly right after boot! I'm not having any issues. Brightness state is saved and recalled correctly, even on reboot. Calculator toggle is also working perfectly!

I think you might have done it, man! Here's the continuation of error.log
[error.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/15463445/error.log)

Will keep testing for a day or two and report back if I find something.

Thank you man!

ldrahnik, 2024-05-31:
@irvng-wav Can I understand that you found nothing wrong?

irvng-wav, 2024-06-01:
Hey @ldrahnik! The only thing I found is that sometimes the brightness level resets to the maximum. This happened like two times when I left the computer turned off for maybe a day. Also recently I've been using mostly Windows in this laptop due to my job, so sometimes when it's been a while (a day or more) since I used Linux and I log back in to it, the numpad brightness is reset. I don't know if that's normal.

Other than that it's been a stable experience. If you want you can go ahead and close this issue and I can open a new issue for that small thing if that's something you feel it's worth looking at.

Thank you!

ldrahnik, 2024-06-01:
@irvng-wav Yes, create another issue please, thank you.

closed 2024-06-01 by ldrahnik

ldrahnik, 2024-06-12:
@irvng-wav With the latest commit should be fixed setting up full brightness when is the laptop started.