asus-linux-drivers/asus-numberpad-driver #183: [Layout Request] Layout Name

furkanzdm, 2024-05-24:
### Layout Name

G512LU

### Your device (cat /sys/devices/virtual/dmi/id/product_name)

ROG Strix G512LU_G512LU

### Supported Devices

G513

### Numpad Layout (photo/picture or code)

[image]

### Additional Info

Uses the same layout as G513 of G15 2022

### Acknowledgements

- [X] This layout doesn't currently exist and I have read the [Layout section](https://github.com/asus-linux-drivers/asus-numberpad-driver?tab=readme-ov-file#layouts)

kamack38, 2024-05-24:
Doesn't the G533 layout work?

furkanzdm, 2024-05-24:
> Doesn't the G533 layout work?

As of the moment, i get no device found error. So i assume it doesn't work.

ldrahnik, 2024-05-24:
@furkanzdm Hi, could you please attach the entire output of the install script?

furkanzdm, 2024-05-24:
> @furkanzdm Hi, could you please attach here the entire output of the install script?

I was using Ubuntu 6.8 lowlatency kernel, and had some driver issues with nvidia too. So switched kernel to 6.8 generic, and all is well. G513 detected as suitable layout and successfully installed. Its working completely fine at the moment.

ldrahnik, 2024-05-24:
Can be reopened.

closed 2024-05-24 by ldrahnik