asus-linux-drivers/asus-numberpad-driver #184: [Bug]: Calculator key does not show the calculator

DavidLazarescu, 2024-06-27:
### Describe the bug

I am on kde plasma and followed the install instructions. Showing the calculator via:
`sed -i "s/enabled = 0/enabled = 1/g" /usr/share/asus-numberpad-driver/numberpad_dev` works well, but tapping the button on the top-right does not.

Is there any manual key binding that I'm supposed to setup?

### Expected behavior

The calculator to show upon pressing the button.

### Desktop

- Program version (latest, built from git)
- Distro: Manjaro
- Kernel: 6.9.3-3
- DE: Plasma

ldrahnik, 2024-06-27:
@DavidLazarescu Did you hold the top right icon for at least 1s (by default, configurable)? Without the driver log file `/var/log/asus-numberpad-driver/` with logged attempts of activation I can do nothing.

DavidLazarescu, 2024-06-27:
Yes, I did hold it for more than one second. Is this the log you are requesting?-

ldrahnik, 2024-06-27:
@DavidLazarescu ?

DavidLazarescu, 2024-06-27:
Sorry, there was a problem with the upload, I mean this file:
[error.log](https://github.com/user-attachments/files/16015445/error.log)

ldrahnik, 2024-06-27:
@DavidLazarescu And config file `numberpad_dev`?

DavidLazarescu, 2024-06-27:
I have tried reinstalling it, but it still does not work. Here is my install file: 
[install-27-06-2024-17-50-26.log](https://github.com/user-attachments/files/16017786/install-27-06-2024-17-50-26.log)

and here is numberpad_dev:
```
[main]
numpad_disables_sys_numlock = 1
disable_due_inactivity_time = 0
touchpad_disables_numpad = 1
key_repetitions = 0
multitouch = 0
one_touch_key_rotation = 0
activation_time = 1
sys_numlock_enables_numpad = 1
top_left_icon_activation_time = 1
top_left_icon_slide_func_activation_radius = 550
top_left_icon_slide_func_activates_numpad = 1
top_right_icon_slide_func_activation_radius = 550
enabled_touchpad_pointer = 3
press_key_when_is_done_untouch = 1
enabled = 0
default_backlight_level = 0x01
brightness = 0x48
top_left_icon_brightness_func_disabled = 0
distance_to_move_only_pointer = 0
idled = 0
idle_brightness = 30
idle_enabled = 0
idle_time = 10
```

DavidLazarescu, 2024-06-27:
btw, sometimes the numpad is randomly showing up for some reason.3

ldrahnik, 2024-06-29:
@DavidLazarescu The config used by driver when is installed can be found on path `/usr/share/asus-numberpad-driver/numberpad_dev`. Or the driver was started manually so was used `~/`?

DavidLazarescu, 2024-06-30:
> @DavidLazarescu The config used by driver when is installed can be found on path `/usr/share/asus-numberpad-driver/numberpad_dev`. Or the driver was started manually so was used `~/`?

I installed it following the instructions, here is the file at /usr/share/asus-numberpad-driver/numberpad_dev:
```
[main]
numpad_disables_sys_numlock = 1
disable_due_inactivity_time = 0
touchpad_disables_numpad = 1
key_repetitions = 0
multitouch = 0
one_touch_key_rotation = 0
activation_time = 1
sys_numlock_enables_numpad = 1
top_left_icon_activation_time = 1
top_left_icon_slide_func_activation_radius = 550
top_left_icon_slide_func_activates_numpad = 1
top_right_icon_slide_func_activation_radius = 550
enabled_touchpad_pointer = 3
press_key_when_is_done_untouch = 1
enabled = 0
default_backlight_level = 0x01
brightness = 0x48
top_left_icon_brightness_func_disabled = 0
distance_to_move_only_pointer = 0
idled = 0
idle_brightness = 30
idle_enabled = 1
idle_time = 10
```

ldrahnik, 2024-07-02:
@DavidLazarescu Please, increase manually these 2 config values to avoid random activation by the hands palm:

```
top_left_icon_slide_func_activation_radius = 550
top_right_icon_slide_func_activation_radius = 550
```

Drivers default changed to `1200px`. Thank you for reporting.

DavidLazarescu, 2024-07-02:
Thanks. I have done that, but it still does not fix the activation issues.

ldrahnik, 2024-07-16:
@DavidLazarescu The logged behavior is weird. Unfortunately, I have had no luck with replication, yet.

ldrahnik, 2024-07-28:
@DavidLazarescu Please, try the latest master.

> @DavidLazarescu btw, sometimes the numpad is randomly showing up for some reason.3

It should be fixed.

ldrahnik, 2024-09-16:
Can be reopened.

closed 2024-09-16 by ldrahnik