asus-linux-drivers/asus-numberpad-driver #185: [Bug]: Driver doesn't start without GDM.

RomanSyunyaev, 2024-07-07:
### Describe the bug

Driver was working properly with GDM.
After removing GDM\reinstalling asus-numberpad-driver, systemd failed to start the driver. 
Contents of the error.log are attached below.

The problem seems to be related to the fact that default $XDG_SESSION_TYPE on my system is tty.
The following workaround fixes the problem:
```
$ cat ~/.xinitrc
export XDG_SESSION_TYPE=x11
exec gnome-session
systemctl restart --user asus_numberpad_driver@$USER.service
```

### Expected behavior

Driver is expected to start either after system boot or after X11 start.
This is probably not exactly the driver bug, but adding the workaround to the README might be a good idea.

### Relevant log output

```shell
2024-07-07 13:44:10,597 ERROR X11 load keymap listener error. Exiting
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 462, in load_keymap_listener_x11
    event = display.next_event()
            ^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages/Xlib/display.py", line 187, in next_event
    return self.display.next_event()
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages/Xlib/protocol/display.py", line 217, in next_event
    self.send_and_recv(event = True)
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages/Xlib/protocol/display.py", line 609, in send_and_recv
    raise self.socket_error
Xlib.error.ConnectionClosedError: Display connection closed by server
2024-07-07 13:44:10,599 ERROR Listening touchpad events unexpectedly failed
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 2435, in <module>
    listen_touchpad_events()
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 1952, in listen_touchpad_events
    for e in d_t.events():
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages/libevdev/device.py", line 542, in events
    ev = self._libevdev.next_event(flags)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages/libevdev/_clib.py", line 889, in next_event
    rc = self._next_event(self._ctx, flags, ctypes.byref(ev))
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 2355, in signal_handler
    raise Exception()
Exception
2024-07-07 13:44:10,602 INFO Clean up started
2024-07-07 13:44:10,665 INFO Clean up finished
2024-07-07 13:44:10,665 INFO Exiting
2024-07-07 13:44:11,916 ERROR X11 detected but not connected succesfully to the display :0. Exiting
2024-07-07 13:44:12,357 ERROR X11 detected but not connected succesfully to the display :0. Exiting
2024-07-07 13:45:18,044 ERROR xdg session type can not be empty. Exiting
2024-07-07 13:45:18,430 ERROR xdg session type can not be empty. Exiting
2024-07-07 13:45:18,768 ERROR xdg session type can not be empty. Exiting
2024-07-07 13:45:19,123 ERROR xdg session type can not be empty. Exiting
2024-07-07 13:45:19,472 ERROR xdg session type can not be empty. Exiting
```

### Desktop

- Program version [ff04344]
- Distro [Debian bookworm]
- Kernel [6.1.0-22-amd64]

RomanSyunyaev, 2024-07-07:
Follow up.
Slight fix to the .xinitrc workaround:
```
export XDG_SESSION_TYPE=x11
bash -c "sleep 5 && systemctl restart --user asus_numberpad_driver@$USER.service" &
exec gnome-session
```

ldrahnik, 2024-07-07:
@RomanSyunyaev What returns `$ loginctl` when is GDM not used? Something like (not tested):

```
$ loginctl
 show-session $(loginctl | grep $(whoami) | awk '{print $1}')
```

RomanSyunyaev, 2024-07-07:
Thank you for quick response!

```
Id=1
User=1000
Name=ras
Timestamp=Mon 2024-07-08 02:11:58 +05
TimestampMonotonic=166110404
VTNr=1
Seat=seat0
TTY=tty1
Remote=no
Service=login
Scope=session-1.scope
Leader=1109
Audit=1
Type=tty
Class=user
Active=yes
State=active
IdleHint=no
IdleSinceHint=1720386714657658
IdleSinceHintMonotonic=162588942
LockedHint=no

Id=2
User=1000
Name=ras
Timestamp=Mon 2024-07-08 02:11:58 +05
TimestampMonotonic=166208060
VTNr=0
Remote=no
Service=systemd-user
Leader=1182
Audit=2
Type=unspecified
Class=manager
Active=yes
State=active
IdleHint=no
IdleSinceHint=0
IdleSinceHintMonotonic=0
LockedHint=no
```

ldrahnik, 2024-07-08:
@RomanSyunyaev Do you have `/etc/X11/xinit/xinitrc.d/50-systemd-user.sh` - the script should set up env variables `DISPLAY` and `XAUTHORITY` with no GDM.

And with no env var `WAYLAND_DISPLAY` set up I think I could assume that is used `x11`.

Could you post here the output of `$ env` with no GDM, for sure.

RomanSyunyaev, 2024-07-08:
I don't have `/etc/X11/xinit/xinitrc.d/50-systemd-user.sh`
```
$ ls /etc/X11/xinit/
xinitrc  xinputrc  xserverrc
```

I think that `DISPLAY` and `XAUTHORITY` are set by D-BUS:
```
Xsession: X session started for ras at Mon Jul  8 08:41:23 PM +05 2024
dbus-update-activation-environment: setting DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1000/bus
dbus-update-activation-environment: setting DISPLAY=:0
dbus-update-activation-environment: setting XAUTHORITY=/home/ras/.Xauthority
dbus-update-activation-environment: setting XDG_CURRENT_DESKTOP=GNOME
localuser:ras being added to access control list
dbus-update-activation-environment: setting GTK_MODULES=gail:atk-bridge
dbus-update-activation-environment: setting QT_ACCESSIBILITY=1
```

Yes, I think that if `WAYLAND_DISPLAY` is not set, that means that `x11` is being used.
The problem is (if I understood you correctly) that systemd tries to start the driver after the system boot. At that moment neither `WAYLAND_DIAPLAY` nor `DISPLAY` is populated and there is no way to tell which one is going to be used by user before he runs `startx` command.

This is my `$ env` after the system boot
```
SHELL=/bin/bash
LANGUAGE=en_US:en
CREDENTIALS_DIRECTORY=/run/credentials/getty@tty1.service
MEMORY_PRESSURE_WRITE=c29tZSAyMDAwMDAgMjAwMDAwMAA=
XDG_SEAT=seat0
PWD=/home/ras
LOGNAME=ras
XDG_SESSION_TYPE=tty
SYSTEMD_EXEC_PID=1086
MOTD_SHOWN=pam
HOME=/home/ras
LANG=en_US.UTF-8
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=00:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.avif=01;35:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.webp=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:*~=00;90:*#=00;90:*.bak=00;90:*.old=00;90:*.orig=00;90:*.part=00;90:*.rej=00;90:*.swp=00;90:*.tmp=00;90:*.dpkg-dist=00;90:*.dpkg-old=00;90:*.ucf-dist=00;90:*.ucf-new=00;90:*.ucf-old=00;90:*.rpmnew=00;90:*.rpmorig=00;90:*.rpmsave=00;90:
MEMORY_PRESSURE_WATCH=/sys/fs/cgroup/system.slice/system-getty.slice/getty@tty1.service/memory.pressure
INVOCATION_ID=58ab1ae3cc5241eaba11d3376b7116d7
XDG_SESSION_CLASS=user
TERM=linux
USER=ras
SHLVL=1
XDG_VTNR=1
XDG_SESSION_ID=1
XDG_RUNTIME_DIR=/run/user/1000
XDG_DATA_DIRS=/usr/share/gnome:/usr/local/share/:/usr/share/
HUSHLOGIN=FALSE
PATH=/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games
DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1000/bus
MAIL=/var/mail/ras
_=/usr/bin/env
```

This is my `env` after `startx`:
```
$ env
SHELL=/bin/bash
SESSION_MANAGER=local/RAS:@/tmp/.ICE-unix/1479,unix/RAS:/tmp/.ICE-unix/1479
COLORTERM=truecolor
XDG_MENU_PREFIX=gnome-
GNOME_DESKTOP_SESSION_ID=this-is-deprecated
GNOME_KEYRING_CONTROL=/run/user/1000/keyring
LANGUAGE=en_US:en
SSH_AUTH_SOCK=/run/user/1000/keyring/ssh
CREDENTIALS_DIRECTORY=/run/credentials/getty@tty1.service
MEMORY_PRESSURE_WRITE=c29tZSAyMDAwMDAgMjAwMDAwMAA=
XMODIFIERS=@im=ibus
PWD=/etc/init.d
LOGNAME=ras
XDG_SESSION_TYPE=tty
[... 1 lines cut ...]
XAUTHORITY=/home/ras/.Xauthority
WINDOWPATH=1
MOTD_SHOWN=pam
HOME=/home/ras
LANG=en_US.UTF-8
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=00:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.avif=01;35:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.webp=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01 [... cut ...]
MEMORY_PRESSURE_WATCH=/sys/fs/cgroup/system.slice/system-getty.slice/getty@tty1.service/memory.pressure
VTE_VERSION=7006
GNOME_TERMINAL_SCREEN=/org/gnome/Terminal/screen/12cac451_923e_4761_ab40_e514b2f4e97b
XDG_SESSION_CLASS=user
TERM=xterm-256color
USER=ras
GNOME_TERMINAL_SERVICE=:1.125
DISPLAY=:0
SHLVL=1
QT_IM_MODULE=ibus
XDG_RUNTIME_DIR=/run/user/1000
XDG_DATA_DIRS=/usr/share/gnome:/usr/local/share/:/usr/share/
HUSHLOGIN=FALSE
PATH=/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games
MAIL=/var/mail/ras
_=/usr/bin/env
OLDPWD=/home/ras
```

ldrahnik, 2024-07-09:
I have created [branch](https://github.com/asus-linux-drivers/asus-numberpad-driver/tree/not-used-gdm), let me know, please.

RomanSyunyaev, 2024-07-09:
I think there is a typo somewhere.

```
Do you want install systemctl service? [y/N]y
install_service.sh: line 93: syntax error near unexpected token `esac'
install_service.sh: line 93: `esac'
```

Did not have a chance yet to look for the typo myself though, sorry!

RomanSyunyaev, 2024-07-09:
Hmm, for some reason the problem persists.
```
2024-07-09 17:45:08,812 ERROR xdg session type can not be empty. Exiting
2024-07-09 17:45:09,363 ERROR xdg session type can not be empty. Exiting
2024-07-09 17:45:09,643 ERROR xdg session type can not be empty. Exiting
2024-07-09 17:45:10,153 ERROR xdg session type can not be empty. Exiting
2024-07-09 17:45:10,707 ERROR xdg session type can not be empty. Exiting
```
And now even  `systemctl restart --user asus_numberpad_driver@$USER.service` does not help: the driver starts, but it fails to detect the top_right_icon area (numlock).

ldrahnik, 2024-07-12:
@RomanSyunyaev Try the latest version of branch:

```
$ sudo rm -rf /usr/lib/systemd/user/asus_numberpad_driver@.service
$ bash install_service.sh # you should see setting up: (SET) env var XDG_SESSION_TYPE: x11
$ cat /usr/lib/systemd/user/asus_numberpad_driver@.service
...
Environment="XDG_SESSION_TYPE=x11" # check this row
...
```

RomanSyunyaev, 2024-07-16:
The question sounds like a good option to me, yes.

ldrahnik, 2024-07-16:
Does miss to you the `x11` value in env var `XDG_SESSION_TYPE` somewhere else?

RomanSyunyaev, 2024-07-16:
@ldrahnik  I am not sure if I understood the question, but one of the replies in the link you've sent me above was

"_It's commonly set by desktop managers like gdm or sddm. I don't think it's actually recommended by any xdg spec so the XDG prefix **is probably some wishful thinking.**_"

So no, I usually do not need x11 value to be set up. However, I can do it in .xinitrc or something if required. Before posting the bug report this was exactly what I did -- I set up my xinitrc to set x11 value and restart the driver after that.

ldrahnik, 2024-07-16:
@RomanSyunyaev I committed that question.

ldrahnik, 2024-07-17:
Can be reopened.

RomanSyunyaev, 2024-07-20:
Sorry for being slow, but that did not really work.
For some reason I still have to `systemctl restart --user asus_numberpad_driver@$USER.service` after `startx` in order to launch the driver.

ldrahnik, 2024-07-21:
@RomanSyunyaev What does say error log of driver? (`/var/log/asus-numberpad-driver/`) Why was not started? Drivers systemctl service should be in loop of exitings with value 1 and after `startx` automatically starts.

RomanSyunyaev, 2024-07-21:
@ldrahnik The problem is that there is nothing in the logs after the `startx`.

This is the content of `error.log` after the boot:
```
2024-07-21 13:34:45,236 ERROR X11 load keymap listener error. Exiting
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 462, in load_keymap_listener_x11
    event = display.next_event()
            ^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.11/site-packages/Xlib/display.py", line 187, in next_event
    return self.display.next_event()
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.11/site-packages/Xlib/protocol/display.py", line 217, in next_event
    self.send_and_recv(event = True)
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.11/site-packages/Xlib/protocol/display.py", line 609, in send_and_recv
    raise self.socket_error
Xlib.error.ConnectionClosedError: Display connection closed by server
2024-07-21 13:34:45,242 ERROR Listening touchpad events unexpectedly failed
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 2436, in <module>
    listen_touchpad_events()
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 1953, in listen_touchpad_events
    for e in d_t.events():
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.11/site-packages/libevdev/device.py", line 542, in events
    ev = self._libevdev.next_event(flags)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.11/site-packages/libevdev/_clib.py", line 889, in next_event
    rc = self._next_event(self._ctx, flags, ctypes.byref(ev))
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 2356, in signal_handler
    raise Exception()
Exception
2024-07-21 13:34:45,250 INFO Clean up started
2024-07-21 13:34:45,365 INFO Clean up finished
2024-07-21 13:34:45,365 INFO Exiting
2024-07-21 13:34:45,980 ERROR X11 detected but not connected succesfully to the display :0. Exiting
2024-07-21 13:35:19,258 ERROR X11 detected but not connected succesfully to the display :0. Exiting
2024-07-21 13:35:19,669 ERROR X11 detected but not connected succesfully to the display :0. Exiting
2024-07-21 13:35:20,021 ERROR X11 detected but not connected succesfully to the display :0. Exiting
2024-07-21 13:35:20,567 ERROR X11 detected but not connected succesfully to the display :0. Exiting
2024-07-21 13:35:21,095 ERROR X11 detected but not connected succesfully to the display :0. Exiting
```
And nothing more after `startx`.

ldrahnik, 2024-07-21:
@RomanSyunyaev Status of `systemctl` service says what.

ldrahnik, 2024-07-21:
@RomanSyunyaev Enable debug loggings might help, add this env var to the service: `Environment="LOG=DEBUG"`

RomanSyunyaev, 2024-07-21:
> @RomanSyunyaev What does say status of `systemctl` service.

```
$ systemctl status --user asus_numberpad_driver@$USER.service

× asus_numberpad_driver@ras.service - Asus NumberPad Driver
     Loaded: loaded (/usr/lib/systemd/user/asus_numberpad_driver@.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Sun 2024-07-21 13:35:21 BST; 2h 23min ago
   Duration: 165ms
    Process: 1681 ExecStart=/usr/share/asus-numberpad-driver/.env/bin/python3 /usr/share/asus-numberpad-driver/numberpad.py up5401ea /usr/share/asus-numberpad-driver/ (code=exited, status=1/FAILURE)
   Main PID: 1681 (code=exited, status=1/FAILURE)
        CPU: 1.389s

Jul 21 13:35:21 ras systemd[1468]: asus_numberpad_driver@ras.service: Failed with result 'exit-code'.
Jul 21 13:35:21 ras systemd[1468]: asus_numberpad_driver@ras.service: Consumed 1.389s CPU time.
Jul 21 13:35:21 ras systemd[1468]: asus_numberpad_driver@ras.service: Scheduled restart job, restart counter is at 5.
Jul 21 13:35:21 ras systemd[1468]: Stopped asus_numberpad_driver@ras.service - Asus NumberPad Driver.
Jul 21 13:35:21 ras systemd[1468]: asus_numberpad_driver@ras.service: Consumed 1.389s CPU time.
Jul 21 13:35:21 ras systemd[1468]: asus_numberpad_driver@ras.service: Start request repeated too quickly.
Jul 21 13:35:21 ras systemd[1468]: asus_numberpad_driver@ras.service: Failed with result 'exit-code'.
Jul 21 13:35:21 ras systemd[1468]: Failed to start asus_numberpad_driver@ras.service - Asus NumberPad Driver.
```

RomanSyunyaev, 2024-07-21:
> @RomanSyunyaev Enabling debug logs might help. Add this line / env var to the service: `Environment="LOG=DEBUG"` Reinstall.

Sorry, could you explain what do you mean?
Should I add the line manually to the `/usr/lib/systemd/user/asus_numberpad_driver@.service`?
Or should I add it to the 'install_service.sh` ?

PS: While experimenting with this, I have noticed something interesting. On some occasions the driver starts, on some it doesn't. I am not quite familiar with systemctl services inner workings -- is it possible to add some sort of delay to make it sure that the driver starts after x11 is loaded properly?

ldrahnik, 2024-07-21:
@RomanSyunyaev The line add to the file in the repository: `asus_numberpad_driver.x11.service`, somewhere under `[Service]` and reinstall driver.

ldrahnik, 2024-07-21:
@RomanSyunyaev Delay to thé service can be created with this line `ExecStartPre=/bin/sleep 30` under `[Service]`.

RomanSyunyaev, 2024-07-21:
> @RomanSyunyaev The line add to the file in the repository: `asus_numberpad_driver.x11.service`, somewhere under `[Service]` and reinstall driver.

Thanks!
This is the reulting `error.log`
```
2024-07-21 16:41:40,123 INFO X11 detected and connected succesfully to the display :0
2024-07-21 16:41:40,124 INFO Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
2024-07-21 16:41:40,124 INFO Set keyboard 0 from H: Handlers=sysrq kbd leds event0
2024-07-21 16:41:40,155 INFO Detecting touchpad from string: "N: Name="ASUP1415:00 093A:300C Touchpad""
2024-07-21 16:41:40,155 INFO Set touchpad device id 2 from S: Sysfs=/devices/platform/AMDI0010:03/i2c-2/i2c-ASUP1415:00/0018:093A:300C.0001/input/input13
2024-07-21 16:41:40,155 INFO Set touchpad id 10 from H: Handlers=mouse1 event10
2024-07-21 16:41:40,156 INFO Touchpad min-max: x 0-3996, y 0-2242
2024-07-21 16:41:40,156 INFO Numpad min-max: x 200-3796, y 200-2162
2024-07-21 16:41:40,156 DEBUG X11 will try to load keymap
2024-07-21 16:41:40,157 DEBUG X11 loaded keymap succesfully
2024-07-21 16:41:40,157 DEBUG {'Num_Lock': KEY_NUMLOCK:69, '0': KEY_0:11, '1': KEY_1:2, '2': KEY_2:3, '3': KEY_3:4, '4': KEY_4:5, '5': KEY_5:6, '6': KEY_6:7, '7': KEY_7:8, '8': KEY_8:9, '9': KEY_9:10, 'a': [KEY_RIGHTALT:100, KEY_A:30], 'b': [KEY_RIGHTALT:100, KEY_B:48], 'c': [KEY_RIGHTALT:100, KEY_C:46], 'd': [KEY_RIGHTALT:100, KEY_D:32], 'e': [KEY_RIGHTALT:100, KEY_E:18], 'f': [KEY_RIGHTALT:100, KEY_F:33], 'Shift_L': KEY_LEFTSHIFT:42, 'Control_L': KEY_LEFTCTRL:29, 'u': [KEY_RIGHTALT:100, KEY_U:22], 'space': KEY_SPACE:57, 'slash': KEY_102ND:86, 'BackSpace': KEY_BACKSPACE:14, 'asterisk': [KEY_LEFTSHIFT:42, KEY_8:9], 'minus': KEY_MINUS:12, 'percent': [KEY_LEFTSHIFT:42, KEY_5:6], 'period': KEY_SLASH:53, 'Return': KEY_ENTER:28, 'plus': [KEY_LEFTSHIFT:42, KEY_EQUAL:13], 'equal': KEY_EQUAL:13, 'Alt_R': KEY_RIGHTALT:100}
2024-07-21 16:41:40,659 INFO Setting up for config file key: "brightness" with value: "0x48"
2024-07-21 16:41:40,659 DEBUG Writting to config file: "<_io.TextIOWrapper name='/usr/share/asus-numberpad-driver/numberpad_dev' mode='w' encoding='UTF-8'>"
2024-07-21 16:41:41,845 DEBUG X11 will try to load keymap
2024-07-21 16:41:41,845 DEBUG X11 loaded keymap succesfully
2024-07-21 16:41:41,845 DEBUG {'Num_Lock': KEY_NUMLOCK:69, '0': KEY_0:11, '1': KEY_1:2, '2': KEY_2:3, '3': KEY_3:4, '4': KEY_4:5, '5': KEY_5:6, '6': KEY_6:7, '7': KEY_7:8, '8': KEY_8:9, '9': KEY_9:10, 'a': [KEY_RIGHTALT:100, KEY_A:30], 'b': [KEY_RIGHTALT:100, KEY_B:48], 'c': [KEY_RIGHTALT:100, KEY_C:46], 'd': [KEY_RIGHTALT:100, KEY_D:32], 'e': [KEY_RIGHTALT:100, KEY_E:18], 'f': [KEY_RIGHTALT:100, KEY_F:33], 'Shift_L': KEY_LEFTSHIFT:42, 'Control_L': KEY_LEFTCTRL:29, 'u': [KEY_RIGHTALT:100, KEY_U:22], 'space': KEY_SPACE:57, 'slash': KEY_102ND:86, 'BackSpace': KEY_BACKSPACE:14, 'asterisk': [KEY_LEFTSHIFT:42, KEY_8:9], 'minus': KEY_MINUS:12, 'percent': [KEY_LEFTSHIFT:42, KEY_5:6], 'period': KEY_SLASH:53, 'Return': KEY_ENTER:28, 'plus': [KEY_LEFTSHIFT:42, KEY_EQUAL:13], 'equal': KEY_EQUAL:13, 'Alt_R': KEY_RIGHTALT:100}
2024-07-21 16:41:41,845 DEBUG X11 will try to load keymap
2024-07-21 16:41:41,846 DEBUG X11 loaded keymap succesfully
2024-07-21 16:41:41,846 DEBUG {'Num_Lock': KEY_NUMLOCK:69, '0': KEY_0:11, '1': KEY_1:2, '2': KEY_2:3, '3': KEY_3:4, '4': KEY_4:5, '5': KEY_5:6, '6': KEY_6:7, '7': KEY_7:8, '8': KEY_8:9, '9': KEY_9:10, 'a': [KEY_RIGHTALT:100, KEY_A:30], 'b': [KEY_RIGHTALT:100, KEY_B:48], 'c': [KEY_RIGHTALT:100, KEY_C:46], 'd': [KEY_RIGHTALT:100, KEY_D:32], 'e': [KEY_RIGHTALT:100, KEY_E:18], 'f': [KEY_RIGHTALT:100, KEY_F:33], 'Shift_L': KEY_LEFTSHIFT:42, 'Control_L': KEY_LEFTCTRL:29, 'u': [KEY_RIGHTALT:100, KEY_U:22], 'space': KEY_SPACE:57, 'slash': KEY_102ND:86, 'BackSpace': KEY_BACKSPACE:14, 'asterisk': [KEY_LEFTSHIFT:42, KEY_8:9], 'minus': KEY_MINUS:12, 'percent': [KEY_LEFTSHIFT:42, KEY_5:6], 'period': KEY_SLASH:53, 'Return': KEY_ENTER:28, 'plus': [KEY_LEFTSHIFT:42, KEY_EQUAL:13], 'equal': KEY_EQUAL:13, 'Alt_R': KEY_RIGHTALT:100}
2024-07-21 16:41:41,871 DEBUG X11 will try to load keymap
2024-07-21 16:41:41,872 DEBUG X11 loaded keymap succesfully
2024-07-21 16:41:41,872 DEBUG {'Num_Lock': KEY_NUMLOCK:69, '0': KEY_0:11, '1': KEY_1:2, '2': KEY_2:3, '3': KEY_3:4, '4': KEY_4:5, '5': KEY_5:6, '6': KEY_6:7, '7': KEY_7:8, '8': KEY_8:9, '9': KEY_9:10, 'a': [KEY_RIGHTALT:100, KEY_A:30], 'b': [KEY_RIGHTALT:100, KEY_B:48], 'c': [KEY_RIGHTALT:100, KEY_C:46], 'd': [KEY_RIGHTALT:100, KEY_D:32], 'e': [KEY_RIGHTALT:100, KEY_E:18], 'f': [KEY_RIGHTALT:100, KEY_F:33], 'Shift_L': KEY_LEFTSHIFT:42, 'Control_L': KEY_LEFTCTRL:29, 'u': [KEY_RIGHTALT:100, KEY_U:22], 'space': KEY_SPACE:57, 'slash': KEY_102ND:86, 'BackSpace': KEY_BACKSPACE:14, 'asterisk': [KEY_LEFTSHIFT:42, KEY_8:9], 'minus': KEY_MINUS:12, 'percent': [KEY_LEFTSHIFT:42, KEY_5:6], 'period': KEY_SLASH:53, 'Return': KEY_ENTER:28, 'plus': [KEY_LEFTSHIFT:42, KEY_EQUAL:13], 'equal': KEY_EQUAL:13, 'Alt_R': KEY_RIGHTALT:100}
2024-07-21 16:41:41,912 DEBUG X11 will try to load keymap
2024-07-21 16:41:41,912 DEBUG X11 loaded keymap succesfully
2024-07-21 16:41:41,912 DEBUG {'Num_Lock': KEY_NUMLOCK:69, '0': KEY_0:11, '1': KEY_1:2, '2': KEY_2:3, '3': KEY_3:4, '4': KEY_4:5, '5': KEY_5:6, '6': KEY_6:7, '7': KEY_7:8, '8': KEY_8:9, '9': KEY_9:10, 'a': [KEY_RIGHTALT:100, KEY_A:30], 'b': [KEY_RIGHTALT:100, KEY_B:48], 'c': [KEY_RIGHTALT:100, KEY_C:46], 'd': [KEY_RIGHTALT:100, KEY_D:32], 'e': [KEY_RIGHTALT:100, KEY_E:18], 'f': [KEY_RIGHTALT:100, KEY_F:33], 'Shift_L [... cut ...]
2024-07-21 16:42:25,509 ERROR X11 detected but not connected succesfully to the display :0. Exiting
2024-07-21 16:42:25,944 ERROR X11 detected but not connected succesfully to the display :0. Exiting
2024-07-21 16:42:26,205 ERROR X11 detected but not connected succesfully to the display :0. Exiting
2024-07-21 16:42:26,701 ERROR X11 detected but not connected succesfully to the display :0. Exiting
2024-07-21 16:42:27,213 ERROR X11 detected but not connected succesfully to the display :0. Exiting
```

I'll try the `sleep` now.

RomanSyunyaev, 2024-07-21:
> @RomanSyunyaev Delay to thé service can be created with this line `ExecStartPre=/bin/sleep 30` under `[Service]`.

That seems to work, thank you! I have tested it 5 times and it loaded correctly every time.

One little correction. It has to be:
```
ExecStartPre=/bin/sleep 4
```
Otherwise there seems to be a conflict with
```
TimeoutSec=5
```

ldrahnik, 2024-07-27:
@RomanSyunyaev Does work for you to adding `RestartSec=1` as well? Instead of `ExecStartPre=/bin/sleep 4`. The reason is that I would like to avoid having a delay before the start when starting the driver using the command `$ systemctl restart driver_service` or `$ systemctl start driver_service`

ldrahnik, 2024-07-27:
Can be reopened.

closed 2024-07-27 by ldrahnik

RomanSyunyaev, 2024-07-27:
@ldrahnik That seems to work as well, thank you.