asus-linux-drivers/asus-numberpad-driver #186: [Bug]: Wont launch anything/Install error?

Asdf1qaz, 2024-07-15:
### Describe the bug

I have tried to install it on the two new laptops that I have. Both are Asus Vivobook S 14 Flip (TN3402Y).
Both running Linux Mint 21.3 and kernal 6.5.0-41.
One went fine and it worked right off the bat. The other now so much. 
It did chuck an error about "ERROR: Failed building wheel for xkbcommon" But I think it did it on the other working one too?

### Expected behavior

It installed as runs?

### Relevant log output

```shell
k@k-Vivobook:~$ cd asus-numberpad-driver/
k@k-Vivobook:~/asus-numberpad-driver$ ./install.sh 
[sudo] password for k:         
groupadd: group 'numberpad' already exists
Added group numberpad to current user

Reading package lists...
Building dependency tree...
Reading state information...
gcc is already the newest version (4:11.2.0-1ubuntu1).
ibus is already the newest version (1.5.26-4).
libevdev2 is already the newest version (1.12.1+dfsg-1).
libxkbcommon-dev is already the newest version (1.4.0-1).
xinput is already the newest version (1.6.3-1build2).
i2c-tools is already the newest version (4.3-2build1).
python3-virtualenv is already the newest version (20.13.0+ds-2).
curl is already the newest version (7.81.0-1ubuntu1.16).
libxml2-utils is already the newest version (2.9.13+dfsg-1ubuntu0.4).
python3-dev is already the newest version (3.10.6-1~22.04).
The following package was automatically installed and is no longer required:
  ninja-build
Use 'sudo apt autoremove' to remove it.
0 to upgrade, 0 to newly install, 0 to remove and 2 not to upgrade.

groupadd: group 'input' already exists
groupadd: group 'i2c' already exists
groupadd: group 'uinput' already exists
Added groups input, i2c, uinput, numberpad to current user
uinput module loaded
i2c-dev module loaded
uinput module added to auto loaded modules
Udev rules reloaded and triggered

Testing interface i2c-0: failed
Testing interface i2c-1: success (adr 0x15)
The detection was successful. Touchpad with NumberPad found: i2c-1

Default config will be autocreated during the first run and available for futher modifications here:
/usr/share/asus-numberpad-driver/numberpad_dev

created virtual environment CPython3.10.12.final.0-64 in 179ms
  creator CPython3Posix(dest=/usr/share/asus-numberpad-driver/.env, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=/home/k/.local/share/virtualenv)
    added seed packages: pip==22.0.2, setuptools==59.6.0, wheel==0.37.1
  activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
Requirement already satisfied: pip in /usr/share/asus-numberpad-driver/.env/lib/python3.10/site-packages (22.0.2)
Collecting pip
  Using cached pip-24.1.2-py3-none-any.whl (1.8 MB)
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 22.0.2
    Uninstalling pip-22.0.2:
      Successfully uninstalled pip-22.0.2
Successfully installed pip-24.1.2
Requirement already satisfied: setuptools in /usr/share/asus-numberpad-driver/.env/lib/python3.10/site-packages (59.6.0)
Collecting setuptools
[... 127 lines cut ...]

Calculator app

Do you want try to install toggling script for XF86Calculator key? [y/N]y

Detected gnome-calculator
Setting up for gnome-calculator
Toggling script for calculator app gnome-calculator has been installed.

Power supply saver

By default is idle functionality disabled and may be manually enabled in config file later. Or by installing this rule can be enabled by every detected change of power supply mode to battery mode.

Idle mode is configured to be enabled after 10s of inactivity and to decrease 30% of brightness.

Do you want install the rule for idle functionality? [y/N]n

Installation finished succesfully

Reboot is required. Do you want reboot now? [y/N]
```

### Desktop

Linux Mint 21.3 and kernal 6.5.0-41

ldrahnik, 2024-07-16:
@Asdf1qaz Python versions are the same? `CPython3.10.12.final.0-64`?

Asdf1qaz, 2024-07-17:
Sorry, still learning linux. Where are the install logs for this stored? and the python installed is:

k@k-Vivobook:~$ python3
Python 3.10.12 (main, Mar 22 2024, 16:50:05) [GCC 11.4.0] on linux

ldrahnik, 2024-07-17:
@Asdf1qaz `/var/log/asus-numberpad-driver`

Shrimp22, 2024-07-19:
@ldrahnik I have same issue This is my instalation log 
Reading package lists...
Building dependency tree...
Reading state information...
gcc is already the newest version (4:11.2.0-1ubuntu1).
ibus is already the newest version (1.5.26-4).
libevdev2 is already the newest version (1.12.1+dfsg-1).
libxkbcommon-dev is already the newest version (1.4.0-1).
xinput is already the newest version (1.6.3-1build2).
i2c-tools is already the newest version (4.3-2build1).
python3-virtualenv is already the newest version (20.13.0+ds-2).
curl is already the newest version (7.81.0-1ubuntu1.16).
libxml2-utils is already the newest version (2.9.13+dfsg-1ubuntu0.4).
python3-dev is already the newest version (3.10.6-1~22.04).
The following packages were automatically installed and are no longer required:
  apg gnome-control-center-faces gnome-online-accounts libcolord-gtk1
  libfreerdp-server2-2 libgnome-bg-4-1 libgsound0 libgssdp-1.2-0
  libgupnp-1.2-1 libgupnp-av-1.0-3 libgupnp-dlna-2.0-4 libluajit-5.1-2
  libluajit-5.1-common libmsgpackc2 libntfs-3g89 librygel-core-2.6-2
  librygel-db-2.6-2 librygel-renderer-2.6-2 librygel-server-2.6-2 libtermkey1
  libtree-sitter0 libunibilium4 libvncserver1 libvterm0 libwpe-1.0-1
  libwpebackend-fdo-1.0-1 linux-headers-5.15.0-113
  linux-headers-5.15.0-113-generic linux-headers-6.5.0-18-generic
  linux-hwe-6.5-headers-6.5.0-18 linux-image-5.15.0-113-generic
  linux-image-6.5.0-18-generic linux-modules-5.15.0-113-generic
  linux-modules-6.5.0-18-generic linux-modules-extra-5.15.0-113-generic
  linux-modules-extra-6.5.0-18-generic lua-luv mobile-broadband-provider-info
  neovim-runtime network-manager-gnome python3-certifi python3-greenlet
  python3-macaroonbakery python3-msgpack python3-neovim python3-protobuf
  python3-pymacaroons python3-pynvim python3-requests python3-rfc3339
  python3-tz rygel
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.

groupadd: group 'input' already exists
groupadd: group 'i2c' already exists
groupadd: group 'uinput' already exists
Added groups input, i2c, uinput, numberpad to current user
uinput module loaded
i2c-dev module loaded
uinput module added to auto loaded modules
Udev rules reloaded and triggered

Testing interface i2c-0: failed
Testing interface i2c-1: failed
Testing interface i2c-2: success (adr 0x15)
The detection was successful. Touchpad with NumberPad found: i2c-2

Default config will be autocreated during the first run and available for futher modifications here:
/usr/share/asus-numberpad-driver/numberpad_dev

created virtual environment CPython3.10.12.final.0-64 in 108ms
  creator CPython3Posix(dest=/usr/share/asus-numberpad-driver/.env, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=/home/ranko/.local/share/virtualenv)
    added seed packages: pip==22.0.2, setuptools==59.6.0, wheel==0.37.1
  activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
install.sh: line 104: /usr/share/asus-numberpad-driver/.env/bin/activate: No such file or directory
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: pip in /home/ranko/.local/lib/python3.10/site-packages (24.1.2)
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: setuptools in /home/ranko/.local/lib/python3.10/site-packages (71.0.3)
Defaulting to user installation because normal site-packages is not writeable
Collecting libevdev (from -r requirements.txt (line 1))
  Using cached libevdev-0.11-py3-none-any.whl
Requirement already satisfied: numpy in /home/ranko/.local/lib/python3.10/site-packages (from -r requirements.txt (line 2)) (2.0.0)
Collecting pyinotify (from -r requirements.txt (line 3))
  Using cached pyinotify-0.9.6-py3-none-any.whl
Collecting python-xlib (from -r requirements.txt (line 4))
  Using cached python_xlib-0.33-py2.py3-none-any.whl.metadata (6.2 kB)
Collecting smbus2 (from -r requirements.txt (line 5))
[... 41 lines cut ...]
      creating build/temp.linux-x86_64-cpython-310
      building 'xkbcommon._ffi' extension
      creating build/temp.linux-x86_64-cpython-310/build
      creating build/temp.linux-x86_64-cpython-310/build/temp.linux-x86_64-cpython-310
      x86_64-linux-gnu-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I/usr/include/python3.10 -c build/temp.linux-x86_64-cpython-310/xkbcommon._ffi.c -o build/temp.linux-x86_64-cpython-310/build/temp.linux-x86_64-cpython-310/xkbcommon._ffi.o
      build/temp.linux-x86_64-cpython-310/xkbcommon._ffi.c: In function ‘_cffi_const_XKB_CONTEXT_NO_SECURE_GETENV’:
      build/temp.linux-x86_64-cpython-310/xkbcommon._ffi.c:1017:12: error: ‘XKB_CONTEXT_NO_SECURE_GETENV’ undeclared (first use in this function); did you mean ‘_cffi_const_XKB_CONTEXT_NO_SECURE_GETENV’?
       1017 |   int n = (XKB_CONTEXT_NO_SECURE_GETENV) <= 0;
            |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
            |            _cffi_const_XKB_CONTEXT_NO_SECURE_GETENV
      build/temp.linux-x86_64-cpython-310/xkbcommon._ffi.c:1017:12: note: each undeclared identifier is reported only once for each function it appears in
      error: command '/usr/bin/x86_64-linux-gnu-gcc' failed with exit code 1
      [end of output]
  
[... cut ...]
ERROR: ERROR: Failed to build installable wheels for some pyproject.toml based projects (xkbcommon)
pyenv-virtualenv: deactivate must be sourced. Run 'source deactivate' instead of 'deactivate'

NumberPad layout

The automatically recommended NumberPad layout for this laptop (ASUS Zenbook 14 UM3406HA_UM3406HA) is up5401ea. Do you want to use the up5401ea layout? (The photo of the recommended NumberPad layout can be found here https://github.com/asus-linux-drivers/asus-numberpad-driver#up5401ea) [y/N]

I am using ubuntu 22.04 kernel version: 6.9.3-060903-generic

sde1000, 2024-07-20:
The system you are building on has a version of libxkbcommon that is lower than 1.5, but your requirements.txt file is requesting the latest version of python-xkbcommon, which now requires at least libxkbcommon 1.5.

In the project's requirements.txt file, specify `xkbcommon<1.1` instead of just `xkbcommon`.

ldrahnik, 2024-07-20:
@Shrimp22 I had the same issue after clearing the cache:

```
sudo rm -rf /usr/share/asus-numberpad-driver/.env
```

The version is now fixed to be lower than `xkbcommon<1.1` as advised @sde1000. 

Can be reopened.

closed 2024-07-20 by ldrahnik