asus-linux-drivers/asus-numberpad-driver #187: [Bug]: Top buttons don't work and numbers don't match

Triw-12, 2024-07-25:
### Describe the bug

1st problem: the buttons for changing numpad brightness and for activating/deactivating the numpad simply don't work, even with a long press.

2nd problem: Numbers return other characters that would correspond if I had pressed the keyboard keys without shift. For example, when I press 7, the character "è" appears (I have an azerty keyboard). Strangely, this problem does not appear in the search bars. Strangely enough, this problem does not appear in the search bars, but in applications such as word processing or the calculator, the problem does appear.

### Expected behavior

I'd like the keyboard to switch on/off when I press the top keys, and to be able to adjust the contrast.
I'd like the keys to display numbers correctly.

### Relevant log output

```shell
Input driver version is 1.0.1
Input device ID: bus 0x0 vendor 0x0 product 0x0 version 0x0
Input device name: "ASUP1415:00 093A:300C NumberPad"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 2 (KEY_1)
    Event code 3 (KEY_2)
    Event code 4 (KEY_3)
    Event code 5 (KEY_4)
    Event code 6 (KEY_5)
    Event code 7 (KEY_6)
    Event code 8 (KEY_7)
    Event code 9 (KEY_8)
    Event code 10 (KEY_9)
    Event code 11 (KEY_0)
    Event code 13 (KEY_EQUAL)
    Event code 14 (KEY_BACKSPACE)
    Event code 16 (KEY_Q)
    Event code 18 (KEY_E)
    Event code 22 (KEY_U)
    Event code 28 (KEY_ENTER)
    Event code 29 (KEY_LEFTCTRL)
    Event code 32 (KEY_D)
    Event code 33 (KEY_F)
    Event code 40 (KEY_APOSTROPHE)
    Event code 42 (KEY_LEFTSHIFT)
    Event code 43 (KEY_BACKSLASH)
    Event code 46 (KEY_C)
    Event code 48 (KEY_B)
    Event code 51 (KEY_COMMA)
    Event code 52 (KEY_DOT)
    Event code 57 (KEY_SPACE)
    Event code 69 (KEY_NUMLOCK)
    Event code 140 (KEY_CALC)
    Event code 272 (BTN_LEFT)
    Event code 273 (BTN_RIGHT)
    Event code 274 (BTN_MIDDLE)
Properties:
Testing ... (interrupt to exit)
Event: time 1721924934.463091, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 1
Event: time 1721924934.463091, -------------- SYN_REPORT ------------
Event: time 1721924934.463153, type 1 (EV_KEY), code 11 (KEY_0), value 1
Event: time 1721924934.463153, -------------- SYN_REPORT ------------
Event: time 1721924934.463272, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 0
Event: time 1721924934.463272, -------------- SYN_REPORT ------------
Event: time 1721924934.463300, type 1 (EV_KEY), code 11 (KEY_0), value 0
Event: time 1721924934.463300, -------------- SYN_REPORT ------------
àEvent: time 1721924935.321111, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 1
Event: time 1721924935.321111, -------------- SYN_REPORT ------------
Event: time 1721924935.321144, type 1 (EV_KEY), code 2 (KEY_1), value 1
Event: time 1721924935.321144, -------------- SYN_REPORT ------------
Event: time 1721924935.321203, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 0
Event: time 1721924935.321203, -------------- SYN_REPORT ------------
Event: time 1721924935.321217, type 1 (EV_KEY), code 2 (KEY_1), value 0
Event: time 1721924935.321217, -------------- SYN_REPORT ------------
[... 126 lines cut ...]
sys_numlock_enables_numpad = 1
top_left_icon_activation_time = 1
top_left_icon_slide_func_activation_radius = 1200
top_left_icon_slide_func_activates_numpad = 1
top_right_icon_slide_func_activation_radius = 1200
enabled_touchpad_pointer = 3
press_key_when_is_done_untouch = 1
enabled = 1
default_backlight_level = 0x01
brightness = 0x48
top_left_icon_brightness_func_disabled = 0
distance_to_move_only_pointer = 0
idled = 0
idle_brightness = 30
idle_enabled = 1
idle_time = 10

(all options are activated, model up5401ea, computer ASUS Zenbook 14 UM3406HA_UM3406HA)
```

### Desktop

- Program version [e.g. 57c742a]
- Distro [e.g. Arch Linux]
- Kernel [e.g. 5.18.15.arch1-2]

ldrahnik, 2024-07-27:
@Triw-12  Hi, thank you for creating an issue. Could you please attach the install and error logs, both you can find under the directory `/var/log/asus-numberpad-driver`.

Triw-12, 2024-07-27:
[error.log](https://github.com/user-attachments/files/16398728/error.log)
[install-25-07-2024-18-23-00.log](https://github.com/user-attachments/files/16398729/install-25-07-2024-18-23-00.log)

ldrahnik, 2024-07-27:
@Triw-12 Please add to file `asus_numberpad_driver.wayland.service` this line `Environment="LOG=DEBUG"` under `[Service]` and reinstall. In the error log, you now should see the mapping of keys sent by your Wayland server to this driver. Find something like but for Wayland:

```
2024-07-21 16:54:48,636 DEBUG X11 will try to load keymap
2024-07-21 16:54:48,637 DEBUG X11 loaded keymap succesfully
2024-07-21 16:54:48,637 DEBUG {'Num_Lock': KEY_NUMLOCK:69, '0': KEY_0:11, '1': KEY_1:2, '2': KEY_2:3, '3': KEY_3:4, '4': KEY_4:5, '5': KEY_5:6, '6': KEY_6:7, '7': KEY_7:8, '8': KEY_8:9, '9': KEY_9:10, 'a': KEY_A:30, 'b': KEY_B:48, 'c': KEY_C:46, 'd': KEY_D:32, 'e': KEY_E:18, 'f': KEY_F:33, 'Shift_L': KEY_LEFTSHIFT:42, 'Control_L': KEY_LEFTCTRL:29, 'u': KEY_U:22, 'space': KEY_SPACE:57, 'slash': KEY_SLASH:53, 'BackSpace': KEY_BACKSPACE:14, 'asterisk': [KEY_LEFTSHIFT:42, KEY_8:9], 'minus': KEY_MINUS:12, 'percent': [KEY_LEFTSHIFT:42, KEY_5:6], 'period': KEY_DOT:52, 'Return': KEY_ENTER:28, 'plus': [KEY_LEFTSHIFT:42, KEY_EQUAL:13], 'equal': KEY_EQUAL:13}
```

or post here entire log.

Triw-12, 2024-07-27:
```
2024-07-27 14:44:21,169 DEBUG Wayland will try to load keymap
2024-07-27 14:44:21,199 DEBUG Wayland loaded keymap succesfully
2024-07-27 14:44:21,199 DEBUG {'Num_Lock': KEY_NUMLOCK:69, '0': [KEY_LEFTSHIFT:42, KEY_0:11], '1': [KEY_LEFTSHIFT:42, KEY_1:2], '2': [KEY_LEFTSHIFT:42, KEY_2:3], '3': [KEY_LEFTSHIFT:42, KEY_3:4], '4': [KEY_LEFTSHIFT:42, KEY_4:5], '5': [KEY_LEFTSHIFT:42, KEY_5:6], '6': [KEY_LEFTSHIFT:42, KEY_6:7], '7': [KEY_LEFTSHIFT:42, KEY_7:8], '8': [KEY_LEFTSHIFT:42, KEY_8:9], '9': [KEY_LEFTSHIFT:42, KEY_9:10], 'a': KEY_Q:16, 'b': KEY_B:48, 'c': KEY_C:46, 'd': KEY_D:32, 'e': KEY_E:18, 'f': KEY_F:33, 'Shift_L': KEY_LEFTSHIFT:42, 'Control_L': KEY_LEFTCTRL:29, 'u': KEY_U:22, 'space': KEY_SPACE:57, 'slash': [KEY_LEFTSHIFT:42, KEY_DOT:52], 'BackSpace': KEY_BACKSPACE:14, 'asterisk': KEY_BACKSLASH:43, 'minus': KEY_6:7, 'percent': [KEY_LEFTSHIFT:42, KEY_APOSTROPHE:40], 'period': [KEY_LEFTSHIFT:42, KEY_COMMA:51], 'Return': KEY_ENTER:28, 'plus': [KEY_LEFTSHIFT:42, KEY_EQUAL:13], 'equal': KEY_EQUAL:13}
```

ldrahnik, 2024-07-27:
@Triw-12 Corresponds that with the configuration inside your Wayland compositor?

ldrahnik, 2024-07-27:
@Triw-12 About 1st problem, does work slide gesture to the center? To activating? Please, attach here clean driver log with attempt.

Triw-12, 2024-07-28:
Thank you for your prompt replies and sorry mine were a bit slow.

> @Triw-12 Corresponds that to the configuration inside your Wayland compositor? Keymap layout? In other words Is that correct? (I do not know which one - compositor - you use, e.g. hyprland)
> 
> _EDIT:_ Do you have multiple keyboard layouts? Try install English with querty and switch between azerty and querty. Because you have enabled debugging you should now see in driver log layout Is changed and even key map should be changed every time.

I have the impression that the shift is not properly configured:
it should be at 52 rather than 42 (I'm running GNOME)
[wayland.txt](https://github.com/user-attachments/files/16403874/wayland.txt)

Triw-12, 2024-07-28:
> @Triw-12 About 1st problem, does work slide gesture to the center? To activating? Please, attach me here clean driver log with attempt.
> 
> **EDIT:** try on the latest master, please

By touching the numlock for more than a second :
```
2024-07-28 15:43:58,665 INFO Touched top_right_icon area (numlock) in time: 1722174238.6650488
2024-07-28 15:43:58,940 INFO Un-touched top_right_icon area (representing numlock key) in time: 1722174238.9404347
2024-07-28 15:43:59,141 INFO Un-touched top_right_icon area (representing numlock key) in time: 1722174239.1409836
2024-07-28 15:43:59,984 INFO Un-touched top_right_icon area (representing numlock key) in time: 1722174239.9849148
2024-07-28 15:44:00,305 INFO Un-touched top_right_icon area (representing numlock key) in time: 1722174240.3049948
2024-07-28 15:44:00,584 INFO Un-touched top_right_icon area (representing numlock key) in time: 1722174240.5846543
2024-07-28 15:44:00,905 INFO Un-touched top_right_icon area (representing numlock key) in time: 1722174240.9049573
2024-07-28 15:44:00,945 INFO Un-touched top_right_icon area (representing numlock key) in time: 1722174240.944957
2024-07-28 15:44:00,984 INFO Un-touched top_right_icon area (representing numlock key) in time: 1722174240.9844425
2024-07-28 15:44:01,064 INFO Un-touched top_right_icon area (representing numlock key) in time: 1722174241.0644221
2024-07-28 15:44:01,144 INFO Un-touched top_right_icon area (representing numlock key) in time: 1722174241.1442242
2024-07-28 15:44:01,224 INFO Un-touched top_right_icon area (representing numlock key) in time: 1722174241.2247372
```
Numpad remains off

By locking the numpad manually (change enable from `numberpad_dev`) and sliding the second button:
```
2024-07-28 15:50:13,986 INFO Touched top_left_icon area in time: 1722174613.9867203
2024-07-28 15:50:14,268 INFO Un-touched top_left_icon area in time: 1722174614.26861
2024-07-28 15:50:14,348 INFO Un-touched top_left_icon area in time: 1722174614.3488176
2024-07-28 15:50:14,552 INFO Un-touched top_left_icon area in time: 1722174614.5523615
2024-07-28 15:50:14,712 INFO Un-touched top_left_icon area in time: 1722174614.712445
2024-07-28 15:50:14,793 INFO Un-touched top_left_icon area in time: 1722174614.7930832
2024-07-28 15:50:14,833 INFO Un-touched top_left_icon area in time: 1722174614.8330245
2024-07-28 15:50:14,872 INFO Un-touched top_left_icon area in time: 1722174614.8724062
2024-07-28 15:50:15,033 INFO Un-touched top_left_icon area in time: 1722174615.0330153
2024-07-28 15:50:15,073 INFO Un-touched top_left_icon area in time: 1722174615.0729878
2024-07-28 15:50:15,113 INFO Un-touched top_left_icon area in time: 1722174615.1129735
2024-07-28 15:50:15,153 INFO Un-touched top_left_icon area in time: 1722174615.1529942
```

Full logs since reinstallation (I think there was an error during installation, which is why I'm putting it here):
[error.log](https://github.com/user-attachments/files/16403940/error.log)

ldrahnik, 2024-07-28:
Thank you for your prompt replies and sorry mine were a bit slow.

> @Triw-12 Corresponds that to the configuration inside your Wayland compositor? Keymap layout? In other words Is that correct? (I do not know which one - compositor - you use, e.g. hyprland)
>
> _EDIT:_ [...]

I have the impression that the shift is not properly configured:
it should be at 52 rather than 42 (I'm running GNOME)
[wayland.txt](https://github.com/user-attachments/files/16403874/wayland.txt)

@Triw-12 Which Azerty keyboard layout do you use? Is modified? or is it the same as the original one? I would like to replicate your problem.

[image]

ldrahnik, 2024-07-28:
> @Triw-12 Strangely, this problem does not appear in the search bars. Strangely enough, this problem does not appear in the search bars, but in applications such as word processing or the calculator, the problem does appear.

@Triw-12 Can you please confirm that manually printing numbers using shift+numeric key above letters line QWERTY works? If I understood you correctly it works only in some places?

ldrahnik, 2024-07-28:
@Triw-12 Have you installed distribution in clean way? Or you modified / installed wayland compositor by your own?

ldrahnik, 2024-07-28:
@Triw-12 Install `dconf` and look here:

[image]

Is your keyboard current?

Triw-12, 2024-07-28:
Here is my keyboard configuration :
# KEYBOARD CONFIGURATION FILE

# Consult the keyboard(5) manual page.

XKBMODEL="pc105"
XKBLAYOUT="fr"
XKBVARIANT="latin9"
XKBOPTIONS=""

BACKSPACE="guess"

> > @Triw-12 Strangely, this problem does not appear in the search bars. Strangely enough, this problem does not appear in the search bars, but in applications such as word processing or the calculator, the problem does appear.
> 
> @Triw-12 Can you please confirm that manually printing numbers using shift+numeric key above letters line QWERTY works? If I understood you correctly it works only in some places?
> 
> **EDIT**: I would still recommend to you install QWERTY keyboard and try to switch between them.

I've tried it with an English keyboard and it works for numbers but not for operators (+, -, *, %, =).
Unfortunately, I've tried with other keyboards but the numpad doesn't work anymore since I put the new configuration, but I get a new error in error.log :
```
Exception in thread Thread-6 (check_gnome_layout):
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 950, in check_gnome_layout
    layout_is_active = keyboard_state.layout_index_is_active(mru_layout_index, xkb.StateComponent.XKB_STATE_LAYOUT_EFFECTIVE)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.11/site-packages/xkbcommon/xkb.py", line 1221, in layout_index_is_active
    raise XKBInvalidLayoutIndex()
xkbcommon.xkb.XKBInvalidLayoutIndex
```
I've run a sudo evtest and the numpad is detected, but when I press the keys, nothing happens...

> @Triw-12 Have you installed distribution in clean way? Or you modified / installed wayland compositor by your own?

I only installed the operating system a week ago (getting the numpad to work was pretty much the first thing I tested), so I'm unlikely to have changed any settings.

ldrahnik, 2024-08-03:
> I don't understand what you want me to do...

You can install `dconf` on Ubuntu with the command: `sudo apt install dconf-editor`. For arch try find out. Then open and go here `/org/gnome/desktop/input-sources/` and post the screenshot what do you see here.

ldrahnik, 2024-08-04:
> 2nd problem: Numbers return other characters that would correspond if I had pressed the keyboard keys without shift. For example, when I press 7, the character "è" appears (I have an azerty [...]

@Triw-12 The second problem appears only after this NumberPad driver installation?

Triw-12, 2024-08-08:
> @Triw-12 Install `dconf`. On Ubuntu with the command: `sudo apt install dconf-editor`. For arch try find out. It may be under the same package name. Then open `dconf` and go here `/org/gnome/desktop/input-sources/` and post the screenshot of what do you see here.
> 
> Please, use the latest master. Not printing numbers at all should be solved by reverting the commit mentioned above.

[image]

The latest master has been solved the problem of not printing numbers !

> @Triw-12 The second problem appears only after this NumberPad driver installation?

Yes, I haven't found any other project that lets you use the numpad with this model.

ldrahnik, 2024-08-17:
@Triw-12 I detected an error in the code when using this driver on distribution using wayland and that uses `mru-sources`. Was not prioritized active layout from source `mru-sources` and was not reloaded keymap. Attempt to fix I have committed to the branch https://github.com/asus-linux-drivers/asus-numberpad-driver/tree/187-characters-do-not-match. Please, test it and let me know.

Triw-12, 2024-08-17:
Sorry, but I have the impression that nothing has changed...

Triw-12, 2024-08-17:
The strange thing is that the problem doesn't appear on certain search bars (such as gnome's application search) and when it doesn't work, it's the buttons that are engaged that are good, it's just the shift that doesn't work (after that I'm not sure how your code works)...

[image]

[image]

ldrahnik, 2024-08-17:
@Triw-12 Hm. Please, do this in the mentioned branch:

```
$ systemctl stop --user asus_numberpad_driver@$USER.service # stop running service
$ cd asus-numberpad-driver
$ LOG=DEBUG /usr/share/asus-numberpad-driver/.env/bin/python3 /usr/share/asus-numberpad-driver/numberpad.py "up5401ea" # run with debug logs
```

And try to change the keyboard layouts during. How do you change them? On the top panel? Open `dconf` application and keep watching whether is the most recently used (`mru_sources` here `/org/gnome/desktop/input-sources/mru-sources`) the same as is displayed on the top panel.

 It the last visible keymap in the log correct for the currently selected keyboard layout?

What is the name of the application when it does not work? I have found no problematic app in virtual box wayland gnome ubuntu 22.04.

ldrahnik, 2024-08-17:
@Triw-12 Do not you have shift as part of specific shortcut inside that app for which it does not work?

ldrahnik, 2024-08-18:
@Triw-12 In short, I would focus what sequence of keys do send physical keyboard (`libinput debug-events`) for one specific problematics charakter and what send this driver.

Triw-12, 2024-08-18:
After testing, the keyboard change is well recognized (thanks to the upper panel) and seems correct:
For French AZERTY : 
```
2024-08-18 15:01:47,828 DEBUG {'Num_Lock': KEY_NUMLOCK:69, '0': [KEY_LEFTSHIFT:42, KEY_0:11], '1': [KEY_LEFTSHIFT:42, KEY_1:2], '2': [KEY_LEFTSHIFT:42, KEY_2:3], '3': [KEY_LEFTSHIFT:42, KEY_3:4], '4': [KEY_LEFTSHIFT:42, KEY_4:5], '5': [KEY_LEFTSHIFT:42, KEY_5:6], '6': [KEY_LEFTSHIFT:42, KEY_6:7], '7': [KEY_LEFTSHIFT:42, KEY_7:8], '8': [KEY_LEFTSHIFT:42, KEY_8:9], '9': [KEY_LEFTSHIFT:42, KEY_9:10], 'a': KEY_Q:16, 'b': KEY_B:48, 'c': KEY_C:46, 'd': KEY_D:32, 'e': KEY_E:18, 'f': KEY_F:33, 'Shift_L': KEY_LEFTSHIFT:42, 'Control_L': KEY_LEFTCTRL:29, 'u': KEY_U:22, 'space': KEY_SPACE:57, 'slash': [KEY_LEFTSHIFT:42, KEY_DOT:52], 'BackSpace': KEY_BACKSPACE:14, 'asterisk': KEY_BACKSLASH:43, 'minus': KEY_6:7, 'percent': [KEY_LEFTSHIFT:42, KEY_APOSTROPHE:40], 'period': [KEY_LEFTSHIFT:42, KEY_COMMA:51], 'Return': KEY_ENTER:28, 'plus': [KEY_LEFTSHIFT:42, KEY_EQUAL:13], 'equal': KEY_EQUAL:13}
```
For French BEPO : 
```
2024-08-18 15:13:01,908 DEBUG {'Num_Lock': KEY_NUMLOCK:69, '0': [KEY_LEFTSHIFT:42, KEY_0:11], '1': [KEY_LEFTSHIFT:42, KEY_1:2], '2': [KEY_LEFTSHIFT:42, KEY_2:3], '3': [KEY_LEFTSHIFT:42, KEY_3:4], '4': [KEY_LEFTSHIFT:42, KEY_4:5], '5': [KEY_LEFTSHIFT:42, KEY_5:6], '6': [KEY_LEFTSHIFT:42, KEY_6:7], '7': [KEY_LEFTSHIFT:42, KEY_7:8], '8': [KEY_LEFTSHIFT:42, KEY_8:9], '9': [KEY_LEFTSHIFT:42, KEY_9:10], 'a': KEY_A:30, 'b': KEY_Q:16, 'c': KEY_H:35, 'd': KEY_I:23, 'e': KEY_F:33, 'f': KEY_SLASH:53, 'Shift_L': KEY_LEFTSHIFT:42, 'Control_L': KEY_LEFTCTRL:29, 'u': KEY_S:31, 'space': KEY_SPACE:57, 'slash': KEY_9:10, 'BackSpace': KEY_BACKSPACE:14, 'asterisk': KEY_0:11, 'minus': KEY_8:9, 'percent': KEY_EQUAL:13, 'period': KEY_V:47, 'Return': KEY_ENTER:28, 'plus': KEY_7:8, 'equal': KEY_MINUS:12}
```

Shift-click also works correctly : 
For a click with the numpad (with libinput): 
```
-event17  KEYBOARD_KEY            +62.683s	*** (-1) pressed
 event17  KEYBOARD_KEY            +62.684s	*** (-1) pressed
 event17  KEYBOARD_KEY            +62.684s	*** (-1) released
 event17  KEYBOARD_KEY            +62.684s	*** (-1) released
```
For a physical click :
```
-event0   KEYBOARD_KEY            +66.762s	*** (-1) pressed
 event0   KEYBOARD_KEY            +66.942s	*** (-1) pressed
 event0   KEYBOARD_KEY            +67.060s	*** (-1) released
 event0   KEYBOARD_KEY            +67.066s	*** (-1) released
```
The problem only appears in certain applications where the shift is not necessarily a shortcut: 
- Terminal
- [GNOME calculator](https://wiki.gnome.org/Apps/Calculator)
- LibreOffice

But does not appear in Mozilla or the search bar for GNOME applications

ldrahnik, 2024-09-16:
Can be reopened.

ldrahnik, 2024-09-22:
@Triw-12 Problem with top buttons was discussed also in this issue https://github.com/asus-linux-drivers/asus-numberpad-driver/issues/169 and fix was already commited to the master branch. Could be related to your first problem.

Triw-12, 2024-09-22:
Yes, it works! Don't hesitate to let me know if you think you've solved the second problem!

ldrahnik, 2024-09-29:
@Triw-12 Can you change the keyboard layout (e.g. from `en` to `fr`) inside the terminal by command? Please, do that and then answer:

1. Terminal was changed from the state well-printed characters sent by NumberPad to not-well-printed character
3. Behaviour inside other apps like text editor, libre office, ... changed / not changed
4. App groups that not-well-printed and well-printed characters are inverted now
5. Or describe what happened

Triw-12, 2024-09-30:
When I change keyboards on the command line, the changes are not applied and the problem is always the same as if the command line did absolutely nothing (but dconf saved the changes but not the top panel). The not well-printed characters have not been interchanged
If I switch to the upper panel, however, it works fine and the keyboard changes smoothly.

benj5378, 2024-10-27:
@Triw-12, can you try out #211?

closed 2024-10-29 by ldrahnik