asus-linux-drivers/asus-numberpad-driver #188: [Bug]:

Gonzalo1985, 2024-07-29:
### Describe the bug

I get this message:
Testing interface i2c-0: failed
Testing interface i2c-1: failed
The detection was not successful. Touchpad with NumberPad not found.

This message I get after run:
$ bash install.sh

or this:
$ bash install_device_check.sh

Could you help me?

### Expected behavior

I want to have access to the NumberPad from Ubuntu in my Asus Zenbook model UX3402V

### Desktop

- Program version [e.g. 57c742a]
- Distro [e.g. Arch Linux]
- Kernel [e.g. 5.18.15.arch1-2]

ldrahnik, 2024-07-29:
@Gonzalo1985 Hey, on your model sith Windows NumberPad works, right? I need to see logs from `/var/log/asus-numberpad-driver` and your dsdt table.

```
$ sudo apt-get install iasl
$ cd <work-dir>
$ sudo cp /sys/firmware/acpi/tables/DSDT DSDT
$ sudo iasl -d DSDT
$ sudo gedit DSDT.dsl
```

Gonzalo1985, 2024-07-29:
Hello. I was able to install the drivers now. I can see the NumberPad in my asus, but when I push the numbers doesn't work.

This is my last log:

Leyendo lista de paquetes...
Creando árbol de dependencias...
Leyendo la información de estado...
ibus ya está en su versión más reciente (1.5.29-2).
libevdev2 ya está en su versión más reciente (1.13.1+dfsg-1build1).
curl ya está en su versión más reciente (8.5.0-2ubuntu10.1).
xinput ya está en su versión más reciente (1.6.4-1build1).
i2c-tools ya está en su versión más reciente (4.3-4build2).
python3-dev ya está en su versión más reciente (3.12.3-0ubuntu1).
python3-virtualenv ya está en su versión más reciente (20.25.0+ds-2).
libxml2-utils ya está en su versión más reciente (2.9.14+dfsg-1.3ubuntu3).
libxkbcommon-dev ya está en su versión más reciente (1.6.0-1build1).
gcc ya está en su versión más reciente (4:13.2.0-7ubuntu1).
0 actualizados, 0 nuevos se instalarán, 0 para eliminar y 39 no actualizados.

groupadd: el grupo «input» ya existe
groupadd: el grupo «i2c» ya existe
groupadd: el grupo «uinput» ya existe
Added groups input, i2c, uinput, numberpad to current user
uinput module loaded
i2c-dev module loaded
uinput module added to auto loaded modules
Udev rules reloaded and triggered

Testing interface i2c-0: success (adr 0x15)
The detection was successful. Touchpad with NumberPad found: i2c-0

Default config will be autocreated during the first run and available for futher modifications here:
/usr/share/asus-numberpad-driver/numberpad_dev

created virtual environment CPython3.12.3.final.0-64 in 342ms
  creator CPython3Posix(dest=/usr/share/asus-numberpad-driver/.env, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(download=False, pip=bundle, via=copy, app_data_dir=/home/gonzalo/.local/share/virtualenv)
    added seed packages: pip==24.0
  activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
Requirement already satisfied: pip in /usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages (24.0)
Collecting pip
  Using cached pip-24.2-py3-none-any.whl.metadata (3.6 kB)
Using cached pip-24.2-py3-none-any.whl (1.8 MB)
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 24.0
    Uninstalling pip-24.0:
      Successfully uninstalled pip-24.0
Successfully installed pip-24.2
Collecting setuptools
  Using cached setuptools-71.1.0-py3-none-any.whl.metadata (6.6 kB)
Using cached setuptools-71.1.0-py3-none-any.whl (2.3 MB)
Installing collected packages: setuptools
Successfully installed setuptools-71.1.0
Collecting libevdev (from -r requirements.txt (line 1))
  Using cached libevdev-0.11-py3-none-any.whl
Collecting numpy (from -r requirements.txt (line 2))
  Using cached numpy-2.0.1-cp312-cp312-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (60 kB)
Collecting pyinotify (from -r requirements.txt (line 3))
  Using cached pyinotify-0.9.6-py3-none-any.whl
Collecting python-xlib (from -r requirements.txt (line 4))
  Using cached python_xlib-0.33-py2.py3-none-any.whl.metadata (6.2 kB)
Collecting smbus2 (from -r requirements.txt (line 5))
  Using cached smbus2-0.4.3-py2.py3-none-any.whl.metadata (7.1 kB)
Collecting pyasyncore (from -r requirements.txt (line 6))
  Using cached pyasyncore-1.0.4-py3-none-any.whl.metadata (3.7 kB)
Collecting pywayland (from -r requirements.txt (line 7))
  Using cached pywayland-0.4.18-cp312-cp312-manylinux_2_28_x86_64.whl.metadata (4.4 kB)
Collecting xkbcommon<1.1 (from -r requirements.txt (line 8))
  Using cached xkbcommon-1.0.1-cp312-cp312-linux_x86_64.whl
[... 71 lines cut ...]
Udev rules reloaded and triggered

Calculator app

Do you want try to install toggling script for XF86Calculator key? [y/N]
Detected gnome-calculator
Setting up for gnome-calculator
Toggling script for calculator app gnome-calculator has been installed.

Power supply saver

By default is idle functionality disabled and may be manually enabled in config file later. Or by installing this rule can be enabled by every detected change of power supply mode to battery mode.

Idle mode is configured to be enabled after 10s of inactivity and to decrease 30% of brightness.

Do you want install the rule for idle functionality? [y/N]
Rule 80-numberpad-power-supply applied
Udev rules reloaded and triggered

Installation finished succesfully

Reboot is required. Do you want reboot now? [y/N]

Thank you
Gonzalo

ldrahnik, 2024-07-29:
@Gonzalo1985 IMHO your NumberPad layout is `up5401ea` or post the website with your product to check.

Gonzalo1985, 2024-07-29:
I also tried that option but it didn't work at the moment. Sorry, I didn't understand the "post the website with your product to check". I bought this notebook in Corte Ingles, Spain: https://www.elcorteingles.es/electronica/A46492617-portatil-asus-zenbook-14-oled-ux3402va-km208w-i5-16gb-512gb-ssd-14-w11/

With the original Windows 11 that the notebook came with the NumberPad worked just fine, but when I change the Windows to Ubuntu I had this problem.

Thank you,
Gonzalo

ldrahnik, 2024-07-29:
@Gonzalo1985 Your correct NumberPad layout Is up5401. Recommended one. 

I need to see error log from `/var/log/asus-numberpad-driver`.

Gonzalo1985, 2024-07-29:
Thank you @ldrahnik. Here is the error log. I have installed again the up5401 model driver:

2024-07-29 08:48:20,494 INFO Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
2024-07-29 08:48:20,494 INFO Set keyboard 3 from H: Handlers=sysrq kbd event3 leds
2024-07-29 08:48:20,507 INFO Detecting touchpad from string: "N: Name="ASUE140D:00 04F3:31B9 Touchpad""
2024-07-29 08:48:20,508 INFO Set touchpad device id 0 from S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-ASUE140D:00/0018:04F3:31B9.0001/input/input8
2024-07-29 08:48:20,508 INFO Set touchpad id 5 from H: Handlers=mouse1 event5
2024-07-29 08:48:20,520 INFO Touchpad min-max: x 0-3987, y 0-2238
2024-07-29 08:48:20,520 INFO Numpad min-max: x 40-3947, y 80-2158
2024-07-29 08:48:20,521 ERROR Wayland load keymap listener error. Exiting
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 442, in load_keymap_listener_wayland
    display_wayland.connect()
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages/pywayland/client/display.py", line 131, in connect
    raise ValueError("Unable to connect to display")
ValueError: Unable to connect to display
2024-07-29 08:48:20,523 ERROR Listening touchpad events unexpectedly failed
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 2385, in <module>
    sleep(0.5)
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 2361, in signal_handler
    raise Exception()
Exception
2024-07-29 08:48:20,524 INFO Clean up started
2024-07-29 08:48:20,558 INFO Clean up finished
2024-07-29 08:48:20,559 INFO Exiting
2024-07-29 08:48:22,121 INFO Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
2024-07-29 08:48:22,121 INFO Set keyboard 3 from H: Handlers=sysrq kbd event3 leds
2024-07-29 08:48:22,131 INFO Detecting touchpad from string: "N: Name="ASUE140D:00 04F3:31B9 Touchpad""
2024-07-29 08:48:22,131 INFO Set touchpad device id 0 from S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-ASUE140D:00/0018:04F3:31B9.0001/input/input8
2024-07-29 08:48:22,131 INFO Set touchpad id 5 from H: Handlers=mouse1 event5
2024-07-29 08:48:22,131 INFO Touchpad min-max: x 0-3987, y 0-2238
2024-07-29 08:48:22,132 INFO Numpad min-max: x 40-3947, y 80-2158
2024-07-29 08:59:08,125 INFO Touched top_right_icon area (numlock) in time: 1722254348.1252372
2024-07-29 08:59:11,576 INFO Touched top_right_icon area (numlock) in time: 1722254351.576629
2024-07-29 08:59:12,581 INFO The numpad numlock was pressed longer than the activation time and will be activated: 1.0041477680206299
2024-07-29 08:59:12,581 INFO Activation time: 1.0
2024-07-29 08:59:12,601 INFO System numlock activated
2024-07-29 08:59:12,601 INFO Numpad activated
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ASUE140D:00 04F3:31B9 Touchpad
2024-07-29 08:59:12,632 INFO Setting up for config file key: "enabled" with value: "True"
2024-07-29 08:59:12,633 INFO check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/numberpad_dev wd=1 >
2024-07-29 08:59:23,766 INFO Touched top_right_icon area (numlock) in time: 1722254363.766506
2024-07-29 08:59:24,771 INFO The numpad numlock was pressed longer than the activation time and will be activated: 1.0042378902435303
2024-07-29 08:59:24,771 INFO Activation time: 1.0
2024-07-29 08:59:24,797 INFO System numlock deactivated
2024-07-29 08:59:24,797 INFO Numpad deactivated
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ASUE140D:00 04F3:31B9 Touchpad
2024-07-29 08:59:24,824 INFO Setting up for config file key: "enabled" with value: "False"
2024-07-29 08:59:24,825 INFO check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/numberpad_dev wd=1 >
2024-07-29 09:19:32,203 INFO Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
2024-07-29 09:19:32,203 INFO Set keyboard 3 from H: Handlers=sysrq kbd event3 leds
2024-07-29 09:19:32,211 INFO Detecting touchpad from string: "N: Name="ASUE140D:00 04F3:31B9 Touchpad""
2024-07-29 09:19:32,211 INFO Set touchpad device id 0 from S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-ASUE140D:00/0018:04F3:31B9.0001/input/input8
2024-07-29 09:19:32,211 INFO Set touchpad id 5 from H: Handlers=mouse1 event5
2024-07-29 09:19:32,212 INFO Touchpad min-max: x 0-3987, y 0-2238
2024-07-29 09:19:32,212 INFO Numpad min-max: x 200-3787, y 200-2158
2024-07-29 09:19:33,219 INFO Setting up for config file key: "brightness" with value: "0x48"
2024-07-29 09:19:38,300 INFO check_config_values_changes: detected external change of config file -> loading changes
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=sededi7Wh path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/sededi7Wh wd=1 >
<Event cookie=4808 dir=False mask=0x80 maskname=IN_MOVED_TO name=numberpad_dev path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/numberpad_dev wd=1 >
[... cut ...]
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ASUE140D:00 04F3:31B9 Touchpad
2024-07-29 09:21:52,886 INFO Setting up for config file key: "enabled" with value: "False"
2024-07-29 09:21:52,887 INFO check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/numberpad_dev wd=1 >

Thanks again!
Gonzalo

ldrahnik, 2024-07-30:
@Gonzalo1985 And what js wrong? Could you describe more, please.

Gonzalo1985, 2024-07-30:
Yes @ldrahnik of course. The problem is when I press de numbers in the pad doesn't work. I can see the pad is active but I press number 2 and nothing, I press number 4 and nothing. Happens with all the pad.

ldrahnik, 2024-07-30:
@Gonzalo1985 E.g. top left icon for increasing brightness works? And activating works?

Gonzalo1985, 2024-07-30:
@ldrahnik yes, the top left icon works just fine. I write here the entire log from the beginning of my trying:

2024-07-28 22:16:23,561 INFO Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
2024-07-28 22:16:23,561 INFO Set keyboard 3 from H: Handlers=sysrq kbd event3 leds
2024-07-28 22:16:23,570 INFO Detecting touchpad from string: "N: Name="ASUE140D:00 04F3:31B9 Touchpad""
2024-07-28 22:16:23,571 INFO Set touchpad device id 0 from S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-ASUE140D:00/0018:04F3:31B9.0001/input/input8
2024-07-28 22:16:23,571 INFO Set touchpad id 5 from H: Handlers=mouse1 event5
2024-07-28 22:16:23,571 INFO Touchpad min-max: x 0-3987, y 0-2238
2024-07-28 22:16:23,571 INFO Numpad min-max: x 200-3787, y 200-2158
2024-07-28 22:16:24,577 INFO Setting up for config file key: "brightness" with value: "0x48"
2024-07-28 22:16:27,340 INFO check_config_values_changes: detected external change of config file -> loading changes
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=sed6ar073 path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/sed6ar073 wd=1 >
2024-07-28 22:16:27,443 INFO check_config_values_changes: detected external change of config file -> loading changes
<Event cookie=4484 dir=False mask=0x80 maskname=IN_MOVED_TO name=numberpad_dev path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/numberpad_dev wd=1 >
2024-07-28 22:16:32,914 ERROR Wayland load keymap listener error. Exiting
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 448, in load_keymap_listener_wayland
    while not stop_threads and display_wayland.dispatch(block=True) != -1:
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages/pywayland/utils.py", line 30, in wrapper
    return func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages/pywayland/client/display.py", line 197, in dispatch
    raise RuntimeError(f"Failed with error: {err}")
RuntimeError: Failed with error: 32
2024-07-28 22:16:32,918 ERROR Listening touchpad events unexpectedly failed
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 2441, in <module>
    listen_touchpad_events()
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 1958, in listen_touchpad_events
    for e in d_t.events():
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages/libevdev/device.py", line 536, in events
    ev = self._libevdev.next_event(flags)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages/libevdev/_clib.py", line 889, in next_event
    rc = self._next_event(self._ctx, flags, ctypes.byref(ev))
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 2361, in signal_handler
    raise Exception()
Exception
2024-07-28 22:16:32,920 INFO Clean up started
2024-07-28 22:16:32,929 INFO Clean up finished
2024-07-28 22:16:32,929 INFO Exiting
2024-07-28 22:16:55,138 INFO Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
2024-07-28 22:16:55,139 INFO Set keyboard 3 from H: Handlers=sysrq kbd event3 leds
2024-07-28 22:16:55,148 INFO Detecting touchpad from string: "N: Name="ASUE140D:00 04F3:31B9 Touchpad""
2024-07-28 22:16:55,148 INFO Set touchpad device id 0 from S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-ASUE140D:00/0018:04F3:31B9.0001/input/input8
2024-07-28 22:16:55,148 INFO Set touchpad id 5 from H: Handlers=mouse1 event5
2024-07-28 22:16:55,149 INFO Touchpad min-max: x 0-3987, y 0-2238
2024-07-28 22:16:55,149 INFO Numpad min-max: x 200-3787, y 200-2158
2024-07-28 22:16:55,150 ERROR Wayland load keymap listener error. Exiting
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 442, in load_keymap_listener_wayland
    display_wayland.connect()
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages/pywayland/client/display.py", line 131, in connect
    raise ValueError("Unable to connect to display")
ValueError: Unable to connect to display
2024-07-28 22:16:55,152 ERROR Listening touchpad events unexpectedly failed
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 2385, in <module>
    sleep(0.5)
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 2361, in signal_handler
    raise Exception()
Exception
2024-07-28 22:16:55,153 INFO Clean up started
2024-07-28 22:16:55,166 INFO Clean up finished
2024-07-28 22:16:55,166 INFO Exiting
2024-07-28 22:16:56,635 INFO Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
2024-07-28 22:16:56,636 INFO Set keyboard 3 from H: Handlers=sysrq kbd event3 leds
2024-07-28 22:16:56,648 INFO Detecting touchpad from string: "N: Name="ASUE140D:00 04F3:31B9 Touchpad""
[... 826 lines cut ...]
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 442, in load_keymap_listener_wayland
    display_wayland.connect()
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages/pywayland/client/display.py", line 131, in connect
    raise ValueError("Unable to connect to display")
[... cut ...]
2024-07-30 08:25:01,203 INFO Detecting touchpad from string: "N: Name="ASUE140D:00 04F3:31B9 Touchpad""
2024-07-30 08:25:01,203 INFO Set touchpad device id 0 from S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-ASUE140D:00/0018:04F3:31B9.0001/input/input8
2024-07-30 08:25:01,204 INFO Set touchpad id 5 from H: Handlers=mouse1 event5
2024-07-30 08:25:01,204 INFO Touchpad min-max: x 0-3987, y 0-2238
2024-07-30 08:25:01,204 INFO Numpad min-max: x 200-3787, y 200-2158

I hope it helps! I don't know what to do.

ldrahnik, 2024-07-30:
@Gonzalo1985 When did you try to press number 2 or something else on NumberPad? Specific time. I see nothing wrong.

Gonzalo1985, 2024-07-30:
@ldrahnik in this moment I try to press any number and it doesn't work. But here in the log I see some errors:

2024-07-29 17:10:39,525 INFO Set touchpad id 5 from H: Handlers=mouse1 event5
2024-07-29 17:10:39,525 INFO Touchpad min-max: x 0-3987, y 0-2238
2024-07-29 17:10:39,525 INFO Numpad min-max: x 200-3787, y 200-2158
2024-07-29 17:18:59,568 ERROR Wayland load keymap listener error. Exiting
Traceback (most recent call last):
File "/usr/share/asus-numberpad-driver/numberpad.py", line 448, in load_keymap_listener_wayland
while not stop_threads and display_wayland.dispatch(block=True) != -1:
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages/pywayland/utils.py", line 30, in wrapper
return func(self, *args, **kwargs)
^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages/pywayland/client/display.py", line 197, in dispatch
raise RuntimeError(f"Failed with error: {err}")
RuntimeError: Failed with error: 32
2024-07-29 17:18:59,570 ERROR Listening touchpad events unexpectedly failed
Traceback (most recent call last):
File "/usr/share/asus-numberpad-driver/numberpad.py", line 2441, in
listen_touchpad_events()
File "/usr/share/asus-numberpad-driver/numberpad.py", line 1958, in listen_touchpad_events
for e in d_t.events():
File "/usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages/libevdev/device.py", line 542, in events
ev = self._libevdev.next_event(flags)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages/libevdev/_clib.py", line 889, in next_event
rc = self._next_event(self._ctx, flags, ctypes.byref(ev))
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/share/asus-numberpad-driver/numberpad.py", line 2361, in signal_handler
raise Exception()

ldrahnik, 2024-07-30:
@Gonzalo1985 I see nothing wrong here. Please add to this file `asus_numberpad_driver.wayland.service` line `Environment="LOG=DEBUG"` under `[Service]` and reinstall and post error log again.

ldrahnik, 2024-07-31:
@Gonzalo1985 Please, try the latest master.

Gonzalo1985, 2024-07-31:
Oh yes, great! Now is working. Thank you very much @ldrahnik

closed 2024-07-31 by ldrahnik