asus-linux-drivers/asus-numberpad-driver #189: [Bug]: No numberpad trackpad device found

Pryme-Minister, 2024-07-31:
### Describe the bug

I have tried to install this using all 3 of the install methods, and all of them end up with a failure due to no numpad trackpad being found. 

I have managed to get the pad working with a different driver, so I know I have supported hardware but that driver breaks other things on my system so I would rather not use it. 

### Expected behavior

Trackpad driver install complete and working.

### Relevant log output

```shell
groupadd: group 'numberpad' already exists
Added group numberpad to current user

Reading package lists...
Building dependency tree...
Reading state information...
gcc is already the newest version (4:11.2.0-1ubuntu1).
ibus is already the newest version (1.5.26-4).
libevdev2 is already the newest version (1.12.1+dfsg-1).
libxkbcommon-dev is already the newest version (1.4.0-1).
xinput is already the newest version (1.6.3-1build2).
i2c-tools is already the newest version (4.3-2build1).
python3-virtualenv is already the newest version (20.13.0+ds-2).
curl is already the newest version (7.81.0-1ubuntu1.16).
libxml2-utils is already the newest version (2.9.13+dfsg-1ubuntu0.4).
python3-dev is already the newest version (3.10.6-1~22.04).
0 upgraded, 0 newly installed, 0 to remove and 65 not upgraded.

groupadd: group 'input' already exists
groupadd: group 'i2c' already exists
groupadd: group 'uinput' already exists
Added groups input, i2c, uinput, numberpad to current user
uinput module loaded
i2c-dev module loaded
uinput module added to auto loaded modules
Udev rules reloaded and triggered

Testing interface i2c-1: failed
Testing interface i2c-2: failed
The detection was not successful. Touchpad with NumberPad not found
```

### Desktop

- Pop OS! 22.04 LTS
- Asus Vivobook K3402Z

ldrahnik, 2024-07-31:
@Pryme-Minister Hello, thank you for creating an issue. Could you attach here your DSDT table, please:

```
$ sudo apt-get install iasl
$ cd <work-dir>
$ sudo cp /sys/firmware/acpi/tables/DSDT DSDT
$ sudo iasl -d DSDT
$ sudo gedit DSDT.dsl
```

Pryme-Minister, 2024-07-31:
Intel ACPI Component Architecture
ASL+ Optimizing Compiler/Disassembler version 20200925
Copyright (c) 2000 - 2020 Intel Corporation

File appears to be binary: found 175392 non-ASCII characters, disassembling
Binary file appears to be a valid ACPI table, disassembling
Input file DSDT, Length 0x839B1 (539057) bytes
ACPI: DSDT 0x0000000000000000 0839B1 (v02 _ASUS_ Notebook 01072009 INTL 20200717)
Pass 1 parse of [DSDT]
Pass 2 parse of [DSDT]
Parsing Deferred Opcodes (Methods/Buffers/Packages/Regions)

Parsing completed
Disassembly completed
ASL Output:    DSDT.dsl - 3626899 bytes

 iASL Warning: There were 234 external control methods found during
 disassembly, but only 490 were resolved (4294967040 unresolved). Additional
 ACPI tables may be required to properly disassemble the code. This
 resulting disassembler output file may not compile because the
 disassembler did not know how many arguments to assign to the
 unresolved methods. Note: SSDTs can be dynamically loaded at
 runtime and may or may not be available via the host OS.

 To specify the tables needed to resolve external control method
 references, the -e option can be used to specify the filenames.
 Example iASL invocations:
     iasl -e ssdt1.aml ssdt2.aml ssdt3.aml -d dsdt.aml
     iasl -e dsdt.aml ssdt2.aml -d ssdt1.aml
     iasl -e ssdt*.aml -d dsdt.aml

 In addition, the -fe option can be used to specify a file containing
 control method external declarations with the associated method
 argument counts. Each line of the file must be of the form:
     External (<method pathname>, MethodObj, <argument count>)
 Invocation:
     iasl -fe refs.txt -d dsdt.aml

Pryme-Minister, 2024-07-31:
Not sure if I did that right, never done this before.

Pryme-Minister, 2024-07-31:
Git wont let me upload the files that were output.

ldrahnik, 2024-07-31:
Do it in the sams way how https://github.com/asus-linux-drivers/asus-numberpad-driver/issues/161#issuecomment-1900457098

Pryme-Minister, 2024-07-31:
[DSDT.zip](https://github.com/user-attachments/files/16445605/DSDT.zip)

Pryme-Minister, 2024-07-31:
I was able to zip it, hopefully that's what you need.

ldrahnik, 2024-07-31:
Hm, what is the output of the command `$ i2cdetect -l`?

Try to run manually this command for found numbers, I mean for found I2C buses - whether is backlight of NumberPad enabled:

```
$ i2ctransfer -f -y $NUMBER w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad
```

disable it again you can then using (only previous last differs)
```
$ i2ctransfer -f -y $NUMBER w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x00 0xad
```

It should be probably one of the `Synopsys DesignWare I2C adapter` mapped to number 1,2.

Pryme-Minister, 2024-07-31:
Error: Unsupported option "-1"!
Usage: i2cdetect [-y] [-a] [-q|-r] I2CBUS [FIRST LAST]
       i2cdetect -F I2CBUS
       i2cdetect -l
  I2CBUS is an integer or an I2C bus name
  If provided, FIRST and LAST limit the probing range.

ldrahnik, 2024-07-31:
@Pryme-Minister I wrote `l` as Lenka not `1` as number 1.

Pryme-Minister, 2024-07-31:
I get command not found when doing the other two.

Pryme-Minister, 2024-07-31:
I typed it out instead of copy paste and used "1"

Pryme-Minister, 2024-07-31:
i2c-0	smbus     	SMBus I801 adapter at efa0      	SMBus adapter
i2c-1	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-2	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-3	i2c       	i915 gmbus dpa                  	I2C adapter
i2c-4	i2c       	i915 gmbus dpb                  	I2C adapter
i2c-5	i2c       	i915 gmbus dpc                  	I2C adapter
i2c-6	i2c       	i915 gmbus tc1                  	I2C adapter
i2c-7	i2c       	i915 gmbus tc2                  	I2C adapter
i2c-8	i2c       	i915 gmbus tc3                  	I2C adapter
i2c-9	i2c       	i915 gmbus tc4                  	I2C adapter
i2c-10	i2c       	i915 gmbus tc5                  	I2C adapter
i2c-11	i2c       	i915 gmbus tc6                  	I2C adapter
i2c-12	i2c       	AUX A/DDI A/PHY A               	I2C adapter
i2c-13	i2c       	AUX USBC1/DDI TC1/PHY TC1       	I2C adapter
i2c-14	i2c       	AUX USBC2/DDI TC2/PHY TC2       	I2C adapter
i2c-15	i2c       	AUX USBC3/DDI TC3/PHY TC3       	I2C adapter
i2c-16	i2c       	AUX USBC4/DDI TC4/PHY TC4       	I2C adapter

ldrahnik, 2024-07-31:
@Pryme-Minister The package `i2c-tools` not installed `i2ctransfer`? Try the latest master, should be fixed, when is missing tool `i2ctransfer`, the i2c check is now skipped.

Pryme-Minister, 2024-07-31:
Ok progress! The pad now illuminates with a long press in the right corner, pulls up the calculator with a swipe in the right, and adjusts brightness with a long press in the left corner. Also, my system restarts normally now which the other driver was breaking. However the number pad is not working. I assume I need to activate some sort of "num lock" but there is no option on my keyboard.

ldrahnik, 2024-07-31:
@Pryme-Minister Please, try the latest master.

Pryme-Minister, 2024-07-31:
You are fantastic sir! All functionality is restored.

closed 2024-07-31 by ldrahnik

QushyQushy, 2024-09-16:
I have the same issue on the same machine. Running

`sudo i2ctransfer -f -y 0 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad`

makes the touchpad light up but no functionality.