asus-linux-drivers/asus-numberpad-driver #190: [Bug]: ERROR X11 detected but not connected succesfully to the display on UX3402VA-KM553W with up5401ea layout

nsuffys, 2024-08-05:
### Describe the bug

Hello,

I have a freshly installed Manjaro Linux and the driver on my Zenbook 14 OLED UX3402VA-KM553W and have this error : "ERROR X11 detected but not connected succesfully to the display :0. Exiting".
I do not know what to do.

### Expected behavior

Permission cacess problem to X11 ?

### Relevant log output

```shell
2024-08-05 10:57:28,130 INFO X11 detected and connected succesfully to the display :0
2024-08-05 10:57:28,131 INFO Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
2024-08-05 10:57:28,132 INFO Set keyboard 3 from H: Handlers=sysrq kbd leds event3
2024-08-05 10:57:28,170 INFO Detecting touchpad from string: "N: Name="ASUE140D:00 04F3:31B9 Touchpad""
2024-08-05 10:57:28,171 INFO Set touchpad device id 15 from S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-15/i2c-ASUE140D:00/0018:04F3:31B9.0001/input/input13
2024-08-05 10:57:28,171 INFO Set touchpad id 10 from H: Handlers=event10 mouse1
2024-08-05 10:57:28,172 INFO Touchpad min-max: x 0-3987, y 0-2238
2024-08-05 10:57:28,172 INFO Numpad min-max: x 200-3787, y 200-2158
2024-08-05 10:57:28,683 INFO Setting up for config file key: "brightness" with value: "0x48"
2024-08-05 10:57:52,232 ERROR X11 load keymap listener error. Exiting
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 462, in load_keymap_listener_x11
    event = display.next_event()
            ^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages/Xlib/display.py", line 187, in next_event
    return self.display.next_event()
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages/Xlib/protocol/display.py", line 217, in next_event
    self.send_and_recv(event = True)
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages/Xlib/protocol/display.py", line 609, in send_and_recv
    raise self.socket_error
Xlib.error.ConnectionClosedError: Display connection closed by server
2024-08-05 10:57:52,235 ERROR Listening touchpad events unexpectedly failed
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 2441, in <module>
    listen_touchpad_events()
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 1958, in listen_touchpad_events
    for e in d_t.events():
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages/libevdev/device.py", line 536, in events
    ev = self._libevdev.next_event(flags)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages/libevdev/_clib.py", line 889, in next_event
    rc = self._next_event(self._ctx, flags, ctypes.byref(ev))
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 2361, in signal_handler
    raise Exception()
Exception
2024-08-05 10:57:52,239 INFO Clean up started
2024-08-05 10:57:52,260 INFO Clean up finished
2024-08-05 10:57:52,260 INFO Exiting
2024-08-05 10:58:16,132 ERROR X11 detected but not connected succesfully to the display :0. Exiting
2024-08-05 10:58:17,825 ERROR X11 detected but not connected succesfully to the display :0. Exiting
2024-08-05 10:58:19,361 ERROR X11 detected but not connected succesfully to the display :0. Exiting
2024-08-05 10:58:20,654 ERROR X11 detected but not connected succesfully to the display :0. Exiting
2024-08-05 10:58:21,911 ERROR X11 detected but not connected succesfully to the display :0. Exiting
2024-08-05 10:58:23,246 ERROR X11 detected but not connected succesfully to the display :0. Exiting
2024-08-05 10:58:24,672 ERROR X11 detected but not connected succesfully to the display :0. Exiting
2024-08-05 10:58:25,896 ERROR X11 detected but not connected succesfully to the display :0. Exiting
```

### Desktop

Manjaro Linux
Linux 6.10.0-3-MANJARO
Numpad driver v6.3.2

ldrahnik, 2024-08-05:
@nsuffys Hello, do you use `x11`? What Is the output of `echo $DISPLAY`?

nsuffys, 2024-08-05:
Hello,

I use Xorg 

```
ps -e | grep tty
    775 tty2     00:00:42 Xorg
```
Others infos :

```
echo $DISPLAY~ 
:0~
```

and

```
echo $WAYLAND_DISPLAY~  
/home/suffys
```

ldrahnik, 2024-08-05:
@nsuffys What is the output of `$ cat /usr/lib/systemd/user/asus_numberpad_driver@.service`?

nsuffys, 2024-08-05:
Hmmm

.Xauthority is not present in my home directory but is generated at boot by SSDM with a random suffix in /tmp . I found an explaination here : https://github.com/sddm/sddm/issues/1621 (but I don't know anithing about it).

```
cat /usr/lib/systemd/user/asus_numberpad_driver@.service                                                                                                            

[Unit]
Description=Asus NumberPad Driver
StartLimitBurst=20
StartLimitIntervalSec=300

[Service]
Type=simple
ExecStart=/usr/share/asus-numberpad-driver/.env/bin/python3 /usr/share/asus-numberpad-driver/numberpad.py up5401ea /usr/share/asus-numberpad-driver/
StandardOutput=append:/var/log/asus-numberpad-driver/error.log
StandardError=append:/var/log/asus-numberpad-driver/error.log
TimeoutSec=5
Restart=on-failure
RestartSec=1
Environment="DISPLAY=:0"
Environment="XAUTHORITY=/tmp/xauth_obsMtJ"
Environment="DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1000/bus"
Environment="XDG_RUNTIME_DIR=/run/user/1000"
Environment="XDG_SESSION_TYPE=x11"

[Install]
WantedBy=default.target
```

ldrahnik, 2024-08-05:
@nsuffys Is file readable for drivers service?

nsuffys, 2024-08-05:
@ldrahnik - If I'm not mistaken, no. Every time I reboot, a new file with those permissions.

`-rw-------  1 suffys suffys   98  5 août  17:37 xauth_lqngal`

ldrahnik, 2024-08-05:
@nsuffys Mine has the same permissions: 
```
ls -la | grep Xauthority
-rw-------  1 ldrahnik ldrahnik    61 Aug  5 20:17 .Xauthority
```
Problem is when is installed NumberPad driver, to the service is saved name of the file:

```
Environment="XAUTHORITY=/tmp/xauth_obsMtJ"
```

and when the driver starts, the file is not found because each boot is the xauth file renamed, e.g. to `xauth_lqngal`.

ldrahnik, 2024-08-05:
Related commit https://github.com/sddm/sddm/commit/cb5c07b4c5bafa27b21981a8f4039eadbe0fafdf

nsuffys, 2024-08-05:
@ldrahnik That's right !

I set the actual name of the "/tmp/xauth_xxxx" file in the config file "/usr/lib/systemd/user/asus_numberpad_driver@.service" and run "bash install_service.sh" and it's ok  :+1: 

What an adventure! (for a "newbie")
[image]

ldrahnik, 2024-08-05:
@nsuffys Could you please install this branch [90-not-stable-name-of-xauthority](https://github.com/asus-linux-drivers/asus-numberpad-driver/tree/190-not-stable-name-of-xauthority) and look how looks like your log file `$ cat /var/log/asus-numberpad-driver/error.log`, please?

nsuffys, 2024-08-06:
Hello,

Just tried with this branch (completely uninstalled the original) and bad news (more errors lines)

```
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/numberpad_dev wd=1 >
2024-08-06 10:04:11,153 ERROR X11 detected but not connected succesfully to the display :0. Exiting
2024-08-06 10:04:12,811 ERROR X11 detected but not connected succesfully to the display :0. Exiting
2024-08-06 10:04:14,305 ERROR X11 detected but not connected succesfully to the display :0. Exiting
2024-08-06 10:04:15,799 ERROR X11 detected but not connected succesfully to the display :0. Exiting
2024-08-06 10:04:17,324 ERROR X11 detected but not connected succesfully to the display :0. Exiting
2024-08-06 10:04:18,850 ERROR X11 detected but not connected succesfully to the display :0. Exiting
2024-08-06 10:04:20,325 ERROR X11 detected but not connected succesfully to the display :0. Exiting
2024-08-06 10:04:21,785 ERROR X11 detected but not connected succesfully to the display :0. Exiting
2024-08-06 10:04:23,247 ERROR X11 detected but not connected succesfully to the display :0. Exiting
2024-08-06 10:04:24,708 ERROR X11 detected but not connected succesfully to the display :0. Exiting
2024-08-06 10:04:25,971 ERROR X11 detected but not connected succesfully to the display :0. Exiting
2024-08-06 10:04:27,216 ERROR X11 detected but not connected succesfully to the display :0. Exiting
2024-08-06 10:04:28,465 ERROR X11 detected but not connected succesfully to the display :0. Exiting
2024-08-06 10:04:29,708 ERROR X11 detected but not connected succesfully to the display :0. Exiting
2024-08-06 10:04:30,975 ERROR X11 detected but not connected succesfully to the display :0. Exiting
2024-08-06 10:04:32,210 ERROR X11 detected but not connected succesfully to the display :0. Exiting
2024-08-06 10:04:33,454 ERROR X11 detected but not connected succesfully to the display :0. Exiting
2024-08-06 10:04:34,716 ERROR X11 detected but not connected succesfully to the display :0. Exiting
2024-08-06 10:04:35,962 ERROR X11 detected but not connected succesfully to the display :0. Exiting
2024-08-06 10:04:37,210 ERROR X11 detected but not connected succesfully to the display :0. Exiting
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 51, in <module>
    xauthority_row_with_file = subprocess.check_output(['grep', 'Xauthority'], stdin=xauth_info.stdout)
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/subprocess.py", line 466, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['grep', 'Xauthority']' returned non-zero exit status 1.
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 51, in <module>
    xauthority_row_with_file = subprocess.check_output(['grep', 'Xauthority'], stdin=xauth_info.stdout)
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/subprocess.py", line 466, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['grep', 'Xauthority']' returned non-zero exit status 1.
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 51, in <module>
    xauthority_row_with_file = subprocess.check_output(['grep', 'Xauthority'], stdin=xauth_info.stdout)
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/subprocess.py", line 466, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['grep', 'Xauthority']' returned non-zero exit status 1.
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 51, in <module>
    xauthority_row_with_file = subprocess.check_output(['grep', 'Xauthority'], stdin=xauth_info.stdout)
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/subprocess.py", line 466, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['grep', 'Xauthority']' returned non-zero exit status 1.
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 51, in <module>
    xauthority_row_with_file = subprocess.check_output(['grep', 'Xauthority'], stdin=xauth_info.stdout)
[... cut ...]
ExecStart=/usr/share/asus-numberpad-driver/.env/bin/python3 /usr/share/asus-numberpad-driver/numberpad.py up5401ea /usr/share/asus-numberpad-driver/
StandardOutput=append:/var/log/asus-numberpad-driver/error.log
StandardError=append:/var/log/asus-numberpad-driver/error.log
TimeoutSec=5
Restart=on-failure
RestartSec=1
Environment="DISPLAY=:0"
Environment="XAUTHORITY=/tmp/xauth_yyDoAM"
Environment="DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1000/bus"
Environment="XDG_RUNTIME_DIR=/run/user/1000"
Environment="XDG_SESSION_TYPE=x11"

[Install]
WantedBy=default.target
```

And my file in /tmp is "xauth_lrbZhr"

ldrahnik, 2024-08-06:
@nsuffys What is your output of this:

```
$ xauth info | grep Authority | awk '{print $3}'
/home/ldrahnik/.Xauthority
```

nsuffys, 2024-08-06:
@ldrahnik  - Ok, I've the second result

```
xauth info | grep Authority | awk '{print $3}'
/tmp/xauth_ybbNNf

```

And the error log. No more news from X11 error, but xauth search in /home directory

```
xauth:  file /home/suffys/.Xauthority does not exist
2024-08-06 11:54:25,234 INFO b'Authority file:       /home/suffys/.Xauthority\n'
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 51, in <module>
    xauthority_row_with_file = subprocess.check_output(['grep', 'Xauthority'], stdin=xauth_info.stdout)
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/subprocess.py", line 466, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['grep', 'Xauthority']' returned non-zero exit status 1.
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 51, in <module>
    xauthority_row_with_file = subprocess.check_output(['grep', 'Xauthority'], stdin=xauth_info.stdout)
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/subprocess.py", line 466, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['grep', 'Xauthority']' returned non-zero exit status 1.
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 51, in <module>
    xauthority_row_with_file = subprocess.check_output(['grep', 'Xauthority'], stdin=xauth_info.stdout)
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/subprocess.py", line 466, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['grep', 'Xauthority']' returned non-zero exit status 1.
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 51, in <module>
    xauthority_row_with_file = subprocess.check_output(['grep', 'Xauthority'], stdin=xauth_info.stdout)
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/subprocess.py", line 466, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['grep', 'Xauthority']' returned non-zero exit status 1.
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 51, in <module>
    xauthority_row_with_file = subprocess.check_output(['grep', 'Xauthority'], stdin=xauth_info.stdout)
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/subprocess.py", line 466, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['grep', 'Xauthority']' returned non-zero exit status 1.
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 51, in <module>
    xauthority_row_with_file = subprocess.check_output(['grep', 'Xauthority'], stdin=xauth_info.stdout)
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/subprocess.py", line 466, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... 120 lines cut ...]
  File "/usr/lib/python3.12/subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['grep', 'Xauthority']' returned non-zero exit status 1.
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 51, in <module>
    xauthority_row_with_file = subprocess.check_output(['grep', 'Xauthority'], stdin=xauth_info.stdout)
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/subprocess.py", line 466, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
[... cut ...]
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/subprocess.py", line 466, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['grep', 'Xauthority']' returned non-zero exit status 1.

```

nsuffys, 2024-08-06:
@ldrahnik and I found this second file after a reboot
[image]

ldrahnik, 2024-08-06:
@nsuffys Looks like the tool `$ xauth` finds the xauthority file in the directory defined by `$HOME` but the name is probably hardcoded `.XAuthority` so now I tried to use `glob`, please try the latest version of mentioned branch.

nsuffys, 2024-08-06:
@ldrahnik It's now OK. I restarted three times and everything works. Power on, power off, numeric keys all is OK. There are errors in the log but it doesn't affect anything. systemd log looks good.

[image]

```
2024-08-06 18:21:01,222 ERROR X11 load keymap listener error. Exiting
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 479, in load_keymap_listener_x11
    event = display.next_event()
            ^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages/Xlib/display.py", line 187, in next_event
    return self.display.next_event()
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages/Xlib/protocol/display.py", line 217, in next_event
    self.send_and_recv(event = True)
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages/Xlib/protocol/display.py", line 609, in send_and_recv
    raise self.socket_error
Xlib.error.ConnectionClosedError: Display connection closed by server
2024-08-06 18:21:01,224 ERROR Listening touchpad events unexpectedly failed
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 2458, in <module>
    listen_touchpad_events()
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 1975, in listen_touchpad_events
    for e in d_t.events():
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages/libevdev/device.py", line 542, in events
    ev = self._libevdev.next_event(flags)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages/libevdev/_clib.py", line 889, in next_event
    rc = self._next_event(self._ctx, flags, ctypes.byref(ev))
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 2378, in signal_handler
    raise Exception()
Exception
2024-08-06 18:21:01,226 INFO Clean up started
2024-08-06 18:21:01,258 INFO Clean up finished
2024-08-06 18:21:01,258 INFO Exiting
2024-08-06 18:21:21,272 INFO X11 has xauth file in /tmp folder (filename is changed each reboot) /tmp/xauth_EwWibX
2024-08-06 18:21:21,284 INFO X11 detected and connected succesfully to the display :0
2024-08-06 18:21:21,285 INFO Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
2024-08-06 18:21:21,286 INFO Set keyboard 3 from H: Handlers=sysrq kbd leds event3
2024-08-06 18:21:21,331 INFO Detecting touchpad from string: "N: Name="ASUE140D:00 04F3:31B9 Touchpad""
2024-08-06 18:21:21,331 INFO Set touchpad device id 15 from S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-15/i2c-ASUE140D:00/0018:04F3:31B9.0001/input/input13
2024-08-06 18:21:21,332 INFO Set touchpad id 10 from H: Handlers=event10 mouse1
2024-08-06 18:21:21,332 INFO Touchpad min-max: x 0-3987, y 0-2238
2024-08-06 18:21:21,332 INFO Numpad min-max: x 200-3787, y 200-2158
2024-08-06 18:21:34,885 INFO Touched top_right_icon area (numlock) in time: 1722961294.8859184
2024-08-06 18:21:35,888 INFO The numpad numlock was pressed longer than the activation time and will be activated: 1.002321720123291
2024-08-06 18:21:35,888 INFO Activation time: 1.0
2024-08-06 18:21:35,935 INFO System numlock activated
2024-08-06 18:21:35,935 INFO Numpad activated
2024-08-06 18:21:35,944 INFO Old device at /dev/input/event16 (/sys/devices/virtual/input/input19)
2024-08-06 18:21:35,946 INFO New device at /dev/input/event18 (/sys/devices/virtual/input/input23)
2024-08-06 18:21:35,955 INFO Setting up for config file key: "enabled" with value: "True"
2024-08-06 18:21:35,955 INFO check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/numberpad_dev wd=1 >
2024-08-06 18:21:36,271 INFO Un-touched top_right_icon area (representing numlock key) in time: 1722961296.2710915
2024-08-06 18:21:37,327 INFO Pressed numpad key
2024-08-06 18:21:37,328 INFO [KEY_LEFTSHIFT:42, KEY_0:11]
2024-08-06 18:21:37,328 INFO Unpressed numpad key
2024-08-06 18:21:37,328 INFO [KEY_LEFTSHIFT:42, KEY_0:11]
2024-08-06 18:21:37,730 INFO Pressed numpad key
2024-08-06 18:21:37,730 INFO [KEY_LEFTSHIFT:42, KEY_1:2]
2024-08-06 18:21:37,730 INFO Unpressed numpad key
2024-08-06 18:21:37,730 INFO [KEY_LEFTSHIFT:42, KEY_1:2]
2024-08-06 18:21:38,126 INFO Pressed numpad key
2024-08-06 18:21:38,126 INFO [KEY_LEFTSHIFT:42, KEY_2:3]
2024-08-06 18:21:38,126 INFO Unpressed numpad key
2024-08-06 18:21:38,126 INFO [KEY_LEFTSHIFT:42, KEY_2:3]
2024-08-06 18:21:38,480 INFO Pressed numpad key
2024-08-06 18:21:38,481 INFO [KEY_LEFTSHIFT:42, KEY_3:4]
[... 30 lines cut ...]
2024-08-06 18:21:45,832 INFO Numpad deactivated
2024-08-06 18:21:45,854 INFO Setting up for config file key: "enabled" with value: "False"
2024-08-06 18:21:45,855 INFO check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/numberpad_dev wd=1 >
2024-08-06 18:21:46,081 INFO Un-touched top_right_icon area (representing numlock key) in time: 1722961306.0810864
2024-08-06 18:23:06,490 INFO Touched top_right_icon area (numlock) in time: 1722961386.490351
[... cut ...]
2024-08-06 18:23:12,935 INFO Setting up for config file key: "enabled" with value: "False"
2024-08-06 18:23:12,935 INFO check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/numberpad_dev wd=1 >
2024-08-06 18:23:13,253 INFO Un-touched top_right_icon area (representing numlock key) in time: 1722961393.2531543

```

ldrahnik, 2024-08-06:
It was merged into the master branch. Can be reopened.

closed 2024-08-06 by ldrahnik