asus-linux-drivers/asus-numberpad-driver #192: [Bug]: The detection was not successful. Touchpad with NumberPad not found

PrincessMagicalLove, 2024-08-08:
### Describe the bug

After trying to figure out what was happening, I decided to seek for some help. 
I already searched for past issues that might help me with this, but none of them seem to give me an answer to my problem.
I'm really sorry if this is something obvious, i just want this to work properly.
[DSDT.txt](https://github.com/user-attachments/files/16537393/DSDT.txt)

Also here is the output of the command i2cdetect -l

i2c-0	unknown   	Synopsys DesignWare I2C adapter 	N/A
i2c-1	unknown   	AMDGPU DM i2c hw bus 0          	N/A
i2c-2	unknown   	AMDGPU DM i2c hw bus 1          	N/A
i2c-3	unknown   	AMDGPU DM aux hw bus 0          	N/A
i2c-4	unknown   	SMBus PIIX4 adapter port 0 at 0b00	N/A
i2c-5	unknown   	SMBus PIIX4 adapter port 2 at 0b00	N/A
i2c-6	unknown   	SMBus PIIX4 adapter port 1 at 0b20	N/A

### Expected behavior

Numberpad working

### Relevant log output

```shell
groupadd: group 'numberpad' already exists
Added group numberpad to current user

Last metadata expiration check: 0:44:55 ago on Wed 07 Aug 2024 08:55:51 PM CST.
Package ibus-1.5.30-5.fc40.x86_64 is already installed.
Package libevdev-1.13.2-2.fc40.x86_64 is already installed.
Package curl-8.6.0-10.fc40.x86_64 is already installed.
Package xinput-1.6.4-2.fc40.x86_64 is already installed.
Package i2c-tools-4.3-10.fc40.x86_64 is already installed.
Package python3-devel-3.12.4-1.fc40.x86_64 is already installed.
Package python3-virtualenv-20.21.1-16.fc40.noarch is already installed.
Package libxml2-2.12.8-1.fc40.x86_64 is already installed.
Package libxkbcommon-devel-1.6.0-2.fc40.x86_64 is already installed.
Package gcc-14.2.1-1.fc40.x86_64 is already installed.
Dependencies resolved.
Nothing to do.
Complete!

groupadd: group 'input' already exists
groupadd: group 'i2c' already exists
groupadd: group 'uinput' already exists
Added groups input, i2c, uinput, numberpad to current user
uinput module loaded
i2c-dev module loaded
uinput module added to auto loaded modules
Udev rules reloaded and triggered

Testing interface i2c-0: failed
The detection was not successful. Touchpad with NumberPad not found
```

### Desktop

-OS: Fedora Linux 40 (Workstation Edition) x86_64 
- Host: Vivobook Go E1404FA_E1404FA 1.0

ldrahnik, 2024-08-08:
@PrincessMagicalLove Hi, no problem. Are you sure your model has touchpad with NumberPad? Was activated on Windows? Could you send me photo from the internet where Is NumberPad on your model activated?

ldrahnik, 2024-08-08:
@PrincessMagicalLove What is the output of `/proc/bus/input/devices`?

ldrahnik, 2024-08-08:
@PrincessMagicalLove You should see something like this:

```
$ i2cdetect -l
i2c-0	smbus     	SMBus I801 adapter at efa0      	SMBus adapter
i2c-1	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-2	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-3	i2c       	i915 gmbus dpa                  	I2C adapter
i2c-4	i2c       	i915 gmbus dpb                  	I2C adapter
i2c-5	i2c       	i915 gmbus dpc                  	I2C adapter
i2c-6	i2c       	i915 gmbus tc1                  	I2C adapter
i2c-7	i2c       	i915 gmbus tc2                  	I2C adapter
i2c-8	i2c       	i915 gmbus tc3                  	I2C adapter
i2c-9	i2c       	i915 gmbus tc4                  	I2C adapter
i2c-10	i2c       	i915 gmbus tc5                  	I2C adapter
i2c-11	i2c       	i915 gmbus tc6                  	I2C adapter
i2c-12	i2c       	AUX A/DDI A/PHY A               	I2C adapter
i2c-13	i2c       	AUX USBC1/DDI TC1/PHY TC1       	I2C adapter
i2c-14	i2c       	AUX USBC2/DDI TC2/PHY TC2       	I2C adapte
```

you have a lot of `unknown` or `N/A`:

```
 $ i2cdetect -l

i2c-0 unknown Synopsys DesignWare I2C adapter N/A
i2c-1 unknown AMDGPU DM i2c hw bus 0 N/A
i2c-2 unknown AMDGPU DM i2c hw bus 1 N/A
i2c-3 unknown AMDGPU DM aux hw bus 0 N/A
i2c-4 unknown SMBus PIIX4 adapter port 0 at 0b00 N/A
i2c-5 unknown SMBus PIIX4 adapter port 2 at 0b00 N/A
i2c-6 unknown SMBus PIIX4 adapter port 1 at 0b20 N/A
```

Try to fix this at first.

PrincessMagicalLove, 2024-08-09:
@ldrahnik omg, acutally fixing the unknown and N/A made the driver instalation work, thank you so much, I didn't realize that I was supposed to see a completely different output from i2cdetect -l

closed 2024-08-09 by PrincessMagicalLove

ldrahnik, 2024-08-09:
@PrincessMagicalLove Nice. And reason behind the `UNKNOWN` and `N/A`?