asus-linux-drivers/asus-numberpad-driver #194: [Layout Request] UM3402 ZenBook 14  - UM3402YAR-KN473W not detected, just bought new.

neocogent, 2024-08-14:
### Layout Name

UM3402

### Your device (cat /sys/devices/virtual/dmi/id/product_name)

Zenbook UM3402YAR

### Supported Devices

Looks like maybe: up5401ea, but not detected.

### Numpad Layout (photo/picture or code)

[image]

### Additional Info

Just downloaded driver and ran install.sh script. Output was,

```
groupadd: group 'input' already exists
groupadd: group 'i2c' already exists
Added groups input, i2c, uinput, numberpad to current user
uinput module loaded
i2c-dev module loaded
uinput module added to auto loaded modules
Udev rules reloaded and triggered

Testing interface i2c-0: failed
Testing interface i2c-1: failed
The detection was not successful. Touchpad with NumberPad not found
```
Came here and saw some info, tried i2cdetect -l, output is:

`~/Downloads/asus-numberpad-driver-6.3.2$ i2cdetect -l
i2c-0	unknown   	Synopsys DesignWare I2C adapter 	N/A
i2c-1	unknown   	Synopsys DesignWare I2C adapter 	N/A
i2c-2	unknown   	SMBus PIIX4 adapter port 0 at 0b00	N/A
i2c-3	unknown   	SMBus PIIX4 adapter port 2 at 0b00	N/A
i2c-4	unknown   	SMBus PIIX4 adapter port 1 at 0b20	N/A
i2c-5	unknown   	AMDGPU DM i2c hw bus 0          	N/A
i2c-6	unknown   	AMDGPU DM i2c hw bus 1          	N/A
i2c-7	unknown   	AMDGPU DM i2c hw bus 2          	N/A
i2c-8	unknown   	AMDGPU DM i2c hw bus 3          	N/A
i2c-9	unknown   	AMDGPU DM aux hw bus 0          	N/A
i2c-10	unknown   	AMDGPU DM aux hw bus 2          	N/A
i2c-11	unknown   	AMDGPU DM aux hw bus 3          	N/A
`

Saw past issue with similar output and they fixed it but I have no idea how to make "unknown" items into known. Saw past issue with UM3402 but was incomplete. Maybe I can help complete layout? Looks to me like same as **up5401ea** but is not detected as such. 

I welcome any help getting this driver to work please. Thx.

### Acknowledgements

- [X] This layout doesn't currently exist and I have read the [Layout section](https://github.com/asus-linux-drivers/asus-numberpad-driver?tab=readme-ov-file#layouts)

ldrahnik, 2024-08-14:
@neocogent The problem is with i2c. Not with missing layout. I mean with mentioned `unknown`  fields. I have not replicated the same problem, unfortunatelly. May be associated with newest package versions for handling i2c communication.

neocogent, 2024-08-14:
Is there anything I can do that may help with the i2c problem? 

Can convert this issue to more general i2c one rather than a layout request.

neocogent, 2024-08-14:
I gather from my i2cdetect listing that the touch pad must be one of the first two, since the others have names that surely are not and Synaptic makes touch pads. So maybe I can try the driver by assigning i2c-0 and i2c-1 in turn and see if either one works? Sounds possible but I don't knwo the steps to do that.

ldrahnik, 2024-08-14:
@neocogent Was your laptop at least one time restarted since you tried the first installation?

ldrahnik, 2024-08-14:
@neocogent What is the output with `sudo i2cdetect -l`?

ldrahnik, 2024-08-14:
@neocogent Please, post here entire output of `$ cat /proc/bus/input/devices`

ldrahnik, 2024-08-14:
@neocogent 

> I gather from my i2cdetect listing that the touch pad must be one of the first two, since the others have names that surely are not and Synopsys makes touch pads. So maybe I can try the driver by assigning i2c-0 and i2c-1 in turn and see if either one works? Sounds possible but I don't know the steps to do that.

Try to run the driver directly as `$ python3 /usr/share/asus-numberpad-driver/numberpad.py 'up5401ea'`.

neocogent, 2024-08-14:
Thank you for your time and help! 

> @neocogent Was your laptop at least one time restarted since you tried the first installation of this driver?

No. Not 100% sure but I think have been only suspend/resume cycles. You know Linux - who ever needs to reboot... 

> @neocogent What is the output with sudo $ sudo i2cdetect -l?

```
i2c-0	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-1	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-2	smbus     	SMBus PIIX4 adapter port 0 at 0b00	SMBus adapter
i2c-3	smbus     	SMBus PIIX4 adapter port 2 at 0b00	SMBus adapter
i2c-4	smbus     	SMBus PIIX4 adapter port 1 at 0b20	SMBus adapter
i2c-5	i2c       	AMDGPU DM i2c hw bus 0          	I2C adapter
i2c-6	i2c       	AMDGPU DM i2c hw bus 1          	I2C adapter
i2c-7	i2c       	AMDGPU DM i2c hw bus 2          	I2C adapter
i2c-8	i2c       	AMDGPU DM i2c hw bus 3          	I2C adapter
i2c-9	i2c       	AMDGPU DM aux hw bus 0          	I2C adapter
i2c-10	i2c       	AMDGPU DM aux hw bus 2          	I2C adapter
i2c-11	i2c       	AMDGPU DM aux hw bus 3          	I2C adapter
```

> cat /proc/bus/input/devices

```
I: Bus=0019 Vendor=0000 Product=0005 Version=0000
N: Name="Lid Switch"
P: Phys=PNP0C0D/button/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:31/PNP0C09:00/PNP0C0D:00/input/input0
U: Uniq=
H: Handlers=event0 
B: PROP=0
B: EV=21
B: SW=1

I: Bus=0019 Vendor=0000 Product=0001 Version=0000
N: Name="Power Button"
P: Phys=PNP0C0C/button/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0C:00/input/input1
U: Uniq=
H: Handlers=kbd event1 
B: PROP=0
B: EV=3
B: KEY=8000 10000000000000 0

I: Bus=0011 Vendor=0001 Product=0001 Version=ab83
N: Name="AT Translated Set 2 keyboard"
P: Phys=isa0060/serio0/input0
S: Sysfs=/devices/platform/i8042/serio0/input/input2
U: Uniq=
H: Handlers=sysrq kbd event2 leds 
B: PROP=0
B: EV=120013
B: KEY=402000000 3803078f800d001 feffffdfffefffff fffffffffffffffe
B: MSC=10
B: LED=7

I: Bus=0019 Vendor=0000 Product=0006 Version=0000
N: Name="Video Bus"
P: Phys=LNXVIDEO/video/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:11/LNXVIDEO:00/input/input3
U: Uniq=
H: Handlers=kbd event3 
B: PROP=0
B: EV=3
B: KEY=3e000b00000000 0 0 0

I: Bus=0018 Vendor=04f3 Product=2f2c Version=0100
N: Name="ELAN9008:00 04F3:2F2C"
[... 219 lines cut ...]
pip install xkbcommon
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.12/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.

```

but I guess python is being run in some **venv** type thing which I'm not very familiar with. I tried apt install python3-xkbcommon but that is not found.Also tried **sudo pip** for above and got same result.

ldrahnik, 2024-08-14:
@neocogent What is the output of `$ type i2cdetect`?

ldrahnik, 2024-08-14:
@neocogent Please, try to install the latest master.

ldrahnik, 2024-08-14:
Can be reopened.

closed 2024-08-14 by ldrahnik

neocogent, 2024-08-14:
Yes! Install script completed and rebooted.

Num Pad works great (after I figured out have to keep finger on symbol for a bit). 

Thank you so much for helping with this and even more so for taking the time to make a driver in the first place. It remained the one thing that did not work out-of-box for Mint 22 Linux on this new (to me) laptop.