asus-linux-drivers/asus-numberpad-driver #195: Calculator open multiple instances

encrustace, 2024-08-17:
### Short Description

If the calculator is already opened, it open another one.

Expected behaviour, it should toggle the app, if calculator already opened then close it and if closed then open it.

ldrahnik, 2024-08-17:
@encrustace Hi, this driver send Calc key and associated script for Calc key do toggling. I need detailed description which calculator and use case step by step.

encrustace, 2024-08-17:
System: Arch Linux with Gnome Desktop
Used install.sh script to install.
I am using default script provided by this setup.
Using gnome calculator from flatpak.

ldrahnik, 2024-08-17:
@encrustace And this script in your environment do not do toggling? https://github.com/asus-linux-drivers/asus-numberpad-driver/blob/master/scripts/gnome_calculator_toggle.sh

encrustace, 2024-08-17:
@ldrahnik  It does work when I installed gnome-calculator from pacman library, but was not working with flatpak one, I guess it does not have support of flatpak version, I can close the issue, or can open to track if you plan to give support of flatpak gnome app.

closed 2024-08-17 by encrustace

ldrahnik, 2024-08-18:
@encrustace Try the latest master branch.

encrustace, 2024-08-18:
@ldrahnik  I tried but there was an issue, I have created a [PR](https://github.com/asus-linux-drivers/asus-numberpad-driver/pull/197)
Please have a look.

ldrahnik, 2024-08-18:
@encrustace Please, try the latest master. In case still does not work, reopen PR, I will check it and merge it. (sorry, I was in a hurry and committed not working version)

encrustace, 2024-08-18:
@ldrahnik  Yes, now it is working like butter, Thanks