asus-linux-drivers/asus-numberpad-driver #196: [Bug]: Key repetitions is not working

akaibukai, 2024-08-18:
### Describe the bug

I noticed that the taps on the numpad are handled with a noticeable (and annoying) delay/latency..
Having the key repeitions disabled by default does not help for example when I want to delete a bunch of numbers/characters..
So I wanted at least to have repetitions so I did setup in `/usr/share/asus-numberpad-driver/numberpad_dev`, the folloing line:
```toml
key_repetitions = 1
```

But it doesn't work (even after a reboot).
Other changes done in this file is working however, like the `activation_time` (which is easy to spot if the changes are taken into account for example)

### Expected behavior

Holding the tap on a key should repeat the key.
I don't see myslef taping many same numbers in a row, but I wanted this feature particularly for the backspace.

### Relevant log output

```shell
N/A
```

### Desktop

- Program version: 56d817b (latest at the time of writing)
- Distro: Fedora
- Kernel: 6.10.4-100.fc39.x86_64

ldrahnik, 2024-08-18:
@akaibukai Hi, thank you for creating an issue. I think you missed the info (*EDIT*: doc is missing, sorry) that keys are sent when the finger is released (`press_key_when_is_done_untouch`) from the spot and the reason why is this behaviour by default is because than can be achieved movement with pointer (mouse equivalent) without sending any character - only when is movement longer than defined width (`distance_to_move_only_pointer`) or reached grid column / row line.

ldrahnik, 2024-08-18:
@akaibukai Can be reopened.

closed 2024-08-18 by ldrahnik

akaibukai, 2024-08-18:
Hi @ldrahnik!
Thank you very much for your quick response!
Indeed, setting up `press_key_when_is_done_untouch = 0` did the trick!
Not only the repetition works but also it's way faster!

PS: I was already disabling the touchpad with `enabled_touchpad_pointer = 0`..
Then I wanted to try with the option set back to 3 to allow movement..
But it seems having `press_key_when_is_done_untouch = 0` and  `enabled_touchpad_pointer = 3` is just breaking altogether the touchpad even after the numpad is disabled (no lights)

So I set back to `enabled_touchpad_pointer = 0` because it's the setting I want anyway..
I just wanted to document in case it is a bug (or maybe I just did soemthing wrong)..

In any case, thank you again and for sure you can keep the issue closed!

Have a nice day!