asus-linux-drivers/asus-numberpad-driver #198: [Bug]: Won't work in ROG Zephyrus du0 16(GX650PY)

fjordanrv, 2024-09-11:
### Describe the bug

Hi I follow all the steps but the numpad doesn't work, can you confirm me that this driver is working for my laptop model?

### Expected behavior

The numpad should appear but is now working.

### Desktop

- Distro Fedora 40
- Gnome 46
- Kernel (Linux 6.10.8-200.fc40.x86_64 x86_64)

ldrahnik, 2024-09-11:
@fjordanrv Hi, I need to see driver logs located `var/logs` and the output from the installation.

fjordanrv, 2024-09-12:
Hi, this is the error log.

2024-09-11 13:03:34,429 INFO Detecting touchpad from string: "N: Name="ELAN9009:00 04F3:41D9 Touchpad""
2024-09-11 13:03:34,429 INFO Set touchpad device id 1 from S: Sysfs=/devices/platform/AMDI0010:01/i2c-1/i2c-ELAN9009:00/0018:04F3:41D9.0002/input/input12
2024-09-11 13:03:34,429 INFO Set touchpad id 7 from H: Handlers=mouse2 event7
2024-09-11 13:03:34,429 INFO Detecting keyboard from string: "N: Name="ASUE140F:00 04F3:32A2 Keyboard""
2024-09-11 13:03:34,429 INFO Set keyboard 10 from H: Handlers=sysrq kbd leds event10
2024-09-11 13:03:34,454 INFO Touchpad min-max: x 0-3984, y 0-1152
2024-09-11 13:03:34,454 INFO Numpad min-max: x 20-3964, y 0-1152
2024-09-11 13:03:34,455 ERROR Wayland load keymap listener error. Exiting
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 448, in load_keymap_listener_wayland
    display_wayland.connect()
  File "/usr/share/asus-numberpad-driver/.env/lib64/python3.12/site-packages/pywayland/client/display.py", line 131, in connect
    raise ValueError("Unable to connect to display")
ValueError: Unable to connect to display
2024-09-11 13:03:34,456 ERROR Listening touchpad events unexpectedly failed
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 2390, in <module>
    sleep(0.5)
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 2366, in signal_handler
    raise Exception()
Exception
2024-09-11 13:03:34,456 INFO Clean up started
2024-09-11 13:03:34,478 INFO Clean up finished
2024-09-11 13:03:34,478 INFO Exiting
2024-09-11 13:03:35,671 INFO Detecting touchpad from string: "N: Name="ELAN9009:00 04F3:41D9 Touchpad""
2024-09-11 13:03:35,672 INFO Set touchpad device id 1 from S: Sysfs=/devices/platform/AMDI0010:01/i2c-1/i2c-ELAN9009:00/0018:04F3:41D9.0002/input/input12
2024-09-11 13:03:35,672 INFO Set touchpad id 7 from H: Handlers=mouse2 event7
2024-09-11 13:03:35,672 INFO Detecting keyboard from string: "N: Name="ASUE140F:00 04F3:32A2 Keyboard""
2024-09-11 13:03:35,672 INFO Set keyboard 10 from H: Handlers=sysrq kbd leds event10
2024-09-11 13:03:35,691 INFO Touchpad min-max: x 0-3984, y 0-1152
2024-09-11 13:03:35,692 INFO Numpad min-max: x 20-3964, y 0-1152
2024-09-11 13:03:37,194 INFO Setting up for config file key: "top_left_icon_brightness_func_disabled" with value: "True"
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ELAN9009:00 04F3:41D9 Touchpad
2024-09-11 13:03:37,936 ERROR Error during sending via i2c: "[Errno 121] Remote I/O error"
2024-09-11 13:03:37,936 ERROR Error during sending via i2c: "[Errno 121] Remote I/O error"
2024-09-11 13:03:37,936 INFO Setting up for config file key: "enabled" with value: "True"
2024-09-11 13:03:37,936 INFO check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/numberpad_dev wd=1 >
2024-09-11 13:03:38,036 INFO Numpad activated
2024-09-11 13:06:10,231 INFO check_config_values_changes: detected external change of config file -> loading changes
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad.py path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/numberpad.py wd=1 >
2024-09-11 13:06:17,715 INFO Detecting touchpad from string: "N: Name="ELAN9009:00 04F3:41D9 Touchpad""
2024-09-11 13:06:17,715 INFO Set touchpad device id 1 from S: Sysfs=/devices/platform/AMDI0010:01/i2c-1/i2c-ELAN9009:00/0018:04F3:41D9.0002/input/input12
2024-09-11 13:06:17,715 INFO Set touchpad id 7 from H: Handlers=mouse2 event7
2024-09-11 13:06:17,715 INFO Detecting keyboard from string: "N: Name="ASUE140F:00 04F3:32A2 Keyboard""
2024-09-11 13:06:17,715 INFO Set keyboard 10 from H: Handlers=sysrq kbd leds event10
2024-09-11 13:06:17,726 INFO Touchpad min-max: x 0-3984, y 0-1152
2024-09-11 13:06:17,726 INFO Numpad min-max: x 20-3964, y 0-1152
2024-09-11 13:06:18,729 INFO Setting up for config file key: "top_left_icon_brightness_func_disabled" with value: "True"
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ELAN9009:00 04F3:41D9 Touchpad
2024-09-11 13:06:19,259 ERROR Error during sending via i2c: "[Errno 121] Remote I/O error"
2024-09-11 13:06:19,259 ERROR Error during sending via i2c: "[Errno 121] Remote I/O error"
2024-09-11 13:06:19,259 INFO Setting up for config file key: "enabled" with value: "True"
2024-09-11 13:06:19,259 INFO check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/numberpad_dev wd=1 >
2024-09-11 13:06:19,359 INFO Numpad activated
2024-09-11 13:06:21,133 INFO check_config_values_changes: detected external change of config file -> loading changes
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=sedsrUOFW path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/sedsrUOFW wd=1 >
2024-09-11 13:06:21,234 INFO check_config_values_changes: detected external change of config file -> loading changes
[... cut ...]
2024-09-11 13:07:14,277 ERROR Error during sending via i2c: "[Errno 121] Remote I/O error"
2024-09-11 13:07:14,277 INFO Setting up for config file key: "enabled" with value: "True"
2024-09-11 13:07:14,277 INFO check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/numberpad_dev wd=1 >
2024-09-11 13:07:14,377 INFO Numpad activated

fjordanrv, 2024-09-12:
And the output of the installation:

Última comprobación de caducidad de metadatos hecha hace 1:02:58, el mié 11 sep 2024 12:03:11.
El paquete ibus-1.5.30-6.fc40.x86_64 ya está instalado.
El paquete libevdev-1.13.3-1.fc40.x86_64 ya está instalado.
El paquete curl-8.6.0-10.fc40.x86_64 ya está instalado.
El paquete xinput-1.6.4-2.fc40.x86_64 ya está instalado.
El paquete i2c-tools-4.3-10.fc40.x86_64 ya está instalado.
El paquete python3-devel-3.12.5-2.fc40.x86_64 ya está instalado.
El paquete python3-virtualenv-20.21.1-16.fc40.noarch ya está instalado.
El paquete libxml2-2.12.8-1.fc40.x86_64 ya está instalado.
El paquete libxkbcommon-devel-1.6.0-2.fc40.x86_64 ya está instalado.
El paquete gcc-14.2.1-1.fc40.x86_64 ya está instalado.
Dependencias resueltas.
Nada por hacer.
¡Listo!

groupadd: el grupo «input» ya existe
groupadd: el grupo «i2c» ya existe
groupadd: el grupo «uinput» ya existe
Added groups input, i2c, uinput, numberpad to current user
uinput module loaded
i2c-dev module loaded
uinput module added to auto loaded modules
Udev rules reloaded and triggered

Testing interface i2c-0: success (adr 0x15)
The detection was successful. Touchpad with NumberPad found: i2c-0

In system remains config file from previous installation. Do you want replace that config with default config? [y/N]
created virtual environment CPython3.12.5.final.0-64 in 214ms
  creator CPython3Posix(dest=/usr/share/asus-numberpad-driver/.env, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(extra_search_dir=/usr/share/python-wheels,download=False, pip=bundle, via=copy, app_data_dir=/home/fedorafj/.local/share/virtualenv)
    added seed packages: libevdev==0.11, pip==23.3.2, pyasyncore==1.0.4, pycparser==2.22, pyinotify==0.9.6, python_xlib==0.33, setuptools==74.1.2, six==1.16.0, smbus2==0.4.3
  activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
Requirement already satisfied: pip in /usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages (23.3.2)
Collecting pip
  Using cached pip-24.2-py3-none-any.whl.metadata (3.6 kB)
Using cached pip-24.2-py3-none-any.whl (1.8 MB)
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 23.3.2
    Uninstalling pip-23.3.2:
      Successfully uninstalled pip-23.3.2
Successfully installed pip-24.2
Requirement already satisfied: setuptools in /usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages (74.1.2)
Requirement already satisfied: libevdev in /usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages (from -r requirements.txt (line 1)) (0.11)
Requirement already satisfied: numpy in /usr/share/asus-numberpad-driver/.env/lib64/python3.12/site-packages (from -r requirements.txt (line 2)) (2.1.1)
Requirement already satisfied: pyinotify in /usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages (from -r requirements.txt (line 3)) (0.9.6)
Requirement already satisfied: python-xlib in /usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages (from -r requirements.txt (line 4)) (0.33)
Requirement already satisfied: smbus2 in /usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages (from -r requirements.txt (line 5)) (0.4.3)
Requirement already satisfied: pyasyncore in /usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages (from -r requirements.txt (line 6)) (1.0.4)
Requirement already satisfied: pywayland in /usr/share/asus-numberpad-driver/.env/lib64/python3.12/site-packages (from -r requirements.txt (line 7)) (0.4.18)
Requirement already satisfied: xkbcommon<1.1 in /usr/share/asus-numberpad-driver/.env/lib64/python3.12/site-packages (from -r requirements.txt (line 8)) (1.0.1)
Requirement already satisfied: six>=1.10.0 in /usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages (from python-xlib->-r requirements.txt (line 4)) (1.16.0)
Requirement already satisfied: cffi>=1.12.0 in /usr/share/asus-numberpad-driver/.env/lib64/python3.12/site-packages (from pywayland->-r requirements.txt (line 7)) (1.17.1)
Requirement already satisfied: pycparser in /usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages (from cffi>=1.12.0->pywayland->-r requirements.txt (line 7)) (2.22)

NumberPad layout

The automatically recommended NumberPad layout for this laptop (ROG Zephyrus Duo 16 GX650PY_GX650PY) is gx551. Do you want to use the gx551 layout? (The photo of the recommended NumberPad layout can be found here https://github.com/asus-linux-drivers/asus-numberpad-driver#gx551) [y/N]
Selected key layout: gx551

Systemctl service

Do you want install systemctl service? [y/N]
LAYOUT_NAME: gx551
CONFIG_FILE_DIR_PATH: /usr/share/asus-numberpad-driver

env var DISPLAY: :0
[... 24 lines cut ...]

Do you want install the rule for external keyboard? [y/N]
Rule 90-numberpad-external-keyboard.rules applied
Udev rules reloaded and triggered

Calculator app

Do you want try to install toggling script for XF86Calculator key? [y/N]
Detected gnome-calculator
Setting up for gnome-calculator
Toggling script for calculator app gnome-calculator has been installed.

Power supply saver

By default is idle functionality disabled and may be manually enabled in config file later. Or by installing this rule can be enabled by every detected change of power supply mode to battery mode.

Idle mode is configured to be enabled after 10s of inactivity and to decrease 30% of brightness.

Do you want install the rule for idle functionality? [y/N]
Rule 80-numberpad-power-supply applied
Udev rules reloaded and triggered

Installation finished succesfully

Reboot is required. Do you want reboot now? [y/N]

fjordanrv, 2024-09-12:
I: Bus=0019 Vendor=0000 Product=0005 Version=0000
N: Name="Lid Switch"
P: Phys=PNP0C0D/button/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:3f/PNP0C09:00/PNP0C0D:00/input/input0
U: Uniq=
H: Handlers=event0 
B: PROP=0
B: EV=21
B: SW=1

I: Bus=0019 Vendor=0000 Product=0001 Version=0000
N: Name="Power Button"
P: Phys=PNP0C0C/button/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0C:00/input/input1
U: Uniq=
H: Handlers=kbd event1 
B: PROP=0
B: EV=3
B: KEY=8000 10000000000000 0

I: Bus=0019 Vendor=0000 Product=0003 Version=0000
N: Name="Sleep Button"
P: Phys=PNP0C0E/button/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0E:00/input/input2
U: Uniq=
H: Handlers=kbd event2 
B: PROP=0
B: EV=3
B: KEY=4000 0 0

I: Bus=0018 Vendor=04f3 Product=41d9 Version=0100
N: Name="ELAN9009:00 04F3:41D9"
P: Phys=i2c-ELAN9009:00
S: Sysfs=/devices/platform/AMDI0010:01/i2c-1/i2c-ELAN9009:00/0018:04F3:41D9.0002/input/input6
U: Uniq=
H: Handlers=mouse0 event3 
B: PROP=2
B: EV=1b
B: KEY=400 0 0 0 0 0
B: ABS=3273800000000003
B: MSC=20

I: Bus=0018 Vendor=04f3 Product=41d9 Version=0100
N: Name="ELAN9009:00 04F3:41D9 UNKNOWN"
P: Phys=i2c-ELAN9009:00
S: Sysfs=/devices/platform/AMDI0010:01/i2c-1/i2c-ELAN9009:00/0018:04F3:41D9.0002/input/input7
U: Uniq=
H: Handlers=event4 
B: PROP=0
B: EV=9
B: ABS=10000000000

I: Bus=0018 Vendor=04f3 Product=41d9 Version=0100
N: Name="ELAN9009:00 04F3:41D9 UNKNOWN"
P: Phys=i2c-ELAN9009:00
S: Sysfs=/devices/platform/AMDI0010:01/i2c-1/i2c-ELAN9009:00/0018:04F3:41D9.0002/input/input8
U: Uniq=
H: Handlers=event5 
B: PROP=0
B: EV=100001

I: Bus=0018 Vendor=04f3 Product=41d9 Version=0100
N: Name="ELAN9009:00 04F3:41D9 Stylus"
P: Phys=i2c-ELAN9009:00
S: Sysfs=/devices/platform/AMDI0010:01/i2c-1/i2c-ELAN9009:00/0018:04F3:41D9.0002/input/input9
U: Uniq=
H: Handlers=mouse1 event6 
B: PROP=2
B: EV=1b
B: KEY=c03 0 0 0 0 0
[... 192 lines cut ...]
B: SW=10

I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="HD-Audio Generic Headphone"
P: Phys=ALSA
S: Sysfs=/devices/pci0000:00/0000:00:08.1/0000:09:00.6/sound/card1/input52
U: Uniq=
H: Handlers=event24 
B: PROP=0
B: EV=21
B: SW=4

I: Bus=0005 Vendor=04e8 Product=7021 Version=0001
N: Name="BT5.0 Keyboard"
P: Phys=bc:f4:d4:e2:8f:42
S: Sysfs=/devices/virtual/misc/uhid/0005:04E8:7021.000B/input/input56
U: Uniq=dd:77:54:fa:e9:46
H: Handlers=sysrq kbd leds mouse6 event25 js0 
B: PROP=0
B: EV=12001f
B: KEY=101f 0 3f00033fff 0 0 483ffff17aff32d bfd5444600000000 ff0001 130ff38b17d007 ffff7bfad9415fff ffbeffdfffefffff fffffffffffffffe
B: REL=1943
B: ABS=100000000
B: MSC=10
B: LED=1f

ldrahnik, 2024-09-13:
@fjordanrv Your DSDT table, please.

fjordanrv, 2024-09-13:
[Results.zip](https://github.com/user-attachments/files/16993766/Results.zip)

ldrahnik, 2024-09-14:
> 2024-09-11 13:03:37,936 ERROR Error during sending via i2c: "[Errno 121] Remote I/O error"

Weird. I would recommend you to run manually commands for NumberPad activation, deactivation and changing brightness.

ldrahnik, 2024-09-15:
@fjordanrv Did you try?

ldrahnik, 2024-09-16:
Can be reopened.

QushyQushy, 2024-09-16:
I have the same issue on the same machine. Running

`sudo i2ctransfer -f -y 0 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad`

makes the touchpad light up but no functionality.

ldrahnik, 2024-09-17:
@fjordanrv @QushyQushy It is an already-known problem that involves picking the wrong touchpad device on duo laptops. I tried to simplify detection that problem should not occur anymore. Confirm it works now, please.

fjordanrv, 2024-09-17:
you are a genius now works!!....Thanks a lot!!. I used the normal installation with the bash file and now works perfectly!.

QushyQushy, 2024-09-17:
Can also confirm, it now works on CachyOS. Is there any way this can be installed on an immutable OS like Bazzite?

ldrahnik, 2024-09-17:
@QushyQushy I have no exp with immutable OS's. But this implementation of driver is just python script (args are dir for config - will be autocreated with defaults and layout name) which has to be somehow auto started after booting - that Is role of systemctl service.

QushyQushy, 2024-09-17:
> @QushyQushy I have no exp with immutable OS's. But this implementation of driver is just python script (args are dir for config - will be autocreated with defaults and layout name) that has to be somehow auto started after booting and in the background - role of systemctl service. During installation is modified a system environment a little bit like permissions for example. Has to be than set up manually when is not used `install.sh` or you can try skip only systemctl service installation - I do not know immutable OS's dirs structure so I do not know driver can be installed in the same way.

I managed to get it working. It's pretty straightforward and simple but quite dangerous as well so I won't give instructions. Perhaps the Bazzite team will include it in one of their future builds. I'll make  a feature request for it.

ldrahnik, 2024-11-04:
@QushyQushy 
> Perhaps the Bazzite team will include it in one of their future builds. I'll make a feature request for it.

Have you created that feature request?

ldrahnik, 2024-11-04:
@raphpa 

> Would have helped if the file paths were configurable in a central place though ;)

Now should be possible do this:

```bash
# ENV VARS (with the defaults)
INSTALL_DIR_PATH="/usr/share/asus-numberpad-driver"
CONFIG_FILE_DIR_PATH="$INSTALL_DIR_PATH"
CONFIG_FILE_NAME="numberpad_dev"
LOGS_DIR_PATH="/var/log/asus-numberpad-driver" # only for install and uninstall logs
SERVICE_INSTALL_DIR_PATH="/usr/lib/systemd/user"
INSTALL_UDEV_DIR_PATH="/usr/lib/udev"

# e.g. for BazziteOS (https://github.com/asus-linux-drivers/asus-numberpad-driver/issues/198)
$ INSTALL_DIR_PATH="/home/$USER/.local/share/asus-numberpad-driver"\
INSTALL_UDEV_DIR_PATH="/etc/udev/rules.d/"\
SERVICE_INSTALL_DIR_PATH="/home/$USER/.config/systemd/user/"\
bash install.sh|uninstall.sh
```

Could you please test it and let me know :)

ldrahnik, 2024-11-04:
@raphpa 

> Remove package installation from the script, else it will error out. Comment out the line sudo dnf -y install ...

BazziteOS has installed `dnf` but is not used? I do not understand.

QushyQushy, 2024-11-04:
> @QushyQushy
> 
> > Perhaps the Bazzite team will include it in one of their future builds. I'll make a feature request for it.
> 
> Have you created that feature request?

Yes, [this is it](https://github.com/ublue-os/bazzite/issues/1681).

I will also test the new paths setting and let you know how easy it is. :)

raphpa, 2024-11-04:
> Now should be possible do this:
> 
> ```shell
> # e.g. for BazziteOS (https://github.com/asus-linux-drivers/asus-numberpad-driver/issues/198)
> $ INSTALL_DIR_PATH="/home/$USER/.local/share/asus-numberpad-driver"\
> INSTALL_UDEV_DIR_PATH="/etc/udev/rules.d/"\
> SERVICE_INSTALL_DIR_PATH="/home/$USER/.config/systemd/user/"\
> bash install.sh|uninstall.sh
> ```
> 
> Could you please test it and let me know :)

Seems to work, however the INSTALL_UDEV_DIR_PATH path should not include `rules.d/`.

> BazziteOS has installed `dnf` but is not used? I do not understand. If you think I can change in the repository something else according to your installation, let me know or directly create PR :)

On an atomic OS update basically all system files are thrown away and replaced with the new files. So you are actually not supposed to modify any of them, including installing packages. You are supposed to just use Flatpaks or install stuff into a Distrobox. However I have not found out yet how to reliably access hardware inside Distrobox and have services run out of them.

So the last resort is to use `rpm-ostree` to install packages, which creates file system layers on top of the system files and those layers are replayed on a system update. See https://universal-blue.discourse.group/t/installing-and-managing-software-on-bazzite/35

Speaking of which, for building the python env the preconditions seem to have changed with the update to Fedora 41, the needed packages are now to be installed with `rpm-ostree install xinput virtualenv python3-devel wayland-protocols-devel wayland-devel systemd-devel`

ldrahnik, 2024-11-06:
> Seems to work, however the `INSTALL_UDEV_DIR_PATH` path should not include `rules.d/.`

https://github.com/asus-linux-drivers/asus-numberpad-driver/commit/dafb609436408c66eb5e0ce181ed75859823cbf8

> Speaking of which, for building the python env the preconditions seem to have changed with the update to Fedora 41, the needed packages are now to be installed with rpm-ostree install xinput virtualenv python3-devel wayland-protocols-devel wayland-devel systemd-devel

https://github.com/asus-linux-drivers/asus-numberpad-driver/commit/8669a8962a24c7b55fdd90b8e95c3646e2feacac

ldrahnik, 2024-11-16:
Can be reopened.

closed 2024-11-16 by ldrahnik

fjordanrv, 2024-11-17:
Hi again, 
After the new update of fedora the numberpad is not working again, can you check the error and fixed again. 

thanks in advance.

Attached the error.log
[error.log](https://github.com/user-attachments/files/17790724/error.log)

fjordanrv, 2024-11-17:
Yes, I used the same procedure than the last time.

$ git clone https://github.com/asus-linux-drivers/asus-numberpad-driver
$ cd asus-numberpad-driver
$ bash install.sh

I will try to delete everthing and do it again, let you know if works.

fjordanrv, 2024-11-18:
Hi,
Hi I uninstalled and delete everything, and create the file via GIT link and execute bash intstall and everything ends without error, but still not working the numberpad.

Any idea?.

fjordanrv, 2024-11-18:
[install-18-11-2024-11-36-18.log](https://github.com/user-attachments/files/17799425/install-18-11-2024-11-36-18.log)

ldrahnik, 2024-11-18:
[image]

fjordanrv, 2024-11-18:
Hi, I installed the wayland protocols but still not working. 
[install-18-11-2024-15-15-50.log](https://github.com/user-attachments/files/17801974/install-18-11-2024-15-15-50.log)

raphpa, 2024-11-18:
Install `wayland-devel`

https://packages.fedoraproject.org/pkgs/wayland/wayland-devel/fedora-41.html

ldrahnik, 2024-11-18:
@fjordanrv Solved the problem? I could add `wayland-devel` dependency to install script when is env variable `XDG_SESSION_TYPE=wayland`