asus-linux-drivers/asus-numberpad-driver #199: [Bug]: Clicks triggered when NumberPad is active, on KDE Plasma with Wayland

sleddev, 2024-09-13:
### Describe the bug

When the NumberPad is turned on, and `enabled_touchpad_pointer` is set to 3, the touchpad still registers normal clicks at the same time when I press on the numbers.

### Expected behavior

The touchpad's tap-to-click feature is disabled while the NumberPad is active (how it works on Gnome and/or X11)

### Desktop

- Program version [e.g. 57c742a]: commit 1914f82
- Distro [e.g. Arch Linux]: Fedora 40
- Kernel [e.g. 5.18.15.arch1-2]: 6.10.7-200.fc40.x86_64

sleddev, 2024-09-13:
I have made a fork where I implemented tap-to-click toggling on wayland KDE with qdbus, but I'm not sure how to make it so it still works on gnome and x11

But it does work, so it could be useful: [here's my fork](https://github.com/sleddev/asus-numberpad-driver-qdbus)

ldrahnik, 2024-09-15:
@sleddev Your approach with `qdbus` could  be helpful to other not supported feature for KDE, detecting whether is touchpad enabled or disabled:

> - Disabling the Touchpad (e.g. Fn+special key) disables by default the NumberPad as well (*this functionality presently supports only `xinput` from `xorg` and `gnome` via `gsettings`* - can be disabled)

sleddev, 2024-09-15:
> @sleddev Your approach with `qdbus` could be helpful to other not supported feature for KDE, detecting whether is touchpad enabled or disabled and than disabling NumberPad aswell:
> 
> > * Disabling the Touchpad (e.g. Fn+special key) disables by default the NumberPad as well (_this functionality presently supports only `xinput` from `xorg` and `gnome` via `gsettings`_ - can be disabled)
> 
> Feel free to look at that. I do not use KDE, personally so I can not test it but I would definitely merge it.

I've implemented that feature and opened a [pull request](https://github.com/asus-linux-drivers/asus-numberpad-driver/pull/201) :)

ldrahnik, 2025-03-14:
@sleddev @Lull3n And what about this? It works in Virtualbox. 

```
Operating System: Fedora Linux 41
KDE Plasma Version: 6.2.1
KDE Frameworks Version: 6.7.0
Qt Version: 6.7.2
Kernel Version: 6.13.6-200.fc41.x86_64 (64-bit)
Graphics Platform: Wayland
Processors: 4 × 11th Gen Intel® Core™ i5-1135G7 @ 2.40GHz
Memory: 5.7 GiB of RAM
Graphics Processor: llvmpipe
Manufacturer: innotek GmbH
Product Name: VirtualBox
System Version: 1.2
 
ldrahnik@fedora:~$ sudo libinput debug-events
-event0   DEVICE_ADDED            Power Button                      seat0 default group1  cap:k
-event1   DEVICE_ADDED            Sleep Button                      seat0 default group2  cap:k
-event5   DEVICE_ADDED            VirtualBox mouse integration      seat0 default group3  cap:p left scroll-nat calib
-event4   DEVICE_ADDED            VirtualBox USB Tablet             seat0 default group4  cap:p left scroll-nat calib
-event2   DEVICE_ADDED            AT Translated Set 2 keyboard      seat0 default group5  cap:k
-event3   DEVICE_ADDED            ImExPS/2 Generic Explorer Mouse   seat0 default group6  cap:p left scroll-nat scroll-button
-event2   KEYBOARD_KEY            +0.000s       *** (-1) pressed
 event2   KEYBOARD_KEY            +0.115s       *** (-1) pressed
^C
ldrahnik@fedora:~$ ^C
ldrahnik@fedora:~$ ^C
ldrahnik@fedora:~$ qdbus org.kde.KWin /org/kde/KWin/InputDevice/event4 org.freedesktop.DBus.Properties.Set org.kde.KWin.InputDevice "enabled" "false"
 
ldrahnik@fedora:~$ qdbus org.kde.KWin /org/kde/KWin/InputDevice/event4 org.freedesktop.DBus.Properties.Set org.kde.KWin.InputDevice "enabled" "true"
 
ldrahnik@fedora:~$ 
```

Related to https://github.com/asus-linux-drivers/asus-dialpad-driver/issues/6.

sleddev, 2025-03-14:
> @sleddev @Lull3n And what about this? It works in Virtualbox. 

I can't test it right now, but actually I don't remember this bug happening again since my last comment

ldrahnik, 2025-03-16:
Not replicated. Can be reopened.

closed 2025-03-16 by ldrahnik