asus-linux-drivers/asus-numberpad-driver #202: [Feature Request] Package for Nix OS

scientiac, 2024-09-20:
### Short Description

Unlike other distros we cannot manually setup everything, it would be great to have a declarative service for nixos.

ldrahnik, 2024-09-20:
@scientiac I have no exp with distro like NIX OS. Can you summarize what you had to do - I mean to get this driver working on distro Nix OS?

scientiac, 2024-09-20:
I haven't been able to use it in nixos, hence the request.
For some weeks let it be an issue. I'll try to contribute when I am free.

ldrahnik, 2024-09-20:
@scientiac (y) Do not hesitate to ask about anything here. What I could do is try to run the driver on NIX OS in VirtualBox. Does work NIX OS in VirtualBox?

scientiac, 2024-10-05:
I have been working on it for some days now and it has come to a working state on my laptop:

To test on nixos you need to have flakes enabled, then use the following example to write your configuration:

<details>
<summary>Flakes Configuration</summary>
<br>

https://github.com/scientiac/asus-numberpad-driver
My repo (which, for now is a fork of this repo until the module is complete) contains a flake that exposes a NixOS Module that manages and offers options for auto-cpufreq. To use it, add the flake as an input to your `flake.nix` file and enable the module:

```nix 
# flake.nix

{

    inputs = {
        # ---Snip---
        asus-numberpad-driver = {
          url = "github:scientiac/asus-numberpad-driver";
          inputs.nixpkgs.follows = "nixpkgs";
        };
        # ---Snip---
    }

    outputs = {nixpkgs, asus-numberpad-driver, ...} @ inputs: {
        nixosConfigurations.HOSTNAME = nixpkgs.lib.nixosSystem {
            specialArgs = { inherit inputs; };
            modules = [
                ./configuration.nix
                asus-numberpad-driver.nixosModules.default
            ];
        };
    } 
}
```
Then you can enable the program in your `configuration.nix` file:
```nix
# configuration.nix

{inputs, pkgs, ...}: {
  # ---Snip---
  # Enable Asus Numpad Service
  services.asus-numberpad-driver = {
    enable = true;
    wayland = true; # whether to enable wayland or x11
    layout = "up5401ea"; # your numpad layout
    config = {
      "top_left_icon_brightness_func_max_min_only" = "1";  # your custom configurations
    };
  };
  # ---Snip---
}
```
</details>

ldrahnik, 2024-10-06:
@scientiac Good job(y) (I have not tested that yet). As I said I have not exp with this distro, so my question is: Does exist tool for changing files `configurarion.nix` and `flame.nix`?

scientiac, 2024-10-06:
> @scientiac Good job(y) (I have not tested that yet). As I said I have not exp with this distro, so my question is: Does exist associated tool for changing files `configurarion.nix` and `flame.nix`? I would like to implement support for NixOS to the file `install.sh` and `uninstall.sh`.

Oh, no, You can see a flake.nix and a nix directory on my fork which in itself works like an installation script for NixOS only.

Where users in other distros run install.sh to install the program imperatively, NixOS takes a declarative approach for managing services and installing programs hence, the above steps I have mentioned are equivalent to running the install.sh script in NixOS. The configuration files can be written in multiple ways and in multiple styles so there is no "tool" for changing configuration.nix and flake.nix, here the file"configuration.nix" itself can be renamed, and it is like a convention to name it that way. 

For uninstallation simply removing those lines from configuration and rebuilding the system will work perfectly.

---
Even if there was a way to integrate it in the installation script, Nix users prefer the flake approach because it goes along the nix philosophy and everything in their system is enabled by changing those files consciously.

ldrahnik, 2024-10-06:
@scientiac So for example mentioning existence of these files in Readme.md under block with install inatructions would be just ok. Could you open PR?

scientiac, 2024-10-07:
@ldrahnik Yes, that'll do. I need to give it some finishing touches before I open a PR. Would be great if someone tested and ensured that it works for other layouts as well because I have no way of testing them.

ldrahnik, 2024-11-03:
NixOS support init was merged. Please create a new issue.

closed 2024-11-03 by ldrahnik