asus-linux-drivers/asus-numberpad-driver #204: [Bug]: control process exited with error code

rudrab, 2024-09-21:
### Describe the bug

Hi,
I am trying to install the numpad driver for 	the zenbook UX3405MA.308.
I am getting error:
Job for asus_numberpad_driver@rudra.service failed because the control process exited with error code.
```
systemctl --user status asus_numberpad_driver@rudra.service 
× asus_numberpad_driver@rudra.service - Asus NumberPad Driver
     Loaded: loaded (/usr/lib/systemd/user/asus_numberpad_driver@.service; enabled; preset: disabled)
    Drop-In: /usr/lib/systemd/user/service.d
             └─10-timeout-abort.conf
     Active: failed (Result: exit-code) since Sat 2024-09-21 15:19:05 IST; 41s ago
   Duration: 22ms
 Invocation: c945b197bbd94b32a5bf45e48af8294f
    Process: 3792 ExecStart=/usr/share/asus-numberpad-driver/.env/bin/python3 /usr/share/asus-numberpad-driver/numberpad.py up5401ea /usr/share/asus->
   Main PID: 3792 (code=exited, status=203/EXEC)

Sep 21 15:19:05 roddur systemd[2486]: asus_numberpad_driver@rudra.service: Scheduled restart job, restart counter is at 21.
Sep 21 15:19:05 roddur systemd[2486]: asus_numberpad_driver@rudra.service: Start request repeated too quickly.
Sep 21 15:19:05 roddur systemd[2486]: asus_numberpad_driver@rudra.service: Failed with result 'exit-code'.
Sep 21 15:19:05 roddur systemd[2486]: Failed to start asus_numberpad_driver@rudra.service - Asus NumberPad Driver.
```

The complete log files are attached
[error.log](https://github.com/user-attachments/files/17083640/error.log)
[install-21-09-2024-14-14-33.log](https://github.com/user-attachments/files/17083641/install-21-09-2024-14-14-33.log)

### Expected behavior

The numpad should activated when put finger on the top right corner of the mouse pad.

### Desktop

- Program version #1ace375
- Distro Fedora 41
- Kernel 6.11

uname -a
Linux roddur 6.11.0-63.fc41.x86_64 #1 SMP PREEMPT_DYNAMIC Sun Sep 15 17:48:54 UTC 2024 x86_64 GNU/Linux

ldrahnik, 2024-09-21:
@rudrab 

```Not detected package manager. Driver may not work properly because required packages have not been installed.```

What is your package manager? Feel free contribute and add support for your package manager. That means find equivalent of dependencies at start of file `install.sh` in root of repository for your package manager.

ldrahnik, 2024-09-21:
@rudrab Try to alias `dnf5` to `dnf` using command `alias dnf=dnf5` locally and than try to install again, post error log.

rudrab, 2024-09-21:
adding the alias is not changing the behaviour. 
Also, though dnf5 is the dnf version, `dnf`  itself is working properly:
Just copying the installation line from install.sh can install the dependencies:
```
sudo dnf install ibus libevdev curl xinput i2c-tools python3-devel python3-virtualenv libxml2 libxkbcommon-devel gcc
Updating and loading repositories:
Repositories loaded.
Package "ibus-1.5.31~beta2-1.fc41.x86_64" is already installed.
Package "libevdev-1.13.3-1.fc41.x86_64" is already installed.
Package "curl-8.9.1-2.fc41.x86_64" is already installed.
Package "python3-devel-3.13.0~rc2-2.fc41.x86_64" is already installed.
Package "libxml2-2.12.8-2.fc41.x86_64" is already installed.
Package "libxkbcommon-devel-1.7.0-4.fc41.x86_64" is already installed.
Package "gcc-14.2.1-3.fc41.x86_64" is already installed.

Package                                           Arch         Version                                            Repository                      Size
Installing:
 i2c-tools                                        x86_64       4.3-13.fc41                                        fedora                     212.7 KiB
 python3-virtualenv                               noarch       20.21.1-22.fc41                                    fedora                     694.8 KiB
 xinput                                           x86_64       1.6.4-3.fc41                                       fedora                      62.2 KiB
Installing dependencies:
 libi2c                                           x86_64       4.3-13.fc41                                        fedora                      41.1 KiB
 python-setuptools-wheel                          noarch       69.2.0-8.fc41                                      fedora                     750.4 KiB
 python-wheel-wheel                               noarch       1:0.43.0-4.fc41                                    fedora                      65.1 KiB
 python3-distlib                                  noarch       0.3.8-3.fc41                                       fedora                       1.2 MiB
 python3-filelock                                 noarch       3.15.4-2.fc41                                      fedora                      90.2 KiB

```

ldrahnik, 2024-09-21:
@rudrab `sudo dnf help` does not work anymore with this newest version? On Fedora41?

ldrahnik, 2024-09-21:
@rudrab Commited support for `-help`. Any other error? Does driver works?

rudrab, 2024-09-21:
Yes.. The driver worked partially... i.e. Its switched on. But its not toggling... So I am not geeting my touchpad back and can't search gpogle. I remember there was a config file where I had the setup that 5 sec idle will disable the numpad. But I can't search now.
Can you please tell me the file name and location and the tag line to disable the numpad when idle?

ldrahnik, 2024-09-21:
@rudrab Could you post here error log than? File is located `/usr/share/asus-numberpad-driver/numberpad_dev`. You can try also `sed -i "s/enabled = 1/enabled = 0/g" /usr/share/asus-numberpad-driver/numberpad_dev`

ldrahnik, 2024-09-21:
I am closing and not toggling please here https://github.com/asus-linux-drivers/asus-numberpad-driver/issues/169

closed 2024-09-21 by ldrahnik

kamack38, 2024-09-21:
> @rudrab Commited support for `-help`. Any other error? Does driver works?

Wouldn't it be better to just run `which dnf` to check whether the specified binary exists?

ldrahnik, 2024-09-21:
@kamack38 I was thinking the same but I had no time check cross distribution compatibility.

rudrab, 2024-09-21:
Both `dnf` and `dnf5` is allowed command

kamack38, 2024-09-21:
> @kamack38 I was thinking the same but I had no time check cross distribution compatibility (of `which`).

`whereis` is mostly the same, and it is required by systemd and base (at least on Arch) so I think it exists in most of the distributions.

ldrahnik, 2024-09-22:
@kamack38 I changed looking for `help` subcommands with usage of builtin `$ command`:

> Many operating systems have a which that doesn't even set an exit status, meaning the if which foo won't even work there and will always report that foo exists, even if it doesn't (note that some POSIX shells appear to do this for hash too).
[https://stackoverflow.com/questions/592620/how-can-i-check-if-a-program-exists-from-a-bash-script]

And replaced `which` during creating project specific env for current python3 using `virtualenv` because of mentioned above + it is external dependency and the package `which` was missing. [e.g. https://ioflood.com/blog/install-which-command-linux/]