asus-linux-drivers/asus-numberpad-driver #205: [Bug]: ON/OFF issue with e210ma layout

erfan4323, 2024-09-22:
### Describe the bug

Hi, I hope you are doing well
I have a **Vivobook_ASUSLaptop M1405YA_M1405YA** laptop with an **e210ma** number pad layout
So I did the README, and it went well. The problem is that when I press the **ON/OFF** part of it, it won't work all the time
Sometimes it gets on, and sometimes it doesn't (it happens when you want to turn it off too)

### Expected behavior

I think there is a point that I have to press, but I can't find it
Or maybe it gets on/off with a long delay
I'm not sure
It should be working whenever I press around the ON/OFF section right away

### Relevant log output

```shell
I tried to debug it with libinput and when I clicked around the ON/OFF area, I got these outputs:
 event11  KEYBOARD_KEY            +33.046s	*** (-1) released
 event11  KEYBOARD_KEY            +33.046s	*** (-1) released
```

### Desktop

- Program - version v6.3.4
- Distro - Arch Linux
- Kernel - 6.10.10-arch1-1

ldrahnik, 2024-09-22:
@erfan4323 Please, add in file `numberpad.py` line in row `2041`: `log.info(e)`, reinstall and post here clean (clean it when reinstalling manually) error log from `/var/logs/asus-numberpad-driver/error.log` and post here with description like: 3 times worked, another one not or something like that.

ldrahnik, 2024-09-22:
@erfan4323 Also would be helpful enable debug logs (e.g. adding `LOG=DEBUG` to systemd service as environment variable and than reinstall)

ldrahnik, 2024-09-22:
@erfan4323 Activation/deactivation works with slide gesture to the center from the top right corner or when you touch and hold that corner for longer time than is set up activation time (1s default)

erfan4323, 2024-09-22:
Ok, I think I didn't know I could hold it for 1s, my bad
I reinstalled it, and it works
thank you for your time

closed 2024-09-22 by erfan4323