asus-linux-drivers/asus-numberpad-driver #206: [Bug]: When running manually the config file is wrongly created and the program errors on first run.

scientiac, 2024-10-03:
### Describe the bug

When I run the command 

./numberpad.py up5401ea ~/numberpad_dev

it makes a config file on home named "numberpad_devnumberpad_dev " while i already have an empty "numberpad_deb" file on ~

❯ cat numberpad_devnumberpad_dev 
[main]
numpad_disables_sys_numlock = 1
disable_due_inactivity_time = 0
touchpad_disables_numpad = 1
key_repetitions = 0
multitouch = 0
one_touch_key_rotation = 0
activation_time = 1
sys_numlock_enables_numpad = 1
top_left_icon_activation_time = 1
top_left_icon_slide_func_activation_radius = 1200
top_left_icon_slide_func_activates_numpad = 1
top_right_icon_slide_func_activation_radius = 1200
enabled_touchpad_pointer = 3
press_key_when_is_done_untouch = 1
enabled = 0
default_backlight_level = 0x01
brightness = 0x48
top_left_icon_brightness_func_disabled = 0
distance_to_move_only_pointer = 0
idled = 0
idle_brightness = 30
idle_enabled = 0
idle_time = 10

it also errors on first time turning on the numpadpad with long press gesture

### Expected behavior

It should use the previously available config file and also if it creates it should create the right one.

### Relevant log output

```shell
❯ ./numberpad.py up5401ea ~/numberpad_dev
2024-10-03 20:13:58,257 INFO Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
2024-10-03 20:13:58,258 INFO Set keyboard 0 from H: Handlers=sysrq kbd leds event0
2024-10-03 20:13:58,258 INFO Detecting touchpad from string: "N: Name="ASUP1415:00 093A:300C Touchpad""
2024-10-03 20:13:58,258 INFO Set touchpad device id 2 from S: Sysfs=/devices/platform/AMDI0010:03/i2c-2/i2c-ASUP1415:00/0018:093A:300C.0001/input/input11
2024-10-03 20:13:58,258 INFO Set touchpad id 8 from H: Handlers=event8 mouse1
2024-10-03 20:13:58,259 INFO Touchpad min-max: x 0-3996, y 0-2242
2024-10-03 20:13:58,259 INFO Numpad min-max: x 200-3796, y 200-2162
2024-10-03 20:13:59,265 INFO Setting up for config file key: "brightness" with value: "0x48"
[2024-10-03 20:13:59,882 pyinotify ERROR] add_watch: cannot watch /home/scientiac/numberpad_dev WD=-1, Errno=No such file or directory (ENOENT)
2024-10-03 20:13:59,882 ERROR add_watch: cannot watch /home/scientiac/numberpad_dev WD=-1, Errno=No such file or directory (ENOENT)
2024-10-03 20:14:00,429 INFO Touched top_right_icon area (numlock) in time: 1727965740.4291005
2024-10-03 20:14:01,434 INFO The numpad numlock was pressed longer than the activation time and will be activated: 1.0045814514160156
2024-10-03 20:14:01,434 INFO Activation time: 1.0
2024-10-03 20:14:01,436 ERROR qdbus get failed
Traceback (most recent call last):
  File "/home/scientiac/Experiments/working-reference-touchpad-driver/./numberpad.py", line 745, in qdbusGet
    result = subprocess.check_output(cmd).rstrip()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/zy22nlap6gznpjhqry4w0hgw5mf62q3f-python3-3.11.9/lib/python3.11/subprocess.py", line 466, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/zy22nlap6gznpjhqry4w0hgw5mf62q3f-python3-3.11.9/lib/python3.11/subprocess.py", line 548, in run
    with Popen(*popenargs, **kwargs) as process:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/zy22nlap6gznpjhqry4w0hgw5mf62q3f-python3-3.11.9/lib/python3.11/subprocess.py", line 1026, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
[... 92 lines cut ...]
  File "/nix/store/rfr2g3hk3w3m3j342m5z8gsy3gyg6lsl-python3.11-libevdev-0.11/lib/python3.11/site-packages/libevdev/device.py", line 542, in events
    ev = self._libevdev.next_event(flags)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/rfr2g3hk3w3m3j342m5z8gsy3gyg6lsl-python3.11-libevdev-0.11/lib/python3.11/site-packages/libevdev/_clib.py", line 889, in next_event
    rc = self._next_event(self._ctx, flags, ctypes.byref(ev))
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
KeyboardInterrupt
2024-10-03 20:14:50,947 INFO Clean up started
2024-10-03 20:14:50,947 INFO System numlock deactivated
2024-10-03 20:14:50,949 INFO Setting up for config file key: "idled" with value: "False"
2024-10-03 20:14:51,050 INFO Cancelled numpad idle
2024-10-03 20:14:51,050 INFO Cancelled idle
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ASUP1415:00 093A:300C Touchpad
2024-10-03 20:14:51,076 INFO Setting up for config file key: "enabled" with value: "False"
2024-10-03 20:14:51,177 INFO Numpad deactivated
2024-10-03 20:14:51,178 INFO Clean up finished
2024-10-03 20:14:51,178 INFO Exiting
```

### Desktop

- Program version [c07b6b8]
- Distro [NixOS]
- Kernel [6.10.9-zen1]

ldrahnik, 2024-10-03:
@scientiac Thank you for your effort. Arg is config file **directory** so the correct value is `~/`. Does it work?

scientiac, 2024-10-03:
@ldrahnik what about the program erroring on first use ?

yes giving it the directory worked fine, thanks

scientiac, 2024-10-03:
So it's just a fallback or refresh of sorts, because the driver works fine after it successfully starts. It's fine to have the error then?

scientiac, 2024-10-03:
Thanks for the clarification.

ldrahnik, 2024-10-03:
@scientiac Commited.