asus-linux-drivers/asus-numberpad-driver #207: [Feature Request]: Parity with Default Brightness Behavior

scientiac, 2024-10-04:
### Short Description

A configuration option when enabled should change the brightness action (long press) to have only the minimum and the maximum brightness value (Only two brightness modes instead of multiple), similar to the default behavior of the driver on windows.

### Additional Info

This is because while changing(increasing) brightness via the gesture there is no clear indication that the brightness is full and I always accidentally reach the lowest value again and again.

ldrahnik, 2024-10-05:
@scientiac Please try the latest master and let me know.

scientiac, 2024-10-05:
@ldrahnik Is it set as a default behavior? or do i have to enable using the config option? How do I enable it?

ldrahnik, 2024-10-05:
@scientiac You can enable it by row (option) in config `top_left_icon_brightness_func_max_min_only = 1`. Is missing any config option? Then will be used default from file `numberpad.py`. It does not make sense to me set this as default behaviour. as windows driver has. The main reason is that the driver remembers the last brightness and why limit levels when fit one can be choosen only once.

ldrahnik, 2024-10-05:
In short: I would just pre-create config with only this line as you did and other options should be than autofilled with default values.

scientiac, 2024-10-05:
> @ldrahnik
> 
> I am not quite sure about how passing the config values work. I have passed the directory while running the script with the single configuration of the proposed change:
> 
> the command run by systemd (on nixos) is:
> 
> ```
> /nix/store/zy22nlap6gznpjhqry4w0hgw5mf62q3f-python3-3.11.9/bin/python3 /nix/store/6d0yxkdr556p62bjsqrq6v2ylv7q4bld-python3.11-asus-numberpad-driver-6.3.4/share/asus-numberpad-driver/numberpad.py up5401ea /etc/asus-numberpad-driver
> ```
> 
> where the config is located here:
> 
> /etc/asus-numberpad-driver/numberpad_dev
> 
> ```
> [main]
> top_left_icon_brightness_func_max_min_only = 1
> ```

Setting the configuration as above is still making top left icon gradually change brightness instead of switching between max and min, is there something I should be doing differently?

ldrahnik, 2024-10-05:
> without writing the remaining config options 

The config is 2-way-sync. That means the driver writes to the config as well as the user. Config reflects current driver state for example by config option `enabled`. Config is reloaded by driver each 0.5s. Btw. writting defaults to config can be helpful in case when you look at the config file and then you do not have to check doc for the correct config option name, you just change value 0/1.

ldrahnik, 2024-10-05:
> Setting the configuration as above is still making top left icon gradually change brightness instead of switching between max and min, is there something I should be doing differently?

@scientiac Have you reinstalled the driver? It works for me.

scientiac, 2024-10-05:
The reason why the config wasn't working was because 

```
/nix/store/zy22nlap6gznpjhqry4w0hgw5mf62q3f-python3-3.11.9/bin/python3 /nix/store/6d0yxkdr556p62bjsqrq6v2ylv7q4bld-python3.11-asus-numberpad-driver-6.3.4/share/asus-numberpad-driver/numberpad.py up5401ea /etc/asus-numberpad-driver
```

the directory argument needed to have a trailing slash at the end

like this :
```
<python path> <script path>  <device model> /etc/asus-numberpad-driver/
```

It was a stupid error from my side.

The update works!

closed 2024-10-05 by scientiac