asus-linux-drivers/asus-numberpad-driver #210: [Bug]: Modifier keys not sent properly

benj5378, 2024-10-27:
### Describe the bug

On my BX425JA using up5401 layout, pressing keys sent by modifiers don't work.

For instance pressing...
`/` results in `7`
`*` results in `'`
`=` results in `0`

This is because
`/` is sent as `[KEY_LEFTSHIFT:42, KEY_7:8]` which does correspond to `/` on my keyboard layout - however, for some reason, the modifier key isn't received. (Also, why don't we just sent `KEY_SLASH`?

### Expected behavior

`/` results in `/`
`*` results in `*`
`=` results in `=`

### Relevant log output

```shell
{'Num_Lock': KEY_NUMLOCK:69, '0': KEY_0:11, '1': KEY_1:2, '2': KEY_2:3, '3': KEY_3:4, '4': KEY_4:5, '5': KEY_5:6, '6': KEY_6:7, '7': KEY_7:8, '8': KEY_8:9, '9': KEY_9:10, 'a': KEY_A:30, 'b': KEY_B:48, 'c': KEY_C:46, 'd': KEY_D:32, 'e': KEY_E:18, 'f': KEY_F:33, 'Shift_L': KEY_LEFTSHIFT:42, 'Control_L': KEY_LEFTCTRL:29, 'u': KEY_U:22, 'space': KEY_SPACE:57, 'slash': [KEY_LEFTSHIFT:42, KEY_7:8], 'BackSpace': KEY_BACKSPACE:14, 'asterisk': [KEY_LEFTSHIFT:42, KEY_BACKSLASH:43], 'minus': KEY_SLASH:53, 'percent': [KEY_LEFTSHIFT:42, KEY_5:6], 'period': KEY_DOT:52, 'Return': KEY_ENTER:28, 'plus': KEY_MINUS:12, 'equal': [KEY_LEFTSHIFT:42, KEY_0:11]}

```
Input device name: "ASUE140A:00 04F3:3134 NumberPad"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 2 (KEY_1)
    Event code 3 (KEY_2)
    Event code 4 (KEY_3)
    Event code 5 (KEY_4)
    Event code 6 (KEY_5)
    Event code 7 (KEY_6)
    Event code 8 (KEY_7)
    Event code 9 (KEY_8)
    Event code 10 (KEY_9)
    Event code 11 (KEY_0)
    Event code 12 (KEY_MINUS)
    Event code 14 (KEY_BACKSPACE)
    Event code 18 (KEY_E)
    Event code 22 (KEY_U)
    Event code 28 (KEY_ENTER)
    Event code 29 (KEY_LEFTCTRL)
    Event code 30 (KEY_A)
    Event code 32 (KEY_D)
    Event code 33 (KEY_F)
    Event code 42 (KEY_LEFTSHIFT)
    Event code 43 (KEY_BACKSLASH)
    Event code 46 (KEY_C)
    Event code 48 (KEY_B)
    Event code 52 (KEY_DOT)
    Event code 53 (KEY_SLASH)
    Event code 57 (KEY_SPACE)
    Event code 69 (KEY_NUMLOCK)
    Event code 140 (KEY_CALC)
    Event code 272 (BTN_LEFT)
    Event code 273 (BTN_RIGHT)
    Event code 274 (BTN_MIDDLE)
```
```

### Desktop

- Program version [e.g. 57c742a]
- Distro [e.g. Arch Linux]
- Kernel [e.g. 5.18.15.arch1-2]

ldrahnik, 2024-10-27:
@benj5378 Hello, thank you for creating an issue. A related problem was discussed (not solved yet) here https://github.com/asus-linux-drivers/asus-numberpad-driver/issues/187. 

Do you use Wayland/X11?
Could you please confirm it does not work even immediately when the laptop finishes booting? E.g. login page?

benj5378, 2024-10-27:
@ldrahnik thanks for quick response.

After a lot of debugging, I found something interesting

Doesn't work:
```
Event: time 1730034126.510311, -------------- SYN_REPORT ------------
Event: time 1730034128.638561, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 1
Event: time 1730034128.638561, -------------- SYN_REPORT ------------
Event: time 1730034128.811548, type 1 (EV_KEY), code 8 (KEY_7), value 1
Event: time 1730034128.811548, -------------- SYN_REPORT ------------
Event: time 1730034128.863731, type 1 (EV_KEY), code 8 (KEY_7), value 0
Event: time 1730034128.863731, -------------- SYN_REPORT ------------
Event: time 1730034128.966722, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 0
```

Does work
```
Event: time 1730038557.635306, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 1
Event: time 1730038557.635306, -------------- SYN_REPORT ------------
Event: time 1730038557.635332, type 1 (EV_KEY), code 8 (KEY_7), value 1
Event: time 1730038557.635332, -------------- SYN_REPORT ------------
Event: time 1730038557.635347, type 1 (EV_KEY), code 8 (KEY_7), value 0
Event: time 1730038557.635347, -------------- SYN_REPORT ------------
Event: time 1730038557.635362, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 0
Event: time 1730038557.635362, -------------- SYN_REPORT ------------
Event: time 1730038557.635376, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 1
Event: time 1730038557.635376, -------------- SYN_REPORT ------------
```

> Could you please confirm it does not work even immediately when the laptop finishes booting? E.g. login page?

Might not be relevant due to above founds.

> Do you use multiple keyboard character layouts that print mentioned characters using and not using a modifier?

I don't think so? I only have one layout set

benj5378, 2024-10-27:
I fixed the issue, it's related to timing. I'll submit a PR shortly

benj5378, 2024-10-27:
@ldrahnik, here is some really interesting, that indicates that in some parts of the OS, the shift key actually detected and considered. Let me show my findings using https://github.com/eyelash/tutorials/blob/master/libinput.c 

Using normal keyboard:
Pressing `7` gives `the key 7 was pressed`
Pressing `7 + Shift `gives `the key & was pressed` (the `&` is just due to unset layout in code)

Okay, so we know if the shift key is considered, libinput will yield `&`.

Now let's try with the numberpad.py driver:
Pressing `/` that sends `7 + Shift` gives `the key & was pressed`

So this means, that the events are sent correctly. And libinput correctly receives `7 + Shift` at once even we no delay. So it's somewhere else in the OS, that the translation doesn't happen without a delay???

Even though it's libinput that forwards keys to Wayland. How is it possible for libinput see the keys as a combination, but when they reach Wayland client, they are not a combination anymore??

[image]

ldrahnik, 2024-10-27:
@benj5378 Each Wayland client can have different keymap so key may be translated to wrong one. Only one idea what I have and was mentioned in related issue.

benj5378, 2024-10-27:
>  so key may be translated to wrong one

@ldrahnik, well, it can't be that it is translated to wrong one - because we know it's translated with delay.

We just know, that it is translated as expect, when reaching libinput.
And that once it is forwarded to Wayland Client, it isn't translated at all.

PR #211 (2024-10-29): Fix internal modifier keys
Fixes #210

I really don't get why this is needed. I thought the whole idea behind `EV_SYN` was to avoid physical delays? Is this related to Wayland or something?
commit b673715c (2024-10-29): Fix internal modifier keys

Fixes #210
closed 2024-10-29 by ldrahnik