asus-linux-drivers/asus-numberpad-driver #212: [Bug]: Backlight doesn't turn on Zenbook UX3402VA

ivan0rdonez, 2024-10-29:
### Describe the bug

I run PopOS 22.04 LTS on a Zenbook UX3402VA_UX3402VA. I installed executing `bash install.sh` without errors but the backlight of the numpad/mousepad never turned on. 

I discover that the padnum does actually work since I can input numbers and symbols with the touchpad after long pressing the top right corner. 

I tried the faq's steps, I executed `bash install_device_check.sh` to get the device id and address, they are the default ones, and tried to send different signals as recommended to no avail:

```
i2ctransfer -f -y 1 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x60 0xad
i2ctransfer -f -y 1 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad
```

And also bright levels.

I noticed an error in the /var/log/asus-numberpad-driver/error.log 

```
2024-10-29 18:28:32,323 ERROR X11 load keymap listener error. Exiting
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 468, in load_keymap_listener_x11
    event = display.next_event()
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.10/site-packages/Xlib/display.py", line 187, in next_event
    return self.display.next_event()
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.10/site-packages/Xlib/protocol/display.py", line 217, in next_event
    self.send_and_recv(event = True)
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.10/site-packages/Xlib/protocol/display.py", line 609, in send_and_recv
    raise self.socket_error
Xlib.error.ConnectionClosedError: Display connection closed by server
2024-10-29 18:28:32,326 ERROR Listening touchpad events unexpectedly failed
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 2626, in <module>
    listen_touchpad_events()
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 2127, in listen_touchpad_events
    for e in d_t.events():
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.10/site-packages/libevdev/device.py", line 542, in events
    ev = self._libevdev.next_event(flags)
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.10/site-packages/libevdev/_clib.py", line 889, in next_event
    rc = self._next_event(self._ctx, flags, ctypes.byref(ev))
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 2546, in signal_handler
    raise Exception()
Exception
2024-10-29 18:28:32,328 INFO Clean up started
2024-10-29 18:28:32,343 INFO Clean up finished
2024-10-29 18:28:32,343 INFO Exiting
2024-10-29 18:40:38,097 ERROR X11 detected but not connected succesfully to the display :1. Exiting
2024-10-29 18:40:39,489 INFO X11 detected and connected succesfully to the display :1
2024-10-29 18:40:39,491 INFO Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
```

Finally I tried with the recommended layout (`ux433fa`) but it doesn`t look like the one in my instruction booklet or the pictures of the asus store, since I never booted windows I don't know for sure, so I tried with `up5401ea.py` and `[up5401ea-3145.py` since it is the one that looks more similar to the manual, but indeed the mapping is different.

### Expected behavior

Backlight should turn on when I longpress the upper right corner of my mousepad

### Relevant log output

```shell
var/log/asus-numberpad-driver/error.log 

2024-10-29 18:28:17,859 INFO Touched top_right_icon area (numlock) in time: 1730194097.8590577
2024-10-29 18:28:18,861 INFO The numpad numlock was pressed longer than the activation time and will be activated: 1.0017402172088623
2024-10-29 18:28:18,861 INFO Activation time: 1.0
2024-10-29 18:28:18,892 INFO System numlock deactivated
2024-10-29 18:28:18,892 INFO Numpad deactivated
2024-10-29 18:28:18,942 INFO Setting up for config file key: "enabled" with value: "False"
2024-10-29 18:28:18,943 INFO check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/numberpad_dev wd=1 >
2024-10-29 18:28:32,323 ERROR X11 load keymap listener error. Exiting
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 468, in load_keymap_listener_x11
    event = display.next_event()
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.10/site-packages/Xlib/display.py", line 187, in next_event
    return self.display.next_event()
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.10/site-packages/Xlib/protocol/display.py", line 217, in next_event
    self.send_and_recv(event = True)
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.10/site-packages/Xlib/protocol/display.py", line 609, in send_and_recv
    raise self.socket_error
Xlib.error.ConnectionClosedError: Display connection closed by server
2024-10-29 18:28:32,326 ERROR Listening touchpad events unexpectedly failed
Traceback (most recent call last):
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 2626, in <module>
    listen_touchpad_events()
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 2127, in listen_touchpad_events
    for e in d_t.events():
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.10/site-packages/libevdev/device.py", line 542, in events
    ev = self._libevdev.next_event(flags)
  File "/usr/share/asus-numberpad-driver/.env/lib/python3.10/site-packages/libevdev/_clib.py", line 889, in next_event
    rc = self._next_event(self._ctx, flags, ctypes.byref(ev))
  File "/usr/share/asus-numberpad-driver/numberpad.py", line 2546, in signal_handler
    raise Exception()
Exception
2024-10-29 18:28:32,328 INFO Clean up started
2024-10-29 18:28:32,343 INFO Clean up finished
2024-10-29 18:28:32,343 INFO Exiting
2024-10-29 18:40:38,097 ERROR X11 detected but not connected succesfully to the display :1. Exiting
2024-10-29 18:40:39,489 INFO X11 detected and connected succesfully to the display :1
2024-10-29 18:40:39,491 INFO Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
```

### Desktop

- Program version: mainbranch
- Distro Pop!_OS 22.04 LTS x86_64 
- Kernel 6.9.3-76060903-generic

ldrahnik, 2024-10-29:
@ivan0rdonez Hello, thank you for creating an issue. What was the output of `$ bash install_device_check.s`?

ldrahnik, 2024-10-31:
Can be reopened.

closed 2024-10-31 by ldrahnik