asus-linux-drivers/asus-numberpad-driver #213: [Bug]: Normally shifted symbols are missing shift

alercah, 2024-10-31:
### Describe the bug

I just got a new UM3402. Thanks for making the installation so painless and making this just work. Really well done work on this firmware.

But unfortunately I've a funny bug: it seems that the faked Shift key isn't working correctly. So if I press + I get =, if I press * I get 8, and if I press % I get 5. The remaining symbols work.

I use KDE with fcitx5.

### Expected behavior

I expected the +, %, and * symbols to produce the correct symbol.

### Relevant log output

```shell
Output of journalctl -b 0 -g numberpad:

Oct 31 05:22:23 fedora (udev-worker)[1174]: ACAD: Process '/usr/share/asus-numberpad-driver/udev/power_supply_is_ac.sh' failed with exit code 2.
Oct 31 05:22:31 fedora systemd[1967]: Created slice app-asus_numberpad_driver.slice - Slice /app/asus_numberpad_driver.
Oct 31 05:22:31 fedora systemd[1967]: Started asus_numberpad_driver@alercah.service - Asus NumberPad Driver.
Oct 31 05:22:31 fedora systemd[1967]: asus_numberpad_driver@alercah.service: Main process exited, code=exited, status=1/FAILURE
Oct 31 05:22:31 fedora systemd[1967]: asus_numberpad_driver@alercah.service: Failed with result 'exit-code'.
Oct 31 05:22:31 fedora systemd[1967]: asus_numberpad_driver@alercah.service: Consumed 2.126s CPU time.
Oct 31 05:22:32 fedora systemd[1967]: asus_numberpad_driver@alercah.service: Scheduled restart job, restart counter is at 1.
Oct 31 05:22:32 fedora systemd[1967]: Started asus_numberpad_driver@alercah.service - Asus NumberPad Driver.
Oct 31 05:22:33 fedora kernel: input: ASUE140D:00 04F3:31B9 NumberPad as /devices/virtual/input/input31
```

### Desktop

- Program version v6.3.4
- Fedora 40
- Kernel 6.11.5-200.fc40.x86_64

ldrahnik, 2024-10-31:
@alercah Hi, thank you for your kind words. Try to remove `wayland` specific condition here in case you use `x11` and reinstall:

```
1676          if display_wayland:
1677            sleep(0.005)
```

ldrahnik, 2024-10-31:
`Numberpad.py` and `/var/log/asus-numberpad-driver/`

ldrahnik, 2024-10-31:
@alercah Change it in cloned `git` repository and then reinstall.

ldrahnik, 2024-11-01:
@alercah Does this branch work as expected? https://github.com/asus-linux-drivers/asus-numberpad-driver/pull/215

ldrahnik, 2024-11-03:
Can be reopened.

closed 2024-11-03 by ldrahnik