asus-linux-drivers/asus-numberpad-driver #216: [Bug]: Changes to nix files broke the service.

scientiac, 2024-11-03:
### Describe the bug

You've removed journal path and added more python dependencies which aren't available in nixpkgs breaking the whole nixos build configuration, hence users will be unable to make changes to the system until the flake input is removed from their configuration.

### Expected behavior

Systemd service for the driver should start the service normally.

### Desktop

- Program version [c77675b]
- Distro [e.g. NixOS]
- Kernel [6.11.5-zen1]

scientiac, 2024-11-03:
I synced my fork to be on par with this after the merge and changes you have done today.

Had to immediately revert to my last commit, and it fixed the installation from my system.

I have made some changes in the following files:

nix/default.nix (removed `systemd-python` package since it doesn't exist.)
nix/module.nix (Fix systemd-service not starting on reboot)
README.md (Included the updated configuration for toggling the calculator.)

ldrahnik, 2024-11-03:
@scientiac Thank you for letting me know. I overlooked not synchronized `wantedBy` with source services. About logging into journal via `from systemd.journal import JournalHandler` is problem not existing pip package or library on which it depends?

scientiac, 2024-11-03:
I'm replying from bed via mobile right now (will be verifying tomorrow morning), the package name in nixpkgs for `python-systemd` is:

https://search.nixos.org/packages?channel=24.05&from=0&size=50&sort=relevance&type=packages&query=python311Packages.systemd

so instead of  `python-systemd` just putting `systemd` should work.

ldrahnik, 2024-11-04:
@scientiac Does it work now?

closed 2024-11-04 by ldrahnik