asus-linux-drivers/asus-numberpad-driver #218: [Bug]: Followed instructions on ASUS ROG STRIX SCAR III G512LV but service is non-functioning.

DaanSelen, 2024-11-06:
### Describe the bug

Numpad does not activate with the touch button. Also the Systemd service fails without any useful error message.
I only changed the paths in the service file because the previous message was "could not find python3".

### Expected behavior

To have the red grid light up and the numpad functional.

### Relevant log output

```shell
dselen@DS-STRIX:/usr/lib/systemd/user$ systemctl status --user asus_numberpad_driver@$USER.service
● asus_numberpad_driver@dselen.service - Asus NumberPad Driver
     Loaded: loaded (/usr/lib/systemd/user/asus_numberpad_driver@.service; enabled; preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Thu 2024-11-07 00:55:21 CET; 425ms ago
    Process: 9141 ExecStart=/usr/share/asus-numberpad-driver/.env/bin/python3 /usr/share/asus-numberpad-driver/numberpad.py up5401ea /usr/share/asus-numberpad-driver/ (code=exited, status=1/FAILURE)
   Main PID: 9141 (code=exited, status=1/FAILURE)
        CPU: 44ms
dselen@DS-STRIX:/usr/lib/systemd/user$ systemctl status --user asus_numberpad_driver@$USER.service
● asus_numberpad_driver@dselen.service - Asus NumberPad Driver
     Loaded: loaded (/usr/lib/systemd/user/asus_numberpad_driver@.service; enabled; preset: enabled)
     Active: active (running) since Thu 2024-11-07 00:55:23 CET; 1ms ago
   Main PID: 9146 (python3)
      Tasks: 1 (limit: 18822)
     Memory: 280.0K
        CPU: 0
     CGroup: /user.slice/user-1000.slice/user@1000.service/app.slice/app-asus_numberpad_driver.slice/asus_numberpad_driver@dselen.service
             └─9146 /usr/share/asus-numberpad-driver/.env/bin/python3 /usr/share/asus-numberpad-driver/numberpad.py up5401ea /usr/share/asus-numberpad-driver/

nov 07 00:55:23 DS-STRIX systemd[1108]: Started asus_numberpad_driver@dselen.service - Asus NumberPad Driver.
dselen@DS-STRIX:/usr/lib/systemd/user$ systemctl status --user asus_numberpad_driver@$USER.service
● asus_numberpad_driver@dselen.service - Asus NumberPad Driver
     Loaded: loaded (/usr/lib/systemd/user/asus_numberpad_driver@.service; enabled; preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Thu 2024-11-07 00:55:23 CET; 956ms ago
    Process: 9146 ExecStart=/usr/share/asus-numberpad-driver/.env/bin/python3 /usr/share/asus-numberpad-driver/numberpad.py up5401ea /usr/share/asus-numberpad-driver/ (code=exited, status=1/FAILURE)
   Main PID: 9146 (code=exited, status=1/FAILURE)
        CPU: 41ms
dselen@DS-STRIX:/usr/lib/systemd/user$ systemctl status --user asus_numberpad_driver@$USER.service
● asus_numberpad_driver@dselen.service - Asus NumberPad Driver
     Loaded: loaded (/usr/lib/systemd/user/asus_numberpad_driver@.service; enabled; preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Thu 2024-11-07 00:55:24 CET; 660ms ago
    Process: 9149 ExecStart=/usr/share/asus-numberpad-driver/.env/bin/python3 /usr/share/asus-numberpad-driver/numberpad.py up5401ea /usr/share/asus-numberpad-driver/ (code=exited, status=1/FAILURE)
   Main PID: 9149 (code=exited, status=1/FAILURE)
        CPU: 43ms
```

### Desktop

- Program version [e.g. 8669a89]
- Distro [e.g. Debian 12]
- Kernel [e.g. Linux DS-STRIX 6.1.0-26-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.112-1 (2024-09-30) x86_64 GNU/Linux]

ldrahnik, 2024-11-07:
Please, attach driver logs.

ldrahnik, 2024-11-07:
@DaanSelen Can be reopened.

DaanSelen, 2024-11-07:
```
dselen@DS-STRIX:/var/log/asus-numberpad-driver$ cat install-07-11-2024-22-48-02.log 
Reading package lists...
Building dependency tree...
Reading state information...
ibus is already the newest version (1.5.27-5).
libevdev2 is already the newest version (1.13.0+dfsg-1).
curl is already the newest version (7.88.1-10+deb12u7).
xinput is already the newest version (1.6.3-1).
i2c-tools is already the newest version (4.3-2+b3).
python3-dev is already the newest version (3.11.2-1+b1).
python3-virtualenv is already the newest version (20.17.1+ds-1).
libxml2-utils is already the newest version (2.9.14+dfsg-1.3~deb12u1).
libxkbcommon-dev is already the newest version (1.5.0-1).
gcc is already the newest version (4:12.2.0-3).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

groupadd: group 'input' already exists
groupadd: group 'i2c' already exists
groupadd: group 'uinput' already exists
Added groups input, i2c, uinput, numberpad to current user
uinput module loaded
i2c-dev module loaded
uinput module added to auto loaded modules
Udev rules reloaded and triggered

The i2cdetect or i2ctransfer tool not found to proceed initial test whether any i2c device react like NumberPad

Default config will be autocreated during the first run and available for futher modifications here:
/usr/share/asus-numberpad-driver/numberpad_dev

created virtual environment CPython3.11.2.final.0-64 in 186ms
  creator CPython3Posix(dest=/usr/share/asus-numberpad-driver/.env, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=/home/dselen/.local/share/virtualenv)
    added seed packages: pip==23.0.1, setuptools==66.1.1, wheel==0.38.4
  activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
Requirement already satisfied: pip in /usr/share/asus-numberpad-driver/.env/lib/python3.11/site-packages (23.0.1)
Collecting pip
  Using cached pip-24.3.1-py3-none-any.whl (1.8 MB)
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 23.0.1
    Uninstalling pip-23.0.1:
      Successfully uninstalled pip-23.0.1
Successfully installed pip-24.3.1
Requirement already satisfied: setuptools in /usr/share/asus-numberpad-driver/.env/lib/python3.11/site-packages (66.1.1)
Collecting setuptools
  Using cached setuptools-75.3.0-py3-none-any.whl.metadata (6.9 kB)
Using cached setuptools-75.3.0-py3-none-any.whl (1.3 MB)
Installing collected packages: setuptools
  Attempting uninstall: setuptools
    Found existing installation: setuptools 66.1.1
    Uninstalling setuptools-66.1.1:
      Successfully uninstalled setuptools-66.1.1
Successfully installed setuptools-75.3.0
Collecting libevdev (from -r requirements.txt (line 1))
  Using cached libevdev-0.11.tar.gz (27 kB)
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
Collecting numpy (from -r requirements.txt (line 2))
  Using cached numpy-2.1.3-cp311-cp311-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (62 kB)
Collecting pyinotify (from -r requirements.txt (line 3))
  Using cached pyinotify-0.9.6.tar.gz (60 kB)
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
Collecting python-xlib (from -r requirements.txt (line 4))
  Using cached python_xlib-0.33-py2.py3-none-any.whl.metadata (6.2 kB)
Collecting smbus2 (from -r requirements.txt (line 5))
  Using cached smbus2-0.5.0-py2.py3-none-any.whl.metadata (6.9 kB)
Collecting pyasyncore (from -r requirements.txt (line 6))
[... 94 lines cut ...]
 - Numlock key does not activate NumberPad (config value is set to sys_numlock_enables_numpad=0)
 - Numberpad disactivation does not disable Numlock (config value is set to numpad_disables_sys_numlock=0)

In summary when an external keyboard is connected then NumberPad activation is not linked to Numlock state and vice versa.

Do you want install the rule for external keyboard? [y/N]
Calculator app

Do you want try to install toggling script for XF86Calculator key? [y/N]
Power supply saver

By default is idle functionality disabled and may be manually enabled in config file later. Or by installing this rule can be enabled by every detected change of power supply mode to battery mode.

Idle mode is configured to be enabled after 10s of inactivity and to decrease 30% of brightness.

Do you want install the rule for idle functionality? [y/N]
Rule 80-numberpad-power-supply applied
Udev rules reloaded and triggered

Installation finished succesfully

Reboot is required. Do you want reboot now? [y/N]
dselen@DS-STRIX:/var/log/asus-numberpad-driver$
```
Following the logs

DaanSelen, 2024-11-07:
```
dselen@DS-STRIX:~/asus-numberpad-driver$ export PATH=/usr/sbin:$PATH
dselen@DS-STRIX:~/asus-numberpad-driver$ bash install_device_check.sh 
Testing interface i2c-1: success (adr 0x38)
The detection was successful. Touchpad with NumberPad found: i2c-1
```

ldrahnik, 2024-11-08:
@DaanSelen What have you changed and where for `python3`? Because I see service from yesterday as `running` and `python3` set up to virtual environment as usually that means `/usr/share/asus-numberpad-driver/.env/bin/python3`.

ldrahnik, 2024-11-08:
@DaanSelen Do you hold the icon for at least 1s?

ldrahnik, 2024-11-08:
> 
```
dselen@DS-STRIX:~/asus-numberpad-driver$ export PATH=/usr/sbin:$PATH
dselen@DS-STRIX:~/asus-numberpad-driver$ bash install_device_check.sh 
Testing interface i2c-1: success (adr 0x38)
The detection was successful. Touchpad with NumberPad found: i2c-1
```

@DaanSelen https://github.com/asus-linux-drivers/asus-numberpad-driver/commit/18acef39f5f34391082cde54bdbe51651254a212

JorchMantilla, 2024-11-08:
hello, same here. Part of a log file says:

Collecting pywayland (from -r requirements.txt (line 7))
  Downloading pywayland-0.4.18.tar.gz (247 kB)
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'error'
  error: subprocess-exited-with-error
  
  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [56 lines of output]
      running dist_info
      creating /tmp/pip-modern-metadata-ukihghh0/pywayland.egg-info
      writing /tmp/pip-modern-metadata-ukihghh0/pywayland.egg-info/PKG-INFO
      writing dependency_links to /tmp/pip-modern-metadata-ukihghh0/pywayland.egg-info/dependency_links.txt
      writing entry points to /tmp/pip-modern-metadata-ukihghh0/pywayland.egg-info/entry_points.txt
      writing requirements to /tmp/pip-modern-metadata-ukihghh0/pywayland.egg-info/requires.txt
      writing top-level names to /tmp/pip-modern-metadata-ukihghh0/pywayland.egg-info/top_level.txt
      writing manifest file '/tmp/pip-modern-metadata-ukihghh0/pywayland.egg-info/SOURCES.txt'
      reading manifest file '/tmp/pip-modern-metadata-ukihghh0/pywayland.egg-info/SOURCES.txt'
      reading manifest template 'MANIFEST.in'
      adding license file 'LICENSE'
      writing manifest file '/tmp/pip-modern-metadata-ukihghh0/pywayland.egg-info/SOURCES.txt'
      creating '/tmp/pip-modern-metadata-ukihghh0/pywayland-0.4.18.dist-info'
      Traceback (most recent call last):
        File "/usr/share/asus-numberpad-driver/.env/lib/python3.13/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
          ~~~~^^
        File "/usr/share/asus-numberpad-driver/.env/lib/python3.13/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ~~~~^^^^^^^^^^^^^^^^^^^^^^^^
        File "/usr/share/asus-numberpad-driver/.env/lib/python3.13/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 149, in prepare_metadata_for_build_wheel
          return hook(metadata_directory, config_settings)
        File "/tmp/pip-build-env-0jnf4bv6/overlay/lib/python3.13/site-packages/setuptools/build_meta.py", line 376, in prepare_metadata_for_build_wheel
          self.run_setup()
          ~~~~~~~~~~~~~~^^
        File "/tmp/pip-build-env-0jnf4bv6/overlay/lib/python3.13/site-packages/setuptools/build_meta.py", line 319, in run_setup
          exec(code, locals())
          ~~~~^^^^^^^^^^^^^^^^
        File "<string>", line 175, in <module>
        File "/tmp/pip-build-env-0jnf4bv6/overlay/lib/python3.13/site-packages/setuptools/__init__.py", line 117, in setup
          return distutils.core.setup(**attrs)
                 ~~~~~~~~~~~~~~~~~~~~^^^^^^^^^
        File "/tmp/pip-build-env-0jnf4bv6/overlay/lib/python3.13/site-packages/setuptools/_distutils/core.py", line 183, in setup
          return run_commands(dist)
        File "/tmp/pip-build-env-0jnf4bv6/overlay/lib/python3.13/site-packages/setuptools/_distutils/core.py", line 199, in run_commands
          dist.run_commands()
          ~~~~~~~~~~~~~~~~~^^
        File "/tmp/pip-build-env-0jnf4bv6/overlay/lib/python3.13/site-packages/setuptools/_distutils/dist.py", line 954, in run_commands
          self.run_command(cmd)
          ~~~~~~~~~~~~~~~~^^^^^
        File "/tmp/pip-build-env-0jnf4bv6/overlay/lib/python3.13/site-packages/setuptools/dist.py", line 999, in run_command
          super().run_command(command)
          ~~~~~~~~~~~~~~~~~~~^^^^^^^^^
        File "/tmp/pip-build-env-0jnf4bv6/overlay/lib/python3.13/site-packages/setuptools/_distutils/dist.py", line 973, in run_command
          cmd_obj.run()
          ~~~~~~~~~~~^^
        File "/tmp/pip-build-env-0jnf4bv6/overlay/lib/python3.13/site-packages/setuptools/command/dist_info.py", line 98, in run
          bdist_wheel = self.get_finalized_command('bdist_wheel')
        File "/tmp/pip-build-env-0jnf4bv6/overlay/lib/python3.13/site-packages/setuptools/_distutils/cmd.py", line 303, in get_finalized_command
          cmd_obj.ensure_finalized()
          ~~~~~~~~~~~~~~~~~~~~~~~~^^
        File "/tmp/pip-build-env-0jnf4bv6/overlay/lib/python3.13/site-packages/setuptools/_distutils/cmd.py", line 111, in ensure_finalized
          self.finalize_options()
          ~~~~~~~~~~~~~~~~~~~~~^^
        File "<string>", line 78, in finalize_options
      AssertionError: Specified Wayland protocol file, /usr/share/wayland/wayland.xml, does not exist please specify valid protocol file
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

JorchMantilla, 2024-11-08:
[install-08-11-2024-15-41-46.log](https://github.com/user-attachments/files/17684055/install-08-11-2024-15-41-46.log)

DaanSelen, 2024-11-08:
> Systemd

Okay after 1 full second it works! I am going to look to shortening this.

ldrahnik, 2024-11-09:
@JorchMantilla Try the lower version of Python (I use it without any problem `3.10.12`) and in case still does not work please create another issue. Unfortunately for my distribution does not work installing Python `3.13-dev` using `pyenv`.

closed 2024-11-09 by ldrahnik