asus-linux-drivers/asus-numberpad-driver #221: [Bug]: ERROR AL INSTALAR WAYLAND FERORA 41

JorchMantilla, 2024-11-25:
### Describe the bug

Al instalar systemctl se produce el error

### Expected behavior

Que haya soporte para wayland

### Relevant log output

```shell
jorchu@fedora:~/Descargas/asus-numberpad-driver-6.4.0$ bash install_service.sh
Systemctl service

Do you want install systemctl service? [y/N]y

LAYOUT_NAME: up5401ea
CONFIG_FILE_DIR_PATH: /usr/share/asus-numberpad-driver

env var DISPLAY: :0
env var WAYLAND_DISPLAY: wayland-0
env var AUTHORITY: /run/user/1000/.mutter-Xwaylandauth.F3Z5X2
env var XDG_RUNTIME_DIR: /run/user/1000
env var DBUS_SESSION_BUS_ADDRESS: unix:path=/run/user/1000/bus
env var XDG_SESSION_TYPE: wayland

ERROR LOG FILE: /var/log/asus-numberpad-driver/error.log

Unfortunatelly you will not be able use feature: Disabling Touchpad (e.g. Fn+special key) disables NumberPad aswell, at this moment is supported only X11
Asus numberpad driver service placed
Systemctl daemon reloaded
Created symlink '/home/jorchu/.config/systemd/user/default.target.wants/asus_numberpad_driver@jorchu.service' → '/usr/lib/systemd/user/asus_numberpad_driver@.service'.
Asus numberpad driver service enabled
Asus numberpad driver service started
```

### Desktop

- Fedora 41
- Linux fedora 6.11.8-300.fc41.x86_64 #1 SMP PREEMPT_DYNAMIC Thu Nov 14 20:37:39 UTC 2024 x86_64 GNU/Linux

ldrahnik, 2024-11-25:
@JorchMantilla Hi, thank your for creating an issue. I do not see anything wrong. Could you post log file `/var/log/asus-numberpad-driver/error.log` and output of `install.sh` script.

JorchMantilla, 2024-11-25:
Updating and loading repositories:
Repositories loaded.
Package "ibus-1.5.31~rc1-2.fc41.x86_64" is already installed.
Package "libevdev-1.13.3-1.fc41.x86_64" is already installed.
Package "curl-8.9.1-2.fc41.x86_64" is already installed.
Package "xinput-1.6.4-3.fc41.x86_64" is already installed.
Package "i2c-tools-4.3-13.fc41.x86_64" is already installed.
Package "python3-devel-3.13.0-1.fc41.x86_64" is already installed.
Package "python3-virtualenv-20.21.1-25.fc41.noarch" is already installed.
Package "libxml2-2.12.8-2.fc41.x86_64" is already installed.
Package "libxml2-2.12.8-2.fc41.i686" is already installed.
Package "libxkbcommon-devel-1.7.0-4.fc41.x86_64" is already installed.
Package "gcc-14.2.1-3.fc41.x86_64" is already installed.

Nothing to do.

groupadd: el grupo «input» ya existe
groupadd: el grupo «i2c» ya existe
groupadd: el grupo «uinput» ya existe
Added groups input, i2c, uinput, numberpad to current user
uinput module loaded
i2c-dev module loaded
uinput module added to auto loaded modules
Udev rules reloaded and triggered

Testing interface i2c-0: failed
Testing interface i2c-1: success (adr 0x15)
The detection was successful. Touchpad with NumberPad found: i2c-1

Default config will be autocreated during the first run and available for futher modifications here:
/usr/share/asus-numberpad-driver/numberpad_dev

created virtual environment CPython3.13.0.final.0-64 in 125ms
  creator CPython3Posix(dest=/usr/share/asus-numberpad-driver/.env, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(extra_search_dir=/usr/share/python-wheels,download=False, pip=bundle, via=copy, app_data_dir=/home/jorchu/.local/share/virtualenv)
    added seed packages: pip==24.2
  activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
Requirement already satisfied: pip in /usr/share/asus-numberpad-driver/.env/lib/python3.13/site-packages (24.2)
Collecting pip
  Using cached pip-24.3.1-py3-none-any.whl.metadata (3.7 kB)
Using cached pip-24.3.1-py3-none-any.whl (1.8 MB)
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 24.2
    Uninstalling pip-24.2:
      Successfully uninstalled pip-24.2
Successfully installed pip-24.3.1
Collecting setuptools
  Using cached setuptools-75.3.0-py3-none-any.whl.metadata (6.9 kB)
Using cached setuptools-75.3.0-py3-none-any.whl (1.3 MB)
Installing collected packages: setuptools
Successfully installed setuptools-75.3.0
Collecting libevdev (from -r requirements.txt (line 1))
  Using cached libevdev-0.11.tar.gz (27 kB)
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'done'
Collecting numpy (from -r requirements.txt (line 2))
  Using cached numpy-2.1.3-cp313-cp313-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (62 kB)
Collecting pyinotify (from -r requirements.txt (line 3))
  Using cached pyinotify-0.9.6.tar.gz (60 kB)
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'done'
[... 109 lines cut ...]

The application of this rule results in the following changes if an external keyboard is connected:

 - Numlock key does not activate NumberPad (config value is set to sys_numlock_enables_numpad=0)
 - Numberpad disactivation does not disable Numlock (config value is set to numpad_disables_sys_numlock=0)

In summary when an external keyboard is connected then NumberPad activation is not linked to Numlock state and vice versa.

Do you want install the rule for external keyboard? [y/N]
Calculator app

Do you want try to install toggling script for XF86Calculator key? [y/N]
Power supply saver

By default is idle functionality disabled and may be manually enabled in config file later. Or by installing this rule can be enabled by every detected change of power supply mode to battery mode.

Idle mode is configured to be enabled after 10s of inactivity and to decrease 30% of brightness.

Do you want install the rule for idle functionality? [y/N]
Rule 80-numberpad-power-supply applied
Udev rules reloaded and triggered

Installation finished succesfully

Reboot is required. Do you want reboot now? [y/N]

ldrahnik, 2024-11-25:
@JorchMantilla Please, try install package `wayland-devel`.

JorchMantilla, 2024-11-25:
Ok, uninstalled. After that I installed:
sudo dfn install wayland-devel.
Everything is ok.
Install again: bash install.sh
Reboot and after that it does not function.

The new log:

Updating and loading repositories:
Repositories loaded.
Package "ibus-1.5.31-1.fc41.x86_64" is already installed.
Package "libevdev-1.13.3-1.fc41.x86_64" is already installed.
Package "curl-8.9.1-2.fc41.x86_64" is already installed.
Package "xinput-1.6.4-3.fc41.x86_64" is already installed.
Package "i2c-tools-4.3-13.fc41.x86_64" is already installed.
Package "python3-devel-3.13.0-1.fc41.x86_64" is already installed.
Package "python3-virtualenv-20.21.1-25.fc41.noarch" is already installed.
Package "libxml2-2.12.8-2.fc41.x86_64" is already installed.
Package "libxml2-2.12.8-2.fc41.i686" is already installed.
Package "libxkbcommon-devel-1.7.0-4.fc41.x86_64" is already installed.
Package "gcc-14.2.1-3.fc41.x86_64" is already installed.

Nothing to do.

groupadd: el grupo «input» ya existe
groupadd: el grupo «i2c» ya existe
groupadd: el grupo «uinput» ya existe
Added groups input, i2c, uinput, numberpad to current user
uinput module loaded
i2c-dev module loaded
uinput module added to auto loaded modules
Udev rules reloaded and triggered

Testing interface i2c-0: failed
Testing interface i2c-1: success (adr 0x15)
The detection was successful. Touchpad with NumberPad found: i2c-1

Default config will be autocreated during the first run and available for futher modifications here:
/usr/share/asus-numberpad-driver/numberpad_dev

created virtual environment CPython3.13.0.final.0-64 in 138ms
  creator CPython3Posix(dest=/usr/share/asus-numberpad-driver/.env, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(extra_search_dir=/usr/share/python-wheels,download=False, pip=bundle, via=copy, app_data_dir=/home/jorchu/.local/share/virtualenv)
    added seed packages: pip==24.2
  activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
Requirement already satisfied: pip in /usr/share/asus-numberpad-driver/.env/lib/python3.13/site-packages (24.2)
Collecting pip
  Using cached pip-24.3.1-py3-none-any.whl.metadata (3.7 kB)
Using cached pip-24.3.1-py3-none-any.whl (1.8 MB)
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 24.2
    Uninstalling pip-24.2:
      Successfully uninstalled pip-24.2
Successfully installed pip-24.3.1
Collecting setuptools
  Downloading setuptools-75.6.0-py3-none-any.whl.metadata (6.7 kB)
Downloading setuptools-75.6.0-py3-none-any.whl (1.2 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.2/1.2 MB 17.6 MB/s eta 0:00:00
Installing collected packages: setuptools
Successfully installed setuptools-75.6.0
Collecting libevdev (from -r requirements.txt (line 1))
  Using cached libevdev-0.11.tar.gz (27 kB)
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'done'
Collecting numpy (from -r requirements.txt (line 2))
[... 104 lines cut ...]

Do you want install the rule for external keyboard? [y/N]
Rule 90-numberpad-external-keyboard.rules applied
Udev rules reloaded and triggered

Calculator app

Do you want try to install toggling script for XF86Calculator key? [y/N]
Detected gnome-calculator
Setting up for gnome-calculator
Toggling script for calculator app gnome-calculator has been installed.

Power supply saver

By default is idle functionality disabled and may be manually enabled in config file later. Or by installing this rule can be enabled by every detected change of power supply mode to battery mode.

Idle mode is configured to be enabled after 10s of inactivity and to decrease 30% of brightness.

Do you want install the rule for idle functionality? [y/N]
Rule 80-numberpad-power-supply applied
Udev rules reloaded and triggered

Installation finished succesfully

Reboot is required. Do you want reboot now? [y/N]

ldrahnik, 2024-11-25:
@JorchMantilla Install `pkg-config libsystemd-dev` aswell.

ldrahnik, 2024-11-25:
@JorchMantilla Does it work?

JorchMantilla, 2024-11-25:
yes!
Thanks a lot6

closed 2024-11-25 by JorchMantilla