asus-linux-drivers/asus-numberpad-driver #223: [Bug]: Another backlight not turning on. Zenbook pro ux6404. Numpad/dialpad won't display.

zeroz41, 2024-11-27:
### Describe the bug

No backlight display lighting up numbers for Asus Dialpad.
No errors noticed in logs so far, installation passed. Rebooted machine.
Sorry didn't find an exact previous issue match and thought this could be appropriate.

What works: 
- Enabling numpad by long press top right of pad
- Calculator Gesture opens my default Calc App
- Pressing parts of the touchpad does seem to correspond with keys that "Should" be appears with backlight. (was able to 'type' into KCalc Application)
- CIRCLE LED indicating dialpad is enabled DOES light up(on top left of pad) when numberpad is enabled.

What doesn't work:
No numberpad shows up when enabled. No light at all besides previously mentioned circle enablement LED.

btw same behavior whether or not external keyboard plugged it. Also no difference if enableing/disabling manually without systemd service.

### Expected behavior

1. Reboot machine
2. Sysd user service active and no errors
3. I enable the numberpad by press holding top right of touchpad.
4. An enablement circle appears on top left of touchpad.
5. An inluminated set of numbers appears on my Asus laptops touch pad.
6. I can visually identify the numbers to type them into something.

### Relevant log output

```shell
bash install log:

cat /var/log/asus-numberpad-driver/install-27-11-2024-14-33-48.log 
warning: ibus-1.5.31-1 is up to date -- skipping
warning: libevdev-1.13.3-1 is up to date -- skipping
warning: curl-8.11.0-3 is up to date -- skipping
warning: xorg-xinput-1.6.4-2 is up to date -- skipping
warning: i2c-tools-4.4-1 is up to date -- skipping
warning: python-3.12.7-1 is up to date -- skipping
warning: python-virtualenv-20.27.1-2 is up to date -- skipping
warning: libxml2-2.13.4-1 is up to date -- skipping
warning: libxkbcommon-1.7.0-2 is up to date -- skipping
warning: gcc-14.2.1+r134+gab884fffe3fc-1 is up to date -- skipping
 there is nothing to do

groupadd: group 'input' already exists
groupadd: group 'i2c' already exists
groupadd: group 'uinput' already exists
Added groups input, i2c, uinput, numberpad to current user
uinput module loaded
i2c-dev module loaded
uinput module added to auto loaded modules
Udev rules reloaded and triggered

Testing interface i2c-4: failed
Testing interface i2c-5: success (adr 0x15)
The detection was successful. Touchpad with NumberPad found: i2c-5

In system remains config file from previous installation. Do you want replace that config with default config? [y/N]
created virtual environment CPython3.12.7.final.0-64 in 183ms
  creator CPython3Posix(dest=/usr/share/asus-numberpad-driver/.env, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(download=False, pip=bundle, via=copy, app_data_dir=/home/tj/.local/share/virtualenv)
    added seed packages: cffi==1.17.1, libevdev==0.11, numpy==2.1.3, pip==24.3.1, pyasyncore==1.0.4, pycparser==2.22, pyinotify==0.9.6, python_xlib==0.33, pywayland==0.4.18, setuptools==75.6.0, six==1.16.0, smbus2==0.5.0, systemd_python==235, xkbcommon==1.0.1
  activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
Requirement already satisfied: pip in /usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages (24.3.1)
Requirement already satisfied: setuptools in /usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages (75.6.0)
Requirement already satisfied: libevdev in /usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages (from -r requirements.txt (line 1)) (0.11)
Requirement already satisfied: numpy in /usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages (from -r requirements.txt (line 2)) (2.1.3)
Requirement already satisfied: pyinotify in /usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages (from -r requirements.txt (line 3)) (0.9.6)
Requirement already satisfied: python-xlib in /usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages (from -r requirements.txt (line 4)) (0.33)
Requirement already satisfied: smbus2 in /usr/share/asus-numberpad-driver/.env/lib/python3.12/site-packages (from -r requirements.txt (line 5)) (0.5.0)
[... 128 lines cut ...]
i2c-8   i2c             i915 gmbus dpa                          I2C adapter
i2c-9   i2c             i915 gmbus dpb                          I2C adapter
i2c-10  i2c             i915 gmbus dpc                          I2C adapter
i2c-11  i2c             i915 gmbus tc1                          I2C adapter
i2c-12  i2c             i915 gmbus tc2                          I2C adapter
i2c-13  i2c             i915 gmbus tc3                          I2C adapter
i2c-14  i2c             i915 gmbus tc4                          I2C adapter
i2c-15  i2c             i915 gmbus tc5                          I2C adapter
i2c-16  i2c             i915 gmbus tc6                          I2C adapter
i2c-17  i2c             AUX A/DDI A/PHY A                       I2C adapter
i2c-18  i2c             AUX USBC1/DDI TC1/PHY TC1               I2C adapter
i2c-19  i2c             AUX USBC2/DDI TC2/PHY TC2               I2C adapter

------------------------
bash install_device_check.sh
Testing interface i2c-4: failed
Testing interface i2c-5: success (adr 0x15)
The detection was successful. Touchpad with NumberPad found: i2c-5
```

### Desktop

- Program version [e.g. 57c742a]
- Distro [e.g. Arch Linux]
- Kernel [e.g. 5.18.15.arch1-2]

zeroz41, 2024-11-27:
potentially worth mentioning too, i installed on KDE wayland. If switching to X11 service fails. But i'm not worried about that for now, and is not the issue ;)

ldrahnik, 2024-11-27:
@zeroz41 Hm. I would try reboot after installation + switch to Windows in case you have dual boot whether it works there.  Also, post here dsdt table please:

```
$ sudo apt-get install iasl
$ cd <work-dir>
$ sudo cp /sys/firmware/acpi/tables/DSDT DSDT
$ sudo iasl -d DSDT
$ sudo gedit DSDT.dsl
```

zeroz41, 2024-11-27:
Hello thanks for response. With the driver active: 
i2ctransfer -f -y 5 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 $<0x01> 0xad turns on the "circle" led as expected and 0x00 turns it off.
i2clist:
i2c-4   i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-5   i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-6   i2c             Synopsys DesignWare I2C adapter         I2C adapter

Tried messing with 4 and 6 but they seemed to not be active not accept read write.
heres full list.
[tj@lappy asus-numberpad-driver-6.4.0]$ i2cdetect -l
i2c-0   i2c             NVIDIA i2c adapter 1 at 1:00.0          I2C adapter
i2c-1   i2c             NVIDIA i2c adapter 3 at 1:00.0          I2C adapter
i2c-2   i2c             NVIDIA i2c adapter 5 at 1:00.0          I2C adapter
i2c-3   i2c             NVIDIA i2c adapter 6 at 1:00.0          I2C adapter
i2c-4   i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-5   i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-6   i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-7   smbus           SMBus I801 adapter at efa0              SMBus adapter
i2c-8   i2c             i915 gmbus dpa                          I2C adapter
i2c-9   i2c             i915 gmbus dpb                          I2C adapter
i2c-10  i2c             i915 gmbus dpc                          I2C adapter
i2c-11  i2c             i915 gmbus tc1                          I2C adapter
i2c-12  i2c             i915 gmbus tc2                          I2C adapter
i2c-13  i2c             i915 gmbus tc3                          I2C adapter
i2c-14  i2c             i915 gmbus tc4                          I2C adapter
i2c-15  i2c             i915 gmbus tc5                          I2C adapter
i2c-16  i2c             i915 gmbus tc6                          I2C adapter
i2c-17  i2c             AUX A/DDI A/PHY A                       I2C adapter
i2c-18  i2c             AUX USBC1/DDI TC1/PHY TC1               I2C adapter
i2c-19  i2c             AUX USBC2/DDI TC2/PHY TC2               I2C adapter

 I tried i2c 0x15 for 456, where 5 is the only one that accepted the command.
maybe there is a second activation i need to find on device 5? I'm not sure.

Here's dsdt.dsl. its quite large.

[DSDT.dsl.txt](https://github.com/user-attachments/files/17940735/DSDT.dsl.txt)

ldrahnik, 2024-11-27:
@zeroz41 Second activation is 60,61 (more in Readme) but it is implemented in driver.

ldrahnik, 2024-11-27:
@zeroz41 What is the output of `$ cat /proc/bus/input/devices`.

ldrahnik, 2024-11-27:
> potentially worth mentioning too, i installed on KDE wayland. If switching to X11 service fails. But i'm not worried about that for now, and is not the issue ;)

@zeroz41 Unfortunatelly you have to reinstall driver each switch in case you use systemd service.

ldrahnik, 2024-11-27:
@zeroz41 Ah, not NumberPad, dialpad doesn't work. Yeah, it will be necessary to listen communication on Windows. Do you have Windows? I am open to instruct you how.

zeroz41, 2024-11-27:
@ldrahnik roger on reinstall for swithing x11, no big deal. Also i do not have a windows partition.
Here's the devices cat. :)
[cat_devices.txt](https://github.com/user-attachments/files/17940843/cat_devices.txt)

ldrahnik, 2024-11-27:
@zeroz41 On Linux try click on Dialpad while you run these commands: `sudo evtest` or `sudo libinput debug-events`

zeroz41, 2024-11-28:
sure thing. Gere is some output from libinput debug-events

sudo libinput debug-events
-event2   DEVICE_ADDED            Power Button                      seat0 default group1  cap:k
-event23  DEVICE_ADDED            Video Bus                         seat0 default group2  cap:k
-event18  DEVICE_ADDED            Video Bus                         seat0 default group2  cap:k
-event0   DEVICE_ADDED            Lid Switch                        seat0 default group3  cap:S
-event1   DEVICE_ADDED            Power Button                      seat0 default group4  cap:k
-event22  DEVICE_ADDED            BY Tech Gaming Keyboard           seat0 default group5  cap:k
-event25  DEVICE_ADDED            BY Tech Gaming Keyboard System Control seat0 default group5  cap:k
-event26  DEVICE_ADDED            BY Tech Gaming Keyboard Consumer Control seat0 default group5  cap:kp scroll-nat
-event27  DEVICE_ADDED            BY Tech Gaming Keyboard           seat0 default group5  cap:k
-event28  DEVICE_ADDED            BY Tech Gaming Keyboard Mouse     seat0 default group5  cap:p left scroll-nat scroll-button
-event15  DEVICE_ADDED            EVGA Corporation EVGA X12 Gaming Mouse seat0 default group6  cap:p left scroll-nat scroll-button
-event16  DEVICE_ADDED            EVGA Corporation EVGA X12 Gaming Mouse Keyboard seat0 default group6  cap:kp scroll-nat
-event19  DEVICE_ADDED            EVGA Corporation EVGA X12 Gaming Mouse Keyboard seat0 default group6  cap:k
-event3   DEVICE_ADDED            ELAN9008:00 04F3:4116             seat0 default group7  cap:t  size 300x188mm ntouches 10 calib
-event6   DEVICE_ADDED            ELAN9008:00 04F3:4116 Stylus      seat0 default group7  cap:T  size 310x194mm left calib
-event11  DEVICE_ADDED            ASUE1A03:00 04F3:32A0 Mouse       seat0 default group8  cap:p left scroll-nat scroll-button
-event12  DEVICE_ADDED            ASUE1A03:00 04F3:32A0 Touchpad    seat0 default group8  cap:pg  size 129x80mm tap(dl off) left scroll-nat scroll-2fg-edge click-buttonareas-clickfinger dwt-on dwtp-on
-event14  DEVICE_ADDED            Asus WMI hotkeys                  seat0 default group9  cap:k
-event10  DEVICE_ADDED            AT Translated Set 2 keyboard      seat0 default group10 cap:k
-event257  DEVICE_ADDED            ASUE1A03:00 04F3:32A0 NumberPad   seat0 default group11 cap:kp left scroll-nat
-event12  POINTER_MOTION          +0.038s        -0.29/  0.00 ( -2.00/ +0.00)
 event12  GESTURE_HOLD_BEGIN      +0.040s       1
 event12  GESTURE_HOLD_END        +0.051s       1
-event257  KEYBOARD_KEY            +0.055s      *** (-1) pressed
 event257  KEYBOARD_KEY            +0.055s      *** (-1) released
2-event12  POINTER_MOTION          +1.189s        0.00/ -0.46 ( +0.00/ -2.00)
 event12  POINTER_MOTION          +1.196s         0.00/ -0.28 ( +0.00/ -1.00)
 event12  POINTER_MOTION          +1.203s         0.32/ -0.32 ( +1.00/ -1.00)
 event12  POINTER_MOTION          +1.210s        -0.32/  0.00 ( -1.00/ +0.00)
 event12  GESTURE_HOLD_BEGIN      +1.214s       1
 event12  GESTURE_HOLD_END        +1.229s       1
-event257  KEYBOARD_KEY            +1.232s      *** (-1) pressed
 event257  KEYBOARD_KEY            +1.233s      *** (-1) released
--event12  POINTER_MOTION          +4.798s        0.28/ -0.28 ( +1.00/ -1.00)
 event12  POINTER_MOTION          +4.805s         0.32/ -0.32 ( +1.00/ -1.00)
 event12  POINTER_MOTION          +4.819s         0.28/  0.00 ( +1.00/ +0.00)
 event12  POINTER_MOTION          +4.827s         0.24/  0.00 ( +1.00/ +0.00)
 event12  GESTURE_HOLD_BEGIN      +4.828s       1
 event12  GESTURE_HOLD_END        +4.840s       1
-event257  KEYBOARD_KEY            +4.843s      *** (-1) pressed
 event257  KEYBOARD_KEY            +4.844s      *** (-1) released
/-event12  POINTER_MOTION          +6.451s        0.00/ -0.62 ( +0.00/ -2.00)
 event12  POINTER_MOTION          +6.458s         0.00/ -0.68 ( +0.00/ -2.00)
 event12  POINTER_MOTION          +6.465s        -0.34/ -0.34 ( -1.00/ -1.00)
 event12  POINTER_MOTION          +6.472s        -0.34/ -0.34 ( -1.00/ -1.00)
 event12  POINTER_MOTION          +6.479s        -0.34/ -0.34 ( -1.00/ -1.00)
 event12  GESTURE_HOLD_BEGIN      +6.481s       1
 event12  POINTER_MOTION          +6.486s        -0.68/ -0.34 ( -2.00/ -1.00)
 event12  GESTURE_HOLD_END        +6.500s       1
-event257  KEYBOARD_KEY            +6.503s      *** (-1) pressed
 event257  KEYBOARD_KEY            +6.504s      *** (-1) released
9-event12  POINTER_MOTION          +8.181s        2.04/ -0.68 ( +6.00/ -2.00)
 event12  POINTER_MOTION          +8.188s         1.70/ -0.68 ( +5.00/ -2.00)
 event12  POINTER_MOTION          +8.195s         1.70/ -1.36 ( +5.00/ -4.00)
-event257  KEYBOARD_KEY            +8.208s      *** (-1) pressed
 event257  KEYBOARD_KEY            +8.208s      *** (-1) released
3-event12  POINTER_MOTION          +9.448s        0.68/ -1.36 ( +2.00/ -4.00)
-event257  KEYBOARD_KEY            +9.463s      *** (-1) pressed
 event257  KEYBOARD_KEY            +9.463s      *** (-1) released
8-event12  POINTER_MOTION          +10.529s       0.68/ -1.70 ( +2.00/ -5.00)
 event12  POINTER_MOTION          +10.537s        1.02/ -2.38 ( +3.00/ -7.00)
-event257  KEYBOARD_KEY            +10.551s     *** (-1) pressed
 event257  KEYBOARD_KEY            +10.551s     *** (-1) released
6-event12  POINTER_MOTION          +17.688s       0.00/  0.31 ( +0.00/ +1.00)
 event12  POINTER_MOTION          +17.695s        0.00/  0.93 ( +0.00/ +3.00)
[... cut ...]
 event12  GESTURE_HOLD_END        +17.715s      1
-event257  KEYBOARD_KEY            +17.716s     *** (-1) pressed
 event257  KEYBOARD_KEY            +17.717s     *** (-1) released
9-event12  POINTER_MOTION          +18.494s      -0.67/ -0.33 ( -2.00/ -1.00)
 event12  POINTER_MOTION          +18.501s       -0.67/ -0.33 ( -2.00/ -1.00)
 event12  POINTER_MOTION          +18.509s       -0.68/ -0.34 ( -2.00/ -1.00)
 event12  POINTER_MOTION          +18.516s       -1.02/  0.00 ( -3.00/ +0.00)
 event12  GESTURE_HOLD_BEGIN      +18.520s      1

ldrahnik, 2024-11-28:
@zeroz41 So these events are send from Touchpad device but are associated to DialPad? Try `sudo evemu-record /dev/input/event<xyz touchpad device event id>` and try describe what you are doing on DialPad / simple should be for example figure out scan code for middle button inside Dialpad in case it is button and can be pressed.

zeroz41, 2024-11-28:
@ldrahnik It seems I was misled and mistaken about the model I own and it only features ASUS dialpad. I wanted to scour the internet and youtube videos to confirm. I thought it featured both numberpad and dialpad when purchased. 
My apologies for the time sunk but regardless thanks for the support. 
We did at least learn that the same numpad activations also enable disable dialpad led, and gesture support works the same.
So non issue, we can close.

ldrahnik, 2024-11-28:
@zeroz41 This NumberPad driver adds NumberPad functionality with no hardware support except backlight leds. So Dialpad works out of the box? I am open to create new repository if it does make sense.

zeroz41, 2024-11-28:
I will check functionality of dialpad without your driver installed soon.
. MY uncertainty from watching youtube videos was still is if my model/other dialpads even have the full backlight led or just the led that lights up the circle shown here. I haven't confirmed 100%    
[image]

To be honest if you were to create a dialpad driver it looks like much of your numberpad code pretty much just works the same for the gestures and activation. Everything functioned for numpad as expected expect the numberpad actually being visible lol.

From what i've looked at the dialpad basically acts as a rotary knob upon activation around the circle, which the gestures being clockwise or counter clockwise around  the circle to use. While much more advanced on windows, i'd imagine some base functionality of hooking into volume controls or brightness.

ldrahnik, 2024-12-01:
@zeroz41 Ok, two variants od DialPad. And you have this one thwt layes on touchpad. So basically associate shortcuts for currently opened app (somehow) with default shortcuts when is not opened any app would be ok? Have possivility to divide circle by half, to quaters? How it looks like when Is light up? Is there defined grid like NumberPad has? So dividing by other number than grid has would be confusing?

ldrahnik, 2024-12-01:
@zeroz41 I have created repository https://github.com/asus-linux-drivers/asus-dialpad-driver but I will not start until the idea is clear.

zeroz41, 2024-12-01:
Nice! @ldrahnik . I wouldn't want to answer incorrectly as I've not experienced actual usage of asus dialpad on Windows. So makes sense not to start anything now until things are more clear!

And yes seems there is a more physical ASUS DIAL and a touchpad based one ASUS DIALPAD. 
Some other questions is "for dialPAD" is is the actual diapad always in the same relative coordinate accross models? If so then your idea of relative touchpad dimension division could be practicle. Or defined grid as mentioned.
Or is there some special detection by hardware when fingers are placed 'near the dial' vs not near dial, while dialpad is "active".
From what i've seen the gesture activation to turn on for DIALPAD appear to be same as ASUS numberpad.

If i get a windows iso and some time I'd be willing to test some things down the line. Currently have limited capped bandwidth related to where i am.

GLLM, 2025-01-17:
Looking at this feature request ... with my Asus Proart P16 2024 (H7606W) ... whose DialPad I would love to see work in Linux ;-)

ldrahnik, 2025-01-17:
@GLLM You can help me with creating the driver by posting your file `cat /proc/bus/input/devices`, confirming that installing this driver originally for NumberPad lit up the circle of Dialpad on Touchpad and by attempting to locate the position of x/y coordination of circle and button in the middle by listening to the Touchpads (I assume it will be sent by Touchpad device) cursor data:

```
$ sudo libinput debug-events
# or
$ sudo evemu-record # pick up the touchpad if does not exist specific device for Dialpad
```

GLLM, 2025-01-20:
Hello @ldrahnik !

thanks for your swift reply.
Do you want me to open another issue for this model Asus Proart P16 2024 (aka H7606W) ?

Here's the /proc/bus/input/devices output: 
[devices.txt](https://github.com/user-attachments/files/18482051/devices.txt)

I'll add the libinput in a moment

Cheers
GLLM

GLLM, 2025-01-21:
Addendum, the driver currenly emulates a numpad on the touchpad, though there's none ! 😆 
I guess it's expected but it's ... weird/not practical.

Question: can the DialPad driver ever emulate a "real" dial that the OS/DE may rely on for various use cases ?
Such as : scrolling in youtube videos (freetuve or website), timeline scroll in NLE (video editors such as KDEnlive, Resolve, Shotcut ...), alter the volume of a media player or the OS/DE in general.
Thanks

ldrahnik, 2025-01-21:
@GLLM Design is now a question. How design something like emulated dialpad with possibility be configured for different distros and usages inside multiple applications. I think thought wayland protocol and even x11 window manager should be possible catch currently opened window and then like this driver do listen for touchpad events and depends on opened window with specific app use configured app shortcut, with some internal logic to evaluate circle aréna on touchpad as knob. AFAIK Windows emulates NumberPad by C# driver aswell because was sniffed thought I2C communication on Windows only enabling/disabling. About DialPad I do not known about any sniffed I2C communication on Windows but I assume the same. I will try prepare some prototyp of driver to test this Is the way what we want, in next weeks. Or you can do it if you want.

GLLM, 2025-01-22:
Hi @ldrahnik 
Unfortunately, I'm not a developer, so helping to dev the driver or any other such thing is completely out of my skills.

However, I still dual boot to windows, so I can try help about the DialPad behavior inside win11.

thanks
GLLM

ldrahnik, 2025-01-25:
@GLLM @zeroz41 I attach the prototype here, run it in visual studio using `debugpy` or in terminal using `python3`. It should simulate knob (by default is circle divided by 4, without recognized app or without focused any window is implemented clockwise volume up, by default 4 times per entire cycle ++++, counter-clockwise volume down, center button toggling volume mute) and center button according to this image with ONE CHANGE (0,0 is TOP LEFT corner):

[image]

**It does not activate/deactivate at this moment backlight. I know that. I will add it ASAP, which is important for better estimating where the circle is located.**

```
#!/usr/bin/env python3

import logging
import os
import sys
import threading
from time import sleep, time
import Xlib.display
import Xlib.X
from libevdev import EV_ABS, EV_KEY, EV_LED, EV_MSC, EV_SYN, Device, InputEvent, const, device
from smbus2 import SMBus, i2c_msg
from typing import Optional
import re
import math
from pywayland.client import Display
from pywayland.protocol.wayland import WlSeat

# Logging setup
logging.basicConfig(level=logging.INFO)
log = logging.getLogger('asus-dialpad-driver')

# Detect session type
xdg_session_type = os.environ.get('XDG_SESSION_TYPE')
if not xdg_session_type:
    log.error("XDG session type is not set. Exiting.")
    sys.exit(1)

# Setup display for X11
display = None
if xdg_session_type == "x11":
    try:
        display = Xlib.display.Display()
        log.info("X11 session detected and connected.")
    except Exception as e:
        log.error(f"Failed to connect to X11 display: {e}")
        sys.exit(1)

wayland_display = None
if xdg_session_type == "wayland":
    try:
        wayland_display = Display()
        wayland_display.connect(os.environ.get('WAYLAND_DISPLAY'))
        log.info("Wayland session detected and connected.")
    except Exception as e:
        log.error(f"Failed to connect to Wayland display: {e}")
        sys.exit(1)

# Figure out devices from devices file
touchpad: Optional[str] = None
touchpad_name: Optional[str] = None
device_id: Optional[str] = None
device_addr: Optional[int] = None

# Constants
try_times = 5
try_sleep = 0.1

# Look into the devices file #
while try_times > 0:

    touchpad_detected = 0

    with open('/proc/bus/input/devices', 'r') as f:
[... 311 lines cut ...]
                                        emulate_shortcuts(action)
                                last_slice = current_slice
                    else:
                        log.debug(f"Touch outside circle at ({touch_x}, {touch_y}), ignoring.")

    except device.EventsDroppedException:
        for e in dev_t.sync(True):
            pass
    except Exception as e:
        log.error(f"Error in listen_touchpad_events: {e}")

# Initialize the device
initialize_virtual_device()

# Start the touchpad listener in a separate thread
listener_thread = threading.Thread(target=listen_touchpad_events, daemon=True)
listener_thread.start()

# Keep the script running
try:
    while True:
        sleep(1)
except KeyboardInterrupt:
    log.info("Exiting.")
```

zeroz41, 2025-01-26:
@ldrahnik I will try to test this out within the following week if it helps. Thanks for making it.

GLLM, 2025-01-26:
hello @ldrahnik !

Thanks a lot for you script/mockup !!
 
I tried to run the pytho script you sent, on my Fedora 41 KDE laptop.
I have an issue.
I was missing python3-libevdev so I added it.

And now I'm stuck with the below issue :

`gllm@p16:~/Downloads$ sudo py asus_dialpad.py 
[sudo] password for gllm: 
Traceback (most recent call last):
  File "/home/Downloads/asus_dialpad.py", line 11, in <module>
    from smbus2 import SMBus, i2c_msg
ModuleNotFoundError: No module named 'smbus2'
`

I have tried installing python3-i2c-tools but it does not help!
Could you please advise which python lib I need to install so that your script runs ?
I've looked into fedora dnf for smbus or i2c libs but none helped!

Thanks a lot
GLLM

ldrahnik, 2025-01-26:
@GLLM Try to install the same pip packages as for the Numberpad driver, which means file `requirements.txt`

GLLM, 2025-01-31:
Thanks for your reply. I added this but still cannot launch the python script

ldrahnik, 2025-01-31:
@GLLM Post here output please

zeroz41, 2025-02-01:
For my dialpad (Asus zenbook UX6404) (there is a visual indicator on the center and rotary on my device) the center value appears to be ~Distance 380 when tapping the center. Also spun around it a few times. Likely will differ for GLLM's. Is there anything specific to test that could help?
I'm on wayland KDE, And i did notice some spradic volume control happening when rotating around touchpad near dialpad, so thats cool! Just may not be lined up on REAL dimensions yet, but i'm seeing it.

Had small errors on runtime. Here's a diff that appeared to fix it, and add text file logging.

19,26c19
< logging.basicConfig(
<     level=logging.INFO,
<     format='%(asctime)s - %(name)s - %(levelname)s - %(message)s',
<     handlers=[
<         logging.FileHandler('asus-dialpad.log'),
<         logging.StreamHandler()
<     ]
< )
---
> logging.basicConfig(level=logging.INFO)
45d37
< #xdg_session_type == "wayland"
50c42
<         wayland_display.connect()
---
>         wayland_display.connect(os.environ.get('WAYLAND_DISPLAY'))
409c401
<                         log.info("Distance: %f, Center button radius: %f", distance, center_button_radius)
---
>                         log.info(distance, center_button_radius)

[asus-dialpad1.log](https://github.com/user-attachments/files/18629094/asus-dialpad1.log)

GLLM, 2025-02-03:
@ldrahnik 
I get this when running pip  install -r requirements.txt:
```(...) Installing collected packages: smbus2, pyasyncore, numpy, xkbcommon, pywayland
Successfully installed numpy-2.2.2 pyasyncore-1.0.4 pywayland-0.4.18 smbus2-0.5.0 xkbcommon-1.0.1
gllm@p16:~/git/asus-numberpad-driver$ sudo py asus_dialpad.py 
Traceback (most recent call last):
  File "/home/git/asus-numberpad-driver/asus_dialpad.py", line 11, in <module>
    from smbus2 import SMBus, i2c_msg
ModuleNotFoundError: No module named 'smbus2'
```

ldrahnik, 2025-02-15:
So current version of code is below, 1) @zeroz41 with your contribution (logging will be to journal as in numberpad driver and now is IMHO enough send output to terminal) and 2) my attempt of implementation how to find current window title under KDE: ```#!/usr/bin/env python3

import logging
import os
import sys
import threading
from time import sleep, time
import Xlib.display
import Xlib.X
from libevdev import EV_ABS, EV_KEY, EV_LED, EV_MSC, EV_SYN, Device, InputEvent, const, device
from smbus2 import SMBus, i2c_msg
from typing import Optional
import re
import math
from pywayland.client import Display
from pywayland.protocol.wayland import WlSeat
import subprocess
import dbus

# Logging setup
logging.basicConfig(
    format='%(asctime)s %(levelname)s %(message)s',
    level=os.environ.get('LOG', 'INFO')
)
log = logging.getLogger('asus-dialpad-driver')

# Detect session type
xdg_session_type = os.environ.get('XDG_SESSION_TYPE')
if not xdg_session_type:
    log.error("XDG session type is not set. Exiting.")
    sys.exit(1)

# Setup display for X11
display = None
if xdg_session_type == "x11":
    try:
        display = Xlib.display.Display(os.environ.get('DISPLAY'))
        log.info("X11 session detected and connected.")
    except Exception as e:
        log.error(f"Failed to connect to X11 display: {e}")
        sys.exit(1)

wayland_display = None
if xdg_session_type == "wayland":
    try:
        wayland_display = Display(os.environ.get('WAYLAND_DISPLAY'))
        wayland_display.connect()
        log.info("Wayland session detected and connected.")
    except Exception as e:
        log.error(f"Failed to connect to Wayland display: {e}")
        sys.exit(1)

# Figure out devices from devices file
touchpad: Optional[str] = None
touchpad_name: Optional[str] = None
device_id: Optional[str] = None
device_addr: Optional[int] = None

# Constants
try_times = 5
try_sleep = 0.1

# Look into the devices file #
while try_times > 0:

    touchpad_detected = 0

    with open('/proc/bus/input/devices', 'r') as f:
        lines = f.readlines()
        for line in lines:
[... 395 lines cut ...]
initialize_virtual_device()

# Start the touchpad listener in a separate thread
listener_thread = threading.Thread(target=listen_touchpad_events, daemon=True)
listener_thread.start()

# Keep the script running
try:
    while True:
        sleep(1)
except KeyboardInterrupt:
    log.info("Exiting.")```

@zeroz41 What I do not understand is sent value `380`, that means [380;380] like [x;y] middle of circle? Would be helpful test and send me these values from the code that fits your laptop:

```
            # Define circle and slices
            circle_diameter = width / 4
            circle_radius = circle_diameter / 2
            center_button_diameter = circle_diameter / 2
            center_button_radius = center_button_diameter / 2
            padding = circle_radius - center_button_radius
            circle_center_x = padding + circle_radius
            circle_center_y = padding + circle_radius
```

ldrahnik, 2025-02-15:
@GLLM It looks like you installed package `smbus2` for different instance of python in your system. Try instead these commands:

```
$ py -m pip install  -r requirements.txt
$ sudo py asus_dialpad.py 
```

GLLM, 2025-02-16:
hello @ldrahnik 
thanks for your help.
 
I can launch the sript now
- the dialpad C:CCW move trigger a volume up/down and the middle button a mute action. but the dialpad center seems off.  I did not manage to adapt it
- the false numpad still is active, meaning I get numbers as I type on empty location on the trackpad. How can I disable it ?

Thanks
GLLM

ldrahnik, 2025-02-16:
@GLLM Like this stop NumberPad (will be started again next boot automatically or manually start using second command):

```
$ systemctl stop --user asus_numberpad_driver@$USER.service
$ systemctl start --user asus_numberpad_driver@$USER.service
```

GLLM, 2025-02-18:
Thanks @ldrahnik 
it worked but when i stop it, the new dialpad functionality does not work anymore when running asus_dialpad.py

Do you see a current possibility do decouple the numpad feature from the dialpad one ?

Do you need information about what happens when I run dialpad.py ?  Can I help improve it ?

thanks
GLLM

ldrahnik, 2025-02-19:
@GLLM Yeah, the plan is to have a script `asus_dialpad.py` easy-to-use independently to `numberpad.py`. Now only this script from the previous comment exists. If you can, would be helpful to localize the center of the circle and other dimensions as I asked from @zeroz41 in the previous comment, change these values in the code, and then re-run the code so the code fits perfectly the spot of the dial-pad or even photo of your touchpad taken from the mobile or product photo of your laptops touchpad from the websites would be helpful.

```
            # Define circle and slices
            circle_diameter = width / 4
            circle_radius = circle_diameter / 2
            center_button_diameter = circle_diameter / 2
            center_button_radius = center_button_diameter / 2
            padding = circle_radius - center_button_radius
            circle_center_x = padding + circle_radius
            circle_center_y = padding + circle_radius
```

GLLM, 2025-02-20:
> > Do you need information about what happens when I run dialpad.py ? Can I help improve it?
> 
> If you want help, would be helpful to localize the center of the circle and other dimensions as I asked from [@zeroz41](https://github.com/zeroz41) in the previous comment, change these values in the code, and then re-run the code so the code fits perfectly the spot of the dial-pad or even photo of your touchpad taken from the mobile or product photo of your laptops touchpad from the websites would be helpful.
> 
> ```
>             # Define circle and slices
>             circle_diameter = width / 4
>             circle_radius = circle_diameter / 2
>             center_button_diameter = circle_diameter / 2
>             center_button_radius = center_button_diameter / 2
>             padding = circle_radius - center_button_radius
>             circle_center_x = padding + circle_radius
>             circle_center_y = padding + circle_radius
> ```

Deal, I will trial & error that. Thanks.

Here's a photo I found on the web: [Proart P16 20204](https://laptopmedia.com/wp-content/uploads/2024/06/4-1-e1717416756650.jpg)

GLLM, 2025-02-20:
@ldrahnik 
I've done my homework 💪🏻 

I've physically measured (as best I could) the:
* touchpad : very close to 15.0 cm wide (x) by very close to 10 cm high (y)
* dialpad center: 1.85cm from the top (y) and 1.85 cm from the left (x)
* dialpad total diameter: 2.9cm
* dialpad center button diameter: 1.15cm

I checked the reported touchpad size in pixels (added a log.info in the python script): 4762 (x) x 3099 (y)

Hence, I have close to 
* 317.46 px per cm for the X axis (4762/15)
* 309.9 px per cm for the Y axis

So, I get:
* circle_diameter = 919  (used the X ppcm for 2.9cm)
* center_button_diameter = 364 (used the X ppcm for 1.15cm)
* circle_center_x = 586 (used the X ppcm for 1.85cm)
* circle_center_y = 573 (used the Y ppcm for 1.85cm)

I replaced your variable approximation by the above values and it works flawlessly !!! 👍🏻 

I hope this may help, tell me if you need more.

GLLM

ldrahnik, 2025-02-22:
> It seems launching the script from the terminal makes 'xdg_session_type' = tty, then nothing wayland related works.

@GLLM Could you please try this change? When is not detected `x11` the script now assumes it is wayland, like the numberpad driver does:

This:

```
# Setup display for X11
display = None
if xdg_session_type == "x11":
    try:
        display = Xlib.display.Display(os.environ.get('DISPLAY'))
        log.info("X11 session detected and connected.")
    except Exception as e:
        log.error(f"Failed to connect to X11 display: {e}")
        sys.exit(1)

wayland_display = None
if xdg_session_type == "wayland":
    try:
        wayland_display = Display(os.environ.get('WAYLAND_DISPLAY'))
        wayland_display.connect()
        log.info("Wayland session detected and connected.")
    except Exception as e:
        log.error(f"Failed to connect to Wayland display: {e}")
        sys.exit(1)
```

Change to:

```
# Setup display for X11
display = None
wayland_display = None
if xdg_session_type == "x11":
    try:
        display = Xlib.display.Display(os.environ.get('DISPLAY'))
        log.info("X11 session detected and connected.")
    except Exception as e:
        log.error(f"Failed to connect to X11 display: {e}")
        sys.exit(1)
else:
    try:
        wayland_display = Display(os.environ.get('WAYLAND_DISPLAY'))
        wayland_display.connect()
        log.info("Wayland session detected and connected.")
    except Exception as e:
        log.error(f"Failed to connect to Wayland display: {e}")
        sys.exit(1)
```

GLLM, 2025-02-24:
@ldrahnik thanks for the update. 

The dialpad repo you mention is empty, has only one master branch with a single file name 'LICENCE'.
I'm not sure where I can pull the changes you mention from.

What I did is : pull HEAD, uninstalled then reinstalled.
I then added theupdated file( asus_dialpad.py)  but I still have an error:

```
$~/git/asus-numberpad-driver$ sudo py asus_dialpad.py 
2025-02-24 18:23:58,475 ERROR Failed to connect to Wayland display: Unable to connect to display

```

I'm not sure what I'm doing wrong.

ldrahnik, 2025-02-24:
@GLLM
> The dialpad repo you mention is empty, has only one master branch with a single file name 'LICENCE'.
I'm not sure where I can pull the changes you mention from.

 Not pushed yet, maybe during this evening `v0.0.1`.

ldrahnik, 2025-02-25:
@GLLM [Pushed](https://github.com/asus-linux-drivers/asus-dialpad-driver), closing here.

closed 2025-02-25 by ldrahnik