asus-linux-drivers/asus-numberpad-driver #224: [Feature Request] The ability to use hidraw instead of i2c

Stivux, 2024-12-02:
### Short Description

I am using ASUS Zenbook UM3402YA (up5401ea layout) I noticed that brightness setting via i2c change report 13 feature of /dev/hidraw0

You can use https://github.com/hidutils/hid-feature to check your own.

It looks like:
13   │ Vendor Defined Page FF00 / Vendor Usage 0xc4     │  8   │   8..=15  │    0..=255  │   1   │   112 │ 70
  13   │ Vendor Defined Page FF00 / Vendor Usage 0xc4     │  8   │  16..=23  │    0..=255  │   1   │     3 │ 03
  13   │ Vendor Defined Page FF00 / Vendor Usage 0xc4     │  8   │  24..=31  │    0..=255  │   1   │     **0** │ 00
  13   │ Vendor Defined Page FF00 / Vendor Usage 0xc4     │  8   │  32..=39  │    0..=255  │   1   │    **81** │ 51
  
When I turn numpad on, 0 goes into 1
When I change brightness level 81 is changing to other number (81 is 0x41 + 16)

Advantages of such a change are independence from i2c-dev and ability to recognize the latest brightness after reboot.

ldrahnik, 2024-12-02:
@Stivux What I do wrong:

```
drahnik@Zenbook-UP5401EA:~/projects/hid-feature/target/debug$ ./hid-feature list-devices
Available HID devices:
hidraw0       - ELAN9009:00 04F3:2E36
hidraw1       - ASUE140D:00 04F3:31B9
hidraw2       - hid-ishtp 8087:0AC2
```

drahnik@Zenbook-UP5401EA:~/projects/hid-feature/target/debug$ sudo ./hid-feature list /dev/hidraw1

...
  13   │ Vendor Defined Page FF00 / Vendor Usage 0xc4     │  8   │   8..=15  │    0..=255  │   1   │     0 │ 00
  13   │ Vendor Defined Page FF00 / Vendor Usage 0xc4     │  8   │  16..=23  │    0..=255  │   1   │     0 │ 00
  13   │ Vendor Defined Page FF00 / Vendor Usage 0xc4     │  8   │  24..=31  │    0..=255  │   1   │   233 │ e9
  13   │ Vendor Defined Page FF00 / Vendor Usage 0xc4     │  8   │  32..=39  │    0..=255  │   1   │   233 │ e9
Error: unexpected end of file
```

ldrahnik, 2024-12-03:
@Stivux Laptop `UX3405`:

```
drahnikl@Zenbook-UX3405MA:~/projects/hid-feature/target/debug$ sudo ./hid-feature list /dev/hidraw1
Report ┃                      Usage                       ┃ Bits ┃ Bit Range ┃ Value Range ┃ Count ┃ Value ┃ Bytes
━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━╇━━━━━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━╇━━━━━━━╇━━━━━━
  2    │ Digitizers / Contact Count Maximum               │  8   │   8..=15  │    0..=5    │   1   │     5 │ 05
  6    │ Digitizers / Pad Type                            │  8   │   8..=15  │    0..=15   │   1   │     0 │ 00
Error: unexpected end of file
drahnikl@Zenbook-UX3405MA:~/projects/hid-feature/target/debug$ sudo ./hid-feature list-devices
Available HID devices:
hidraw0       - ELAN9008:00 04F3:4213
hidraw1       - ASUP1415:00 093A:300C
hidraw2       - hid-ishtp 8087:0AC2
hidraw3       - E-Signal/A-One USB Gaming Mouse
hidraw4       - E-Signal/A-One USB Gaming Mouse
drahnikl@Zenbook-UX3405MA:~/projects/hid-feature/target/debug$ sudo ./hid-feature list /dev/hidraw1
Report ┃                      Usage                       ┃ Bits ┃ Bit Range ┃ Value Range ┃ Count ┃ Value ┃ Bytes
━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━╇━━━━━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━╇━━━━━━━╇━━━━━━
  2    │ Digitizers / Contact Count Maximum               │  8   │   8..=15  │    0..=5    │   1   │     5 │ 05
  6    │ Digitizers / Pad Type                            │  8   │   8..=15  │    0..=15   │   1   │     0 │ 00
Error: unexpected end of file
drahnikl@Zenbook-UX3405MA:~/projects/hid-feature/target/debug$ 
```

Stivux, 2024-12-03:
@ldrahnik what kernel version are you using? I tested it on Fedora with 6.11.4

ldrahnik, 2024-12-03:
@Stivux I use EOS 7.1:

```
ldrahnik@Zenbook-UP5401EA:~$ uname -r
6.8.0-48-generic
```

and EOS 8:

```
ldrahnik@Zenbook-UX3405MA:~$ uname -r
6.8.0-49-generic
```

ldrahnik, 2024-12-03:
@Stivux What a `python3` version do you use for [pypi.org/project/hid-tools](https://pypi.org/project/hid-tools)? I received these errors:

```
ldrahnik@Zenbook-UP5401EA:~/projects/hid-tools$ sudo pip3 install .
Processing /home/ldrahnik/projects/hid-tools
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [14 lines of output]
      Traceback (most recent call last):
        File "/usr/lib/python3/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 363, in <module>
          main()
        File "/usr/lib/python3/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 345, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
        File "/usr/lib/python3/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 164, in prepare_metadata_for_build_wheel
          return hook(metadata_directory, config_settings)
        File "/tmp/pip-build-env-z4ji7nmq/overlay/local/lib/python3.10/dist-packages/hatchling/build.py", line 117, in prepare_metadata_for_build_wheel
          f.write(builder.config.core_metadata_constructor(builder.metadata))
        File "/tmp/pip-build-env-z4ji7nmq/overlay/local/lib/python3.10/dist-packages/hatchling/metadata/spec.py", line 460, in construct_metadata_file_2_3
          if metadata.core.license:
        File "/tmp/pip-build-env-z4ji7nmq/overlay/local/lib/python3.10/dist-packages/hatchling/metadata/core.py", line 677, in license
          from packaging.licenses import canonicalize_license_expression
      ModuleNotFoundError: No module named 'packaging.licenses'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

and 

```
ldrahnik@Zenbook-UP5401EA:~/projects/hid-tools$ sudo ./hid-feature
Traceback (most recent call last):
  File "/home/ldrahnik/projects/hid-tools/./hid-feature", line 21, in <module>
    import hidtools.cli.feature
  File "/home/ldrahnik/projects/hid-tools/hidtools/cli/feature.py", line 24, in <module>
    from hidtools.hidraw import HidrawDevice
  File "/home/ldrahnik/projects/hid-tools/hidtools/hidraw.py", line 27, in <module>
    from hidtools.hid import ReportDescriptor
  File "/home/ldrahnik/projects/hid-tools/hidtools/hid.py", line 26, in <module>
    from hidtools.hut import HUT, U8, U16, U32
  File "/home/ldrahnik/projects/hid-tools/hidtools/hut.py", line 23, in <module>
    import parse
ModuleNotFoundError: No module named 'parse'
```

Stivux, 2024-12-03:
@ldrahnik Since I'm on Fedora 41, Python 3.13

Stivux, 2024-12-03:
I'm downloading elementary OS iso to check it out on my own

Stivux, 2024-12-03:
I tested it on Elementary OS 8.0. It works too.
[image]

ldrahnik, 2024-12-04:
@Stivux How you installed `hid-features`? I will try tommorow and let you know.

Stivux, 2024-12-04:
@ldrahnik hid-feature is part of hid-tools
I install it on ElementaryOS via virtualenv.
In virtualenv:
```sh
pip install hid-tools
```
Then activate virtualenv as root.

One more observation: if I disable the i2c-dev module I can't see changing of led feature values. They will be constant. But I can enable disable led setting all d000* features.
Like:
```sh
hid-feature set /dev/hidraw0 -f d0000 112 -f d0000 3 -f d0000 1 -f d0000 83 # led on, brightness 0x43. 
```
In this case I can change values (enable/disable led, set brightness), but can't see new value of d0003 for example. Looks like i2c-dev is used to communicate with hidraw devices.

ldrahnik, 2024-12-04:
@Stivux Looks like the same values are displayed to me `0, 0, 233 and 233`. Steps:

```
ldrahnik@Zenbook-UP5401EA:~/projects/hid-feature$ pyenv install 3.9.19
ldrahnik@Zenbook-UP5401EA:~/projects/hid-feature$ pyenv global 3.9.19
ldrahnik@Zenbook-UP5401EA:~/projects/hid-feature$ python3 --version
Python 3.9.19
ldrahnik@Zenbook-UP5401EA:~$ sudo chown ldrahnik:ldrahnik /dev/hidraw*
[sudo] password for ldrahnik:          
(.penv) ldrahnik@Zenbook-UP5401EA:~/projects/hid-feature$ pip install hid-tools
Collecting hid-tools
  Downloading hid_tools-0.8-py3-none-any.whl (106 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 106.5/106.5 KB 279.3 kB/s eta 0:00:00
Collecting pyyaml
  Downloading PyYAML-6.0.2-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (751 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 751.2/751.2 KB 133.6 kB/s eta 0:00:00
Collecting pyroute2
  Downloading pyroute2-0.7.12-py3-none-any.whl (460 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 460.5/460.5 KB 252.9 kB/s eta 0:00:00
Collecting typing-extensions
  Using cached typing_extensions-4.12.2-py3-none-any.whl (37 kB)
Collecting click
  Using cached click-8.1.7-py3-none-any.whl (97 kB)
Collecting libevdev
  Using cached libevdev-0.11-py3-none-any.whl
Collecting parse
  Using cached parse-1.20.2-py2.py3-none-any.whl (20 kB)
Installing collected packages: pyroute2, parse, typing-extensions, pyyaml, libevdev, click, hid-tools
Successfully installed click-8.1.7 hid-tools-0.8 libevdev-0.11 parse-1.20.2 pyroute2-0.7.12 pyyaml-6.0.2 typing-extensions-4.12.2
ldrahnik@Zenbook-UP5401EA:~$ ls -la /dev/hidraw*
crw------- 1 ldrahnik ldrahnik 241, 0 Dec  4 16:28 /dev/hidraw0
crw------- 1 ldrahnik ldrahnik 241, 1 Dec  4 16:28 /dev/hidraw1
crw------- 1 ldrahnik ldrahnik 241, 2 Dec  4 16:28 /dev/hidraw2
(.penv) ldrahnik@Zenbook-UP5401EA:~/projects/hid-feature$ hid-feature list-devices
Available devices:
/dev/hidraw0: ELAN9009:00 04F3:2E36
/dev/hidraw1: ASUE140D:00 04F3:31B9
/dev/hidraw2: hid-ishtp 8087:0AC2
(.penv) ldrahnik@Zenbook-UP5401EA:~/projects/hid-feature$ hid-feature list /dev/hidraw1 --fetch-values | grep d000 
  d0000 |     13 | Vendor Defined Page 1     | 0xff0000c4                                 | [ 0, 255] |     1 |   8  | 0
  d0001 |     13 | Vendor Defined Page 1     | 0xff0000c4                                 | [ 0, 255] |     1 |   8  | 0
  d0002 |     13 | Vendor Defined Page 1     | 0xff0000c4                                 | [ 0, 255] |     1 |   8  | 233
  d0003 |     13 | Vendor Defined Page 1     | 0xff0000c4                                 | [ 0, 255] |     1 |   8  | 233

```

ldrahnik, 2024-12-04:
```
(.penv) ldrahnik@Zenbook-UP5401EA:~/projects/hid-feature$ hid-feature list /dev/hidraw2 --fetch-values > values2
(.penv) ldrahnik@Zenbook-UP5401EA:~/projects/hid-feature$ hid-feature list /dev/hidraw1 --fetch-values > values1
(.penv) ldrahnik@Zenbook-UP5401EA:~/projects/hid-feature$ hid-feature list /dev/hidraw0 --fetch-values > values0
(.penv) ldrahnik@Zenbook-UP5401EA:~/projects/hid-feature$ hid-feature list /dev/hidraw0 --fetch-values > values0_new
(.penv) ldrahnik@Zenbook-UP5401EA:~/projects/hid-feature$ hid-feature list /dev/hidraw1 --fetch-values > values1_new
(.penv) ldrahnik@Zenbook-UP5401EA:~/projects/hid-feature$ hid-feature list /dev/hidraw2 --fetch-values > values2_new
(.penv) ldrahnik@Zenbook-UP5401EA:~/projects/hid-feature$ diff values2 values2_new 
(.penv) ldrahnik@Zenbook-UP5401EA:~/projects/hid-feature$ diff values1 values1_new 
(.penv) ldrahnik@Zenbook-UP5401EA:~/projects/hid-feature$ diff values0 values0_new 
(.penv) ldrahnik@Zenbook-UP5401EA:~/projects/hid-feature$ 

```

ldrahnik, 2024-12-04:
@Stivux What is your output?

````
ldrahnik@Zenbook-UP5401EA:~$ cat /sys/class/hidraw/hidraw1/device/uevent 
DRIVER=hid-multitouch
HID_ID=0018:000004F3:000031B9
HID_NAME=ASUE140D:00 04F3:31B9
HID_PHYS=i2c-ASUE140D:00
HID_UNIQ=
MODALIAS=hid:b0018g0004v000004F3p000031B9
```

Stivux, 2024-12-05:
@ldrahnik 
`journalctl | grep hid`
```
Dec 05 16:14:34 elementary kernel: AMD-Vi: ivrs, add hid:AMDI0020, uid:\_SB.FUR0, rdevid:160
Dec 05 16:14:34 elementary kernel: AMD-Vi: ivrs, add hid:AMDI0020, uid:\_SB.FUR1, rdevid:160
Dec 05 16:14:34 elementary kernel: AMD-Vi: ivrs, add hid:AMDI0020, uid:\_SB.FUR2, rdevid:160
Dec 05 16:14:34 elementary kernel: AMD-Vi: ivrs, add hid:AMDI0020, uid:\_SB.FUR3, rdevid:160
Dec 05 16:14:34 elementary kernel: hid: raw HID events driver (C) Jiri Kosina
Dec 05 16:14:34 elementary kernel: hid-generic 0018:04F3:31B9.0001: input,hidraw0: I2C HID v1.00 Mouse [ASUE140D:00 04F3:31B9] on i2c-ASUE140D:00
Dec 05 16:14:34 elementary kernel: hid-multitouch 0018:04F3:31B9.0001: input,hidraw0: I2C HID v1.00 Mouse [ASUE140D:00 04F3:31B9] on i2c-ASUE140D:00
Dec 05 16:15:42 elementary io.elementary.w[3728]: GFileInfo created without standard::is-hidden
Dec 05 16:15:42 elementary io.elementary.w[3728]: file ../../../gio/gfileinfo.c: line 1632 (g_file_info_get_is_hidden): should not be reached
Dec 05 16:15:42 elementary io.elementary.w[3728]: GFileInfo created without standard::is-hidden
Dec 05 16:15:42 elementary io.elementary.w[3728]: file ../../../gio/gfileinfo.c: line 1632 (g_file_info_get_is_hidden): should not be reached
Dec 05 16:16:47 elementary io.elementary.w[3728]: GFileInfo created without standard::is-hidden
Dec 05 16:16:47 elementary io.elementary.w[3728]: file ../../../gio/gfileinfo.c: line 1632 (g_file_info_get_is_hidden): should not be reached
Dec 05 16:16:47 elementary io.elementary.w[3728]: GFileInfo created without standard::is-hidden
Dec 05 16:16:47 elementary io.elementary.w[3728]: file ../../../gio/gfileinfo.c: line 1632 (g_file_info_get_is_hidden): should not be reached
Dec 05 16:16:47 elementary io.elementary.w[3728]: GFileInfo created without standard::is-hidden
Dec 05 16:16:47 elementary io.elementary.w[3728]: file ../../../gio/gfileinfo.c: line 1632 (g_file_info_get_is_hidden): should not be reached
Dec 05 16:16:47 elementary io.elementary.w[3728]: GFileInfo created without standard::is-hidden
Dec 05 16:16:47 elementary io.elementary.w[3728]: file ../../../gio/gfileinfo.c: line 1632 (g_file_info_get_is_hidden): should not be reached
Dec 05 16:16:49 elementary io.elementary.w[3728]: GFileInfo created without standard::is-hidden
Dec 05 16:16:49 elementary io.elementary.w[3728]: file ../../../gio/gfileinfo.c: line 1632 (g_file_info_get_is_hidden): should not be reached
Dec 05 16:16:49 elementary io.elementary.w[3728]: GFileInfo created without standard::is-hidden
Dec 05 16:16:49 elementary io.elementary.w[3728]: file ../../../gio/gfileinfo.c: line 1632 (g_file_info_get_is_hidden): should not be reached
Dec 05 16:16:49 elementary io.elementary.w[3728]: GFileInfo created without standard::is-hidden
Dec 05 16:16:49 elementary io.elementary.w[3728]: file ../../../gio/gfileinfo.c: line 1632 (g_file_info_get_is_hidden): should not be reached
Dec 05 16:16:49 elementary io.elementary.w[3728]: GFileInfo created without standard::is-hidden
Dec 05 16:16:49 elementary io.elementary.w[3728]: file ../../../gio/gfileinfo.c: line 1632 (g_file_info_get_is_hidden): should not be reached
```
Also

```
elementary@elementary:~$ cat /sys/class/hidraw/hidraw0/device/uevent 
DRIVER=hid-multitouch
HID_ID=0018:000004F3:000031B9
HID_NAME=ASUE140D:00 04F3:31B9
HID_PHYS=i2c-ASUE140D:00
HID_UNIQ=
MODALIAS=hid:b0018g0004v000004F3p000031B9
elementary@elementary:~$ 
```

ldrahnik, 2024-12-05:
@Stivux Yes, I use it for example for NumberPad

Stivux, 2024-12-06:
> @Stivux Yes, I use it for example for NumberPad

I don't have any more ideas why it doesn't work for you. Looks like this issue can be closed.

ldrahnik, 2024-12-06:
@Stivux I am now flashing USB disk and will try live iso OS8 also. On laptop UP5401EA. I will continue.

ldrahnik, 2024-12-06:
@Stivux Unfortuntelly the same. Not replicated change of these values 0, 0, 233, 233.  I am skipping some important step maybe but rather I think you have different .dsdt table because another laptop model or something like that.

ldrahnik, 2024-12-06:
[image]

ldrahnik, 2024-12-06:
```
$ sudo apt-get install iasl
$ cd <work-dir>
$ sudo cp /sys/firmware/acpi/tables/DSDT DSDT
$ sudo iasl -d DSDT
$ sudo gedit DSDT.dsl
```

Stivux, 2024-12-06:
Can you try do as experiment this:
`hid-feature set /dev/hidraw1 -f d0000 112 -f d0001 3 -f d0002 1 -f d0003 83`

ldrahnik, 2024-12-07:
@Stivux 
```
ldrahnik@Zenbook-UP5401EA:~/projects/hid-feature/target/debug$ ./hid-feature --version
hid-feature 0.1.0
ldrahnik@Zenbook-UP5401EA:~/projects/hid-feature/target/debug$ ./hid-feature set /dev/hidraw1 -f d0000 112 -f d0001 3 -f d0002 1 -f d0003 83
error: unexpected argument '-f' found

  tip: to pass '-f' as a value, use '-- -f'

Usage: hid-feature set [OPTIONS] <PATH> [BYTES]...

For more information, try '--help'.
ldrahnik@Zenbook-UP5401EA:~/projects/hid-feature/target/debug$ 
```

ldrahnik, 2024-12-07:
@Stivux 

Mine I2C1 with register `0x0015` here:

```
...
  I2cSerialBusV2 (0x0015, ControllerInitiated, 0x00061A80,
                        AddressingMode7Bit, "\\_SB.PC00.I2C1",
                        0x00, ResourceConsumer, , Exclusive,
                        )
..
```

Full below:

```
    Scope (_SB.PC00.I2C1)
    {
        Device (ETPD)
        {
            Name (_ADR, One)  // _ADR: Address
            Name (ETPH, Package (0x23)
            {
                "ELAN1200", 
                "ELAN1201", 
                "ELAN1203", 
                "ELAN1200", 
                "ELAN1201", 
                "ELAN1300", 
                "ELAN1301", 
                "ELAN1300", 
                "ELAN1301", 
                "ELAN1000", 
                "ELAN1200", 
                "ELAN1200", 
                "ELAN1200", 
                "ELAN1200", 
                "ELAN1200", 
                "ELAN1203", 
                "ELAN1203", 
                "ELAN1201", 
                "ELAN1300", 
                "ELAN1300", 
                "ELAN1200", 
                "ELAN1300", 
                "ELAN1201", 
                "ELAN1200", 
                "ELAN1201", 
                "ELAN1200", 
                "ELAN1401", 
                "ELAN1200", 
                "ELAN1401", 
                "ELAN1401", 
                "ELAN1401", 
                "ELAN1206", 
                "ASUE1407", 
                "ASUE140A", 
                "ASUE140D"
            })
            Name (FTPH, Package (0x0C)
            {
                "FTE1001", 
                "FTE1200", 
                "FTE1200", 
                "FTE1300", 
                "FTE1300", 
                "FTE1201", 
                "FTE1200", 
                "FTE1200", 
                "FTE1201", 
[... 312 lines cut ...]

                            }
                        }
                        Case (One)
                        {
                            Return (One)
                        }
                        Default
                        {
                            Return (Zero)
                        }

                    }
                }
                Else
                {
                    Return (Buffer (One)
                    {
                         0x00                                             // .
                    })
                }
            }
        }
    }
```

ldrahnik, 2024-12-07:
@Stivux Did you try to install this driver? It works? Could you please post the entire output of the `bash install.sh` script.

Stivux, 2024-12-07:
> ldrahnik@Zenbook-UP5401EA:~/projects/hid-feature/target/debug$ ./hid-feature --version
> hid-feature 0.1.0
> ldrahnik@Zenbook-UP5401EA:~/projects/hid-feature/target/debug$ ./hid-feature set /dev/hidraw1 [...]

Sorry. I forgot to mention, that here `hid-feature` from  https://pypi.org/project/hid-tools

How can we understand from dsdt what i2c register need we use?

ldrahnik, 2024-12-08:
@Stivux Well, it works. Backlight was enabled by your command. So thank your for creating this issue! The question is, how it works, I mean for example together with i2c etc. Can you summarize what you know in one post again, please.

ldrahnik, 2024-12-08:
But read values after setting up I can not:

```
$ hid-feature list /dev/hidraw1 --fetch-values | grep d000
  d0000 |     13 | Vendor Defined Page 1     | 0xff0000c4                                 | [ 0, 255] |     1 |   8  | 0
  d0001 |     13 | Vendor Defined Page 1     | 0xff0000c4                                 | [ 0, 255] |     1 |   8  | 0
  d0002 |     13 | Vendor Defined Page 1     | 0xff0000c4                                 | [ 0, 255] |     1 |   8  | 233
  d0003 |     13 | Vendor Defined Page 1     | 0xff0000c4                                 | [ 0, 255] |     1 |   8  | 233
```

But this command works:

```
$ hid-feature set /dev/hidraw1 -f d0000 112 -f d0001 3 -f d0002 1 -f d0003 [81-87]
```

ldrahnik, 2024-12-08:
@Stivux 

> How can we understand from dsdt what i2c register need we use?

In the `numberpad.py` is the code:

```
    try:
        with SMBus(int(device_id)) as bus:
            data = [0x05, 0x00, 0x3d, 0x03, 0x06, 0x00, 0x07, 0x00, 0x0d, 0x14, 0x03, int(value, 16), 0xad]
            msg = i2c_msg.write(device_addr, data)
            bus.i2c_rdwr(msg)
    except Exception as e:
        log.error('Error during sending via i2c: \"%s\"', e)
```

that handles the communication and from .dsdt table `0x15` is address where are written these values sniffed on windows via i2c sniffer tool (someone posted these values on the ubuntu created issue page, it was not sniffed by me)

link to ubuntu issue https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1810183?comments=all
link to comment attachment of sniffed i2c communication https://launchpadlibrarian.net/527767773/ProbeLog.txt
rows:

```
00::03/13/2021-22:23:06.173 [SimplePeripheralBusProbe]Marking SPB request 00003BF8EEF082E8 for reuse, and completing client request 00003BF8EE838318 with STATUS_SUCCESS and bytes=13
[0]0000.0000::03/13/2021-22:23:06.173 [SimplePeripheralBusProbe]device   5: ##00 write   13 -  0000: **05 00 3d 03 06 00 07 00 0d 14 03 01 ad**
[0]0000.0000::03/13/2021-22:23:06.173 [SimplePeripheralBusProbe] [SpbPeripheralCompleteRequestPair] <--
[0]0000.0000::03/13/2021-22:23:06.173 [SimplePeripheralBusProbe] [SpbPeripheralOnCompletion] <--
[3]0004.00F8::03/13/2021-22:23:06.179 [SimplePeripheralBusProbe] [OnRead] --> entry
```

Stivux, 2024-12-08:
@ldrahnik 

1. You can communicate with numpad led via `/dev/hidraw*` file. You need to know which file matches your touchpad.
Looks like it can be found out from the `/proc/bus/input/devices`
For example `Sysfs` for my touchpad is:
 `/devices/platform/AMDI0010:00/i2c-0/i2c-ASUE140D:00/0018:04F3:31B9.0001/input/input8`
At  `/devices/platform/AMDI0010:00/i2c-0/i2c-ASUE140D:00/0018:04F3:31B9.0001/input/input8/device/hidraw/` I found `hidraw0` directory, so `/dev/hidraw0` matches touchpad.
2. You need to know what hid feature report is used to set led state. In our case it is report 13. In other cases it may be different. I haven't figured out how to define it yet. For some reason you can't see any changes of feature reports.
3. Now you can send report to hid device.
In Python it looks like:
```python
# Part of code from https://github.com/vpelletier/python-ioctl-opt/blob/master/ioctl_opt/__init__.py
import fcntl
import ctypes

_IOC_NRBITS = 8
_IOC_TYPEBITS = 8
_IOC_SIZEBITS = 14
_IOC_DIRBITS = 2

_IOC_NRMASK = (1 << _IOC_NRBITS) - 1
_IOC_TYPEMASK = (1 << _IOC_TYPEBITS) - 1
_IOC_SIZEMASK = (1 << _IOC_SIZEBITS) - 1
_IOC_DIRMASK = (1 << _IOC_DIRBITS) - 1

_IOC_NRSHIFT = 0
_IOC_TYPESHIFT = _IOC_NRSHIFT + _IOC_NRBITS
_IOC_SIZESHIFT = _IOC_TYPESHIFT + _IOC_TYPEBITS
_IOC_DIRSHIFT = _IOC_SIZESHIFT + _IOC_SIZEBITS

IOC_NONE = 0
IOC_WRITE = 1
IOC_READ = 2

def IOC(dir, type, nr, size):
    assert dir <= _IOC_DIRMASK, dir
    assert type <= _IOC_TYPEMASK, type
    assert nr <= _IOC_NRMASK, nr
    assert size <= _IOC_SIZEMASK, size
    return (dir << _IOC_DIRSHIFT) | (type << _IOC_TYPESHIFT) | (nr << _IOC_NRSHIFT) | (size << _IOC_SIZESHIFT)

HIDIOCSFEATURE = lambda len: IOC(IOC_WRITE|IOC_READ, ord('H'), 0x06, len)
HIDIOCGFEATURE = lambda len: IOC(IOC_WRITE|IOC_READ, ord('H'), 0x07, len)
file_handler = open("/dev/hidraw0", 'wb')
report = bytearray([0] * 256)
report[0] = 13 # report number
report_arr = (ctypes.c_char * len(report)).from_buffer(report)

report_size = fcntl.ioctl(file_handler, HIDIOCGFEATURE(len(report)), report_arr, True)
report[1] = 112 # it is magic number now
report[2] = 3 # it is magic number now
report[3] = 0 # or 0 to disable led
report[4] = 81 # is 0x41 + 16. 16 is magic number now

report_arr = (ctypes.c_char * len(report)).from_buffer(report)
fcntl.ioctl(file_handler, HIDIOCSFEATURE(report_size), report_arr, True)
```
The advantage is that it works even if i2c_dev and i2c_smbus modules are blacklisted.
Of course we need some research to understand hid report specification.
To be honest, I haven't figured it out yet. I just shared my observations

Stivux, 2025-01-26:
It's me again.
After some tests, I noticed that if I boot into Windows with the numberpad drivers installed, use numberpad and reboot into Linux, then you will see changes of values using `hid-feature` tool. But after some number of reboots on Linux (without booting Windows in between) it "breaks" i.e. you can still control the backlight using hidraw, but you don't see the changes via `hid-feature`. It looks like it's a BIOS or kernel bug. So the way to find out which hid report to work with is to boot Windows with drivers (Windows PE may be suitable for this) and then launch Linux and use `hid-tools`\`hid-feature` as described above.

BTW, I started creating [my own](https://github.com/Stivux/asus_numpad) version of the "driver", using C++20 (Only backlight control is implemented now 😅 ). Since it seems strange to me to use Python for the driver because its overhead. In addition, your implementation consists of almost a single `numberpad.py` and a bunch of bash scripts. I just can't imagine how you'd contribute to that.

ldrahnik, 2025-01-26:
> BTW, I started creating [my own](https://github.com/Stivux/asus_numpad) version of the "driver", using C++20 (Only backlight control is implemented now 😅 ). Since it seems strange to me to use [...]

Its is great you learn how it works. Personally, I have no problem with `i2c` and I have not registered any opened issues I guess up to now someone's laptop does not support `i2c` or something related to `i2c`. It just works, IMHO.

I do not understand your last sentence. Anyway, please, do not create another do not maintained c++ code. These repositories only confuse the community. What the community needs is a maintained kernel module. It could be rust repository, maybe, in future? Elready exists [one](https://github.com/iamkroot/asus-numpad). From my side its is planned but because lack of time https://github.com/asus-linux-drivers/asus-numberpad-driver/issues/45 I atleats first tried to have satisfactorily working python version, first. I think about this project like pilot project, a playground where we can debug problems for as many users as possible before kernel module will be written.

ldrahnik, 2025-03-17:
Not needed. Can be reopened.