asus-linux-drivers/asus-numberpad-driver #225: [Bug]: "Unable to connect to display" on Debian 12

nroach44, 2024-12-03:
### Describe the bug

I've installed Debian 12 with minimal customisations on a Vivo E410, and after performing the install with the defaults (and the suggested correct layout), and the install appears to succeed, but upon a reboot the touchpad still just works as a touchpad and not a numpad.

### Expected behavior

Numpad works; service doesn't restart repeatedly.

### Relevant log output

```shell
Dec 03 15:14:33 Asus-Vivo-E410 systemd[1426]: Started asus_numberpad_driver@nroach44.service - Asus NumberPad Driver.
Dec 03 15:14:33 Asus-Vivo-E410 PackageKit[1013]: search-file transaction /300_acbdeaac from uid 1000 finished with success after 539ms
Dec 03 15:14:33 Asus-Vivo-E410 /usr/share/asus-numberpad-driver/numberpad.py[3087]: Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
Dec 03 15:14:33 Asus-Vivo-E410 /usr/share/asus-numberpad-driver/numberpad.py[3087]: Set keyboard 0 from H: Handlers=sysrq kbd leds event0
Dec 03 15:14:33 Asus-Vivo-E410 /usr/share/asus-numberpad-driver/numberpad.py[3087]: Detecting touchpad from string: "N: Name="ASUE1409:00 04F3:3157 Touchpad""
Dec 03 15:14:33 Asus-Vivo-E410 /usr/share/asus-numberpad-driver/numberpad.py[3087]: Set touchpad device id 1 from S: Sysfs=/devices/pci0000:00/0000:00:17.1/i2c_designware.1/i2c-1/i2c-ASUE1409:00/0018:04F3:3157.0001/input/input8
Dec 03 15:14:33 Asus-Vivo-E410 /usr/share/asus-numberpad-driver/numberpad.py[3087]: Set touchpad id 5 from H: Handlers=mouse1 event5
Dec 03 15:14:33 Asus-Vivo-E410 /usr/share/asus-numberpad-driver/numberpad.py[3087]: Touchpad min-max: x 0-3204, y 0-1833
Dec 03 15:14:33 Asus-Vivo-E410 /usr/share/asus-numberpad-driver/numberpad.py[3087]: Numpad min-max: x 200-3004, y 240-1793
Dec 03 15:14:33 Asus-Vivo-E410 /usr/share/asus-numberpad-driver/numberpad.py[3087]: Wayland load keymap listener error. Exiting
                                                                                    Traceback (most recent call last):
                                                                                      File "/usr/share/asus-numberpad-driver/numberpad.py", line 450, in load_keymap_listener_wayland
                                                                                        display_wayland.connect()
                                                                                      File "/usr/share/asus-numberpad-driver/.env/lib/python3.11/site-packages/pywayland/client/display.py", line 131, in connect
                                                                                        raise ValueError("Unable to connect to display")
                                                                                    ValueError: Unable to connect to display
Dec 03 15:14:33 Asus-Vivo-E410 /usr/share/asus-numberpad-driver/numberpad.py[3087]: Clean up started
Dec 03 15:14:33 Asus-Vivo-E410 /usr/share/asus-numberpad-driver/numberpad.py[3087]: Clean up finished
Dec 03 15:14:33 Asus-Vivo-E410 /usr/share/asus-numberpad-driver/numberpad.py[3087]: Exiting
```

### Desktop

- Program version: `65181e2528c4a4e518c8f9c60effbe2d8e1fbb9d`
- Distro: Debian 12
- Kernel: `6.1.0-28-amd64`

ldrahnik, 2024-12-03:
@benj5378 It looks like driver was installed for not the same user that was logged in. I am right? Is posted part entire log? Could you post install log, please. Try manually start `numberpad.py` file.

nroach44, 2024-12-03:
> It looks like driver was installed for not the same user that was logged in.

Nope, only one user account on the machine (mine).

Log is attached.
[install.log](https://github.com/user-attachments/files/17989098/install.log)

Manually running `/usr/share/asus-numberpad-driver/.env/bin/python3 /usr/share/asus-numberpad-driver/numberpad.py e210ma /usr/share/asus-numberpad-driver/` does seem to work.

ldrahnik, 2024-12-03:
@nroach44 

> Manually running /usr/share/asus-numberpad-driver/.env/bin/python3 /usr/share/asus-numberpad-driver/numberpad.py e210ma /usr/share/asus-numberpad-driver/ does seem to work.

Please post output.

nroach44, 2024-12-03:
````
nroach44@Asus-Vivo-E410:~$ /usr/share/asus-numberpad-driver/.env/bin/python3 /usr/share/asus-numberpad-driver/numberpad.py e210ma /usr/share/asus-numberpad-driver/
2024-12-03 15:44:23,260 INFO Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
2024-12-03 15:44:23,261 INFO Set keyboard 0 from H: Handlers=sysrq kbd leds event0
2024-12-03 15:44:23,285 INFO Detecting touchpad from string: "N: Name="ASUE1409:00 04F3:3157 Touchpad""
2024-12-03 15:44:23,286 INFO Set touchpad device id 1 from S: Sysfs=/devices/pci0000:00/0000:00:17.1/i2c_designware.1/i2c-1/i2c-ASUE1409:00/0018:04F3:3157.0001/input/input8
2024-12-03 15:44:23,286 INFO Set touchpad id 5 from H: Handlers=mouse1 event5
2024-12-03 15:44:23,287 INFO Touchpad min-max: x 0-3204, y 0-1833
2024-12-03 15:44:23,287 INFO Numpad min-max: x 200-3004, y 240-1793
2024-12-03 15:44:24,292 INFO Setting up for config file key: "top_left_icon_brightness_func_disabled" with value: "True"
2024-12-03 15:44:27,478 INFO Touched top_right_icon area (numlock) in time: 1733211867.4786417
2024-12-03 15:44:27,563 INFO Un-touched top_right_icon area (representing numlock key) in time: 1733211867.5638433
2024-12-03 15:44:27,633 INFO Touched top_right_icon area (numlock) in time: 1733211867.6330853
2024-12-03 15:44:27,702 INFO Un-touched top_right_icon area (representing numlock key) in time: 1733211867.7027617
2024-12-03 15:44:28,869 INFO Touched top_right_icon area (numlock) in time: 1733211868.869932 
2024-12-03 15:44:28,987 INFO Un-touched top_right_icon area (representing numlock key) in time: 1733211868.9875681
2024-12-03 15:44:32,517 INFO Touched top_right_icon area (numlock) in time: 1733211872.517329 
2024-12-03 15:44:32,621 INFO Un-touched top_right_icon area (representing numlock key) in time: 1733211872.6214054
2024-12-03 15:44:38,688 INFO Touched top_right_icon area (numlock) in time: 1733211878.6881738
2024-12-03 15:44:38,764 INFO Un-touched top_right_icon area (representing numlock key) in time: 1733211878.764728
2024-12-03 15:44:39,708 INFO Touched top_right_icon area (numlock) in time: 1733211879.7082212
2024-12-03 15:44:40,448 INFO Un-touched top_right_icon area (representing numlock key) in time: 1733211880.4484792
2024-12-03 15:44:43,198 INFO Touched top_right_icon area (numlock) in time: 1733211883.1980233
2024-12-03 15:44:44,198 INFO The numpad numlock was pressed longer than the activation time and will be activated: 1.0000264644622803
2024-12-03 15:44:44,198 INFO Activation time: 1.0
2024-12-03 15:44:44,230 INFO System numlock activated
2024-12-03 15:44:44,230 INFO Numpad activated
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ASUE1409:00 04F3:3157 Touchpad 
2024-12-03 15:44:45,058 INFO Setting up for config file key: "enabled" with value: "True"
2024-12-03 15:44:45,059 INFO check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/numberpad_dev wd=1 >
2024-12-03 15:44:48,299 INFO Pressed numpad key
2024-12-03 15:44:48,299 INFO KEY_9:10
2024-12-03 15:44:48,299 INFO Unpressed numpad key
2024-12-03 15:44:48,299 INFO KEY_9:10
2024-12-03 15:44:48,675 INFO Pressed numpad key
2024-12-03 15:44:48,676 INFO KEY_9:10
2024-12-03 15:44:48,677 INFO Unpressed numpad key
2024-12-03 15:44:48,677 INFO KEY_9:10
2024-12-03 15:44:48,836 INFO Pressed numpad key
2024-12-03 15:44:48,836 INFO KEY_9:10
2024-12-03 15:44:48,836 INFO Unpressed numpad key
2024-12-03 15:44:48,836 INFO KEY_9:10
2024-12-03 15:44:48,968 INFO Pressed numpad key
2024-12-03 15:44:48,969 INFO Unpressed numpad key
2024-12-03 15:44:48,969 INFO KEY_9:10
````

ldrahnik, 2024-12-03:
@nroach44 So it works? Because I see no problem to fix.

ldrahnik, 2024-12-03:
@vazw How was gathered log under section relevant log output? What is state of systemd service, I mean command `$ sudo systemctl --user status asus_number..`

ldrahnik, 2024-12-03:
@nroach44 What is state of systemd service, I mean command `$ sudo systemctl --user status asus_number..`

ldrahnik, 2024-12-03:
@nroach44 Be you I would try modify service And reinstall and restart laptop. Is weird that when is exited code 1 the service is not restarted. How long time take your booting? And you were imediatelly log in? Or spend time on log in page?, how many attemps of restarts are logged?

closed 2024-12-04 by ldrahnik