asus-linux-drivers/asus-numberpad-driver #229: [Bug]: Wrong key in up5401ea

Topbanaana, 2024-12-27:
### Describe the bug

On numpad pressing '*' gives "-" , wrong key association.
(Vivobook_ASUSLaptop M3401QC) recommended layout up5401ea

### Expected behavior

'*' pressed '*' shown

### Relevant log output

```shell
6.4.1 (12.12.2024)
```

### Desktop

- Program version [e.g. 57c742a]
- Distro [e.g. Arch Linux]
- Kernel [e.g. 5.18.15.arch1-2]

ldrahnik, 2024-12-27:
@Topbanaana Hello, Is layout `up5401ea` correct one for your laptop? Please check by eyes.

> Relevant log output
Please copy and paste any relevant log output. The install and uninstall logs are here /var/log/asus-numberpad-driver/*.log. The runtime logs can be found using journalctl -f | grep asus-numberpad-driver or journalctl -f --user-unit asus_numberpad_driver@ldrahnik.service. This will be automatically formatted into code, so no need for backticks.

And provide driver logs.

ldrahnik, 2024-12-28:
Can be reopened.

closed 2024-12-28 by ldrahnik