asus-linux-drivers/asus-numberpad-driver #230: [Feature Request] Disable Calculator Button when touchpad and numberpad are off

Novantric, 2025-01-09:
### Short Description

When the numberpad isn't enabled/visible, prevent the calculator button from working to prevent accidentally opening it.

### Additional Info

Love this project and works perfectly! But would it be possible to disable the calculator button when the numberpad isn't active?
If my left thumb hits the button area the calculator is opened, even when the touchpad mouse is disabled.
If you could I think this would be a really helpful QOL update.

ldrahnik, 2025-01-09:
@Novantric Hello, I do not understand. Could you please describe where the problem is?

> When the numberpad isn't enabled/visible, disable the calculator button region to prevent accidentally opening it.

How would you like to enable/make it visible with this change?

-- 
You accidentally open NumberPad using your left hand thumb? That means top left or right icon?

ldrahnik, 2025-01-09:
@Novantric Please, try the latest master branch.

ldrahnik, 2025-01-09:
Can be reopened.

closed 2025-01-09 by ldrahnik

Novantric, 2025-01-10:
Just reinstalled it and can confirm I see the `top_left_icon_brightness_func_disabled` option.
Setting this to one disables the calculator action, as intended.
However, this seems to have broken opening the calculator when set to 0.

It seems you misunderstood my original issue.
When disabling the touchpad in GNOME (see the icon below), the top left icon's swipe region was still active. This meant that the calculator could be opened when the touchpad was disabled. I believe this is unintentional behavior, as the top right icon region was disabled, or if intentional should be able to be toggled.
[image]

I'm running Ubuntu 24.10 on Wayland.

Lastly, you might want to review this part of the documentation, as it's worded pretty poorly.
[image]

ldrahnik, 2025-01-10:
> @Novantric When disabling the touchpad in GNOME (see the icon below), the top left icon's swipe region was still active. This meant that the calculator could be opened when the touchpad was disabled. I believe this is unintentional behavior, as the top right icon region was disabled, or if intentional should be able to be toggled.
image
> I'm running Ubuntu 24.10 on Wayland.

Does work for you disabling NumberPad when is activated Touchpad? `touchpad_disables_numpad = 1`

ldrahnik, 2025-01-10:
>@Novantric Lastly, you might want to review this part of the documentation, as it's worded pretty poorly.
image

You are right. I removed that sentence.

ldrahnik, 2025-01-10:
@Novantric For disabling at all use `top_left_icon_slide_func_disabled` and not `top_left_icon_brightness_func_disabled=1`. It was typo in my previous post.

Novantric, 2025-01-10:
I know.

Novantric, 2025-01-10:
(My Pseudocode was more wonky than I thought, should be fixed now)

ldrahnik, 2025-01-10:
@Novantric So for you is not enough, for some reason this change/fix https://github.com/asus-linux-drivers/asus-numberpad-driver/commit/b69042675517b7166bf3fc8929017ae937a073b0? In short: that commit cancelled the possibility of triggering slide function of top left icon when is Touchpad disabled (e.g. via Fn+F6 on my laptop).

I do not know what does mean `[` and `]`.  I will try to understand that later.

Novantric, 2025-01-10:
@ldrahnik good implementation, but seems broken.

1 Works.
Values of 0 and 3 don't work at all and act like it's set to 1.
2 Makes 2-4 calculator windows pop up, no matter the numberpad state (but is off when the touchpad is).

ldrahnik, 2025-01-10:
@Novantric Now it should work, I tested all values.

Novantric, 2025-01-10:
Perfect, 2 works now! With that you've basically solved this issue.
If it's not too much trouble, would it be possible to make the calculator gesture work if the touchpad is off but the numberpad is on? for when `touchpad_disables_numpad` is 0.

ldrahnik, 2025-01-10:
@Novantric Reverted.

Novantric, 2025-01-10:
This combined with the new option 2 for `top_left_icon_slide_func_disabled` has the functionality I've been looking for! Thank you so much for your time and efforts :)